openwrt/mt76 #100: MT76 causes Kernel error on LEDE master (ZBT-WG3526)

mattiantonini, 2017-05-01:
Hi everyone,
I'm using LEDE as operating system on a ZBT-WG3526. When I try to create 2 or more SSID at 2.4 GHz, this error occurs:

Mon May  1 20:45:07 2017 kern.warn kernel: [   24.090000] WARNING: CPU: 0 PID: 30 at /home/randd/SME2.0/LEDE/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1219 mt7603_mac_work+0xe8/0x284 [mt7603e]()
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.110000] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_helper xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nfMon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] CPU: 0 PID: 30 Comm: kworker/u8:2 Not tainted 4.4.61 #0
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] Stack : 8fd2b200 80460000 803e59f4 000004c3 8f16d354 00000000 8fc04614 8fd2b218
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 	  00000088 80063fa8 803e59f4 00000000 0000001e 804c367c 80401370 8fd6dd54
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 	  80460000 80061cf4 80470000 804e0000 8046e188 8046e18c 803ea308 8fd6dd54
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 	  80460000 800414d0 8fc04614 8fd6dd8c 00000324 00000000 00000000 00d6dd74
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 	  8f16d26c 8f649f00 8f649e00 30796870 00000000 00000000 00000000 00000000
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 	  ...
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] Call Trace:
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<80016844>] show_stack+0x50/0x84
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<801b86c4>] dump_stack+0x84/0xbc
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<8002bf40>] warn_slowpath_common+0xa0/0xd0
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<8002bff8>] warn_slowpath_null+0x18/0x24
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<8f16d354>] mt7603_mac_work+0xe8/0x284 [mt7603e]
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<8003f100>] process_one_work+0x214/0x358
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<8003fffc>] worker_thread+0x2d8/0x440
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<800447ac>] kthread+0xd8/0xec
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.180000] 
Mon May  1 20:45:07 2017 kern.warn kernel: [   24.280000] ---[ end trace 0f5d231acece27c2 ]---

Any suggestion?

Thanks,
Mattia Antonini

camelcamro, 2017-05-02:
Are you using current trunk ? Or only release version?
On current trunk, there are many cases already fixed (mt76xx).
If is still not working prefect, but mostly no crashes with multiple ssid on 2,4ghz

mattiantonini, 2017-05-02:
Hi @camelcamro,
I have found a solution for my problem. It was related with HostAPd misconfiguration. I have to remove the /var/run/hostapd-phy* files if I change the /etc/config/wireless.

Anyway I am using the current trunk.

Cheers,
Mattia

closed 2017-05-02 by mattiantonini