openwrt/mt76 #101: ZBT-WG3526 MT7603 - WIFI 2.4 Ghz - throughput is very poor

mattiantonini, 2017-05-02:
Hi all,
I am testing Lede with current trunk on a ZBTwg3526. Throughput is very poor if I perform a iPerf test.

Server (WG3526) `iperf -s`
```
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 10.3.231.1 port 5001 connected with 10.3.231.108 port 60952
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-20.9 sec   256 KBytes   100 Kbits/sec
```

Client (laptop) `iperf -c 10.3.231.1 -i 1 -w 128KB`
```
------------------------------------------------------------
Client connecting to 10.3.231.1, TCP port 5001
TCP window size:  256 KByte (WARNING: requested  128 KByte)
------------------------------------------------------------
[  3] local 10.3.231.108 port 60952 connected with 10.3.231.1 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0- 2.0 sec   128 KBytes   524 Kbits/sec
[  3]  2.0- 4.0 sec  0.00 Bytes  0.00 bits/sec
[  3]  4.0- 6.0 sec  0.00 Bytes  0.00 bits/sec
[  3]  6.0- 8.0 sec  0.00 Bytes  0.00 bits/sec
[  3]  8.0-10.0 sec  0.00 Bytes  0.00 bits/sec
[  3] 10.0-12.0 sec  0.00 Bytes  0.00 bits/sec
[  3]  0.0-20.9 sec   256 KBytes   101 Kbits/sec
```

mattiantonini, 2017-05-05:
@nbd168 can you give me some support?

camelcamro, 2017-05-06:
mine is much faster ... don'T know what is your problem ...
pls try first to install current trunk LEDE version, then use default wireless settings ...
or use mine ... or compare it ...
not all modes are fully supported by all devices ... or then the throughput is horrible ...
use mine and retest
/etc/config/wireless

config wifi-device 'radio0'
	option log_level '3'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country 'US'
	option txpower '20'
	option channel '36'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'WLAN-WLAN-INT-5'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option ifname 'wlan1'

config wifi-device 'radio1'
	option log_level '3'
	option channel '11'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option type 'mac80211'
	option hwmode '11g'
	option htmode 'HT40'
	option country 'US'
	option txpower '20'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'WLAN-WLAN-INT'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option ifname 'wlan0'

config wifi-iface
	option device 'radio0'
	option mode 'ap'
	option ssid 'WLAN-Gast5'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option network 'lan_guest'

config wifi-iface
	option device 'radio1'
	option mode 'ap'
	option ssid 'WLAN-Gast'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option network 'lan_guest'
	option ifname 'wlan0-1'

camelcamro, 2017-05-06:
also, check during the test your logs ...
logread -f
maybe kernel crashed cause of wrong/bad wifi config and is restarting always the wifi related libs

mattiantonini, 2017-05-06:
Hi @camelcamro, I tried with your configuration but the problem has not been resolved (on the latest trunk). I know that there are some problem with this device (I found some issues of yours). Can you tell me which version of kernel are you using? hostapd version? or better, can you give me your .config file?
Thanks a lot ;)

camelcamro, 2017-06-02:
pls try to set wlan 2,4GHZ to MODE: "LEGACY" ... this is the only mode which is not crashing if you are sending DATA over it .. or better .. it is also crashing, but not that often as the other MODES (EG: "N" mode with HT20 or/nad HT40)

you will see with "LEGACY" the transfer speed is real throughput ~22mbit 
tested eg: copy from wlan to LAN server 8GB (SMB).. and in average i got 27Mbit/sec ~

the real problem is that the MT7621 device with chipset 7603(eg: zbt3526) is buggy as hell ...

camelcamro, 2017-06-02:
like this crashes which you will get ...

Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.630000] ------------[ cut here ]------------
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.630000] WARNING: CPU: 0 PID: 28 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-05-30-ef5b045d/mt7603_mac.c:1237 mt7603_mac_work+0x204/0x288 [mt7603e]()
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.650000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserFri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] CPU: 0 PID: 28 Comm: kworker/u8:2 Not tainted 4.4.70 #0
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] Stack : 8fd2b100 80490000 80408a8c 000004d5 8ec3550c 00000000 8fc04614 8fd2b118
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]         00000088 80064ffc 80408a8c 00000000 0000001c 804f367c 80424570 8fd79d54
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]         80490000 80062d48 80490000 80510000 804942c8 804942cc 8040d3ac 8fd79d54
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]         80490000 80042524 8fc04614 8fd79d8c 000004fa 00000000 8fc04614 00d79d74
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]         8ec35308 8ffe5700 8ffe5600 30796870 00000000 00000000 00000000 00000000
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]         ...
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] Call Trace:
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<8001692c>] show_stack+0x50/0x84
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<801bba2c>] dump_stack+0x84/0xbc
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<8002cf94>] warn_slowpath_common+0xa0/0xd0
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<8002d04c>] warn_slowpath_null+0x18/0x24
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<8ec3550c>] mt7603_mac_work+0x204/0x288 [mt7603e]
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<80040154>] process_one_work+0x214/0x358
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<80041050>] worker_thread+0x2d8/0x440
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<80045800>] kthread+0xd8/0xec
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.820000]
Fri Jun  2 12:17:41 2017 kern.warn kernel: [ 4109.910000] ---[ end trace fc74a59bd3559497 ]---

Fri Jun  2 12:17:42 2017 kern.warn kernel: [ 4110.980000] ------------[ cut here ]------------
Fri Jun  2 12:17:42 2017 kern.warn kernel: [ 4110.980000] WARNING: CPU: 3 PID: 28 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-05-30-ef5b045d/mt7603_mac.c:1225 mt7603_mac_work+0xe8/0x288 [mt7603e]()
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.000000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserFri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.170000] CPU: 3 PID: 28 Comm: kworker/u8:2 Tainted: G        W       4.4.70 #0
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.170000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000] Stack : 8fd2b100 80490000 80408a8c 000004c9 8ec353f0 00000000 8fc04614 8fd2b118
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000]         00000088 80064ffc 80408a8c 00000003 0000001c 804f367c 80424570 8fd79d54
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000]         80490000 80062d48 80490000 80510000 804942c8 804942cc 8040d3ac 8fd79d54
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000]         80490000 80042524 8fc04614 8fd79d8c 000006ce 00000000 00000000 00d79d74
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000]         8ec35308 8ffe5700 8ffe5600 30796870 00000000 00000000 00000000 00000000
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.180000]         ...
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.210000] Call Trace:
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.220000] [<8001692c>] show_stack+0x50/0x84
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.220000] [<801bba2c>] dump_stack+0x84/0xbc
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.230000] [<8002cf94>] warn_slowpath_common+0xa0/0xd0
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.230000] [<8002d04c>] warn_slowpath_null+0x18/0x24
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.240000] [<8ec353f0>] mt7603_mac_work+0xe8/0x288 [mt7603e]
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.240000] [<80040154>] process_one_work+0x214/0x358
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.250000] [<80041050>] worker_thread+0x2d8/0x440
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.250000] [<80045800>] kthread+0xd8/0xec
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.260000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.260000]
Fri Jun  2 12:17:43 2017 kern.warn kernel: [ 4111.260000] ---[ end trace fc74a59bd3559498 ]---

nbd168, 2017-12-03:
Fixed in the current version

closed 2017-12-03 by nbd168