openwrt/mt76 #104: more kernel errors ....mt7603_mac.c:1231 (Workqueue: phy0 mt7603_mac_work [mt7603e])

camelcamro, 2017-05-17:
current trunk of LEDE with mt7621 (zbt 3526)
is anyone having an idea how to solve ?

d May 17 16:47:53 2017 kern.warn kernel: [28604.420000] ------------[ cut here ]------------
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.420000] WARNING: CPU: 1 PID: 11927 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1231 mt7603_mac_work+0x204/0x284 [mt7603e]()
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.440000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserWed May 17 16:47:53 2017 kern.warn kernel: [28604.610000] CPU: 1 PID: 11927 Comm: kworker/u8:2 Not tainted 4.4.61 #0
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] Stack : 8e70b280 80490000 80408a6c 000004cf 8ec55470 00000000 8fc04614 8e70b298
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] 	  00000088 80064fa8 80408a6c 00000001 00002e97 804f367c 80424530 8cb05d54
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] 	  80490000 80062cf4 80490000 80510000 80494288 8049428c 8040d380 8cb05d54
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] 	  80490000 800424d0 8fc04614 8cb05d8c 000007b3 00000000 00000000 00b05d74
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] 	  8ec5526c 8f032700 8f032600 30796870 00000000 00000000 00000000 00000000
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.620000] 	  ...
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.660000] Call Trace:
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.660000] [<8001692c>] show_stack+0x50/0x84
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.660000] [<801bb6f8>] dump_stack+0x84/0xbc
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.670000] [<8002cf40>] warn_slowpath_common+0xa0/0xd0
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.670000] [<8002cff8>] warn_slowpath_null+0x18/0x24
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.680000] [<8ec55470>] mt7603_mac_work+0x204/0x284 [mt7603e]
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.680000] [<80040100>] process_one_work+0x214/0x358
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.690000] [<80040ffc>] worker_thread+0x2d8/0x440
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.690000] [<800457ac>] kthread+0xd8/0xec
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.700000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.700000] 
Wed May 17 16:47:53 2017 kern.warn kernel: [28604.710000] ---[ end trace cf8d4752e2d858b6 ]---
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.070000] ------------[ cut here ]------------
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.070000] WARNING: CPU: 1 PID: 11927 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1219 mt7603_mac_work+0xe8/0x284 [mt7603e]()
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.090000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserWed May 17 16:47:57 2017 kern.warn kernel: [28608.260000] CPU: 1 PID: 11927 Comm: kworker/u8:2 Tainted: G        W       4.4.61 #0
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] Stack : 8e70b280 80490000 80408a6c 000004c3 8ec55354 00000000 8fc04614 8e70b298
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] 	  00000088 80064fa8 80408a6c 00000001 00002e97 804f367c 80424530 8cb05d54
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] 	  80490000 80062cf4 80490000 80510000 80494288 8049428c 8040d380 8cb05d54
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] 	  80490000 800424d0 8fc04614 8cb05d8c 00000987 00000000 00000000 00b05d74
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] 	  8ec5526c 8f032700 8f032600 30796870 00000000 00000000 00000000 00000000
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.270000] 	  ...
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.310000] Call Trace:
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.310000] [<8001692c>] show_stack+0x50/0x84
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.310000] [<801bb6f8>] dump_stack+0x84/0xbc
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.320000] [<8002cf40>] warn_slowpath_common+0xa0/0xd0
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.320000] [<8002cff8>] warn_slowpath_null+0x18/0x24
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.330000] [<8ec55354>] mt7603_mac_work+0xe8/0x284 [mt7603e]
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.330000] [<80040100>] process_one_work+0x214/0x358
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.340000] [<80040ffc>] worker_thread+0x2d8/0x440
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.340000] [<800457ac>] kthread+0xd8/0xec
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.350000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.350000] 
Wed May 17 16:47:57 2017 kern.warn kernel: [28608.350000] ---[ end trace cf8d4752e2d858b7 ]---

mattiantonini, 2017-05-19:
Hi @camelcamro,
sometimes this error appears even to me. Try this workaround:
`wifi down`
`rm /var/run/hostapd-phy*`
and reboot.

Let me know!

Mattia

techping, 2017-05-19:
@camelcamro These commands can't solve the problem.

camelcamro, 2017-05-24:
i tested much more ... to figure out why there are errors ...
i think, i found out, that it is related to WPA2
if i produce a lot of traffic (copy data via wlan 2,4 GHZ to NAS server .. within 5min i get simular kernel errors.
tried also to remove the MULTISSID, which was solved for ~1-2 months ...
i tried now to disable the WPA2 (no security + MAC security, which is not safe, but for a 5hours tests it was OK)
and result: without security => then i didn't got kernel errors/crashes.
maybe this hint helps devels to find the real issue.
hmm, I'M sorry, that I have not enough knowledge for the wifi sources .. i can'T fix it, but without 2,4GHZ network it is more or less senseless to use this kind of software :(
is really a showstopper for all mt76xx related hardware routers :(
anything, on which i can help to get this solved ?

camelcamro, 2017-06-02:
tried many different options ...
wifi 2,4GHZ on mt7621 is horrible unstable ...
the only thing what is not that bad working is MODE "LEGACY".
it is horrible slow, too, but better as nothing.

mode ("N"), this is always crashing (eg: copy 1GB through the wlan 2,4GHZ)

the issues are not related to multiple SSID .. it makes no differences if i have multiple SSID or not.

it seems to be for me, that the MT7603 is buggy as hell ...

home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1219 mt7603_mac_work+0xe8/0x284

what i fugured out is, that on my MT7620 devices (eg: ZBT-826) the wifi is stable and REAL DL/UP is around 72mbit/sec ..

with MT7621 (eg: zbt 3526), which is having dual chipset for 2,4GHZ and 5GHZ .. there the 2,4GHZ on "LEGACY" mode is having a REAL throughput of 22mbit/sec

is there any chance to support developers with more debug logging infos for the CRASHING of this mt7603 chipset stuff ?
i would like to help to get this done .. let me know if you new a power tester :)

cu camel

mrkiko, 2017-06-24:
Hello.
Latest commits may have changed the situation. Could you try?
I ask this out of curiosity, I am not having particular issues at the 
moment. I saw that kernel error, but the wi-fi network did not crash in my 
experience.
At least, I think.

mrkiko, 2017-06-24:
And BTW - hos can you enable legacy mode? Is it standard openwrt config or 
something device or driver-specific?
And what about mac security?
Thank you very much

camelcamro, 2017-06-24:
i already tried trunk from 1 week ago ...
> Latest commits may have changed the situation. Could you try?
was there a new change ?
mac security ....if i completely disable security, then i don't have the problems or kernel errors ..
so, i guess maybe it is something related to that.
legacy mode ... also tried ...

mrkiko, 2017-06-24:
>     <reply+0101e3b929be046082e69ea833e5fb64ca3831c274e82c5692cf000000011565bff
> thread.[AQHjuZN1r2a4Ydhu5XRRoeSLi5m1f25Sks5sHKP2gaJpZM4NeLrd.gif]

Thanks for the reply.

pparent76, 2017-07-03:
What is "legacy" mode? Do you mean "802.11g" or "802.11b"?

mattiantonini, 2017-07-03:
@pparent76 "Legacy Mode" means a mixed a/b/g behaviour. In my experiance, it works at effective throughput of 22mbps.

From Openwrt wiki (https://wiki.openwrt.org/doc/uci/wireless):
NONE - disables 802.11n rates and enforce the usage of legacy 802.11 b/g/a rates

pparent76, 2017-07-03:
Ok so basicaly with hostapd we should set option:

ieee80211n=0

Right?

mrkiko, 2017-07-03:
Thank you guys for your suggestions and help.
I arrived at the conclusion you should set the UCI parameter "mode" to "NONE", but I did not check how this reflects on the hostapd.conf file.
Thank you all again,
Enrico

pparent76, 2017-07-05:
I got this message even with n-mode disabled, followed by reboot ( I extracted it from /sys/kernel/debug/crashlog )

```
<1>[61615.500000] CPU 0 Unable to handle kernel paging request at virtual address 00000004, epc == 800a0334, ra == 800a091c
<4>[61615.500000] Oops[#1]:
<4>[61615.500000] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.4.74 #0
<4>[61615.500000] task: 8048fe98 ti: 80480000 task.ti: 80480000
<4>[61615.500000] $ 0   : 00000000 804f0000 00000003 00000024
<4>[61615.500000] $ 4   : 8048f100 00000003 00000000 81180e14
<4>[61615.500000] $ 8   : 8048f1cc 8048f1cc 00000000 8fc0df58
<4>[61615.500000] $12   : 11000000 ffffffff 00003151 77a842c0
<4>[61615.500000] $16   : 8048f100 00000003 00000003 00000141
<4>[61615.500000] $20   : 8fc0dddc 8f1c9c6c 8f1c9520 8048fa08
<4>[61615.500000] $24   : 00000000 800135d8                  
<4>[61615.500000] $28   : 80480000 8fc0dcc8 00000000 800a091c
<4>[61615.500000] Hi    : 00000036
<4>[61615.500000] Lo    : 00000000
<4>[61615.500000] epc   : 800a0334 __rmqueue.isra.11+0x78/0x318
<4>[61615.500000] ra    : 800a091c get_page_from_freelist+0x348/0x760
<4>[61615.500000] Status: 11000402      KERNEL EXL 
<4>[61615.500000] Cause : 4080000c (ExcCode 03)
<4>[61615.500000] BadVA : 00000004
<4>[61615.500000] PrId  : 0001992f (MIPS 1004Kc)
<4>[61615.500000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[61615.500000] Process swapper/0 (pid: 0, threadinfo=80480000, task=8048fe98, tls=00000000)
<4>[61615.500000] Stack : 00000003 802afac8 00000000 0000003b 0003aaaa 06080000 ffffffff 8efc33c0
<4>[61615.500000]         00000000 8f2a0c14 8048f100 00000003 00000003 00000141 8fc0dddc 8f1c9c6c
<4>[61615.500000]         8f1c9520 8048fa08 00000000 800a091c 00000000 00000000 00000000 8fc0dd70
<4>[61615.500000]         00000141 8fc35c00 00000000 80056634 00000000 00000001 0000f0af 00000008
<4>[61615.500000]         00000100 fffffff8 80490000 8048fa00 8048f1e4 000000e4 804126c0 8048c880
<4>[61615.500000]         ...
<4>[61615.500000] Call Trace:
<4>[61615.500000] [<800a0334>] __rmqueue.isra.11+0x78/0x318
<4>[61615.500000] [<800a091c>] get_page_from_freelist+0x348/0x760
<4>[61615.500000] [<800a0e10>] __alloc_pages_nodemask+0xdc/0x6ec
<4>[61615.500000] [<800a1574>] __alloc_page_frag+0x50/0x170
<4>[61615.500000] [<8f23c4a0>] mt76_dma_attach+0x1d0/0xee4 [mt76]
<4>[61615.500000] 
<4>[61615.500000] 
<4>[61615.500000] Code: 8ce90004  8cea0000  24030024 <ad490004> 70431802  ad2a0000  24090100  ace00008  ace90000 
<4>[61615.820000] ---[ end trace ef048c7901df3793 ]---

```

mrkiko, 2017-07-06:
I am sorry I have no answer. What version of the driver are you running? Report the commit you are at.
Thanks.

pparent76, 2017-07-08:
It was with commit 70d225123fa106667fb96265fe62162339e36ddc. https://github.com/openwrt/mt76/commit/70d225123fa106667fb96265fe62162339e36ddc

New commits have just been pushed to LEDE I will see if it is better with it.

mrkiko, 2017-07-08:
thank you very much.

camelcamro, 2017-07-08:
I will also test it Today if builds are ready :)
Thx

pparent76, 2017-07-08:
Thing is in order to reproduce crash you need to be patient. While using wifi, in real conditions, it may need ~24hours before seeing any reboot or kernel messages.

pparent76, 2017-07-10:
I still got this with latest version and kernel 4.9.34::
```
[ 7449.950000] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[11317.250000] ------------[ cut here ]------------
[11317.250000] WARNING: CPU: 1 PID: 34 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-07-04-e90a81a5/mt7603_mac.c:1290 mt7603_mac_work+0x124/0x248 [mt7603e]
[11317.270000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
[11317.410000] CPU: 1 PID: 34 Comm: kworker/u8:2 Not tainted 4.9.34 #0
[11317.420000] Workqueue: phy0 mt7603_mac_work [mt7603e]
[11317.420000] Stack : 8fd05200 804e0000 8047b360 8f1ee918 00000009 0000050a 8fc04a18 8fd05218
[11317.430000]         00000088 80069700 8047b360 00000001 00000022 80543824 8047b2cc 8fd81d54
[11317.440000]         804e0000 800a5c28 00000001 80540000 804e9de4 804e9de8 8047ff8c 8fd81d54
[11317.450000]         804e0000 80041374 8fc04a18 8fd81d8c 000012d0 00000000 00000000 00d81d74
[11317.450000]         8f1ed5b8 8ff20d00 8ff20f00 30796870 00000000 00000000 00000000 00000000
[11317.460000]         ...
[11317.460000] Call Trace:
[11317.470000] [<8000f640>] show_stack+0x50/0x84
[11317.470000] [<801d9684>] dump_stack+0x84/0xc0
[11317.480000] [<8002a1a4>] __warn+0xe4/0x118
[11317.480000] [<8002a26c>] warn_slowpath_null+0x1c/0x30
[11317.490000] [<8f1ed6dc>] mt7603_mac_work+0x124/0x248 [mt7603e]
[11317.490000] [<8003f1ac>] process_one_work+0x200/0x348
[11317.500000] [<80040098>] worker_thread+0x2d4/0x45c
[11317.500000] [<80044be8>] kthread+0xd8/0xec
[11317.500000] [<8000a318>] ret_from_kernel_thread+0x14/0x1c
[11317.510000] ---[ end trace a763364bc1f36be8 ]---
```

pparent76, 2017-07-12:
More kernel errors:
```

<4>[11670.870000] ------------[ cut here ]------------
<4>[11670.870000] WARNING: CPU: 1 PID: 0 at mm/slub.c:3843 ksize+0xe0/0xf8
<4>[11670.880000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark
<4>[11670.950000]  nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[11671.020000] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G        W       4.9.34 #0
<4>[11671.020000] Stack : 00000000 00000000 80547b2a 00000041 803ffa44 00000000 00000000 80540000
<4>[11671.030000]         8fc53ddc 804e3ea7 8047b360 00000001 00000000 80543824 8ec91000 00000000
<4>[11671.040000]         00000000 80069700 00000001 80540000 804e9de4 804e9de8 8047ff8c 8fc19944
<4>[11671.050000]         00000003 800a5c28 8ec91000 00000000 8149ce00 00000000 8f280bfc 00c19944
<4>[11671.060000]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[11671.070000]         ...
<4>[11671.070000] Call Trace:
<4>[11671.070000] [<8000f640>] show_stack+0x50/0x84
<4>[11671.080000] [<801d9684>] dump_stack+0x84/0xc0
<4>[11671.080000] [<8002a1a4>] __warn+0xe4/0x118
<4>[11671.080000] [<8002a26c>] warn_slowpath_null+0x1c/0x30
<4>[11671.090000] [<800e3d84>] ksize+0xe0/0xf8
<4>[11671.090000] [<802d1ac8>] pskb_expand_head+0x98/0x23c
<4>[11671.100000] [<8f22935c>] ieee80211_probereq_get+0xb38/0x18fc [mac80211]
<4>[11671.100000] ---[ end trace a640f2b2f85712c6 ]---
<4>[11671.110000] Kernel bug detected[#1]:
<4>[11671.110000] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.34 #0
<4>[11671.110000] task: 804e3fa0 task.stack: 804d4000
<4>[11671.110000] $ 0   : 00000000 80540000 810268c0 00000001
<4>[11671.110000] $ 4   : 00000000 810268d3 000f0000 811c6a00
<4>[11671.110000] $ 8   : 804e0000 801e5904 8fc0df58 8fc0df58
<4>[11671.110000] $12   : 11000000 00000000 00000000 0000024d
<4>[11671.110000] $16   : 8e84f840 8f281500 8f0faae4 8f281b3c
<4>[11671.110000] $20   : 00000002 8f281b38 00000000 000000e8
<4>[11671.110000] $24   : 00000000 8000c1cc                  
<4>[11671.110000] $28   : 804d4000 8fc0dea0 80204060 802ce860
<4>[11671.110000] Hi    : 00000004
<4>[11671.110000] Lo    : 00000000
<4>[11671.110000] epc   : 800e50d0 kfree+0x78/0x1a4
<4>[11671.110000] ra    : 802ce860 __kfree_skb+0x14/0x9c
<4>[11671.110000] Status: 11000403      KERNEL EXL IE 
<4>[11671.110000] Cause : 50800034 (ExcCode 0d)
<4>[11671.110000] PrId  : 0001992f (MIPS 1004Kc)
<4>[11671.110000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark
<4>[11671.110000]  nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
[... cut ...]
<4>[11671.110000] [<802ce860>] __kfree_skb+0x14/0x9c
<4>[11671.110000] [<8f1989ec>] mt76_dma_attach+0x71c/0xc90 [mt76]
<4>[11671.110000] Code: 30630001  24030001  38630001 <00030336> 8c430000  7c630380  10600003  00000000  10000002 
<4>[11671.110000] 
<4>[11671.410000] ---[ end trace a640f2b2f85712c7 ]---

===================================
Time: 1499814343.517515
<4>[11671.110000] 
<4>[11671.410000] ---[ end trace a640f2b2f85712c7 ]---
<0>[11671.420000] Kernel panic - not syncing: Fatal exception in interrupt

```

mrkiko, 2017-07-12:
Mhm... I unfortunately don't know what to tell you. And in my hw I don't have those messages, or at least I did not notice until now.

b2ag, 2017-07-24:
@mirko did you try iperf/iperf3? I have an MT7688AN SoC and getting slow_path warning and wifi stops working really soon when stress testing with iperf3.

b2ag, 2017-07-24:
@mrkiko did you try iperf/iperf3. I have a MT7688AN SoC getting slow_path in mt7603_mac_work warning soon after starting stress testing with iperf3.

mrkiko, 2017-07-24:
First of all, thank you very very much for your suggestions guys.
And for the general support.

Yeah, actually looking at logs i can see kernel warnings too.

My setup: ere is my Netgear R6220 AP. Everyday, the Wi-Fi is enabled at 
06:58 and stays active until 23:58. It's possible it did crash and nobody 
noticed, the AP isn't so heavily used in the MT7603 interface (wlan0).
Instead, it is used more heavily, by me, on the wlan1 interface. I don't 
think I have AC-capable PC devices, only my phone so far. And being it an 
iPhone, I don't know if I can run iperf on it.
Regarding the wlan0 interface, non-AC, I maybe able to test, but i can not 
make promises. I'll let you know.
Thank you very much guys, again.

-------- my dmesg

BusyBox v1.26.2 () built-in shell (ash)

cose NETGEAR
root@eStorm:~# dmesg
Linux version 4.4.74 (mrkiko@) (gcc version 5.4.0 (LEDE GCC 5.4.0 r4504-eaaba94bf6) ) #0 SMP Fri Jun 30 10:14:10 2017
SoC Type: MediaTek MT7621 ver:1 eco:3
bootconsole [early0] enabled
CPU0 revision is: 0001992f (MIPS 1004Kc)
MIPS: machine is Netgear R6220
Determined physical RAM map:
  memory: 08000000 @ 00000000 (usable)
Initrd not found or empty - disabling initrd
Zone ranges:
   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
   HighMem  empty
Movable zone start for each node
Early memory node ranges
   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
On node 0 totalpages: 32768
free_area_init_node: node 0, pgdat 80479000, node_mem_map 81000000
   Normal zone: 256 pages used for memmap
   Normal zone: 0 pages reserved
   Normal zone: 32768 pages, LIFO batch:7
Reserving 0MB of memory at 0MB for crashkernel
VPE topology {2} total 2
Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
PERCPU: Embedded 10 pages/cpu @8110c000 s8400 r8192 d24368 u40960
pcpu-alloc: s8400 r8192 d24368 u40960 alloc=10*4096
pcpu-alloc: [0] 0 [0] 1 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
PID hash table entries: 512 (order: -1, 2048 bytes)
Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Writing ErrCtl register=0007d440
Readback ErrCtl register=0007d440
Memory: 124588K/131072K available (3665K kernel code, 199K rwdata, 836K rodata, 216K init, 255K bss, 6484K reserved, 0K cma-reserved, 0K highmem)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Hierarchical RCU implementation.
 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS:256
clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... 769 lines cut ...]
br-lan: port 3(wlan1) entered forwarding state
device wlan1 left promiscuous mode
br-lan: port 3(wlan1) entered disabled state
device wlan0 left promiscuous mode
br-lan: port 2(wlan0) entered disabled state
MCU message 40 (seq 7) timed out
IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
MCU message 40 (seq 8) timed out
MCU message 8 (seq 9) timed out
MCU message 40 (seq 10) timed out
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
device wlan1 entered promiscuous mode
device wlan0 entered promiscuous mode
MCU message 8 (seq 11) timed out
MCU message 40 (seq 12) timed out
IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
br-lan: port 3(wlan0) entered forwarding state
br-lan: port 3(wlan0) entered forwarding state
MCU message 40 (seq 13) timed out
IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
br-lan: port 2(wlan1) entered forwarding state
br-lan: port 2(wlan1) entered forwarding state
br-lan: port 3(wlan0) entered forwarding state
br-lan: port 2(wlan1) entered forwarding state
root@eStorm:~#

mrkiko, 2017-07-24:
Anyway, we need to reproduce kernel warning with symbols and maybe some 
debug infos enabled, to make them really useful, I guess.

pparent76, 2017-07-24:
How can we enable debug symbols when cross-compiling lede/kernel/driver?

I think problems with mt76 driver on mt7621 devices is at least partially related to this bug:

https://bugs.lede-project.org/index.php?do=details&task_id=804

camelcamro, 2017-07-24:
It can be...
We are currently testing

pparent76, 2017-07-24:
>It's possible it did crash and nobody noticed

Check your uptime with command "uptime" to know that for sure.

mrkiko, 2017-07-25:
Hello! :)
You can enable most relevant options I guess, in the "make menuconfig" 
menu in LEDE.
You can find them in Global build settings -> Kernel build options .

if you wish and have no problem to unbrick your device (I had problems in 
the past with this), you can also give a spin to
make kernel_menuconfig

if my memory serves me well.
And at that point you can customise the kernel even more, should the need 
to do so arise.
Regarding RCU stalls: I am experiencing the same issues on a x86 system. 
:)

Thank you again for all guys.

mrkiko, 2017-07-25:
thank you very much.
Regarding the system uptime, I am pretty sure it didn't crash
04:01:24 up 23 days, 15:04,  load average: 0.00, 0.00, 0.00

... but I was referring to Wi-Fi crashing (e.g.: getting stuck, and 
recovering the morning after when re-inited).
Sorry for me being unclear.
And thanks.

pparent76, 2017-07-25:
I will see what I can enable in my build.

>Regarding RCU stalls: I am experiencing the same issues on a x86 system.
What do you mean? Which x86 system? Here it's a serious problem leading to crashes and freezes.

mrkiko, 2017-07-25:
In my cases, an EEE PC 1101AHgo system, prints various kernel warnings 
when left alone. Now I can't report them because I don't have the system 
at ready disposal. It's a problem I tried to investigate at some point i 
think, but never came to a conclusion. they didn't seem to have 
particularly bad consequences, still I am not sure.
These appeared only when the system was idle. If the system was not idle, 
then those messages would not appear.
Let me know if you're interested, so we may discuss this via other 
communicaiton methods.

mrkiko, 2017-07-26:
LEDE commit 6acb53c5267b96aa4f166c703998eaad3cb5d9fd .
Great work ... congrats to all of you guys.

camelcamro, 2017-07-26:
we are still testing ....
but first tests what I'm doing shows, that it is now MUCH more stable ...

camelcamro, 2017-07-27:
ok, the new version is much more stable as i tested it now really with loadtests and much more ...
throughput is also better on 5ghz (2,4ghz iperf i need to redo)
anyhow ...

still i got kernel error related on mt76 driver (only on 2,4ghz driver)

`Errors found:
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.171691] ------------[ cut here ]------------
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.176636] WARNING: CPU: 2 PID: 9936 at /home/pparent/tmp/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-07-17-3c4c9a64/mt7603_mac.c:1291 mt7603_mac_work+0x124/0x23c [mt7603e]
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.194683] Modules linked in: option ath9k usb_wwan rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192c_common rtl_usb rtl_pci pppoe nf_nat_pptp nf_conntrack_pptp l2tp_ppp iptable_nat ipt_REJECT ipt_MASQUERADE ath9k_common xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_coova xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NET nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_log_ipv4 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt btcoexist ath9k_hw ath fuse sch_cake sch_fq sch_teql em_nbyte sch_dsmark sWed Jul 26 22:16:51 2017 kern.warn kernel: [38087.363720]  nf_nat_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables pppoatm ppp_generic slhc nfsd nfsv3 nfs msdos ip_gre gre ifb l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel tun vfat fat udf crc_itu_t ntfs lockd sunrpc grace minix isofs hfsplus hfs cramfs configfs cifs autofs4 kafs af_rxrpc dns_resolver br2684 atm fscache nls_utf8 nls_iso8859_1 nls_cp437 sha256_generWed Jul 26 22:16:51 2017 kern.warn kernel: [38087.434447] CPU: 2 PID: 9936 Comm: kworker/u8:0 Not tainted 4.9.37 #0
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.440951] Workqueue: phy0 mt7603_mac_work [mt7603e]
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.445992] Stack : 8cde2e00 804d0000 8047060c 8e596918 00000009 0000050b 8fc04698 8cde2e18
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.454355]         00000088 80069b60 8047060c 00000002 000026d0 80533824 80470578 8cd65d54
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.462699]         804d0000 800a7a38 00000001 80530000 804dff04 804dff08 8047521c 8cd65d54
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.471050]         804d0000 80041634 8fc04698 8cd65d8c 0000b635 00000000 00000001 00d65d74
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.479410]         8e5955c4 8fc85000 8ebdcc80 30796870 00000000 00000000 00000000 00000000
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.487772]         ...
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.490218] Call Trace:
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.492704] [<8000f764>] show_stack+0x54/0x88
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.497087] [<801e75c4>] dump_stack+0x84/0xc0
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.501435] [<8002a3e4>] __warn+0xe4/0x118
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.505514] [<8002a4ac>] warn_slowpath_null+0x1c/0x30
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.510621] [<8e5956e8>] mt7603_mac_work+0x124/0x23c [mt7603e]
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.516485] [<8003f46c>] process_one_work+0x200/0x348
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.521520] [<80040358>] worker_thread+0x2d4/0x45c
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.526309] [<80044ea8>] kthread+0xd8/0xec
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.530395] [<8000a438>] ret_from_kernel_thread+0x14/0x1c
Wed Jul 26 22:16:51 2017 kern.warn kernel: [38087.535906] ---[ end trace 53cc8ab663cdd760 ]---
`

pparent76, 2017-07-27:
@camelcamro 
Did you see any rcu_sched yet?

camelcamro, 2017-07-27:
rcu_sched errors seems to be done ..
NO rcu_sched ERRORS any more :)
=> tested with: 
.) inclusive traffic shapping basic and cake
.) tested with LAN, WLAN 2,4Ghz, WLAN 5GHz, multiple SSID

advantage: now also higher throughput on wlan 5.0 Ghz verified
(still need to measure the theoretical throughput with iperf)

2,4GHz performance not tested yet, but i can use also ht20 and ht40 ..before it was only possible to use "legacy mode"  

so, in any case a big advantage now with current trunk snapshot (>= Thu Jul 27 )

slthomason, 2017-07-27:
I have been following this work.  I am working on this hardware: https://lede-project.org/toh/hwdata/zbt/zbt_we1326 which is very similar to what you are working on.  Even on the latest trunk build with these recent changes, I am still getting a lot of problems with the 2G wifi.

I very often get the mt7603_mac_work errors you get above.  However, I cannot get 2G to run stable.  When I run an iperf test against the wireless, withing 60 seconds it will stop sending data and the iperf test fails.  The 2G will stay connected.  DHCP works, but ping to the AP fails, and no data will go over the connection, but the connection stays connected.

Any ideas?  Are you seeing the same?  My current project is using this WE1326, but I am planning to use the WG3526 as well, and seem to get the same results on it as well.

On both models, the 5G seems to be very, very stable.  But on the 2G I am getting a lot of these problems.

camelcamro, 2017-07-27:
Pls let me test 2,4 ghz wifi with newest lede trunk version.
Tomorrow, should be some other 2,4ghz patches also be included in.build ( dts+etc) ...
Then I week retest 2,4ghz ok the different modes.
Most stable is 2,4ghz with legacy mode.
I will keep you up2date

slthomason, 2017-07-27:
I tested both with ht40 and ht20.  Is that what you mean by "legacy"?

slthomason, 2017-07-27:
Ok - I will be testing the WG3526 and the WE1326 both in legacy mode in the meantime.  I eagerly await your patches.  But it does seem that the legacy mode (set as "NONE") seems more stable.  I still see ups and downs in iperf, but ... when I test with HT20 or HT40, it stays connected, but iperf test will crash and if trying to test streaming data, the stream will stop, even tho 2G stays connected.

Look forward to your changes and I can help test on both hardware!

slthomason, 2017-08-17:
@camelcamro any luck?  With mine set to NONE I am stable about 98% of the time (so much better) but not 100% stable, and I still would like to move it to using HT40.  Any leads?

pparent76, 2017-08-21:
What do you mean stable 98% of the time? Do you mean there are 2% when the wifi does not work, but then it starts working again, on its own? Do you experience crash/bug that requires reboot?

slthomason, 2017-08-21:
@pparent76  When I use HT40 or HT20 on the 2G, I get the mt7603_mac_work errors within minutes.  When using NONE, I get the mt7603_mac_work errors usually between 16 to 24 hours.  The only way to recover the router after that is to reboot.  This is on the WE1326, which is essentially a stripped down model of the WG3526.  Same chipsets.  Are you guys not experiencing the same?

camelcamro, 2017-08-21:
also having same issues, as already told mt76x is still buggy.
there were some improements done, but for real production it is still buggy :(
i have a script, which is checking the logs for kernel errors, and then doing a reboot.
as if the errors happen, mostly i reboot is needed :(

we can only hope & pray, that someone of the expert devs are having time to dig into more in detail

kofec, 2017-08-21:
@camelcamro 
can you share your script ? Do you know how often is executed ? Do you try restart wifi (or any other recovery actions) or always reboot ?

pparent76, 2017-08-21:
If you rmmod and insmod the module, does it work? Or only reboot works?

mrkiko, 2017-08-21:
In my case, with R6220, these errors didn't break wi-fi I think.
Or at least - nobody told me.

slthomason, 2017-08-21:
Mine breaks 2G wifi only.  5G still works great.

camelcamro, 2017-08-22:
it is nothing special ....

in my case, I'M checking all 5min for errors in logs, and then restart FW + email
if wanted you can add "reboot" cmd, too

#email + ssl (~1MB) - also see: https://wiki.openwrt.org/doc/howto/smtp.client
opkg install libopenssl ssmtp #--force-reinstall

then via cron job call (below) script all 5min ...

####
`file: /etc/check_ERRORS.sh

#!/bin/sh

# Script that checks for occurrence of error message and email it
#
EMAIL_RECEIVER="youremail@gmail.com"
TMP_EMAIL=/tmp/email_USB.txt

CHECKCOUNT=$(awk -v d1="$(date --date="-6 min" "+%a %b %_d %H:%M:%S %Y")" -v d2="$(date "+%a %b %_d %H:%M:%S %Y")" '$0 > d1 && $0 < d2 || $0 ~ d2' /var/log/messages.log | grep -i "kern\.warn kernel" | grep -ic "Call Trace")

if [ $CHECKCOUNT -gt 0 ]; then
    CHECK=$(awk -v d1="$(date --date="-6 min" "+%a %b %_d %H:%M:%S %Y")" -v d2="$(date "+%a %b %_d %H:%M:%S %Y")" '$0 > d1 && $0 < d2 || $0 ~ d2' /var/log/messages.log | grep -B 20 -A 20 -i "Call Trace")
    DATE=`date +%Y-%m-%d\ %H:%M:%S`
    ROUTER_IP=`wget http://ipinfo.io/ip -qO -`
    echo "To: $EMAIL_RECEIVER" > $TMP_EMAIL
    echo "From: errorcheck@localhost" >> $TMP_EMAIL
    echo "Subject: [${HOSTNAME}] KERNEL error/warnings issue - $DATE" >> $TMP_EMAIL
    echo "" >> $TMP_EMAIL
    echo "Please check ${HOSTNAME} (IP: ${ROUTER_IP})" >> $TMP_EMAIL
    echo "Errors found:" >> $TMP_EMAIL
    echo "$CHECK" >> $TMP_EMAIL
    ssmtp $EMAIL_RECEIVER < $TMP_EMAIL
    rm -f $TMP_EMAIL &> /dev/null

    echo "Subject: [${HOSTNAME}] KERNEL error/warnings issue - $DATE"
    echo "$CHECK"
    logger "Subject: [${HOSTNAME}] KERNEL error/warnings issue - $DATE"

    #restart FW to have new rules after a kernel error
    /etc/init.d/firewall restart
fi
`

pparent76, 2017-08-22:
So restarting firewall, makes the wifi work again?

kofec, 2017-08-22:
I think that wifi is recovering automatically but  FW restart is just for additional logs - to separate crashes

pparent76, 2017-09-13:
>On both models, the 5G seems to be very, very stable

I've been testing 5Ghz wifi card alone. It does not seem to work at all for me. The ssid is brodcasted, i can connect to it, but then I get a very very low trhoughput, and lot of package loss. It's very unusable.

I'm wondering if I do something incorect since some of your reported it working.

Here is my hostapd fire:

```
country_code=FR
interface=wlan1
hw_mode=a
ieee80211d=1
ieee80211h=1
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ssid=Test 5ghz
bridge=br-lan
```

pparent76, 2017-09-13:
Concerning 2.4Ghz error it seems they are triggered by this portion of code:

(mt7603_mac.c:1279)
```
if (WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->tx_dma_check,
					       RESET_CAUSE_TX_BUSY,
					       mt7603_tx_dma_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->rx_dma_check,
					       RESET_CAUSE_RX_BUSY,
					       mt7603_rx_dma_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->tx_hang_check,
					       RESET_CAUSE_TX_HANG,
					       mt7603_tx_hang)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->rx_pse_check,
					       RESET_CAUSE_RX_PSE_BUSY,
					       mt7603_rx_pse_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->beacon_check,
					       RESET_CAUSE_BEACON_STUCK,
					       NULL))) {
		dev->beacon_check = 0;
		dev->tx_dma_check = 0;
		dev->tx_hang_check = 0;
		dev->rx_dma_check = 0;
		dev->rx_pse_check = 0;
		dev->rx_dma_idx = ~0;
		memset(dev->tx_dma_idx, 0xff, sizeof(dev->tx_dma_idx));
		reset = true;
	}
```

If I understand it will, whenever the warning happens, the driver resets the wifi card.

MaFeSa, 2017-11-30:
Seems fixed with latest commit.
Thanks

MaFeSa, 2017-12-04:
For me is solved with the latest commit. 
I no longer have this error on Xiaomi Mi Router 3G

closed 2017-12-04 by nbd168

camelcamro, 2017-12-04:
well, I would also like to test, but all mt7621 images are failed to be built (since 22.11.2017 no snapshot in trunk :(

MaFeSa, 2017-12-04:
> all mt7621 images are failed to be built
No problems for me, try `make clean` before `make`

camelcamro, 2017-12-04:
I meant the snapshot images on lede project download trunk directory, as I don't compile them by myself :(

nbd168, 2017-12-04:
I've fixed the build error and there's a new build now.

[3 more replies]