openwrt/mt76 #107: Very poorly working 2.4 GHz, on ZBT-WG3526

DmtryJS, 2017-06-27:
Hello. 

Im use LEDE trank LEDE Reboot SNAPSHOT r4466-f80963d (270617) 16mb

package version 
kmod-mt76-core - 4.4.74+2017-06-16-70d22512-1
kmod-mt7603 - 4.4.74+2017-06-16-70d22512-1
kmod-mt76x2 - 4.4.74+2017-06-16-70d22512-1

n - mode, WPA2-PSK

2.4 GHz iperf messuring 

Connecting to host 192.168.1.33, port 5201
[  4] local 192.168.1.200 port 30842 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  1.72 MBytes  14.4 Mbits/sec
[  4]   1.00-2.00   sec   819 KBytes  6.71 Mbits/sec
[  4]   2.00-3.00   sec   756 KBytes  6.20 Mbits/sec
[  4]   3.00-4.00   sec  1.85 MBytes  15.5 Mbits/sec
[  4]   4.00-5.00   sec   945 KBytes  7.74 Mbits/sec
[  4]   5.00-6.00   sec  1.54 MBytes  12.9 Mbits/sec
[  4]   6.00-7.00   sec   945 KBytes  7.74 Mbits/sec
[  4]   7.00-8.00   sec  1.35 MBytes  11.4 Mbits/sec
[  4]   8.00-9.00   sec  1.23 MBytes  10.3 Mbits/sec
[  4]   9.00-10.00  sec   819 KBytes  6.71 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  11.9 MBytes  9.96 Mbits/sec                  sender
[  4]   0.00-10.00  sec  11.8 MBytes  9.86 Mbits/sec                  receiver

iperf Done.

The distance between the laptop and the router is 0.5 m, the channel is free

Ready to test day and night, just please fix it )

psyborg55, 2017-06-30:
convert mt76 driver to rt2x00 and continue from there..

DmtryJS, 2017-06-30:
Sorry, I did not understand. )

4 days ago there were changes in the code, today I will try to test, maybe something has changed.

DmtryJS, 2017-07-03:
The version of packages in the repository remained the same, nothing in the work has changed.(

DmtryJS, 2017-07-23:
Hello. 
I'm testing a new version of the driver.

kmod-mt76-core - 4.9.37+2017-07-17-3c4c9a64-1
kmod-mt7603 - 4.9.37+2017-07-17-3c4c9a64-1
kmod-mt76x2 - 4.9.37+2017-07-17-3c4c9a64-1

Now the speed is more stable than before, at least this is an honest 'g' standard, those honest 54 Mbit / s connection ...
But still the connection speed does not rise above 72 Mbit / s.  I'm glad that the correction of bugs little by little happens. 
 I hope that all the same it will turn out to overcome bugs and raise the connection speed to 150) thanks.

nbd168, 2017-12-03:
Should be fixed in the latest version, please test.

MaFeSa, 2017-12-04:
Could it be related to this?
https://github.com/openwrt/mt76/issues/129

MiltonLai, 2017-12-22:
2.4GHz is not working on Newifi-D2 (the hardware specs are mt7621at+mt7603en+mt7612en). Totally no traffics after the client device connected.

MaFeSa, 2017-12-22:
> Totally no traffics after the client device connected.

I do not think it's tied to the wireless driver, maybe it's a wrong switch / vlan configuration.

MiltonLai, 2017-12-22:
It's a fresh install with default configurations. The 5GHz wireless work well so I don't think it a vlan/switch configuration problem.
Maybe I didn't describe the issue correctly, there should be a few traffics go through as I can see in ifconfig command results. But the connection is really slow, I can't even open Luci interface from the client device.

MaFeSa, 2018-01-25:
@nbd168 
Should be fixed in the latest version because hardware is similar to Xiaomi 3G (MT7603+MT7612).

barhom, 2018-02-10:
please close this issue and open a new one if problems still exists on your target (wg3526)