openwrt/mt76 #110: Do you consider adding MT7615 support?

guoxiaoqiao, 2017-07-04:
I'm using Phicomm K2P, which have two MT7615DN wireless chip, currently It is work fine with LEDE except no wireless, Is MT7615 similar to MT76x2? or can you try adding support?

[Device list of MT7615 on wikidevi](http://wikidevi.com/wiki/Special:Ask?title=Special%3AAsk&q=%3Cq%3E%3Cq%3E%5B%5BWI1+chip1+model::~MT7615*%5D%5D+%5B%5BWI1+chip1+model::!~MT7615X*%5D%5D%3C%2Fq%3E+OR+%3Cq%3E%5B%5BWI2+chip1+model::~MT7615*%5D%5D+%5B%5BWI2+chip1+model::!~MT7615X*%5D%5D%3C%2Fq%3E%3C%2Fq%3E&po=%3FEmbedded+system+type=Type%0D%0A%3FFCC+ID%0D%0A%3FManuf%0D%0A%3FManuf+product+model=Manuf.+mdl%0D%0A%3FCPU1+model=CPU1%0D%0A%3FCPU1+clock+speed%0D%0A%3FFLA1+amount=FLA1%0D%0A%3FFLA2+amount=FLA2%0D%0A%3FRAM1+amount=RAM1%0D%0A%3FWI1+chip1+model=WI1+chip1%0D%0A%3FWI1+MIMO+config=WI1+MIMO%0D%0A%3FWI2+chip1+model=WI2+chip1%0D%0A%3FWI2+MIMO+config=WI2+MIMO%0D%0A%3FSupported+802dot11+protocols=PHY+modes%0D%0A%3FOUI%0D%0A%3FOUI+(ethernet)=OUI+(Eth)%0D%0A%3FEstimated+year+of+release=Est.+year&eq=yes&p%5Bformat%5D=broadtable&order%5B0%5D=ASC&sort_num=&order_num=ASC&p%5Blimit%5D=500&p%5Boffset%5D=&p%5Blink%5D=all&p%5Bsort%5D=&p%5Bheaders%5D=show&p%5Bmainlabel%5D=&p%5Bintro%5D=&p%5Boutro%5D=&p%5Bsearchlabel%5D=%E2%80%A6+further+results&p%5Bdefault%5D=&p%5Bclass%5D=sortable+wikitable+smwtable)

nbd168, 2017-07-04:
MT7615 is very different from MT76x2, but I will be working on adding support for it.

guoxiaoqiao, 2017-07-04:
Looking forward to your results, and if necessary, I can provide a Phicomm K2P for your development work.

LiranV, 2017-09-24:
@nbd168 Thank you! That would be great!

closed 2017-10-13 by nbd168

ptpt52, 2017-10-17:
@nbd168 hi, is the code for MT7615 get ready now? I am working on it now. 
Hope we can communicate more

nbd168, 2017-10-17:
Interesting. What exactly are you working on?

ptpt52, 2017-10-17:
a driver for MT7615 wireless chip
How is your progress?

nbd168, 2017-10-17:
I haven't started yet, since I'm still waiting for hardware information

robimarko, 2017-10-27:
It looks someone may have leaked official driver here
https://github.com/dt99/mt7615

MaFeSa, 2017-11-08:
@nbd168
Hi, if you need I can send you a Xiaomi R3P or D-Link DIR-882

NishaYume, 2017-11-11:
Hi,Is expected the support? Im thinking about buyin the r3p xiaomi..if yes

aKazancev96, 2018-03-06:
Hi! Is there someone who try to use with that https://github.com/dt99/mt7615
I`m working on it and have some result. 2.4 and 5 AP is UP, but there are some troubles with BcnCheck and status LEDs are off. My 
I'll be grateful for any help or ideas.

> BcnCheck: no bcn occurs within 15 sec for band 0(ra0)!!
> BcnCheck: bcn recover for band 0(ra0)!!

ti9errr, 2018-05-22:
@aKazancev96, create a ticket on github and see if the developer responds.

zhangwju, 2018-07-24:
@aKazancev96  can share how to solved? i aslo same problem
Thanks!

luuhungit, 2018-09-13:
I like Xiaomi wifi devices, It has high performance hardware with very cheap price.

haswelliris, 2018-12-27:
I found [a recent driver source for mt7615](https://drive.google.com/open?id=1Ck7QHCUxFkSropZ4pyVP70HGW_yoGQSg ). Maybe you can port it to mt76
@nbd168 @LorenzoBianconi @sgruszka @robimarko

OpenGG, 2019-06-25:
Ryder Lee from MediaTek and others have been working on mt7615 since Apr 25, 2019.
What is the status now, has anyone tried it out?

thomassth, 2020-06-16:
Any update on the dev?
Also, since the chipset support has not materialized yet, shouldn't the issue be "Open" not "Closed"?

intelfx, 2020-06-16:
mt7615 is long supported with this driver (albeit the quality of this support is less than stellar, at least with R3P).