openwrt/mt76 #112: WiFi does not work on mt7688!!!

Microdrive, 2017-07-11:
I'm testing WiFi driver mt76 for hkl-7688. The distribution is lede 17.1.1 (linux 4.4.61)

The following problems are identified:

1) The network will not start if the option channel 'auto' is set.

```
wifi up
[ 8699.812482] device wlan0 left promiscuous mode
[ 8699.821369] br-lan: port 2(wlan0) entered disabled state
[ 8701.412245] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 8701.428678] device wlan0 entered promiscuous mode
[ 8708.911262] device wlan0 left promiscuous mode
[ 8708.920242] br-lan: port 2(wlan0) entered disabled state
[ 8710.098096] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 8710.168029] device wlan0 entered promiscuous mode
```

```
If you set option channel '2' then the network starts up
[ 9016.293622] device wlan0 entered promiscuous mode
[ 9016.816045] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 9016.828872] br-lan: port 2(wlan0) entered forwarding state
[ 9016.839810] br-lan: port 2(wlan0) entered forwarding state
[ 9018.838318] br-lan: port 2(wlan0) entered forwarding state

```
2) Sometimes when you try to connect, the kernel crashes. This does not always happen and it is difficult to catch:

```
[10043.448483] br-lan: port 2(wlan0) entered disabled state
[10045.839623] ------------[ cut here ]------------
[10045.848860] WARNING: CPU: 0 PID: 0 at /wrt_lede/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7688/mt76-2017-01-31-3c8caafc/mt7603_mac.c:403 mt7603_mac_fill_rx+0x2cc/0x3b4 [mt7603e]()
[10045.883816] Modules linked in: qcserial pppoe ppp_async option iptable_nat usb_wwan qmi_wwan pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG usbserial usbnet slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt cdc_wdm cdc_acm mt76x2e mt7603e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables eeprom_93cx6 leds_gpio dwc2 ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii
[10046.026195] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.61 #0
[10046.037573] Stack : 80355c14 00000000 00000001 803a0000 8039b0fc 8039ad83 80337560 00000000
          803e3438 00000001 00000001 871769c0 87fe1878 8004ae38 8033ca68 80390000
          00000003 00000001 8033ae6c 87c0ddec 87fe1878 80048da4 00000020 00000000
          00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[10046.108002] Call Trace:
[10046.112843] [<8001486c>] show_stack+0x50/0x84
[10046.121479] [<80025418>] warn_slowpath_common+0xa0/0xd0
[10046.131832] [<800254d0>] warn_slowpath_null+0x18/0x24
[10046.141849] [<874e3d70>] mt7603_mac_fill_rx+0x2cc/0x3b4 [mt7603e]
[10046.153937] [<874e2b50>] mt7603_queue_rx_skb+0xb4/0xfc [mt7603e]
[10046.165853] [<874e8fd8>] mt76_dma_attach+0xd08/0xeec [mt76]
[10046.176890]
[10046.179830] ---[ end trace 299f78472b7bd924 ]---
[10046.189095] ------------[ cut here ]------------
[10046.198412] WARNING: CPU: 0 PID: 0 at compat-wireless-2017-01-31/net/mac80211/rx.c:4224 ieee80211_rx_napi+0xa8/0x8b0 [mac80211]()
[10046.221491] Modules linked in: qcserial pppoe ppp_async option iptable_nat usb_wwan qmi_wwan pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG usbserial usbnet slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt cdc_wdm cdc_acm mt76x2e mt7603e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables eeprom_93cx6 leds_gpio dwc2 ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii
[10046.363865] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.61 #0
[10046.377650] Stack : 80355c14 00000000 00000001 803a0000 8039b0fc 8039ad83 80337560 00000000
          803e3438 877de000 00000000 00000001 87fe1878 8004ae38 8033ca68 80390000
          00000003 877de000 8033ae6c 87c0dda4 87fe1878 80048da4 00000001 00000000
          00000119 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[10046.448066] Call Trace:
[10046.452903] [<8001486c>] show_stack+0x50/0x84
[10046.461536] [<80025418>] warn_slowpath_common+0xa0/0xd0
[10046.471888] [<800254d0>] warn_slowpath_null+0x18/0x24
[10046.481994] [<8762591c>] ieee80211_rx_napi+0xa8/0x8b0 [mac80211]
[10046.494002] [<874e98ec>] mt76_rx_complete+0x80/0x108 [mt76]
[10046.505057] [<874e90e8>] mt76_dma_attach+0xe18/0xeec [mt76]
[10046.516095]
[10046.519035] ---[ end trace 299f78472b7bd925 ]---
[10046.541580] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[... 85 lines cut ...]
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000020  00 00 00 00 20 00 00 00  1c 88 79 52 25 cc 1c 88  |.... .....yR%...|
00000030  79 52 25 cd 22 34 00 20  ff ff 00 01 00 00 00 00  |yR%."4. ........|
00000040  00 00 22 00 00 00 00 00  30 00 00 00 00 00 00 00  |..".....0.......|
00000050  82 00 00 94 40 b6 c0 ca  22 84 83 81 40 ca 22 00  |....@..."...@.".|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  c6 c6 c4 c4 c4 c0 c0 c4  c4 c4 c4 c4 c0 c0 00 00  |................|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000c0  00 00 00 00 00 00 00 00  00 1a 22 2a 31 35 01 35  |.........."*15.5|
000000d0  39 40 46 4d 7f 7f 7f 00  00 00 00 00 00 00 00 00  |9@FM............|
000000e0  11 1d 11 1d 1c 35 1c 35  1e 35 1e 35 17 19 17 19  |.....5.5.5.5....|
000000f0  02 00 00 00 99 80 80 88  0a 00 00 00 00 00 00 00  |................|
00000100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 77 00  |..............w.|
00000130  11 1d 11 1d 15 7f 15 7f  17 7f 17 7f 10 3b 10 3b  |.............;.;|
00000140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00010000
```

psyborg55, 2017-07-22:
your config has         option htmode 'HT20'

change that to HT40 and add option noscan '1'

does it work then in HT40?

Microdrive, 2017-07-24:
Yes, it works!

I do not understand why this did not work for the first time, I definitely installed noscann = 1.

benpicco, 2017-08-04:
I still see those crashes with `HT40` and `noscan '1'` when there is some load on the WiFi for a prolonged time.

The WiFi doesn't recover from that, it stays disconnected until I reboot the device.

imwsy, 2017-08-05:
same as #115 
using netgear r6120 with mt7628an

nbd168, 2017-12-03:
Duplicate issue

closed 2017-12-03 by nbd168