openwrt/mt76 #114: STA XX:XX:XX:XX:XX:XX IEEE 802.11: did not acknowledge authentication response

mrkiko, 2017-07-14:
My Netgear Mediatek 7603 device (in a Netgear R6220 AP) gives this error 
when trying to connect with an Atheros AR9285 Rev 2 device.
this worked at some point, but after restoring the device to original FW 
to unbrick it, and re-flashing LEDE it didn't work. Still, the 
coupled MT7612 works fine, and the MT7603 device works fine with another 
nearer laptop.

Driver is at commit cf4ba1287139835c4a4abc7c744802503fb30659 .

Thank you for your great work.

nbd168, 2017-12-03:
Please try the latest version

Lu-Fi, 2018-03-03:
This also occur in latest versions.

`Sat Mar  3 09:03:01 2018 user.info adblock-[3.5.1]: blocklist with overall 46818 domains loaded successfully (Xiaomi Mi Router 3G, OpenWrt SNAPSHOT r6351-694f0bb)
Sat Mar  3 09:04:58 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: authenticated
Sat Mar  3 09:05:04 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: disassociated
Sat Mar  3 09:05:05 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Mar  3 09:05:26 2018 daemon.notice hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: did not acknowledge authentication response
Sat Mar  3 09:05:26 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: disassociated
Sat Mar  3 09:05:27 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Mar  3 09:07:11 2018 daemon.notice hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: did not acknowledge authentication response
Sat Mar  3 09:07:17 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: disassociated
Sat Mar  3 09:07:18 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Mar  3 09:07:32 2018 daemon.notice hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: did not acknowledge authentication response
Sat Mar  3 09:07:38 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: disassociated
Sat Mar  3 09:07:39 2018 daemon.info hostapd: wlan0-1: STA 28:ed:6a:a4:de:5c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Mar  3 09:07:46 2018 daemon.info hostapd: wlan1: STA 28:ed:6a:a4:de:5c IEEE 802.11: authenticated
Sat Mar  3 09:07:46 2018 daemon.info hostapd: wlan1: STA 28:ed:6a:a4:de:5c IEEE 802.11: associated (aid 4)`

nanpuyue, 2018-03-30:
```
Fri Mar 30 09:59:26 2018 daemon.notice hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Mar 30 09:59:37 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: disassociated
Fri Mar 30 09:59:37 2018 daemon.notice hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Mar 30 09:59:41 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: disassociated due to inactivity
Fri Mar 30 09:59:42 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Mar 30 10:00:17 2018 daemon.notice hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Mar 30 10:00:27 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: disassociated
Fri Mar 30 10:00:27 2018 daemon.notice hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Mar 30 10:00:31 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: disassociated due to inactivity
Fri Mar 30 10:00:32 2018 daemon.info hostapd: wlan0: STA 5c:52:1e:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

nanpuyue, 2018-03-30:
I use `hostapd` instead of `wpad`.

deverton, 2018-05-23:
I'm seeing this issue a lot with one particular client on the network. What would help to diagnose this further? I have some packet captures of the authentication process and it looks like the first three steps of the authentication process complete but the fourth one is either not transmitted, not received, or the driver doesn't seem to be notifying hostapd.

nbd168, 2018-10-25:
Please try the latest version (master only, not 18.06).

deverton, 2018-10-26:
Updating from  `OpenWrt 18.06.0 r7188-b0b5c64c22` to `OpenWrt SNAPSHOT r8380-6026356` and I no longer have this issue. Previously it would take the one bad client on my network 60+ minutes to reconnect and now it's instance.

Thanks!

nbd168, 2018-10-26:
Nice, thanks for testing.

closed 2018-10-26 by nbd168