openwrt/mt76 #115: mt7628 crash very often

imwsy, 2017-08-04:
When the flow exceeds 3MiB / s, the entire wifi will drop

Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.324192] ------------[ cut here ]------------
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.333397] WARNING: CPU: 0 PID: 3737 at /home/ubuntu/LEDE/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7628/mt76-2017-07-17-3c4c9a64/mt7603_mac.c:1279 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.373353] Modules linked in: pppoe ppp_async pppox ppp_generic iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipFri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.513509]  compat xt_set x_tables ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ifb leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.588198] CPU: 0 PID: 3737 Comm: kworker/u2:2 Not tainted 4.9.40 #0
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.600980] Workqueue: phy0 0x8317d4f8 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.612878] Stack : 83803414 831abf18 00000088 8004d0a0 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.629452]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.646026]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.662591]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.679161]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.695744]         ...
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.700575] Call Trace:
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.705108] [<8004d0a0>] 0x8004d0a0
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.712027] [<8000df6c>] 0x8000df6c
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.718935] [<8000df6c>] 0x8000df6c
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.725839] [<80023ac0>] 0x80023ac0
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.732761] [<8317d624>] 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.744151] [<80023b88>] 0x80023b88
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.751057] [<80040a3c>] 0x80040a3c
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.757996] [<8317d624>] 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.769372] [<802ef63c>] 0x802ef63c
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.776283] [<80036ca0>] 0x80036ca0
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.783211] [<80036958>] 0x80036958
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.790206] [<80037704>] 0x80037704
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.797140] [<80037458>] 0x80037458
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.804057] [<80037458>] 0x80037458
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.810968] [<8003bf04>] 0x8003bf04
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.817953] [<8003be2c>] 0x8003be2c
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.824879] [<80009278>] 0x80009278
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.831774] 
Fri Aug  4 09:53:49 2017 kern.warn kernel: [ 1295.834721] ---[ end trace 30431f754926fd75 ]---
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.644263] ------------[ cut here ]------------
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.653472] WARNING: CPU: 0 PID: 6049 at /home/ubuntu/LEDE/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7628/mt76-2017-07-17-3c4c9a64/mt7603_mac.c:1291 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.693443] Modules linked in: pppoe ppp_async pppox ppp_generic iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipFri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.833566]  compat xt_set x_tables ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ifb leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.908266] CPU: 0 PID: 6049 Comm: kworker/u2:3 Tainted: G        W       4.9.40 #0
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.923451] Workqueue: phy0 0x8317d4f8 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.935345] Stack : 83803414 8301af98 00000088 8004d0a0 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.951919]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.968495]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1299.985075]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.001644]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.018228]         ...
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.023057] Call Trace:
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.027594] [<8004d0a0>] 0x8004d0a0
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.034505] [<8000df6c>] 0x8000df6c
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.041399] [<8000df6c>] 0x8000df6c
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.048306] [<80023ac0>] 0x80023ac0
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.055236] [<8317d624>] 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.066627] [<80023b88>] 0x80023b88
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.073532] [<80043adc>] 0x80043adc
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.080459] [<8317d624>] 0x8317d624 [mt7603e@83178000+0x6cc0]
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.091854] [<802ef664>] 0x802ef664
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.098764] [<80036ca0>] 0x80036ca0
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.105700] [<80036958>] 0x80036958
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.112679] [<80037704>] 0x80037704
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.119608] [<80037458>] 0x80037458
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.126538] [<80037458>] 0x80037458
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.133432] [<8003bf04>] 0x8003bf04
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.140353] [<8003be2c>] 0x8003be2c
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.147284] [<80009278>] 0x80009278
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.154181] 
Fri Aug  4 09:53:54 2017 kern.warn kernel: [ 1300.157134] ---[ end trace 30431f754926fd76 ]---

braekling, 2017-08-14:
Hi everbody,

I run into the same issue:

```
[83552.490772] ------------[ cut here ]------------
[83552.495531] WARNING: CPU: 0 PID: 215 at /home/buildbot/slave-lede-local/ramips_mt7628/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7628/mt76-2017-07-17-3c4c9a64/mt7603_mac.c:1291 mt7603_mac_work+0x12c/0x248 [mt7603e]
[83552.515937] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt mt76x2e mt7603e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio gpio_button_hotplug
[83552.574277] CPU: 0 PID: 215 Comm: kworker/u2:2 Not tainted 4.9.40 #0
[83552.580745] Workqueue: phy0 mt7603_mac_work [mt7603e]
[83552.585867] Stack : 83803414 839d6098 00000088 8004d580 8382f0ec 80401307 803b0de0 000000d7
[83552.594390]         803b0d4c 839e7d64 80400000 8007c61c 00000088 8004d580 803b65a4 80400000
[83552.602908]         00000003 839e7d64 80400000 80038dd4 00000088 839e7d9c 000000af 00000000
[83552.611427]         00000000 00000000 830954f8 83b60a00 83b60900 30796870 00000000 00000000
[83552.619926]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[83552.628438]         ...
[83552.630940] Call Trace:
[83552.633444] [<8000e448>] show_stack+0x54/0x88
[83552.637888] [<80023fa0>] __warn+0xe4/0x118
[83552.642063] [<80024068>] warn_slowpath_null+0x1c/0x34
[83552.647201] [<83095624>] mt7603_mac_work+0x12c/0x248 [mt7603e]
[83552.653152] [<80036e38>] process_one_work+0x1ec/0x32c
[83552.658280] [<80037be4>] worker_thread+0x2ac/0x420
[83552.663156] [<8003c3e4>] kthread+0xd8/0xec
[83552.667313] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[83552.672802] ---[ end trace dffbe3c49a08f412 ]---
[83555.610487] MCU message 40 (seq 15) timed out
[83556.650566] MCU message 8 (seq 1) timed out
[83557.690481] MCU message 40 (seq 2) timed out
[83557.694909] ------------[ cut here ]------------
[83557.699643] WARNING: CPU: 0 PID: 215 at /home/buildbot/slave-lede-local/ramips_mt7628/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7628/mt76-2017-07-17-3c4c9a64/mt7603_mac.c:1279 mt7603_mac_work+0x12c/0x248 [mt7603e]
[83557.720032] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt mt76x2e mt7603e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio gpio_button_hotplug
[83557.778356] CPU: 0 PID: 215 Comm: kworker/u2:2 Tainted: G        W       4.9.40 #0
[83557.786058] Workqueue: phy0 mt7603_mac_work [mt7603e]
[83557.791193] Stack : 83803414 839d6098 00000088 8004d580 8382f0ec 80401307 803b0de0 000000d7
[83557.799694]         803b0d4c 839e7d64 80400000 8007c61c 00000088 8004d580 803b65a4 80400000
[83557.808209]         00000003 839e7d64 80400000 80038dd4 00000088 839e7d9c 000000c7 00000000
[83557.816725]         80413110 00000000 830954f8 83b60a00 83b60900 30796870 00000000 00000000
[83557.825241]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[83557.833752]         ...
[83557.836236] Call Trace:
[83557.838740] [<8000e448>] show_stack+0x54/0x88
[83557.843195] [<80023fa0>] __warn+0xe4/0x118
[83557.847356] [<80024068>] warn_slowpath_null+0x1c/0x34
[83557.852507] [<83095624>] mt7603_mac_work+0x12c/0x248 [mt7603e]
[83557.858442] [<80036e38>] process_one_work+0x1ec/0x32c
[83557.863586] [<80037be4>] worker_thread+0x2ac/0x420
[83557.868447] [<8003c3e4>] kthread+0xd8/0xec
[83557.872620] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[83557.878094] ---[ end trace dffbe3c49a08f413 ]---
```

I'm running a TP-Link TL-WR841N **v13**, MT7628NN, see https://wikidevi.com/wiki/TP-LINK_TL-WR841N_v13.x I'm running the LEDE Reboot SNAPSHOT r4684-0b7f760 / LuCI Master (git-17.220.63907-bbcf73f). Snapshot because it's the first LEDE version which supports my WR841N v13.

If you need further information, please feel free to ask.

Any help would be appreciated - thx a lot!

braekling, 2017-08-17:
After switching to Legacy mode (instead of default N), I got no more kernel warnings until now. Apparently, the connection isn't interrupted anymore, although requests still seem to be quite slow.

p4u, 2017-08-21:
I'm using  LinkIt Smart 7688 (mt7688) with Reboot (SNAPSHOT, r4707-3e6d303) downloaded from lede site. Just connected to the device to enable WiFi (set disabled to 0) and inmediatly got crashes:

[dmesg.txt](https://github.com/openwrt/mt76/files/1239940/dmesg.txt)
[logread.txt](https://github.com/openwrt/mt76/files/1239942/logread.txt)

WiFi is not usable at all.

nbd168, 2017-12-03:
Please try the latest version

imwsy, 2017-12-04:
machine info:

root@LEDE:/# cat /proc/cpuinfo
system type             : MediaTek MT7628AN ver:1 eco:2
machine                 : Netgear R6020
processor               : 0
cpu model               : MIPS 24KEc V5.5
BogoMIPS                : 385.84
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

wireless settings:

root@LEDE:/# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/10300000.wmac'
wireless.radio0.disabled='0'
wireless.radio0.country='CN'
wireless.radio0.channel='4'
wireless.radio0.htmode='HT40'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.wps_pushbutton='0'
wireless.default_radio0.encryption='psk'
wireless.default_radio0.disassoc_low_ack='0'
wireless.default_radio0.wpa_group_rekey='600'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='0'
wireless.radio1.country='CN'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.encryption='psk'
wireless.default_radio1.wps_pushbutton='0'
wireless.default_radio1.disassoc_low_ack='0'
wireless.default_radio1.wpa_group_rekey='600'

first test:
after running iperf about 3500 seconds wifi will disconnect; after waiting for a minute can reconnect, dmesg have no error message (only MCU message 40 (seq 15) timed out)

second test:

[  4] 112.5-113.0 sec  3.24 MBytes  54.4 Mbits/sec
[  4] 113.0-113.5 sec  3.40 MBytes  57.1 Mbits/sec
[ 5302.617929] ------------[ cut here ]------------
[ 5302.627176] WARNING: CPU: 0 PID: 0 at /home/ubuntu/LEDE/targets/mt7628-snapshot/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2017-12-03-8e17d363/mt7603_mac.c:384 mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[ 5302.667124] Modules linked in: pppoe ppp_async pppox ppp_generic mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack
xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mang
le iptable_filter ip_tables crc_ccitt compat xt_set x_tables ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_s
et_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink leds_gpio gpio_button_hotplug
[ 5302.817285] CPU: 0 PID: 0 Comm: swapper Not tainted 4.9.65 #0
[ 5302.828662] Stack : 8041751a 00000031 00000000 00000001 803c5034 803c4ca7 80378570 00000000
[ 5302.845242]         80413660 00000180 00000001 00000000 83179908 8004d8c0 8037dd48 803c0000
[ 5302.861812]         00000003 00000180 8037bf10 8380ddd4 83179908 8007cf2c 8041751a 000000d7
[ 5302.878392]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5302.894970]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5302.911571]         ...                        
[ 5302.916417] Call Trace:                        
[ 5302.921271] [<8000e478>] show_stack+0x54/0x88  
[ 5302.929927] [<80023fc0>] __warn+0xe4/0x118     
[ 5302.938053] [<80024088>] warn_slowpath_null+0x1c/0x34
[ 5302.948075] [<83123fd8>] mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[ 5302.960160] [<83122e50>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
[ 5302.972075] [<83128e3c>] mt76_dma_attach+0xb6c/0xc9c [mt76]
[ 5302.983119] ---[ end trace 699abe8c3b34b96a ]---
[  4] 113.5-114.0 sec  1.80 MBytes  30.2 Mbits/sec
[  4] 114.0-114.5 sec  1.81 MBytes  30.4 Mbits/sec

but the connection has no end

imwsy, 2017-12-25:
In the latest version, even the 2.4g wifi disconnected, the kernel information will not be wrong message

imwsy, 2018-04-16:
problems resolved after upgrading to the latest version of mt76

closed 2018-04-16 by imwsy