openwrt/mt76 #118: Crashes with newest version commit: 21f20b4

nicupavel, 2017-09-15:
Running on a Linkit7688 board the crashes seems more frequent with latest commit 21f20b4

I'm testing by powering the board and running and netperf benchmark on local network.

`[ 1828.462714] MCU message 8 (seq 5) timed out
[ 1829.502688] MCU message 40 (seq 6) timed out
[ 1830.542656] MCU message 8 (seq 7) timed out
[ 1831.582638] MCU message 40 (seq 8) timed out
[ 1831.602629] ------------[ cut here ]------------
[ 1831.611844] WARNING: CPU: 0 PID: 1374 at /indevel/rainmachine5/lede-bleedingedge/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2017-09-15-21f20b46/mt7603_mac.c:1279 mt7603_mac_work+0x12c/0x244 [mt7603e]
[ 1831.649753] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mmc_spi mmc_block iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG spidev slhc pwm_bl of_mmc_spi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack mmc_core iptable_mangle iptable_filter ip_tables hid_generic crc7 crc_itu_t crc_ccitt hid evdev input_core i2c_mt7621 mt76x2e mt7603e i2c_dev i2c_core mt76 mac80211 cfg80211 compat ledtrig_oneshot ledtrig_heartbeat ledtrig_gpio ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ehci_platform ehci_hcd leds_gpio ohci_platform ohci_hcd ledtrig_transient rainmachine_valves gpio_button_hotplug usbcore nls_base usb_common
[ 1831.811787] CPU: 0 PID: 1374 Comm: kworker/u2:1 Tainted: G        W       4.9.49 #0
[ 1831.826978] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 1831.836985] Stack : 87c03414 8751cb98 00000088 8004d620 87da1724 803e93c7 8039ab10 0000055e
[ 1831.853576]         8039aa7c 87209d64 803e0000 8007b2bc 00000088 8004d620 803a02d4 803e0000
[ 1831.870153]         00000003 87209d64 803e0000 80038e50 00000088 87209d9c 00000166 00000000
[ 1831.886741]         803fb160 00000000 875cd51c 87ff5e00 87ff5d00 30796870 00000000 00000000
[ 1831.903328]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1831.919913]         ...
[ 1831.924759] Call Trace:
[ 1831.929616] [<8000e448>] show_stack+0x54/0x88
[ 1831.938273] [<80023fa0>] __warn+0xe4/0x118
[ 1831.946395] [<80024068>] warn_slowpath_null+0x1c/0x34
[ 1831.956435] [<875cd648>] mt7603_mac_work+0x12c/0x244 [mt7603e]
[ 1831.968016] [<80036e98>] process_one_work+0x1ec/0x32c
[ 1831.978030] [<80037c58>] worker_thread+0x2ac/0x420
[ 1831.987526] [<8003c484>] kthread+0xd8/0xec
[ 1831.995647] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[ 1832.006345] ---[ end trace 0750c6ff2df6c2dc ]---`

nbd168, 2017-12-03:
Please try the latest version

nicupavel, 2017-12-05:
Built the latest source in LEDE and tested with: 
```
while [ 1 ]; do netperf -H 192.168.12.174 -l 30 -v 1 -f M; date; sleep 5; done
```

After 30 minutes network fails, no crashes in dmesg but these messages are continuously show up:
```
[ 2768.664766] MCU message 8 (seq 9) timed out
[ 2769.704765] MCU message 40 (seq 10) timed out
[ 2770.744753] MCU message 8 (seq 11) timed out
[ 2771.784747] MCU message 40 (seq 12) timed out
[ 2783.714788] MCU message 40 (seq 4) timed out
[ 2784.744786] MCU message 8 (seq 5) timed out
[ 2785.784674] MCU message 40 (seq 6) timed out
[ 2798.344599] MCU message 8 (seq 7) timed out
```

On a second run this happened again but after 5+ hours.

```
uname -a
Linux RainMachine 4.9.65 #0 Mon Dec 4 09:57:40 2017 mips GNU/Linux
```
```
[   12.126869] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[   12.142137] mt76_wmac 10300000.wmac: Firmware Version: _e2_mp
[   12.153626] mt76_wmac 10300000.wmac: Build Time: 20150211175503
```

closed 2018-06-21 by nbd168