openwrt/mt76 #121: Kernel panic on latest version.

pparent76, 2017-10-13:
Tested on version compiled on 10/10/2017 at commit d5e945e9b2ee8f52581c1a6978097cf30c39cfd9 with kernel 4.9.54 wg3526 only 2.4ghz activated:

Happened once out of four days of real-life usage for now.

```
<4>[213439.915417] ------------[ cut here ]------------
<4>[213439.920175] WARNING: CPU: 2 PID: 0 at mm/slub.c:3843 ksize+0xe0/0xf8
<4>[213439.926656] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt em_cmp em_nbyte sch_dsmark sch_pie act_ipt sch_codel sch_gred em_meta sch_teql cls_basic sch_prio em_text sch_sfq act_police sch_fq sch_red
<4>[213439.997399]  act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun cmac mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[213440.068949] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        W       4.9.54 #0
<4>[213440.076139] Stack : 00000000 00000000 80557b2a 00000041 80406a24 00000000 00000000 80550000
<4>[213440.084575]         8fc543bc 804ebea7 80482c5c 00000002 00000000 80553824 8e240800 8ebc3e00
<4>[213440.093003]         00000000 8006b878 00000001 80550000 804f1fa4 804f1fa8 80487844 8fc1d940
<4>[213440.101437]         00000003 800a9ff0 8e240800 8ebc3e00 00000001 00000000 02090220 00c1d940
<4>[213440.109882]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[213440.118352]         ...
<4>[213440.120883] Call Trace:
<4>[213440.123453] [<8000f814>] show_stack+0x54/0x88
<4>[213440.127912] [<801df1dc>] dump_stack+0x8c/0xd0
<4>[213440.132338] [<8002bdc4>] __warn+0xe4/0x118
<4>[213440.136503] [<8002be8c>] warn_slowpath_null+0x1c/0x30
<4>[213440.141635] [<800e884c>] ksize+0xe0/0xf8
<4>[213440.145632] [<802d90e0>] pskb_expand_head+0x98/0x23c
<4>[213440.150860] [<8f2a97e0>] ieee80211_probereq_get+0xb48/0x190c [mac80211]
<4>[213440.157759] ---[ end trace 0ef241925016ea77 ]---
<4>[213440.162608] Kernel bug detected[#1]:
<1>[213440.162788] CPU 2 Unable to handle kernel paging request at virtual address 00004000, epc == 800eb648, ra == 800eb5cc
<4>[213440.176904] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.54 #0
<4>[213440.184092] task: 804ebfa0 task.stack: 804dc000
<4>[213440.188681] $ 0   : 00000000 80550000 81026060 00000001
<4>[213440.193987] $ 4   : 00000000 81026073 000f0000 81183900
<4>[213440.199288] $ 8   : 804f0000 801eb484 00000000 00000000
<4>[213440.204590] $12   : 00000000 00000000 3fff0000 ffffffff
<4>[213440.209890] $16   : 8f237a80 8f101540 8fd366d0 8f101b7c
<4>[213440.215194] $20   : 00000002 8f101b78 00000000 00000091
<4>[213440.220494] $24   : 40000000 0ec14800                  
<4>[213440.225797] $28   : 804dc000 8fc0de9c 80204584 802d5e10
<4>[213440.231098] Hi    : 00000007
<4>[213440.234046] Lo    : 00000000
<4>[213440.237025] epc   : 800e9bb0 kfree+0x78/0x1a4
<4>[213440.241471] ra    : 802d5e10 __kfree_skb+0x14/0xa4
<4>[213440.246320] Status: 11008403     KERNEL EXL IE 
<4>[213440.250581] Cause : 50800034 (ExcCode 0d)
<4>[213440.254653] PrId  : 0001992f (MIPS 1004Kc)
<4>[213440.258809] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt em_cmp em_nbyte sch_dsmark sch_pie act_ipt sch_codel sch_gred em_meta sch_teql cls_basic sch_prio em_text sch_sfq act_police sch_fq sch_red
<4>[213440.329426]  act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun cmac mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
Process swapper/0 (pid: 0, threadinfo=804dc000, task=804ebfa0, tls=00000000)
<4>[213440.407546] Stack : 00000002 8f1015b0 8f101540 8fd366d0 8f1015b0 8f0e49f0 8fc0ded8 8fc0ded3
<4>[213440.415980]         00000001 061e0000 00000000 80580000 0e527920 00000000 00000900 804f0000
<4>[213440.424411]         81492440 8f237a80 8ea6e890 00000001 8f101540 00000001 ffffffff 00000002
<4>[213440.432840]         00000000 804de040 00000100 0000000a 00000018 8f0a2b70 00000000 01453ce8
<4>[213440.441277]         804f0000 804b0ba0 8f102188 8054121c 804e95fc 8003020c 00000004 804de04c
<4>[213440.449709]         ...
<4>[213440.452232] Call Trace:
<4>[213440.454758] [<800e9bb0>] kfree+0x78/0x1a4
<4>[213440.458844] [<802d5e10>] __kfree_skb+0x14/0xa4
<4>[213440.463388] [<8f0e49f0>] mt76_dma_attach+0x720/0xc94 [mt76]
<4>[213440.469024] Code: 30630001  24030001  38630001 <00030336> 8c430000  7c630380  10600003  00000000  10000002 
<4>[213440.478852] 
<4>[213440.480423] Oops[#2]:
<4>[213440.480643] ---[ end trace 0ef241935016ea77 ]---

===================================
Time: 1507810662.799728
<0>[213440.486986] Kernel panic - not syncing: Fatal exception in interrupt
```

pparent76, 2017-10-18:
Again with commit 37654d704772512065cd28b007342758c2a0f14a :  
```

<4>[82713.077406] Stack : 00000000 00000000 80557b2a 00000041 80406a24 00000000 00000000 80550000
<4>[82713.085754]         804ec2bc 804ebea7 80482c5c 00000000 00000000 80553824 00000054 8f295a00
<4>[82713.094099]         00000000 8006b878 00000001 80550000 804f1fa4 804f1fa8 80487844 8fc0dd90
<4>[82713.102444]         00000003 800a9fd0 00000054 8f295a00 0000f7f3 00000007 00000000 00c0dd90
<4>[82713.110789]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[82713.119130]         ...
<4>[82713.121567] Call Trace:
<4>[82713.124029] [<8000f814>] show_stack+0x54/0x88
<4>[82713.128386] [<801df1cc>] dump_stack+0x8c/0xd0
<4>[82713.132727] [<8002bdc4>] __warn+0xe4/0x118
<4>[82713.136803] [<8002be8c>] warn_slowpath_null+0x1c/0x30
<4>[82713.141876] [<8f28c124>] mt7603_mac_fill_rx+0x2b0/0x3a8 [mt7603e]
<4>[82713.147957] [<8f28ae30>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
<4>[82713.153958] [<8f284d6c>] mt76_dma_attach+0xa9c/0xc94 [mt76]
<4>[82713.159601] ---[ end trace 30529a2ab5a2f517 ]---
<4>[82713.164302] ------------[ cut here ]------------
<4>[82713.169064] WARNING: CPU: 0 PID: 0 at backports-2017-10-06/net/mac80211/rx.c:4321 ieee80211_rx_napi+0xa8/0x964 [mac80211]
<4>[82713.180018] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt em_cmp em_nbyte sch_dsmark sch_pie act_ipt sch_codel sch_gred em_meta sch_teql cls_basic sch_prio em_text sch_sfq act_police sch_fq sch_red
<4>[82713.250675]  act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]<4>[82713.321600] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.54 #0
<4>[82713.328698] Stack : 00000000 00000000 80557b2a 00000041 80406a24 00000000 00000000 80550000
<4>[82713.337045]         804ec2bc 804ebea7 80482c5c 00000000 00000000 80553824 00000001 8f295a00
<4>[82713.345391]         00000000 8006b878 00000001 80550000 804f1fa4 804f1fa8 80487844 8fc0dd38
<4>[82713.353736]         00000003 800a9fd0 00000001 8f295a00 0000f997 00000007 00000008 00c0dd38
<4>[82713.362081]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[82713.370424]         ...
<4>[82713.372859] Call Trace:
<4>[82713.375308] [<8000f814>] show_stack+0x54/0x88
<4>[82713.379658] [<801df1cc>] dump_stack+0x8c/0xd0
<4>[82713.383995] [<8002bdc4>] __warn+0xe4/0x118
<4>[82713.388071] [<8002be8c>] warn_slowpath_null+0x1c/0x30
<4>[82713.393195] [<8f325b94>] ieee80211_rx_napi+0xa8/0x964 [mac80211]
<4>[82713.399231] [<8f2858b4>] mt76_rx_complete+0x88/0x118 [mt76]
<4>[82713.404794] [<8f284e7c>] mt76_dma_attach+0xbac/0xc94 [mt76]
<4>[82713.410438] ---[ end trace 30529a2bb5a2f517 ]---
<7>[82713.529757] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:00:03:ab:f9:08:51:08:00 SRC=192.168.182.42 DST=8.8.8.8 LEN=68 TOS=0x00 PREC=0x00 TTL=64 ID=9005 DF PROTO=UDP SPT=12756 DPT=53 LEN=48 MARK=0x100 
<7>[82713.551580] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:00:03:ab:f9:08:51:08:00 SRC=192.168.182.42 DST=218.213.150.44 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=10350 DF PROTO=TCP SPT=42651 DPT=80 SEQ=1215163825 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (020405B40402080A0021C6400000000001030306) MARK=0x100 
<7>[82713.622796] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:00:03:ab:f9:08:51:08:00 SRC=192.168.182.42 DST=218.213.150.57 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=43814 DF PROTO=TCP SPT=58766 DPT=80 SEQ=186871495 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (020405B40402080A0021C6460000000001030306) MARK=0x100 
<7>[82715.397136] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:00:03:ab:f9:08:51:08:00 SRC=192.168.182.42 DST=74.125.203.188 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=17439 DF PROTO=TCP SPT=54797 DPT=443 SEQ=1831109351 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (020405B40402080A0021C6F80000000001030306) MARK=0x100 
<4>[82715.996390] ------------[ cut here ]------------
<4>[82716.001062] WARNING: CPU: 3 PID: 0 at mm/slub.c:3843 ksize+0xe0/0xf8
[... cut ...]
<4>[82716.530989]         ...
<4>[82716.533431] Call Trace:
<4>[82716.535883] [<800e9b90>] kfree+0x78/0x1a4
<4>[82716.539897] [<802d5e00>] __kfree_skb+0x14/0xa4
<4>[82716.544372] [<8f2849f0>] mt76_dma_attach+0x720/0xc94 [mt76]
<4>[82716.549930] Code: 30630001  24030001  38630001 <00030336> 8c430000  7c630380  10600003  00000000  10000002 
<4>[82716.559695] 
<4>[82716.561183] Oops[#2]:
<4>[82716.561433] ---[ end trace 30529a2db5a2f517 ]---

===================================
Time: 1508284734.35088

```

pparent76, 2017-11-06:
with updated kernel:

> <1>[179479.856718] CPU 0 Unable to handle kernel paging request at virtual address 00004000, epc == 800eb804, ra == 8ed99c70
> <4>[179479.867461] Oops[#1]:
> <4>[179479.869825] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.58 #0
> <4>[179479.877021] task: 804edfa0 task.stack: 804de000
> <4>[179479.881624] $ 0   : 00000000 00000001 00004000 00004000
> <4>[179479.886930] $ 4   : 8fc02f00 804eb5fc 81494590 8d2d1078
> <4>[179479.892232] $ 8   : 00000018 00b7dd1b 0033ce14 ffffffff
> <4>[179479.897544] $12   : 00008fa7 00000000 00000000 ffffffff
> <4>[179479.902854] $16   : 02088020 8fc02f00 00000064 00000000
> <4>[179479.908165] $20   : 00000004 fffffffb 00000003 00000000
> <4>[179479.913476] $24   : 00000000 0dbc3800                  
> <4>[179479.918786] $28   : 804de000 8fc0db08 804eb5fc 8ed99c70
> <4>[179479.924100] Hi    : 000105cd
> <4>[179479.927054] Lo    : db98b300
> <4>[179479.930049] epc   : 800eb804 kmem_cache_alloc+0x84/0x14c
> <4>[179479.935464] ra    : 8ed99c70 br_nf_hook_thresh+0x63c/0x93c [br_netfilter]
> <4>[179479.942306] Status: 11008403     KERNEL EXL IE 
> <4>[179479.946579] Cause : 40800008 (ExcCode 02)
> <4>[179479.950656] BadVA : 00004000
> <4>[179479.953610] PrId  : 0001992f (MIPS 1004Kc)
> <4>[179479.957770] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter em_cmp sch_teql em_nbyte sch_pie em_meta sch_gred sch_dsmark act_police cls_basic act_ipt sch_prio em_text sch_codel sch_sfq
> <4>[179480.028341]  sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
> Process swapper/0 (pid: 0, threadinfo=804de000, task=804edfa0, tls=00000000)
> <4>[179480.107480] Stack : 8e90f300 00000000 8050b7c0 00000001 00000004 8e90f840 8fc0dbec 8ed99c70
> <4>[179480.115926]         8edd3088 00000001 00000001 8edd2f00 81491c80 00000000 80000000 8004e202
> <4>[179480.124374]         8fd30000 00000000 00000000 8050b7c0 8ee25680 8ed996f8 804eb5fc 8fc0dbb4
> <4>[179480.132819]         8fc0dbec 8e90f840 00000001 80323668 00000001 000001c5 00000985 80104204
> <4>[179480.141268]         00000000 80510000 8e90f840 8fc0dbec 00000001 80323704 8edd2f00 8004e3dc
> <4>[179480.149716]         ...
> <4>[179480.152244] Call Trace:
> <4>[179480.154776] [<800eb804>] kmem_cache_alloc+0x84/0x14c
> <4>[179480.159829] [<8ed99c70>] br_nf_hook_thresh+0x63c/0x93c [br_netfilter]
> <4>[179480.166365] [<80323668>] nf_iterate+0x68/0xb0
> <4>[179480.170798] [<80323704>] nf_hook_slow+0x54/0xf0
> <4>[179480.175408] [<803d96cc>] br_handle_frame+0x3a0/0x3f4
> <4>[179480.180451] [<802eab48>] __netif_receive_skb_core+0x490/0x970
> <4>[179480.186266] [<802eb9dc>] netif_receive_skb_internal+0x74/0xb8
> <4>[179480.192082] [<802ec318>] napi_gro_receive+0x6c/0xd0
> <4>[179480.197215] [<8f320814>] ieee80211_attach_ack_skb+0x1044/0x19b4 [mac80211]
> <4>[179480.204240] Code: 32058000  8e230014  00431821 <8c670000> 41686000  31080001  000000c0  8f86000c  8e230000 
> <4>[179480.214090] 
> <4>[179480.215942] ---[ end trace 3cd45467e0ab2ab6 ]---
> 
> ===================================
> Time: 1509713076.261675
> <0>[179480.225545] Kernel panic - not syncing: Fatal exception in interrupt

pparent76, 2017-11-11:
With latest lede:

(It does not seem to happen without mt7603 driver)

> <1>[248299.550920] CPU 0 Unable to handle kernel paging request at virtual address 07401008, epc == 8ed7c428, ra == 8ed39794
> <4>[248299.565457] Oops[#1]:
> <4>[248299.567826] CPU: 0 PID: 337 Comm: kworker/0:1 Tainted: G        W       4.9.58 #0
> <4>[248299.575431] Workqueue: events_long nf_conntrack_tuple_taken [nf_conntrack]
> <4>[248299.582368] task: 8fec91a0 task.stack: 8fe4c000
> <4>[248299.586956] $ 0   : 00000000 00000001 00000001 00000000
> <4>[248299.592259] $ 4   : 8e1a3e00 00000200 c67448d6 aab22788
> <4>[248299.597559] $ 8   : 8e1a3e60 8e1a3e60 00000361 132f32dc
> <4>[248299.602861] $12   : 0000e1d6 ffffffff 0000c6bb 00001056
> <4>[248299.608161] $16   : fffffffe 07401000 8ed7eb20 8ed7eb20
> <4>[248299.613463] $20   : 8e1a3e78 8e1a3e00 851d4d00 00000000
> <4>[248299.618763] $24   : 00000000 8ed30420                  
> <4>[248299.624065] $28   : 8fe4c000 8fe4dd90 8ed7c098 8ed39794
> <4>[248299.629366] Hi    : 00001543
> <4>[248299.632314] Lo    : a295e000
> <4>[248299.635274] epc   : 8ed7c428 nf_nat_used_tuple+0xe4/0x2cc [nf_nat]
> <4>[248299.641518] ra    : 8ed39794 __nf_ct_ext_destroy+0x64/0x16c [nf_conntrack]
> <4>[248299.648441] Status: 11008403     KERNEL EXL IE 
> <4>[248299.652703] Cause : c0800008 (ExcCode 02)
> <4>[248299.656773] BadVA : 07401008
> <4>[248299.659721] PrId  : 0001992f (MIPS 1004Kc)
> <4>[248299.663878] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter em_cmp sch_teql em_nbyte sch_pie em_meta sch_gred sch_dsmark act_police cls_basic act_ipt sch_prio em_text sch_codel sch_sfq
> <4>[248299.734308]  sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
> Process kworker/0:1 (pid: 337, threadinfo=8fe4c000, task=8fec91a0, tls=00000000)
> <4>[248299.813645] Stack : 8e1a3e00 00000008 8e1a3e00 8e1a3e00 00000008 00000000 8ef4aa8a 00000004
> <4>[248299.822077]         8ed3e3a0 8e1a3e00 00000018 00000002 8ed40000 00000030 804f0000 8ed39794
> <4>[248299.830506]         8e1a3e00 8ed313dc 81491c80 81491c80 8e1a3e00 8ed3e344 00001543 00000001
> <4>[248299.838936]         8e1a3e40 8ed303d0 81491cb0 804f0000 8ed3e354 00000000 00000001 8ed31d80
> <4>[248299.847366]         81491c80 80546c80 804eb5fc 8fec91a0 8fc52340 00000040 8ed40000 8ed40000
> <4>[248299.855796]         ...
> <4>[248299.858323] Call Trace:
> <4>[248299.860846] [<8ed7c428>] nf_nat_used_tuple+0xe4/0x2cc [nf_nat]
> <4>[248299.866744] Code: 8e44001c  02409825  02842023 <13c0000a> 8e260008  16e00003  00000000  10000002  8e450014 
> <4>[248299.876568] 
> <4>[248299.880406] ---[ end trace b9170092839db77d ]---
> 
> ===================================
> Time: 1510323355.427363
> <4>[248299.876568] 
> <4>[248299.880406] ---[ end trace b9170092839db77d ]---
> <0>[248299.889936] Kernel panic - not syncing: Fatal exception

pparent76, 2017-11-20:
New crash, still not happening without mt76 driver running:

> <1>[547043.037932] CPU 2 Unable to handle kernel paging request at virtual address 07403800, epc == 802fd968, ra == 8031b514
> <4>[547043.048644] Oops[#1]:
> <4>[547043.050999] CPU: 2 PID: 29989 Comm: chilli Tainted: G    B   W       4.9.58 #0
> <4>[547043.058274] task: 8e8cf580 task.stack: 8c948000
> <4>[547043.062864] $ 0   : 00000000 80550008 80590000 00000000
> <4>[547043.068172] $ 4   : 811d9b00 07403800 00000400 00000000
> <4>[547043.073475] $ 8   : 00000000 801ec910 6b8171c0 0016ca69
> <4>[547043.078778] $12   : 0000049a 00000000 00000000 00000047
> <4>[547043.084079] $16   : 8ea6ca00 8e680000 00000001 00000030
> <4>[547043.089385] $20   : 00000000 000000c0 00000200 8f35d924
> <4>[547043.094685] $24   : 00000000 800571e0                  
> <4>[547043.099988] $28   : 8c948000 8c949d18 00000000 8031b514
> <4>[547043.105290] Hi    : 00000005
> <4>[547043.108242] Lo    : 0000001a
> <4>[547043.111211] epc   : 802fd968 rtnl_kfree_skbs+0x18/0x24
> <4>[547043.116420] ra    : 8031b514 fq_codel_reset+0x64/0xc4
> <4>[547043.121529] Status: 11007c03     KERNEL EXL IE 
> <4>[547043.125792] Cause : 4080000c (ExcCode 03)
> <4>[547043.129865] BadVA : 07403800
> <4>[547043.132816] PrId  : 0001992f (MIPS 1004Kc)
> <4>[547043.136971] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter em_cmp sch_teql em_nbyte sch_pie em_meta sch_gred sch_dsmark act_police cls_basic act_ipt sch_prio em_text sch_codel sch_sfq
> <4>[547043.207424]  sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
> Process chilli (pid: 29989, threadinfo=8c948000, task=8e8cf580, tls=77f20dec)
> <4>[547043.286527] Stack : 80543298 804eb5fc 00000001 00000026 8c949e18 8ea6ca00 8ea6ca70 8e713780
> <4>[547043.294964]         804b2ce8 803117a0 00000000 8002fa44 00000200 800480e0 8ea6ca00 80311e80
> <4>[547043.303398]         00000001 8002fb10 8f35c000 8c949e18 8f35c000 0000004b 8c949e18 8031297c
> <4>[547043.311833]         00000000 8004ce18 00000001 00000000 804eb5fc 80312ab4 8c949dcc 8c949e18
> <4>[547043.320265]         8f35c000 804b2ce8 00000001 00000001 00000200 8f35d924 00000000 802e69e0
> <4>[547043.328699]         ...
> <4>[547043.331223] Call Trace:
> <4>[547043.333747] [<802fd968>] rtnl_kfree_skbs+0x18/0x24
> <4>[547043.338611] [<8031b514>] fq_codel_reset+0x64/0xc4
> <4>[547043.343401] [<803117a0>] qdisc_reset+0x24/0x5c
> <4>[547043.347911] [<80311e80>] dev_deactivate_queue.constprop.6+0x58/0xac
> <4>[547043.354243] [<8031297c>] dev_deactivate_many+0x84/0x2d0
> <4>[547043.359562] [<802e69e0>] __dev_close_many+0x88/0xe8
> <4>[547043.364505] [<802e6ac4>] dev_close_many+0x84/0x11c
> <4>[547043.369369] [<802e9bb0>] rollback_registered_many+0x12c/0x438
> <4>[547043.375175] [<802e9ee8>] rollback_registered+0x2c/0x48
> <4>[547043.380382] [<802ec814>] unregister_netdevice_queue+0x7c/0xc4
> <4>[547043.386243] [<8f149c08>] tun_get_socket+0x1a50/0x3c30 [tun]
> <4>[547043.391885] Code: 3c028059  8c43d318  ac44d318 <aca30000> 03e00008  00000000  3c028051  8c42c888  38420001 
> <4>[547043.401716] 
> <4>[547043.403626] ---[ end trace 0bb6fa46f84eec75 ]---
>

pparent76, 2017-11-22:
```
<1>[114127.378808] CPU 2 Unable to handle kernel paging request at virtual address 00004000, epc == 800eb804, ra == 802d9eac
<4>[114127.389548] Oops[#1]:
<4>[114127.391913] CPU: 2 PID: 10995 Comm: chilli Tainted: G        W       4.9.58 #0
<4>[114127.399188] task: 8f11e3e0 task.stack: 8e5c4000
<7>[114127.400595] dotspot-7617IN=br-lan OUT= PHYSIN=wlan0 MAC=78:a3:51:2b:f4:1a:ec:10:7b:5d:cd:21:08:00 SRC=192.168.182.6 DST=216.58.213.174 LEN=52 TOS=0x00 PREC=0x00 TTL=64 ID=55078 DF PROTO=TCP SPT=37527 DPT=80 SEQ=1501442155 ACK=1059030267 WINDOW=354 RES=0x00 ACK URGP=0 OPT (0101080A0004F9A000ADB16A) MARK=0x100 
<4>[114127.431117] $ 0   : 00000000 00000001 00004000 00004000
<4>[114127.436436] $ 4   : 8fc02100 804eb5fc 814accc0 ffffffff
<4>[114127.441736] $ 8   : 804eb5fc 0000e518 8f2ef530 0004b1e3
<4>[114127.447042] $12   : 7fd90c98 771b02c0 77109280 77109280
<4>[114127.452342] $16   : 024000c0 8fc02100 00000130 8fc02100
<4>[114127.457646] $20   : 00000001 8fc258d0 8fafd3a0 771871d8
<4>[114127.462956] $24   : 770cde25 771240b8                  
<4>[114127.468258] $28   : 8e5c4000 8e5c5d58 00000000 802d9eac
<4>[114127.473562] Hi    : 00000002
<4>[114127.476510] Lo    : 00000000
<4>[114127.479496] epc   : 800eb804 kmem_cache_alloc+0x84/0x14c
<4>[114127.484878] ra    : 802d9eac __alloc_skb+0x60/0x184
<4>[114127.489814] Status: 11007c03     KERNEL EXL IE 
<4>[114127.494077] Cause : 40800008 (ExcCode 02)
<4>[114127.498147] BadVA : 00004000
<4>[114127.501095] PrId  : 0001992f (MIPS 1004Kc)
<4>[114127.505253] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter em_cmp sch_teql em_nbyte sch_pie em_meta sch_gred sch_dsmark act_police cls_basic act_ipt sch_prio em_text sch_codel sch_sfq
<4>[114127.575740]  sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
Process chilli (pid: 10995, threadinfo=8e5c4000, task=8f11e3e0, tls=771b1dec)
<4>[114127.654837] Stack : 814aa424 8e5c5d94 8e5c5da8 8e8c4318 8e8c4318 8eeba280 024000c0 802d9eac
<4>[114127.663273]         8e8c413c 802e0029 811c7260 811c7260 805118a0 800ba738 8eeba280 00000080
<4>[114127.671705]         811c8100 8eeba37c 8fb777f8 80349c2c 811c7274 8146a2b4 8e8c4318 00000000
<4>[114127.680137]         00100000 8eeba280 00000080 00000000 8eeba31c 8033b4dc 00000000 800eb190
<4>[114127.688571]         00100000 800d1034 00000000 8e5c5e34 00000001 8eeba280 8fafd380 00000008
<4>[114127.697002]         ...
<4>[114127.699524] Call Trace:
<4>[114127.702066] [<800eb804>] kmem_cache_alloc+0x84/0x14c
<4>[114127.707108] [<802d9eac>] __alloc_skb+0x60/0x184
<4>[114127.711716] [<80349c2c>] tcp_send_fin+0xa0/0x1ec
<4>[114127.716424] [<8033b4dc>] tcp_close+0x3f4/0x428
<4>[114127.720949] [<80362e64>] inet_release+0x60/0x78
<4>[114127.725558] [<802cf0bc>] sock_release+0x2c/0xdc
<4>[114127.730154] [<802cf17c>] sock_close+0x10/0x20
<4>[114127.734590] [<800f21ec>] __fput+0x104/0x1f4
<4>[114127.738847] [<800451d8>] task_work_run+0xb0/0xe8
<4>[114127.743543] [<8002dd14>] do_exit+0x3b4/0x834
<4>[114127.747882] [<8002f034>] do_group_exit+0x4c/0xc8
<4>[114127.752562] [<8002f0c4>] __wake_up_parent+0x0/0x18
<4>[114127.757414] Code: 32058000  8e230014  00431821 <8c670000> 41686000  31080001  000000c0  8f86000c  8e230000 
<4>[114127.767249] 
<4>[114127.769125] ---[ end trace 6bfd9413ebd1878b ]---

===================================
Time: 1511352911.402364
```

nbd168, 2017-12-03:
Please try the latest version

pparent76, 2017-12-04:
Ok I'll test, and I will tell you in few days weather I still see crashes or not.

pparent76, 2017-12-04:
Unfortunately it seems that packages have not been updated for a while in snapshots releases.

https://downloads.lede-project.org/snapshots/targets/ramips/mt7621/packages/

I'm not sure why. I will try to compile new mt76 version with SDK.

MaFeSa, 2017-12-08:
I have 4 days of uptime (Xiaomi 3G) without kernel panic or warnings!
Thanks nbd

pparent76, 2017-12-08:
I also have 4 days of uptime, but we need to wait a little bit more for conclusion, because with last version I got up to 6 days of uptime, and yet the kernel panic happened on average twice a week.

pparent76, 2017-12-11:
Almost 7 days up now, sounds good news, but I will still wait another 7 days, before concluding it's 100% stable.

baohavan, 2017-12-13:
Crash after 4 days up

[434148.177895] br-lan: port 3(wlan1) entered forwarding state
[434148.516861] MCU message 8 (seq 11) timed out
[434149.556866] MCU message 8 (seq 12) timed out
[434150.596877] MCU message 8 (seq 13) timed out
[434151.636854] MCU message 8 (seq 14) timed out
[434152.676883] MCU message 8 (seq 15) timed out
[434153.716842] MCU message 8 (seq 1) timed out
[434154.756837] MCU message 8 (seq 2) timed out
[434155.796816] MCU message 8 (seq 3) timed out
[434156.836812] MCU message 8 (seq 4) timed out
[434157.876803] MCU message 8 (seq 5) timed out
[434158.916801] MCU message 8 (seq 6) timed out
[434160.996775] MCU message 8 (seq 7) timed out
[434161.001286] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[434161.008054] br-lan: port 2(wlan0) entered blocking state
[434161.013537] br-lan: port 2(wlan0) entered forwarding state
[434169.357995] ------------[ cut here ]------------
[434169.362846] WARNING: CPU: 0 PID: 3 at net/core/skbuff.c:4343 skb_try_coalesce+0x20c/0x31c
[434169.371305] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio gpio_button_hotplug
[434169.429205] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G        W       4.9.65 #0
[434169.436709] Stack : 8042751a 00000043 00000000 00000001 8382ccd4 803d5207 80388730 00000003
[434169.445301]         80423660 000010f7 803edcc8 82c7e1c0 fffffffe 8004d720 8038df10 803d0000
[434169.453900]         00000003 000010f7 8038c0d8 8384176c fffffffe 8007b3fc 8042751a 0000004d
[434169.462505]         803e6fa0 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[434169.471108]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[434169.479703]         ...
[434169.482274] Call Trace:
[434169.484862] [<8000e478>] show_stack+0x54/0x88
[434169.489401] [<80023fc0>] __warn+0xe4/0x118
[434169.493649] [<80024088>] warn_slowpath_null+0x1c/0x34
[434169.498896] [<802153e0>] skb_try_coalesce+0x20c/0x31c
[434169.504120] [<8026f2b0>] tcp_try_coalesce+0x4c/0xd0
[434169.509179] [<802744c0>] tcp_data_queue+0x6e0/0xb14
[434169.514217] [<80274d90>] tcp_rcv_established+0x49c/0x628
[434169.519717] [<8027d870>] tcp_v4_do_rcv+0x90/0x1bc
[434169.524578] [<8027fe70>] tcp_v4_rcv+0x6dc/0x9ec
[434169.529291] [<8025c004>] ip_local_deliver_finish+0xd4/0x134
[434169.535034] [<8025c55c>] ip_local_deliver+0xc8/0xe0
[434169.540088] [<8025c82c>] ip_rcv+0x2b8/0x32c
[434169.544423] [<80225544>] __netif_receive_skb_core+0x620/0x708
[434169.550366] [<802f5068>] br_pass_frame_up+0xa4/0x114
[434169.555493] [<802f5598>] br_handle_frame_finish+0x4c0/0x524
[434169.561255] [<802f59ac>] br_handle_frame+0x358/0x3ec
[434169.566378] [<802252bc>] __netif_receive_skb_core+0x398/0x708
[434169.572313] [<80226d08>] napi_gro_receive+0x6c/0xd0
[434169.577508] [<832a01c4>] ieee80211_attach_ack_skb+0xfd4/0x1944 [mac80211]
[434169.584558] [<80083ffc>] get_page_from_freelist+0x450/0x73c
[434169.590319] ---[ end trace a41c644d4dbd53b4 ]---
[434205.719512] device wlan0 left promiscuous mode
[434205.724242] br-lan: port 2(wlan0) entered disabled state
[434207.636553] MCU message 8 (seq 8) timed out
[434207.643546] device wlan1 left promiscuous mode
[434207.648349] br-lan: port 3(wlan1) entered disabled state
[434210.016616] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[434210.027300] br-lan: port 2(wlan1) entered blocking state
[434210.032790] br-lan: port 2(wlan1) entered disabled state
[434210.038805] device wlan1 entered promiscuous mode
[434210.043748] br-lan: port 2(wlan1) entered blocking state
[434210.049275] br-lan: port 2(wlan1) entered forwarding state
[434210.066605] br-lan: port 2(wlan1) entered disabled state
[434210.706245] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[434210.713048] br-lan: port 2(wlan1) entered blocking state
[434210.718574] br-lan: port 2(wlan1) entered forwarding state
[434221.278255] device wlan1 left promiscuous mode
[434221.283260] br-lan: port 2(wlan1) entered disabled state

Source revision: 
root@LEDE:~# opkg list | grep 76
kmod-mt76 - 4.9.65+2017-12-08-e5046560-1
kmod-mt76-core - 4.9.65+2017-12-08-e5046560-1
kmod-mt7603 - 4.9.65+2017-12-08-e5046560-1
kmod-mt76x2 - 4.9.65+2017-12-08-e5046560-1

kofec, 2017-12-18:
More logs for:
`[ 1514.677560] ------------[ cut here ]------------
[ 1514.682337] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4314 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 1514.693367] Rate marked as a VHT rate but data is invalid: MCS: 125, NSS: 0
[ 1514.700308] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat hid evdev i2c_dev ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi
[ 1514.771037]  snd_seq_device snd_hwdep snd soundcore vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc32c_generic crc32_generic
[ 1514.800499] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.67 #0
[ 1514.806389] Stack : 00000000 00000000 805b7b4a 00000033 8043d344 00000000 00000000 805b0000
[ 1514.814736]         805381bc 80537da7 804c3764 00000000 00000000 805b3824 00000001 8e85da00
[ 1514.823081]         00000000 8006a848 00000001 805b0000 8053dec4 8053dec8 804c8358 8fc0dd24
[ 1514.831427]         00000003 800a7270 00000001 8e85da00 00000000 00000000 00000002 00c0dd24
[ 1514.839771]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1514.848114]         ...
[ 1514.850550] Call Trace:
[ 1514.853007] [<8000f814>] show_stack+0x54/0x88
[ 1514.857355] [<801da52c>] dump_stack+0x8c/0xd0
[ 1514.861693] [<8002adc4>] __warn+0xe4/0x118
[ 1514.865768] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[ 1514.870773] [<8ea25c20>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 1514.876872] [<8e915980>] mt76_rx_complete+0x88/0x10c [mt76]
[ 1514.882423] [<8e914e7c>] mt76_dma_attach+0xbac/0xc94 [mt76]
[ 1514.888038] ---[ end trace 225259568dada16f ]---
[ 4636.715634] ------------[ cut here ]------------
[ 4636.720289] WARNING: CPU: 0 PID: 0 at /mnt/openwrt/LEDEXiaomiDVB/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-12-17-68b0cf17/mt7603_mac.c:392 mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[ 4636.738149] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat hid evdev i2c_dev ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi
[ 4636.808888]  snd_seq_device snd_hwdep snd soundcore vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc32c_generic crc32_generic
[ 4636.838332] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.67 #0
[ 4636.845432] Stack : 00000000 00000000 805b7b4a 00000041 8043d344 00000000 00000000 805b0000
[ 4636.853777]         805381bc 80537da7 804c3764 00000000 00000000 805b3824 00000001 8e85da00
[ 4636.862122]         00000000 8006a848 00000001 805b0000 8053dec4 8053dec8 804c8358 8fc0dd84
[ 4636.870468]         00000003 800a7270 00000001 8e85da00 81211e20 00000000 00000000 00c0dd84
[ 4636.878812]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 4636.887156]         ...
[ 4636.889592] Call Trace:
[ 4636.892049] [<8000f814>] show_stack+0x54/0x88
[ 4636.896397] [<801da52c>] dump_stack+0x8c/0xd0
[ 4636.900735] [<8002adc4>] __warn+0xe4/0x118
[ 4636.904811] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[ 4636.909847] [<8e8bc00c>] mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[ 4636.915917] [<8e8baf40>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
[ 4636.921903] [<8e914e30>] mt76_dma_attach+0xb60/0xc94 [mt76]
[ 4636.927523] ---[ end trace 225259568dada170 ]---`

pparent76, 2017-12-18:
On my side it has now rebooted, but without leaving any /sys/kernel/debug/crashlog so I don't know what to think.

kofec, 2017-12-18:
`[ 1292.930907] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4314 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 1292.941932] Rate marked as a VHT rate but data is invalid: MCS: 66, NSS: 0
[ 1292.948790] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat hid evdev i2c_dev ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi
[ 1293.019517]  snd_seq_device snd_hwdep snd soundcore vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc32c_generic crc32_generic
[ 1293.049003] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.67 #0
[ 1293.054892] Stack : 00000000 00000000 805b7b4a 00000033 8043d344 00000000 00000000 805b0000
[ 1293.063239]         805381bc 80537da7 804c3764 00000000 00000000 805b3824 00000001 8e8a5a00
[ 1293.071584]         00000000 8006a848 00000001 805b0000 8053dec4 8053dec8 804c8358 8fc0dd24
[ 1293.079930]         00000003 800a7270 00000001 8e8a5a00 00000000 00000000 00000002 00c0dd24
[ 1293.088275]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1293.096619]         ...
[ 1293.099055] Call Trace:
[ 1293.101513] [<8000f814>] show_stack+0x54/0x88
[ 1293.105861] [<801da52c>] dump_stack+0x8c/0xd0
[ 1293.110199] [<8002adc4>] __warn+0xe4/0x118
[ 1293.114274] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[ 1293.119293] [<8ea25c20>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 1293.125397] [<8e8f1980>] mt76_rx_complete+0x88/0x10c [mt76]
[ 1293.130946] [<8e8f0e7c>] mt76_dma_attach+0xbac/0xc94 [mt76]
[ 1293.136566] ---[ end trace 2c6504fc36af843a ]---`

kallisti5, 2017-12-27:
I see #90 and this issue running libremesh which uses mt76.  Onion Omega 2+.

Kernel panic + hard reboot:
```
# [ 1217.789975] CPU 0 Unable to handle kernel paging request at virtual address 074031ac, epc == 802aceb4, ra == 802aced0
[ 1217.800815] Oops[#1]:
[ 1217.803129] CPU: 0 PID: 2634 Comm: dnsmasq Tainted: G        W       4.4.71 #0
[ 1217.810454] task: 86c95928 ti: 87524000 task.ti: 87524000
[ 1217.815927] $ 0   : 00000000 770509ac 00000001 07403000
[ 1217.821245] $ 4   : 802acd98 00000000 87525c4c 00000002
[ 1217.826557] $ 8   : 8774b17c 00002000 00000000 02000000
[ 1217.831868] $12   : 7ff76aa8 77050990 00000000 770fc2c0
[ 1217.837179] $16   : 07402fcc 8037cff0 8037d080 87525dc0
[ 1217.842489] $20   : 00000000 00000000 00000000 00000025
[ 1217.847796] $24   : 00000000 802c0ef8                  
[ 1217.853109] $28   : 87524000 87525c30 87525da0 802aced0
[ 1217.858421] Hi    : 00000002
[ 1217.861337] Lo    : 00000000
[ 1217.864260] epc   : 802aceb4 0x802aceb4
[ 1217.868148] ra    : 802aced0 0x802aced0
[ 1217.872033] Status: 1100e403 KERNEL EXL IE 
[ 1217.876288] Cause : 00800008 (ExcCode 02)
[ 1217.880350] BadVA : 074031ac
[ 1217.883269] PrId  : 00019655 (MIPS 24KEc)
[ 1217.887329] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_fh
[ 1217.978621] Process dnsmasq (pid: 2634, threadinfo=87524000, task=86c95928, tls=770fdd48)
[ 1217.986908] Stack : 00000000 00000002 00000003 80228e84 00000000 8037cff0 8037d4e4 87525dc0
          00000000 0000000e 00000001 00000000 00000005 00020021 86c50b8c 0000003d
          ffffff82 00000025 00000002 000e0001 877d6f00 00000005 fffffff2 00000000
          00000000 87525cfc 87525da0 8774af00 8037cff0 802a4d14 800ba380 00000100
          00000200 00000018 87525dd0 00000008 00000000 87525da0 80380000 8774af00
          ...
[ 1218.023019] Call Trace:[<80228e84>] 0x80228e84
[ 1218.027582] [<802a4d14>] 0x802a4d14
[ 1218.031117] [<800ba380>] 0x800ba380
[ 1218.034682] [<802a4e00>] 0x802a4e00
[ 1218.038232] [<802a4ebc>] 0x802a4ebc
[ 1218.041797] [<802c1488>] 0x802c1488
[ 1218.045360] [<801fdf3c>] 0x801fdf3c
[ 1218.048932] [<8026d3b0>] 0x8026d3b0
[ 1218.052468] [<800ba380>] 0x800ba380
[ 1218.056003] [<80246a00>] 0x80246a00
[ 1218.059560] [<8026ecd8>] 0x8026ecd8
[ 1218.063130] [<801f36a0>] 0x801f36a0
[ 1218.066675] [<801f55d4>] 0x801f55d4
[ 1218.070210] [<802a2c0c>] 0x802a2c0c
[ 1218.073745] [<801f4200>] 0x801f4200
[ 1218.077306] [<800bc3a8>] 0x800bc3a8
[ 1218.080851] [<801f31c8>] 0x801f31c8
[ 1218.084386] [<800c2a64>] 0x800c2a64
[ 1218.087951] [<800b8700>] 0x800b8700
[ 1218.091495] [<8000682c>] 0x8000682c
[ 1218.095034] [<800069e0>] 0x800069e0
[ 1218.098569] [<800a6af4>] 0x800a6af4
[ 1218.102107] 
[ 1218.103614] 
Code: 02202021  1000000e  24030018 <8e0601e0> 10c00005  27a70030  afa20010  27a5001c  0c0ab2f9 
[ 1218.113756] ---[ end trace 2beb8dc92ed9d410 ]---
[ 1218.122704] Fatal exception: panic in 5 seconds
[ 1223.132075] Kernel panic - not syncing: Fatal exception
```

Boot info:
```

   ____       _             ____
  / __ \___  (_)__  ___    / __ \__ _  ___ ___ ____ _
 / /_/ / _ \/ / _ \/ _ \  / /_/ /  ' \/ -_) _ `/ _ `/
 \____/_//_/_/\___/_//_/  \____/_/_/_/\__/\_, /\_,_/
 W H A T  W I L L  Y O U  I N V E N T ? /___/"
[... 143 lines cut ...]
[   83.996262] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.71 #0
[   84.003333] Stack : 8032087c 00000000 00000001 80370000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[   84.039472] Call Trace:[<800144bc>] 0x800144bc
[   84.044009] [<800144bc>] 0x800144bc
[   84.047544] [<8002506c>] 0x8002506c
[   84.051107] [<876a591c>] 0x876a591c [mac80211@87680000+0x5ff40]
[   84.057139] [<80025124>] 0x80025124
[   84.060681] [<801fdf3c>] 0x801fdf3c
[   84.064261] [<876a591c>] 0x876a591c [mac80211@87680000+0x5ff40]
[   84.070277] [<8007a3cc>] 0x8007a3cc
[   84.073905] [<875f98ec>] 0x875f98ec [mt76@875f8000+0x3c00]
[   84.079531] [<875f90e8>] 0x875f90e8 [mt76@875f8000+0x3c00]
[   84.085133] [<8020c504>] 0x8020c504
[   84.088702] [<800275ec>] 0x800275ec
[   84.092259] [<8004f1a0>] 0x8004f1a0
[   84.095802] [<8004b6b8>] 0x8004b6b8
[   84.099355] [<800047f4>] 0x800047f4
[   84.102899] 
[   84.104410] ---[ end trace 4b2bf5c28b116a13 ]---
```

kallisti5, 2017-12-27:
```
0;root@LiMe-c0fe40: /root@LiMe-c0fe40:/# opkg list | grep 76
dnsmasq - 2.76-8
kmod-sdhci-mt7620 - 4.4.46-1
kmod-sound-mt7620 - 4.4.46-1
procd - 2017-01-26-0f589776-1
ralink-wifi-mt76x8 - 1
```

pparent76, 2018-02-13:
With latest commit ( 246d548b0688b27e0f7dc05c98f3a3255275d1b2 ) , after less than a day of testing: 

```
<4>[12233.844342] Unhandled kernel unaligned access[#1]:
<4>[12233.849143] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.9.77 #0
<4>[12233.855038] task: 8fc540a0 task.stack: 8fc74000
<4>[12233.859542] $ 0   : 00000000 00000001 ffffffff 956728d0
<4>[12233.864763] $ 4   : 8e270000 8fc1dd7c 000000ff 00000000
<4>[12233.869977] $ 8   : 00000054 80521554 804c75fc 00000001
<4>[12233.875196] $12   : 44000000 00000000 00000000 778962c0
<4>[12233.880412] $16   : 95672840 8e838000 8ed5e734 8f23e3c0
<4>[12233.885629] $20   : 8e42e3d4 00000000 956728b0 00000000
<4>[12233.890846] $24   : 00000000 8ed87148                  
<4>[12233.896060] $28   : 8fc74000 8fc1dd50 8e270000 803153a8
<4>[12233.901281] Hi    : 00000000
<4>[12233.904142] Lo    : 00000050
<4>[12233.907034] epc   : 8ed5c304 ipt_do_table+0x130/0x7d0 [ip_tables]
<4>[12233.913127] ra    : 803153a8 nf_iterate+0x68/0xb0
<4>[12233.917804] Status: 11007c03
<1>[12233.920383] CPU 0 Unable to handle kernel paging request at virtual address 0f439098, epc == 8ed5c234, ra == 8ed68090
<4>[12233.931128] KERNEL EXL IE 
Cause : 40800010 (ExcCode 04)
<4>[12233.936519] BadVA : 00000027
<4>[12233.939381] PrId  : 0001992f (MIPS 1004Kc)
<4>[12233.943452] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie sch_gred act_ipt em_cmp sch_prio sch_codel em_text em_meta sch_sfq act_police sch_fq
<4>[12234.013883]  sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
Process swapper/2 (pid: 0, threadinfo=8fc74000, task=8fc540a0, tls=00000000)
<4>[12234.092254] Stack : 00000000 00000000 00000000 0002bb01 00000000 804c75fc 8e270040 00000001
<4>[12234.100598]         80521554 804c0000 00000000 ffffffff 0000049a 804e74a0 8e838000 00000000
<4>[12234.108941]         00000000 00000014 00000000 00040002 8fc1de04 8fc1de3c 8f23e3c0 00000001
<4>[12234.117285]         00000004 00000003 02080020 8e83805c 804c75fc 803153a8 00000000 00000000
<4>[12234.125628]         00000000 00000000 8f23e3c0 8f23e3c0 8fc1de3c 00000001 fffffffb 8031543c
<4>[12234.133972]         ...
<4>[12234.136418] Call Trace:
<4>[12234.138855] [<8ed5c304>] ipt_do_table+0x130/0x7d0 [ip_tables]
<4>[12234.144589] [<803153a8>] nf_iterate+0x68/0xb0
<4>[12234.148926] [<8031543c>] nf_hook_slow+0x4c/0xe0
<4>[12234.153441] [<8031c54c>] ip_rcv+0x2fc/0x320
<4>[12234.157612] [<802dead0>] __netif_receive_skb_core+0x718/0x970
<4>[12234.163335] [<802e0aac>] process_backlog+0x9c/0x160
<4>[12234.168190] [<802e084c>] net_rx_action+0x138/0x2fc
<4>[12234.172971] [<8002e924>] __do_softirq+0x294/0x2e0
<4>[12234.177653] [<8002ec18>] irq_exit+0x7c/0x98
<4>[12234.181824] [<80209670>] plat_irq_dispatch+0xb4/0xdc
<4>[12234.186772] [<8000a968>] except_vec_vi_end+0xb8/0xc4
<4>[12234.191719] [<8000c324>] r4k_wait_irqoff+0x18/0x20
<4>[12234.196514] [<80064134>] cpu_startup_entry+0x104/0x170
<4>[12234.201651] [<800270b4>] not_nmi+0x8c/0x1d8
<4>[12234.205812] Code: 8ec20004  27a5002c  afa2002c <8c420028> afa30030  0040f809  02602025  104000fe  00000000 
<4>[12234.215550] 
<4>[12234.217038] Oops[#2]:
<4>[12234.217281] ---[ end trace 8cc326dfbb1faa61 ]---

===================================
Time: 1518499448.403892
<4>[12234.217281] ---[ end trace 8cc326dfbb1faa61 ]---
<0>[12234.224954] Kernel panic - not syncing: Fatal exception in interrupt
```

pparent76, 2018-02-26:
again:

> <1>[93953.846001] CPU 2 Unable to handle kernel paging request at virtual address 07406000, epc == 802f1338, ra == 8030d214
> <4>[93953.856627] Oops[#1]:
> <4>[93953.858900] CPU: 2 PID: 17309 Comm: chilli Tainted: G        W       4.9.77 #0
> <4>[93953.866089] task: 8eb405e0 task.stack: 8d56c000
> <4>[93953.870592] $ 0   : 00000000 80530008 80570000 00000000
> <4>[93953.875820] $ 4   : 811e2200 07406000 00000400 00000001
> <4>[93953.881035] $ 8   : 00000000 801e6710 8f7b04c0 00000003
> <4>[93953.886256] $12   : 0000049a 00000000 00000000 000000c5
> <4>[93953.891470] $16   : 8e8e5a00 8d730000 00000001 00000030
> <4>[93953.896691] $20   : 00000000 000000c0 00000200 8e2f3924
> <4>[93953.901906] $24   : 00000000 8005630c                  
> <4>[93953.907122] $28   : 8d56c000 8d56dd18 00000000 8030d214
> <4>[93953.912338] Hi    : 00000000
> <4>[93953.915204] Lo    : 00014400
> <4>[93953.918088] epc   : 802f1338 rtnl_kfree_skbs+0x18/0x24
> <4>[93953.923210] ra    : 8030d214 fq_codel_reset+0x64/0xc4
> <4>[93953.928233] Status: 11007c03      KERNEL EXL IE 
> <4>[93953.932407] Cause : 4080000c (ExcCode 03)
> <4>[93953.936396] BadVA : 07406000
> <4>[93953.939260] PrId  : 0001992f (MIPS 1004Kc)
> <4>[93953.943330] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie sch_gred act_ipt em_cmp sch_prio sch_codel em_text em_meta sch_sfq act_police sch_fq
> <4>[93954.013791]  sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
> Process chilli (pid: 17309, threadinfo=8d56c000, task=8eb405e0, tls=7731ddf4)
> <4>[93954.092283] Stack : 00000000 8004bf30 8d56dd20 804e74a0 ffffffea 8e8e5a00 8e8e5a70 8e240180
> <4>[93954.100634]         8048f0b8 80303490 804eb0c0 00000001 00000200 800471f0 8e8e5a00 80303b70
> <4>[93954.108979]         814a9c80 8feaa920 8e2f2000 8d56de18 8e2f2000 00000003 8d56de18 8030466c
> <4>[93954.117329]         00000000 8004bf30 00000001 00000000 804c75fc 803047a4 8d56ddcc 8d56de18
> <4>[93954.125678]         8e2f2000 8048f0b8 00000001 00000001 00000200 8e2f3924 00000000 802da6e0
> <4>[93954.134022]         ...
> <4>[93954.136462] Call Trace:
> <4>[93954.138904] [<802f1338>] rtnl_kfree_skbs+0x18/0x24
> <4>[93954.143678] [<8030d214>] fq_codel_reset+0x64/0xc4
> <4>[93954.148384] [<80303490>] qdisc_reset+0x24/0x5c
> <4>[93954.152809] [<80303b70>] dev_deactivate_queue.constprop.6+0x58/0xac
> <4>[93954.159055] [<8030466c>] dev_deactivate_many+0x84/0x2d0
> <4>[93954.164260] [<802da6e0>] __dev_close_many+0x88/0xe8
> <4>[93954.169119] [<802da7c4>] dev_close_many+0x84/0x11c
> <4>[93954.173893] [<802dd8b0>] rollback_registered_many+0x12c/0x438
> <4>[93954.179619] [<802ddbe8>] rollback_registered+0x2c/0x48
> <4>[93954.184740] [<802e0604>] unregister_netdevice_queue+0x7c/0xc4
> <4>[93954.190522] [<8f119c08>] tun_get_socket+0x1a50/0x3c5c [tun]
> <4>[93954.196082] Code: 3c028057  8c43c178  ac44c178 <aca30000> 03e00008  00000000  3c02804f  8c4281c8  38420001 
> <4>[93954.205824] 
> <4>[93954.207589] ---[ end trace 2838d2334964ea2d ]---
> 
> ===================================
> Time: 1519311805.48049

eoghan2t9, 2018-07-01:
Has anyone fixed this kernel panic as I get the same error on my Afoundry EW1200?

nbd168, 2018-10-25:
Please try the latest version (master only, not 18.06).

eoghan2t9, 2018-10-28:
I'll build and try again and let you know.

swg0101, 2019-01-06:
Here's one:

:~# opkg list | grep mt76
kmod-mt76-core - 4.14.90+2018-12-28-3598046d-1
kmod-mt7603 - 4.14.90+2018-12-28-3598046d-1
kmod-mt76x02-common - 4.14.90+2018-12-28-3598046d-1
kmod-mt76x2 - 4.14.90+2018-12-28-3598046d-1
kmod-mt76x2-common - 4.14.90+2018-12-28-3598046d-1

:~# cat /sys/kernel/debug/crashlog
Time: 1546749840.150939
Modules:        rt2800usb@8e328000+3620 rt2800lib@8e760000+1d90a        pppoe@8e4a4000+2290     ppp_async@8e448000+1be0 rt2x00usb@8e426000+1fb9 rt2x00lib@8e740000+8c33 rndis_host@8e3a0000+1429        pppox@8e3f3000+58a      ppp_generic@8e6d8000+5aa2    mt76x2e@8e3d4000+2a56   mt76x2_common@8e3bc000+2c88     mt76x02_lib@8e630000+9346       mt7603e@8e450000+901e   mt76@8e3d8000+61ff      mac80211@8e500000+6f2c1 iptable_nat@8e3c7000+310        ipt_MASQUERADE@8e3e7000+2f0 ebtable_nat@8e403000+3b0 ebtable_filter@8e422000+3b0     ebtable_broute@8e49a000+350     cp210x@8e37c000+29b0    ch341@8e44a000+11f0     cfg80211@8e4c0000+38d10 cdc_ether@8e442000+13a2 xt_u32@8e432000+430     xt_time@8e430000+730    xt_tcpmss@8e41f000+470       xt_string@8e41c000+310  xt_statistic@8e418000+390       xt_state@8e3e2000+310   xt_recent@8e424000+1c00 xt_quota@8e405000+350   xt_policy@8e402000+7f0  xt_pkttype@8e3ff000+290 xt_owner@8e3fc000+3b0   xt_nat@8e3ec000+650  xt_multiport@8e3e9000+550       xt_mark@8e3e6000+2f0    xt_mac@8e3e3000+2b0     xt_limit@8e3cf000+4f0   xt_length@8e3cc000+310  xt_iprange@8e3ca000+470 xt_hl@8e3c6000+370      xt_helper@8e3c3000+3b0  xt_hashlimit@8e3d0000+2880  xt_esp@8e3b1000+350      xt_ecn@8e3ae000+590     xt_dscp@8e3aa000+430    xt_conntrack@8e3a8000+950       xt_connmark@8e378000+530        xt_connlimit@8e3a2000+1320      xt_connbytes@8e306000+690       xt_comment@8e36b000+230 xt_bpf@8e368000+570  xt_addrtype@8e377000+970        xt_TRACE@8ef76000+250   xt_TEE@8f747000+600     xt_TCPMSS@8e06a000+b30  xt_REDIRECT@8e03f000+310        xt_NFQUEUE@8ef2f000+d90 xt_NFLOG@8e345000+2d0   xt_NETMAP@8e072000+650  xt_LOG@8e17d000+350 xt_LED@8ee45000+5b0      xt_HL@8e0e0000+590      xt_FLOWOFFLOAD@8e022000+b60     xt_DSCP@8e375000+630    xt_CT@8efa8000+ad0      xt_CLASSIFY@8e374000+290        xt_DELUDE@8efd9000+670  xt_TARPIT@8e36e000+1170 ipt_REJECT@8e0d7000+410 xt_tcpudp@8e2fc000+750       xt_CHAOS@8e360000+730   wireguard@8e380000+19f0f        usbserial@8e348000+4bcb usbnet@8e340000+4955    ums_usbat@8e330000+20d0 ums_sddr55@8e2f8000+1790        ums_sddr09@8e300000+2410        ums_karma@8e048000+6f0       ums_jumpshot@8eff4000+1010      ums_isd200@8e0c6000+1760        ums_freecom@8eef0000+8d0        ums_datafab@8e338000+1370       ums_cypress@8e35b000+a70        ums_alauda@8e33c000+2270        ts_kmp@8e35d000+570     ts_fsm@8e09f000+ab0  ts_bm@8e087000+5d0      slhc@8e02c000+149b      rtl8150@8e198000+1ef0   r8152@8e350000+99f0     nfnetlink_queue@8ef70000+2870   nfnetlink_log@8eff2000+1cb2     nf_reject_ipv4@8e0d6000+923     nf_nat_redirect@8e2cb000+5fb    nf_nat_masquerade_ipv4@8ec9d000+67c  nf_conntrack_ipv4@8e336000+1470 nf_nat_ipv4@8e334000+1031       nf_log_ipv4@8e0a2000+dd0        nf_flow_table_hw@8eff7000+920   nf_flow_table@8e320000+365f     nf_dup_ipv6@8efb8000+48d        nf_dup_ipv4@8ee7f000+44d     nf_defrag_ipv4@8ee17000+4b6     nf_conntrack_rtcache@8efbd000+af0       nf_conntrack_netlink@8e308000+57f0      iptable_raw@8e0bc000+2b0        iptable_mangle@8e0af000+3d0     iptable_filter@8efde000+2d0     ipt_ah@8ee5a000+310  ipt_ECN@8e103000+5d0    ip6table_raw@8efb2000+2b0       ip_tables@8efec000+2bcd ebtables@8efe0000+4155  ebt_vlan@8ef9c000+3f0   ebt_stp@8efa3000+790    ebt_snat@8efac000+390   ebt_redirect@8e151000+3b0       ebt_pkttype@8e14b000+270     ebt_nflog@8e01f000+2d0  ebt_mark_m@8efc9000+290 ebt_mark@8e028000+2f0   ebt_log@8ee79000+a40    ebt_limit@8e1b8000+420  ebt_ip6@8e15f000+6f0    ebt_ip@8efd6000+570     ebt_dnat@8efd3000+430   ebt_arpreply@8ef15000+4f0       ebt_arp@8e181000+6b0 ebt_among@8e180000+a50  ebt_802_3@8ef94000+310  crc_itu_t@8e0bb000+42b  crc_ccitt@8e0b8000+42b  compat_xtables@8efc1000+26b     compat@8ef80000+19b7    br_netfilter@8efcc000+3015      arptable_filter@8e0fb000+2b0    arpt_mangle@8ef77000+3d0     arp_tables@8efc4000+2461        act_skbedit@8ef9f000+a20        act_mirred@8ef9b000+f60 em_u32@8e187000+270     cls_u32@8efb4000+2340   cls_tcindex@8efae000+1510       cls_flow@8efaa000+16d0  cls_route@8ef8e000+14c0      cls_fw@8e2fd000+f70     sch_tbf@8ef88000+1730   sch_htb@8ef84000+3740   sch_hfsc@8ef7c000+3a70  sch_ingress@8ee04000+510        ledtrig_usbport@8e205000+b00    cryptodev@8ef60000+80e6 xt_set@8eef4000+1b70    ip_set_list_set@8ee86000+1cb0        ip_set_hash_netiface@8ef58000+6310      ip_set_hash_netport@8ef50000+6230       ip_set_hash_netnet@8ef48000+6750        ip_set_hash_net@8ef40000+5c70   ip_set_hash_netportnet@8ef38000+6c50    ip_set_hash_mac@8e09c000+2930ip_set_hash_ipportnet@8ef30000+67b0     ip_set_hash_ipportip@8ef28000+54f0      ip_set_hash_ipport@8e050000+5170        ip_set_hash_ipmark@8e038000+4e90        ip_set_hash_ip@8e060000+4db0    ip_set_bitmap_port@8e02e000+1830        ip_set_bitmap_ipmac@8e0e8000+1bd0    ip_set_bitmap_ip@8e06e000+1b30  ip_set@8e080000+594a    nfnetlink@8eeae000+1277 ip6t_NPT@8eeb6000+650   ip6t_MASQUERADE@8e11d000+270    nf_nat_masquerade_ipv6@8edc6000+71c     ip6table_nat@8eeef000+310   nf_conntrack_ipv6@8e06c000+15e0  nf_defrag_ipv6@8e076000+13be    nf_nat_ipv6@8ef24000+12d3       nf_nat@8e0fc000+3adc    nf_conntrack@8ef00000+11181     ip6t_rt@8e0cd000+670    ip6t_frag@ [... cut ...]
<4>[99451.839342]         8e3f4bc0 8cd61bc8 8fd5a000 8cd61c74 8ce8b780 8cd61cb4 8f769000 8e5212a4
<4>[99451.847674]         8cdbe000 8cdbe000 8e4f3048 8cd61cb4 00000004 8cd61c54 8e3f4000 8ccbee1c
<4>[99451.856005]         ...
<4>[99451.858441] Call Trace:
<4>[99451.860879] [<8010b3a4>] kmem_cache_alloc+0x128/0x17c
<4>[99451.865960] [<8e55a714>] minstrel_calc_rate_stats+0xc94/0x2b00 [mac80211]
<4>[99451.872743] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010
<4>[99451.882468]
<4>[99451.885000] ---[ end trace 50504bbf23a8b6d1 ]---

nbd168, 2019-01-25:
Please try the latest version, thanks.

closed 2019-03-04 by nbd168

pparent76, 2019-03-04:
All my testing shows that the bug is no longer happening.  Thank's so much!