openwrt/mt76 #122: t76_dma_attach+0xb6c/0xc9c [mt76]

kofec, 2017-10-15:
[52575.828383] ------------[ cut here ]------------
[52575.833224] WARNING: CPU: 0 PID: 1490 at /mnt/openwrt/LEDEXiaomiDangole/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2017-10-12-37654d70/mt76x2_mac.c:254 mt76x2_mac_process_rx+0xb4/0x2d0 [mt76x2e]
[52575.852312] Modules linked in: pppoe ppp_async rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_hashlimit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_itu_t crc_ccitt hid evdev input_core mt76x2e mt7603e i2c_dev i2c_core mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables
[52575.924887]  x_tables ip6_udp_tunnel udp_tunnel tun vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache usbcore nls_base usb_common crc16 aead crypto_null crc32c_generic cryptomgr crc32_generic crypto_hash
[52575.958281] CPU: 0 PID: 1490 Comm: hostapd Not tainted 4.9.54 #0
[52575.964422] Stack : 8042751a 00000034 00000000 00000001 87d4fb3c 803d9207 8038b120 000005d2
[52575.972992]         80423660 000000fe 00000001 00000000 87005900 8004d640 803908e4 803d0000
[52575.981558]         00000003 000000fe 8038ead4 87c0dde8 87005900 8007b2dc 8042751a 000000c6
[52575.990125]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[52575.998679]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[52576.007239]         ...
[52576.009760] Call Trace:
[52576.012281] [<8000e478>] show_stack+0x54/0x88
[52576.016747] [<80023fc0>] __warn+0xe4/0x118
[52576.020949] [<80024088>] warn_slowpath_null+0x1c/0x34
[52576.026139] [<870241b8>] mt76x2_mac_process_rx+0xb4/0x2d0 [mt76x2e]
[52576.032574] [<87020578>] mt76x2_queue_rx_skb+0x84/0xcc [mt76x2e]
[52576.038728] [<877f4e3c>] mt76_dma_attach+0xb6c/0xc9c [mt76]
[52576.044437] ---[ end trace b6226a97328750fe ]---

MaFeSa, 2017-10-16:
```
[352679.006132] ------------[ cut here ]------------
[352679.015522] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x258/0x2fc()
[352679.032501] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[352679.046526] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_mppe ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_cake act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt76x2e mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc32c_generic
[352679.273156] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.4.89 #0
[352679.285096] Stack : 00000000 00000000 804d6862 00000033 00000000 00000000 80480000 804f0000
[352679.285096] 	  87c4bf6c 8047bca3 803f57f0 00000003 00000000 804d367c ffffffff 00000200
[352679.285096] 	  00100000 800651c8 80480000 804f0000 80480318 8048031c 803fa12c 87c21e04
[352679.285096] 	  00000003 80062f14 ffffffff 00000200 00100000 00000000 00000006 00c21e04
[352679.285096] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[352679.285096] 	  ...
[352679.355975] Call Trace:
[352679.361013] [<800169e8>] show_stack+0x54/0x88
[352679.369865] [<801b763c>] dump_stack+0x84/0xbc
[352679.378712] [<8002cfb8>] warn_slowpath_common+0xa0/0xd0
[352679.389275] [<8002d014>] warn_slowpath_fmt+0x2c/0x38
[352679.399322] [<802d090c>] dev_watchdog+0x258/0x2fc
[352679.408865] [<8007427c>] call_timer_fn.isra.4+0x24/0x80
[352679.419431] [<800744d4>] run_timer_softirq+0x1fc/0x25c
[352679.429827] [<8002fbcc>] __do_softirq+0x294/0x2e0
[352679.439355] [<8002feb4>] irq_exit+0x78/0x94
[352679.447852] [<801ecb14>] plat_irq_dispatch+0xb4/0xdc
[352679.457890] 
[352679.461071] ---[ end trace 4a4b77cf639b3759 ]---
[352679.470435] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[352679.482914] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000067
[352679.495045] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=06f12000, max=512, ctx=363, dtx=356, fdx=356, next=363
[352679.516517] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=06f2c000, max=512, calc=447, drx=448
[354240.005616] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[354240.018110] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000067
[354240.030256] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=06f12000, max=512, ctx=50, dtx=28, fdx=28, next=50
[354240.051038] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=05094000, max=512, calc=243, drx=244
```

DIR-860L B1 LEDE 17.01.3 r3540-f67c22e

MaFeSa, 2017-11-30:
For me is fixed, I no longer have this error.
Thanks

closed 2017-12-03 by nbd168