openwrt/mt76 #123: warnings on MT7628AN

dangowrt, 2017-10-18:
```
[ 5229.581876] ------------[ cut here ]------------
[ 5229.586642] WARNING: CPU: 0 PID: 1569 at /var/lib/buildbot/slaves/tictex-02/ramips_mt76x8/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2017-10-12-37654d70/mt7603_mac.c:1285 mt7603_mac_work+0x12c/0x24c [mt7603e]
[ 5229.607389] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt mt76x2e mt7603e mt76 mac80211 batman_adv libcrc32c cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables crc16 crc32c_generic crypto_hash leds_gpio gpio_button_hotplug
[ 5229.670490] CPU: 0 PID: 1569 Comm: kworker/u2:0 Not tainted 4.9.54 #0
[ 5229.677055] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 5229.682189] Stack : 83803414 830e9918 00000088 8004d640 83995c4c 80401307 803b1fe0 00000621
[ 5229.690693]         803b1f4c 8384bd64 80400000 8007cc5c 00000088 8004d640 803b77a4 80400000
[ 5229.699210]         00000003 8384bd64 80400000 80038e70 00000088 8384bd9c 000002ae 00000000
[ 5229.707718]         00000000 00000000 83195534 83baae00 83baad00 30796870 00000000 00000000
[ 5229.716234]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5229.724744]         ...
[ 5229.727227] Call Trace:
[ 5229.729731] [<8000e478>] show_stack+0x54/0x88
[ 5229.734184] [<80023fc0>] __warn+0xe4/0x118
[ 5229.738344] [<80024088>] warn_slowpath_null+0x1c/0x34
[ 5229.743500] [<83195660>] mt7603_mac_work+0x12c/0x24c [mt7603e]
[ 5229.749434] [<80036eb8>] process_one_work+0x1ec/0x32c
[ 5229.754578] [<80037c78>] worker_thread+0x2ac/0x420
[ 5229.759440] [<8003c4a4>] kthread+0xd8/0xec
[ 5229.763613] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[ 5229.769087] ---[ end trace 1083190065f76c2e ]---
[ 5232.061799] ------------[ cut here ]------------
[ 5232.066553] WARNING: CPU: 0 PID: 1569 at /var/lib/buildbot/slaves/tictex-02/ramips_mt76x8/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2017-10-12-37654d70/mt7603_mac.c:1279 mt7603_mac_work+0x12c/0x24c [mt7603e]
[ 5232.087287] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt mt76x2e mt7603e mt76 mac80211 batman_adv libcrc32c cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables crc16 crc32c_generic crypto_hash leds_gpio gpio_button_hotplug
[ 5232.150451] CPU: 0 PID: 1569 Comm: kworker/u2:0 Tainted: G        W       4.9.54 #0
[ 5232.158247] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 5232.163382] Stack : 83803414 830e9918 00000088 8004d640 83995c4c 80401307 803b1fe0 00000621
[ 5232.171897]         803b1f4c 8384bd64 80400000 8007cc5c 00000088 8004d640 803b77a4 80400000
[ 5232.180398]         00000003 8384bd64 80400000 80038e70 00000088 8384bd9c 000002c3 00000000
[ 5232.188912]         804130d0 00000000 83195534 83baae00 83baad00 30796870 00000000 00000000
[ 5232.197429]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5232.205942]         ...
[ 5232.208425] Call Trace:
[ 5232.210929] [<8000e478>] show_stack+0x54/0x88
[ 5232.215383] [<80023fc0>] __warn+0xe4/0x118
[ 5232.219543] [<80024088>] warn_slowpath_null+0x1c/0x34
[ 5232.224701] [<83195660>] mt7603_mac_work+0x12c/0x24c [mt7603e]
[ 5232.230636] [<80036eb8>] process_one_work+0x1ec/0x32c
[ 5232.235781] [<80037c78>] worker_thread+0x2ac/0x420
[ 5232.240677] [<8003c4a4>] kthread+0xd8/0xec
[ 5232.244863] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[ 5232.250337] ---[ end trace 1083190065f76c2f ]---
```

cj1324, 2017-11-28:
Similar problem

```
# cat /proc/cpuinfo
system type             : MediaTek MT7628AN ver:1 eco:2
machine                 : GL-MT300N-V2
processor               : 0
cpu model               : MIPS 24KEc V5.5
BogoMIPS                : 380.92
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
```
```
[275490.520296] ------------[ cut here ]------------
[275490.525122] WARNING: CPU: 0 PID: 9 at /opt/LEDE/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2017-11-16-8f1c8abd/mt7603_mac.c:1288 mt7603_mac_work+0x12c/0x24c [mt7603e]
[275490.542332] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_esp xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ipt_ah ip_tables crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port
[275490.615369]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 tunnel6 tunnel4 tun af_key xfrm_user xfrm_ipcomp xfrm_algo sha256_generic sha1_generic drbg md5 hmac echainiv des_generic deflate zlib_inflate zlib_deflate cbc authenc leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common aead cryptomgr crypto_null crypto_hash
[275490.674492] CPU: 0 PID: 9 Comm: kworker/u2:1 Tainted: G        W       4.9.58 #0
[275490.682105] Workqueue: phy0 mt7603_mac_work [mt7603e]
[275490.687314] Stack : 87c03414 87cae518 00000088 8004d640 87c2ebc4 803eb2e7 8039d4c0 00000009
[275490.695921]         8039d42c 87cc5d64 803f0000 8007b31c 00000088 8004d640 803a2cb4 803f0000
[275490.704575]         00000003 87cc5d64 803f0000 80038e70 00000088 87cc5d9c 0000015d 00000000
[275490.713180]         803fd080 00000000 876ad604 87f52c00 8742af00 30796870 00000000 00000000
[275490.721783]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[275490.730383]         ...
[275490.732952] Call Trace:
[275490.735542] [<8000e478>] show_stack+0x54/0x88
[275490.740070] [<80023fc0>] __warn+0xe4/0x118
[275490.744334] [<80024088>] warn_slowpath_null+0x1c/0x34
[275490.749556] [<876ad730>] mt7603_mac_work+0x12c/0x24c [mt7603e]
[275490.755589] [<80036eb8>] process_one_work+0x1ec/0x32c
[275490.760815] [<80037c78>] worker_thread+0x2ac/0x420
[275490.765764] [<8003c4a4>] kthread+0xd8/0xec
[275490.770012] [<80009378>] ret_from_kernel_thread+0x14/0x1c
[275490.775587] ---[ end trace e8b80deace26abc5 ]---

```

MaFeSa, 2017-11-30:
Seems fixed with latest commits

closed 2017-12-03 by nbd168