openwrt/mt76 #128: How to use MT7612E with build-in eFuse?

wwng2333, 2017-11-28:
Sorry for my bad English.
I have a MT7612E MiniPCIe card, with PA/LNA.But works badly on openwrt.
I contacted the 7612E wireless card manufacturer, the factory has written the data to the on-chip efuse, this area is one-time, cannot be erased after the write. 
So I want to know how to make the driver use build-in efuse instead of reading EEPROM.

psyborg55, 2017-11-29:
by dropping eeprom entry from dts. but you can also cat debug contents of efuse data and compare with eeprom data

wwng2333, 2017-11-30:
Got it,by the way, can you tell me how to get and export data from eFuse?

psyborg55, 2017-11-30:
no i forgot. try searching for my previous comments on github maybe it was written

anonimou0, 2019-02-19:
@wwng2333 , did you get your MT7612E to use the build-in efuse?
Would you mind sharing details on how to do it?

wwng2333, 2019-02-19:
@anonimou0 i also dont know how to do it.

paullai58, 2019-08-21:
Is there any one could help to explain below.
In mt76x02_eeprom.h 

enum mt76x02_eeprom_modes {
        MT_EE_READ,
        MT_EE_PHYSICAL_READ,
};

what different is between MT_EE_READ and MT_EE_PHYSICAL_READ?

Thanks.