openwrt/mt76 #129: 2.4 GHz HT40 - RX Channel width

MaFeSa, 2017-12-01:
Tested with Xiaomi Mi Router 3G and D-Link DIR-860L with LEDE Reboot SNAPSHOT r5438-e966ed2 / LuCI Master (git-17.316.07773-4891dea)

There aren't other nearby networks
[image]

RX channel width stay always locked at 20 MHz also with option noscan 1 and many data sent/received.
[image]

Clients tested:
Lenovo 700-15ISK with Intel AC3165 driver 19.51.8.2
TP-LINK TL-WN881ND v2
D-Link DVA-140 
Honor 6X

MaFeSa, 2017-12-04:
Not solved with the last commits 
https://github.com/openwrt/mt76/commit/d02a05b0b37cf6d6d966a037452acb0cc8ace4dd

nbd168, 2017-12-04:
Is this something that worked for you in earlier versions?

MaFeSa, 2017-12-04:
No I have always had this problem, this morning I also tried without success with a Dell Inspirion 5770

My config
```

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country '00'
	option noscan '1'
	option channel '1'
	option htmode 'HT40'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'psk2+ccmp'
	option key 'password'
	option disassoc_low_ack '0'
	option hidden '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country '00'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option key 'password'
	option ssid 'LEDE 5GHz'
	option hidden '1'
```

Kernel log
```
[    0.000000] Linux version 4.9.65 (massy@massy-linux) (gcc version 5.5.0 (LEDE GCC 5.5.0 r5389-358df5e) ) #0 SMP Sun Dec 3 18:06:24 2017
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3G
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 804fa560, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 12 pages/cpu @8120f000 s17840 r8192 d23120 u49152
[    0.000000] pcpu-alloc: s17840 r8192 d23120 u49152 alloc=12*4096
[... 312 lines cut ...]
[   21.049819] br-lan: port 1(eth0.1) entered disabled state
[   21.056260] device eth0.1 entered promiscuous mode
[   21.069307] br-lan: port 1(eth0.1) entered blocking state
[   21.074763] br-lan: port 1(eth0.1) entered forwarding state
[   21.887051] pppoe-wan: renamed from ppp0
[   23.324058] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   23.747297] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   23.753291] br-lan: port 2(wlan0) entered blocking state
[   23.758665] br-lan: port 2(wlan0) entered disabled state
[   23.764564] device wlan0 entered promiscuous mode
[   23.777140] br-lan: port 3(wlan1) entered blocking state
[   23.782681] br-lan: port 3(wlan1) entered disabled state
[   23.789192] device wlan1 entered promiscuous mode
[   23.794249] br-lan: port 3(wlan1) entered blocking state
[   23.799671] br-lan: port 3(wlan1) entered forwarding state
[   24.329459] br-lan: port 3(wlan1) entered disabled state
[   24.335110] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   24.341690] br-lan: port 2(wlan0) entered blocking state
[   24.347100] br-lan: port 2(wlan0) entered forwarding state
[   24.999381] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   25.006132] br-lan: port 3(wlan1) entered blocking state
[   25.011482] br-lan: port 3(wlan1) entered forwarding state
```

Many thanks!

MaFeSa, 2017-12-04:
Tried now with D-Link DIR-860L B1 and LEDE 17.01.4
Same configuration and devices but with this router is perfect
[image]

So the problem is with Xiaomi 3G and 7603

MaFeSa, 2017-12-08:
Also a TL-WR1043ND v3 as client with LEDE 17.04.1 can't get 300 Mbit/s of RX rate (only 150 at 20MHz), TX rate is fine 300 Mbit/s at 40 MHz
Xiaomi 3G LEDE Reboot SNAPSHOT r5445-04127f0

MaFeSa, 2017-12-08:
Compiled with latest commits but same problem 
OpenWrt %N SNAPSHOT r5483-abdf0de

Why there is OpenWrt?

iancd, 2017-12-13:
I see similar behaviour with latest snapshot build and 7603. I used the scan function and other diagnostic tools to determine there are no overlapping 2.4GHz channels for either primary or secondary choices in my area. When I set to 40Mhz channel width the channel will stay locked at 20Mhz and I'll see the errors below about overlapping BSSes. My other two Atheros and Marvell based APs work fine with 40Mhz width.

My AP
`Linux version 4.9.67 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5484-69d22a6)
SoC Type: MediaTek MT7621 ver:1 eco:3
mt7603e 0000:02:00.0: ASIC revision: 76030010
mt7603e 0000:02:00.0: Firmware Version: ap_pcie
mt7603e 0000:02:00.0: Build Time: 20160107100755`

Error using channel 4/40Mhz:
`daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=4 sec=8 based on overlapping BSSes`

Switching to channel 3/40Mhz - same behaviour
`daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=3 sec=7 based on overlapping BSSes`

MaFeSa, 2017-12-13:
@iancd You can try with "option noscan 1' to force 40 MHz but as said before RX remain anyway at 20 MHz

iancd, 2017-12-13:
@MaFeSa Thanks I had forgotten about that option, deleted my comment above .

lukasz-michalski, 2017-12-14:
@nbd168 in mt7603_set_channel and in mt7603_mcu_set_channel there are hardcoded MT_BW_20 values. Maybe that is the reason?

MaFeSa, 2017-12-15:
Tested now this commit https://github.com/openwrt/mt76/commit/00f7e60e985520f438a70cf7d64c8fba608a6c27
Now clients gets full speed but LuCI indicates always RX 20MHz, is a visual bug?
[image]

MaFeSa, 2017-12-19:
I tried many wireless cards as a client and RX width stay always locked at 20 MHz, I don't know where is the problem... I did a lot of tests with Iperf.

OpenWrt SNAPSHOT r5521-9f8d282 / LuCI Master (git-17.328.04231-802d5b6)

kofec, 2017-12-19:
maybe this is related with stacktrace which I have:
[ 1292.930907] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4314 ieee80211_rx_napi+0x1a4/0x964 [mac80211] [ 1292.941932] **Rate marked as a VHT rate but data is invalid:** MCS: 66, NSS: 0 [ 1292.948790]
Maybe some parameters are not correcly exxhanged

MiltonLai, 2017-12-22:
@MaFeSa It seems the same problem on newifi-d2 (the same chip set). The 2.4g connection is poor in the latest firmware(r5596).

MaFeSa, 2017-12-22:
> The 2.4g connection is poor 
What does poor mean?
I have no speed problems and the connection is stable even with many clients.
The problem is only 20 MHz width RX

lukasz1992, 2018-01-18:
If it did not help, as users report on LEDE forum.

I tried to find differences between mt7603 and mt76x2 drivers, and think about these 3 options:
1) No status->bw is set.

mt76x2_mac_process_rate sets status->bw in line
`switch (FIELD_GET(MT_RXWI_RATE_BW, rate)) {`
In mt7603 I think the similar place could be in mt7603_mac_fill_rx.

2) mt7603_wtbl_update_cap updates Short GI information, but no bandwidth information

3) Maybe the problem is in logic in mt7603_wtbl_set_rate - but I don't know the way it works.
Anyway `w9 |= FIELD_PREP(MT_WTBL2_W9_CHANGE_BW_RATE, bw_idx ? bw_idx - 1 : 7);` in Mediatek code looks diffrent - MT_WTBL2_W9_CHANGE_BW_RATE is always "set" to 7

I think it is better if someone here who have knowledge about this chip could verify my ideas (based on guessing)

lukasz1992, 2018-01-22:
I studied Mediatek sources - in ParseRxVPacket in cmm_data.c, there is a line:

`RxBlk->rx_rate.field.BW = RXV1_1ST_CYCLE->FrMode;`

In mt76, this same field is named MT_RXV1_FRAME_MODE. So I assume adding line:

`status->bw = (FIELD_GET(MT_RXV1_FRAME_MODE, rxdg0)) & 1;`

in mt7603_mac_fill_rx before `rxd += 6` should fix reporting 40MHz.

MaFeSa, 2018-01-22:
@lukasz1992 
I added `status->bw = (FIELD_GET(MT_RXV1_FRAME_MODE, rxdg0)) & 1;` before `rxd += 6`, the file is  mt7603_mac.c but the problem remains.
I did two iPerf tests and RX is always blocked at 20 MHz

Thanks

lukasz1992, 2018-01-22:
@MaFeSa Could you post outputs of:

> cat /etc/config/wireless
> iw dev
> iw phy0 info
> dmesg

MaFeSa, 2018-01-22:
Sure!

cat /etc/config/wireless
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country '00'
	option noscan '1'
	option channel '1'
	option htmode 'HT40'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'psk2+ccmp'
	option key 'password'
	option disassoc_low_ack '0'
	option hidden '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country '00'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option key 'password'
	option ssid 'LEDE 5GHz'
	option hidden '1'
```
iw dev
```
phy#1
        Interface wlan1
                ifindex 19
                wdev 0x100000002
                addr 34:ce:00:64:84:3e
                ssid LEDE 5GHz
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 20.00 dBm
phy#0
        Interface wlan0
                ifindex 17
                wdev 0x2
                addr 34:ce:00:64:84:3d
                ssid LEDE
                type AP
                channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
```
iw phy0 info
```
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
[... 379 lines cut ...]
[   19.980528] device eth0 entered promiscuous mode
[   19.987662] br-lan: port 1(eth0.1) entered blocking state
[   19.993146] br-lan: port 1(eth0.1) entered disabled state
[   19.999702] device eth0.1 entered promiscuous mode
[   20.012420] br-lan: port 1(eth0.1) entered blocking state
[   20.017868] br-lan: port 1(eth0.1) entered forwarding state
[   20.023953] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   20.780896] pppoe-wan: renamed from ppp0
[   21.143396] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   22.487568] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   22.502130] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   22.508229] br-lan: port 2(wlan1) entered blocking state
[   22.513602] br-lan: port 2(wlan1) entered disabled state
[   22.520034] device wlan1 entered promiscuous mode
[   22.526494] br-lan: port 3(wlan0) entered blocking state
[   22.531849] br-lan: port 3(wlan0) entered disabled state
[   22.538186] device wlan0 entered promiscuous mode
[   22.543311] br-lan: port 3(wlan0) entered blocking state
[   22.548737] br-lan: port 3(wlan0) entered forwarding state
[   23.067688] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   23.729952] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   23.736716] br-lan: port 2(wlan1) entered blocking state
[   23.742044] br-lan: port 2(wlan1) entered forwarding state

```

MaFeSa, 2018-01-25:
I'm testing also commits after 12/17/2017 without success.

> Could you also run AP on mt7602 device and from the client side do iw dev <device> scan - so we could compare results?
If you can - please also provide short dump (with airodump-ng) of beacons from both devices.

I'm not practical of airodump-ng :(

nbd168, 2018-01-25:
rx reporting of 40 MHz channel bandwidth is fixed in the latest version

closed 2018-01-25 by nbd168

MaFeSa, 2018-01-25:
Many thanks 😃

camelcamro, 2018-02-18:
hmm, still getting every day a lot of kernel errors like:
with that wireless config: (mt7621 - zbt-3526)
newest trunk

wireless config:
==========
config wifi-iface 'default_radio1'
<------>option device 'radio1'
<------>option mode 'ap'
<------>option encryption 'psk2+ccmp'
<------>option ifname 'wlan0'
<------>option key 'we_love_nbd168:)'
<------>option network 'lan'
<------>option ssid 'NVR'
config wifi-device 'radio1'
<------>option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
<------>option type 'mac80211'
<------>option hwmode '11g'
<------>option country 'US'
<------>option noscan '1'
<------>option legacy_rates '1'
<------>option channel '11'
<------>option txpower '0'

kernel error:
======== 
`Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.311387] ------------[ cut here ]------------
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.316194] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4308 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.327244] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 101 (0x65)
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.336694] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.408082]  nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat f2fs ext4 jbd2 mbcache nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb crc32c_generic crc32_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_coSun Feb 18 04:41:19 2018 kern.warn kernel: [25657.503676] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.77 #0
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.510774] Stack : 00000000 00000000 80557b4a 00000041 8040fc04 00000000 00000000 80550000
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.519118]         804f62bc 804f5ea7 8048cd04 00000000 00000000 80553824 00000030 0000099e
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.527460]         8ea28dd0 8006ab98 00000001 80550000 804fc004 804fc008 80491910 8fc0dc7c
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.535803]         00000003 800a7a50 00000030 0000099e 81490e20 00000000 00000002 00c0dc7c
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.544145]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.552486]         ...
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.554923] Call Trace:
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.557381] [<8000f714>] show_stack+0x54/0x88
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.561724] [<801e5d9c>] dump_stack+0x8c/0xd0
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.566061] [<8002adc4>] __warn+0xe4/0x118
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.570136] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.575129] [<8eda5c7c>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.581235] [<8ed11b78>] mt76_rx_complete+0x18c/0x278 [mt76]
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.586871] [<8ed11e24>] mt76_rx_poll_complete+0x1c0/0x260 [mt76]
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.592941] [<8ed10e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
Sun Feb 18 04:41:19 2018 kern.warn kernel: [25657.598550] ---[ end trace 2b88dae7b13dd42f ]---
Sun Feb 18 04:44:04 2018 kern.info kernel: [25822.444096] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
Sun Feb 18 04:44:07 2018 kern.info kernel: [25825.649695] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up`

or do we have to add manually the option: option disassoc_low_ack '0' to wireless config ?

cu camel

MaFeSa, 2018-02-18:
> option key 'we_love_nbd168:)' 😄 

No disassoc_low_ack '0' is only for disconnections

This is another things...
https://github.com/openwrt/mt76/issues/138