openwrt/mt76 #131: Clients keeps disconnecting due to "excessive missing ACKs"

guisenges, 2017-12-08:
I've just tested latest commit e50465605266118060df1b86eb7641c65fd29c65 with LEDE snapshot version on my TL-WR840N v4 router. I've tested wifi stability using 3 parallel Ubuntu ISO downloads from 3 different laptops and logread messages keeps informing hostapd "IEEE 802.11: disconnected due to excessive missing ACKs". Connection from these 3 laptops often hangs after some of these messages.

Any debug suggestions? My wireless uci config is the following:

config wifi-device 'radio0'
option type 'mac80211'
option channel '11'
option hwmode '11n'
option path 'platform/10300000.wmac'
option htmode 'HT40'
option country 'BR'
option txpower '20'
option disabled '0'

config wifi-iface 'default_radio0'
option device 'radio0'
option network 'lan'
option mode 'ap'
option ssid 'LEDE'
option encryption 'psk2'
option key '1234567890'

TL-WR840N v4 uses MT7628 chipset

dibg, 2018-02-12:
I have the same problem with xiaomi nano and lege 17.01.4, any updates?

karlp, 2018-03-05:
same on miwifi mini, and snapshot openwrt from LEDE_RELEASE="OpenWrt SNAPSHOT r6380-606cea8"

(client is a macbook, phones seem to be fine, and it's on the 5gig handled by the mt76.

Zetanova, 2018-04-05:
For me the same.
With ANI off, it happends frequently.

TP-Link TL-WR1043N/ND v2
LEDE Reboot 17.01.4 r3560-79f57e422d

Zetanova, 2018-04-16:
thx, I am testing now on OpenWrt SNAPSHOT r6690-cd7878e 
for 24h and there is no issue until now.

Will write in two days again.

Zetanova, 2018-04-18:
seams that the problem got resolved.
@nbd168 thx

closed 2018-04-18 by nbd168

LordMrcS, 2019-02-16:
can confirm having this problem with TP-Link TL-WR941N/ND v2