openwrt/mt76 #132: tx power low

Rascal6, 2017-12-12:
ZBT-WG3526   MT7612E +  MT7603E 

on native firmware tx power was higher

5 GHz does not exceed 11 dbm
[image]

2.4GHz does not exceed 0 dbm
[image]

nbd168, 2017-12-12:
The 20 dBm reported by that other firmware is fake. The hardware is not capable of doing 20 dBm. Reported power is now lower, because I fixed reporting the real target tx power of the device.

closed 2017-12-12 by nbd168

Rascal6, 2017-12-12:
maximally 11?
when there was a native firmware, the power was much better

Rascal6, 2017-12-12:
2,4 - max 0 ?

nbd168, 2017-12-12:
For 2.4 GHz, reporting is not implemented yet. From what I can tell, 11 dBm is the limit for VHT80. I have such a device myself, and I've run some tests with both the proprietary driver and my reimplementation. Signal strength on the client looks exactly the same in both cases.