openwrt/mt76 #135: mt7603 continues to interrupt MT7621 cpu after radio0 is disabled

iancd, 2017-12-15:
Test Device:
OpenWrt GCC 5.5.0 r5484-69d22a6 (buildbot)
MediaTek MT7621 Netgear WNDR3700v5

Steps:
1. mt7603/radio0 is enabled only in Luci
2. Run a iPerf3 throughput test to saturate mt7603/radio0
3. Disable mt7603/radio0 in Luci and verify using `iwinfo`
4. Enable mt76x2/radio1 in Luci and verify using `iwinfo`
5. Run another iPerf test to saturate mt76x2/radio1
6. Continually check /proc/interrupts

The mt7603 processor interrupts increment when radio0 is disabled.
If I repeat the steps using the mt76x2/radio1 the mt76x2 processor interrupts **do not** increment in /proc/interrupts when it's radio1 is disabled

Is it expected behaviour for the mt7603 driver to interrupt even when its radio is disabled?

lukasz1992, 2018-01-18:
https://github.com/LorenzoBianconi/mt76/commit/d8a2df530c90462c58a6118f09b1d3e08b94593c#diff-6e87cb0782640c667ceff9a580059d6c ?

iancd, 2018-01-23:
@lukasz1992 I see your commit has been pulled to master thank you. I'll test this once I get a lede buildbot image with the latest mt76 pull.

barhom, 2018-02-06:
@iancd 
If the issue is fixed, then I propose to close this ticket.

iancd, 2018-02-06:
@lukasz1992 @barhom The latest mt76 version was just committed to the LEDE trunk yesterday. I was able to repeat the steps above and test the new version. Unfortunately this issue still exists in the latest image.  The mt7603e continues to increment processor interrupts after disabled in LUCI and the mt76x2e **does not**. If you see anyone sees commits which may fix this in the future let me know and I'll retest and will close if the issue is fixed I will close this issue.

`Linux version 4.9.77 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5996-256477f) ) #0 SMP Mon Feb 5 11:58:27 2018
Tue Feb  6 20:04:35 2018 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3`
```[   11.895858] mt7603e 0000:02:00.0: ASIC revision: 76030010
[   11.941484] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
[   11.952672] mt7603e 0000:02:00.0: Build Time: 20160107100755
[   11.997164] firmware init done
[   12.171072] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   12.179833] mt76x2e 0000:01:00.0: ASIC revision: 76120044
[   12.217262] mt76x2e 0000:01:00.0: ROM patch already applied
[   12.229812] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[   12.240834] mt76x2e 0000:01:00.0: Build: 1
[   12.249009] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[   12.287217] mt76x2e 0000:01:00.0: Firmware running!```

barhom, 2018-02-12:
[image]

@iancd 
Is the following the interrupts you mean?

@nbd168 
Is it expected that interrupts for mt7603e should increase even though its disabled?

nbd168, 2018-02-12:
That's expected. I'm not aware of any reliable way to shut down the radio and bring it back up aside from shutting down the firmware and DMA (which seems a bit excessive)

nbd168, 2018-02-12:
I will continue to look into this to see if there's a way to shut off rx somehow without destabilizing the system

iancd, 2018-02-12:
@barhom Yes these are the interrupts I see. I was investigating the impact of the CPU limitation of the mt7621 platform on wireless throughput for streaming HD video.
@nbd168 If I disabledthe mt76x2e radio in Luci the corresponding interrupt thread also stops but this behaviour is different for the mt7603e. I assumed the mt76x2e driver was more mature but it might be something else.
So currently if the AP is set to dual-band then the mt7603 potentially could be using CPU resources needlessly which could be used by the mt76x2e for higher throughput. No idea if that is the case based on the interrupts.

lukasz1992, 2018-02-13:
My quick workaround idea, easy to implement:

Change channel to 14 in mt7603_mac_stop

lukasz1992, 2018-02-13:
@iancd @nbd168 
I tested on Xiaomi R3G router (LEDE 17.01.4), there is **no** such problem here.

```
root@LEDE:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  8:     102413     102392     102390     102381  MIPS GIC Local   1  timer
 10:     480327          0          0          0  MIPS GIC  10  1e100000.ethernet
 11:     995212          0          0          0  MIPS GIC  11  mt7603e
 29:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 30:          4          0          0          0  MIPS GIC  30  gsw
 31:     185687          0          0          0  MIPS GIC  31  mt76x2e
 33:         12          0          0          0  MIPS GIC  33  serial
 63:       7538          0          0          0  MIPS GIC  63  IPI call
 64:          0      78976          0          0  MIPS GIC  64  IPI call
 65:          0          0     204669          0  MIPS GIC  65  IPI call
 66:          0          0          0     113985  MIPS GIC  66  IPI call
 67:       9887          0          0          0  MIPS GIC  67  IPI resched
 68:          0      22839          0          0  MIPS GIC  68  IPI resched
 69:          0          0      76517          0  MIPS GIC  69  IPI resched
 70:          0          0          0      79025  MIPS GIC  70  IPI resched
ERR:     100721
```

...
```

root@LEDE:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  8:     102905     102884     102882     102873  MIPS GIC Local   1  timer
 10:     480328          0          0          0  MIPS GIC  10  1e100000.ethernet
 11:     995212          0          0          0  MIPS GIC  11  mt7603e
 29:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 30:          4          0          0          0  MIPS GIC  30  gsw
 31:     185938          0          0          0  MIPS GIC  31  mt76x2e
 33:         12          0          0          0  MIPS GIC  33  serial
 63:       7573          0          0          0  MIPS GIC  63  IPI call
 64:          0      79028          0          0  MIPS GIC  64  IPI call
 65:          0          0     204729          0  MIPS GIC  65  IPI call
 66:          0          0          0     114007  MIPS GIC  66  IPI call
 67:       9973          0          0          0  MIPS GIC  67  IPI resched
 68:          0      22954          0          0  MIPS GIC  68  IPI resched
 69:          0          0      76584          0  MIPS GIC  69  IPI resched
 70:          0          0          0      79058  MIPS GIC  70  IPI resched

```

iancd, 2018-02-14:
@lukasz1992 Thanks for your test. I'm using snapshot trunk builds because the drivers have been enhanced over the stable 17.01.4 versions. Just to be clear this is not a serious problem I just thought the behaviour was unusual. I looked over this again and create a gif to show the issue and updated the test steps for anyone wanting to replicate what I've found using latest trunk build .
Just to be clear this unexpected behaviour is only happening when using Luci so this might be a LEDE thing.

lukasz1992, 2018-02-15:
@iancd my check is with that commit https://github.com/openwrt/mt76/commits/lede-17.01
Actually I disabled radio by command `ifconfig wlan1 down`. Did not checked other options.

iancd, 2018-02-15:
@lukasz1992 Yep it seems like disabling the mt7063 radio via Luci is different from ifconfig wlan0 down. No idea why.

barhom, 2018-02-16:
uci set wireless.radio0.disabled=0 && uci commit && wifi reload
This disabled mt7603 for me and interrupts is NOT increasing.

iancd, 2018-04-15:
@barhom I just checked this issue is still present in the latest mt76.  I think you are correct about wifi-iface" level and not "wifi-device" level for Luci Wifi interfaces page. Do you think I should create a GitHub issue about this for the Luci devs? Using the wifi-device level to disable plays very nicely with Luci so I'm not sure why it is not used. Also still not sure why mt7603e and mt76x2e behave differently with iface level or why disabling mt76x2e causes dormant mt7603e to start interrupting. I've summarised the results in a table below. Each time I enable the wifi interface and then shut it down.

| Procedure to disable Wifi radios                                        | uci show wireless &#124; grep 'disabled'    | mt7603e interrupts Y/N| 
|-------------------------------------------------------------------------|--------------------------------------|-----------------------| 
| uci set wireless.@wifi-iface[0].disabled='1'; uci commit wireless; wifi | wireless.default_radio0.disabled='1' | Yes                   | 
| uci set wireless.@wifi-device[0].disabled=1; uci commit wireless; wifi  | wireless.radio0.disabled='1'         | No                    | 
| uci set wireless.@wifi-iface[1].disabled='1'; uci commit wireless; wifi | wireless.default_radio1.disabled='1' | Yes                   | 
| uci set wireless.@wifi-device[1].disabled=1; uci commit wireless; wifi  | wireless.radio1.disabled='1'         | Yes                   | 
| Luci click radio0 'Disable' button                                      | wireless.default_radio0.disabled='1' | Yes                   | 
| Luci click radio1 'Disable' button                                      | wireless.default_radio1.disabled='1' | Yes                   |

iancd, 2018-06-09:
Same behaviour in latest snapshot release June 2018. @barhom Do you think I should contact the Luci owners and ask them if this is expected? Appreciate your thoughts?

barhom, 2018-08-12:
How big of a problem is it that mt7603e interrupts?