openwrt/mt76 #137: Unstable or slow connection on MT7628 (Onion Omega2)

KisImre, 2017-12-28:
I have three Onion Omega2 boards. Using Onion's offical image which has a proprietary wireless driver, all of their connection is stable and the throughput is around 10-11 Mbps. When I replace the firmware with a LEDE snapshot image, the results are very different depending on the particular instance I test. On one of them I can't even get a stable connection, another one gets a boost and it's throughput increases to 18 Mbps and the third one's got a stable connection but it's quite slow. Before I created this issue, I've read the issues of this project and tried the method described in #28. By using this I had a stable connection on the first board too but the performance was still very poor. I tried to play with the EEPROM contents but I didn't get far.
I ran out of ideas so let me know if there's anything I can do in order to solve this issue. Anyway, here are my results below and I also attached the EEPROM and EFUSE dumps to the issue.

OS | mt76 version | E9F5 | E8D3 | 39A9
-- | -- | -- | -- | --
Offical image b160 | Proprietary | 10.2 Mbps | 11.5 Mbps | 10.3 Mbps
LEDE dda2229c52 | master/68b0cf1 | No connection | 18.1 Mbps | 1.1 Mbps
LEDE dda2229c52 | master/68b0cf1, cal free removed | ~300 kbps | 18.2 Mbps | 3.1 Mbps

Test environment:

- TP-Link TL-WR1043N/ND v1 AP (works well with any other device)
- 2.4 GHz/CH11
- Same power supply
- Same placement of boards

[mt7628_eeprom_otp_dumps.zip](https://github.com/openwrt/mt76/files/1589921/mt7628_eeprom_otp_dumps.zip)

MaFeSa, 2017-12-28:
It's probably due to this
https://github.com/openwrt/mt76/issues/129

nbd168, 2018-02-09:
Please try the latest version

KisImre, 2018-02-10:
I'm going to be able to test this on Monday or Tuesday. I'll give you a feedback when I'm done.

adron-s, 2018-07-25:
The problem is still relevant. See my post [here](https://github.com/openwrt/mt76/issues/28#issuecomment-407843021)

brzz, 2018-08-03:
same too ,but use wmm=0 and at 11M mode can stable connection

nbd168, 2019-01-25:
Please test the latest version

KisImre, 2019-02-10:
First of all it works very well and thank you for your work. I'm sorry it took so long for me the test it.
I can confirm that commit c3efb5d89951e260a9947366af6256d92833db87 fixes the problem. The WiFi setup in my home has slightly changed since then so I tested again with all the previous versions too in order to have comparable results. I think the values in the table below speak for themselves.

OS | mt76 version | E9F5 | E8D3 | 39A9
-- | -- | -- | -- | --
Offical image b160 | Proprietary | 13.3 Mbps | 10.0 Mbps | 12.9 Mbps
Offical image b201 | Proprietary | 15.4 Mbps | 11.5 Mbps | 11.5 Mbps
LEDE 38e704be71 | master/e326bc2 | No connection | 7.1 Mbps | No connection
**OpenWRT 9b14c7d** | **master/2afa0d7** | **14.2 Mbps** | **11.5 Mbps** | **12.9 Mbps**

closed 2019-02-10 by KisImre