openwrt/mt76 #138: WARNING Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]

MaFeSa, 2018-01-03:
Xiaomi Router 3G - OpenWrt SNAPSHOT r5521-9f8d282 / LuCI Master (git-17.328.04231-802d5b6)

```
[280396.209050] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4306 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[280396.220161] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 123 (0x7b)
[280396.229693] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uas slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit
[280396.300631]  act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs hfsplus nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc32c_generic
[280396.343399] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.67 #0
[280396.350583] Stack : 00000000 00000000 80537b4a 00000041 803f49a4 00000000 00000000 80530000
[280396.359016]         804d61bc 804d5da7 8046e9d4 00000000 00000000 80533824 00000001 8e625a00
[280396.367448]         00000000 8006a858 00000001 80530000 804dbec4 804dbec8 804735e0 8fc0dd24
[280396.375880]         00000003 800a7280 00000001 8e625a00 81211e20 00000000 00000002 00c0dd24
[280396.384311]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[280396.392741]         ...
[280396.395264] Call Trace:
[280396.397807] [<8000f884>] show_stack+0x54/0x88
[280396.402242] [<801d9dcc>] dump_stack+0x8c/0xd0
[280396.406667] [<8002add4>] __warn+0xe4/0x118
[280396.410829] [<8002ae38>] warn_slowpath_fmt+0x30/0x3c
[280396.415911] [<8e7a5c20>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[280396.422098] [<8e62d980>] mt76_rx_complete+0x88/0x10c [mt76]
[280396.427735] [<8e62ce7c>] mt76_dma_attach+0xbac/0xc94 [mt76]
[280396.433437] ---[ end trace 6068b4b3bcc38bea ]---
[434005.449666] ------------[ cut here ]------------
[434005.454579] WARNING: CPU: 0 PID: 3 at backports-2017-11-01/net/mac80211/rx.c:4306 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[434005.465774] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 103 (0x67)
[434005.475337] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uas slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit
[434005.546310]  act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs hfsplus nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc32c_generic
[434005.589115] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G        W       4.9.67 #0
[434005.596474] Stack : 00000000 00000000 80537b4a 00000043 803f49a4 00000000 00000000 80530000
[434005.604910]         8fc50edc 804d5da7 8046e9d4 00000000 00000003 80533824 00000001 8e625a00
[434005.613342]         00000000 8006a858 00000001 80530000 804dbec4 804dbec8 804735e0 8fc63c04
[434005.621774]         00000003 800a7280 00000001 8e625a00 81211e20 00000000 00000002 00c63c04
[434005.630205]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[434005.638634]         ...
[434005.641157] Call Trace:
[434005.643706] [<8000f884>] show_stack+0x54/0x88
[434005.648139] [<801d9dcc>] dump_stack+0x8c/0xd0
[434005.652566] [<8002add4>] __warn+0xe4/0x118
[434005.656729] [<8002ae38>] warn_slowpath_fmt+0x30/0x3c
[434005.661867] [<8e7a5c20>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[434005.668054] [<8e62d980>] mt76_rx_complete+0x88/0x10c [mt76]
[434005.673694] [<8e62ce7c>] mt76_dma_attach+0xbac/0xc94 [mt76]
[434005.679457] ---[ end trace 6068b4b3bcc38beb ]---
```

Thanks

MaFeSa, 2018-01-31:
Seems fixed with lastest commits, 4 days of uptime without this warning.

kofec, 2018-01-31:
which version do you have ? I have also Xiaomi Router 3G  and not able to attach to 2,4 GHz

MaFeSa, 2018-02-01:
The last one up to this commit https://github.com/openwrt/mt76/commit/e67e7a55fda1b039b451f5ec208204e19c229566
What configuration are you using?
I don't have problems with 2.4 GHz

If there are no warnings in a few days I close the ticket

kofec, 2018-02-01:
root@LEDEXiaomi3G:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option country 'PL'
        option channel '13'
        option legacy_rates '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'WIFI24G'
        option network 'lan'
        option encryption 'psk-mixed'
        option key 'password'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option country 'PL'
        option channel 'auto'
        option htmode 'VHT80'
        option legacy_rates '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'WIFIG5'
        option network 'lan'
        option encryption 'psk-mixed'
        option key 'password'

config wifi-iface
        option device 'radio0'
        option mode 'ap'
        option ssid 'wifi_24G'
        option network 'lan'
        option encryption 'psk-mixed'
        option key 'password'

MaFeSa, 2018-02-01:
Why you have this?
Delete it

```
config wifi-iface
        option device 'radio0'
        option mode 'ap'
        option ssid 'wifi_24G'
        option network 'lan'
        option encryption 'psk-mixed'
        option key 'password'
```

kofec, 2018-02-01:
this is second virtual AP with diffrent SSID but I will try

lumixen, 2018-02-03:
@kofec let us know please if disabling virtual AP helped in your case! I'm quite interested in guest AP feature.

kofec, 2018-02-03:
Hi, I will confirm tomorrow. But meantime I found on lede forum that I
should disable legacy rate and low level ack

MaFeSa, 2018-02-07:
I can no longer reproduce this warning, probably is fixed with latest commits.
Thanks nbd.

closed 2018-02-07 by MaFeSa