openwrt/mt76 #139: MT7612U support

pellerb, 2018-01-04:
Hi,
I bought EDUP EP-AC1605 WiFi adapters and these are equipped with MediaTek MT7612U controllers. I would like to use them with OpenWRT-based routers and USB3 connection (to have an additional WiFi device).
It seems that this driver doesn't support MT7612U controller yet. The router detects a new USB device connection when I plug in, but it doesn't recognize this adapter even if kmod-mt76x2 driver package is installed.
Could you please add MT7612U support in parallel with MT7612E?
It would be also very nice to distribute compiled version of the package too.
Thanks,
Balázs

benVolatiles, 2018-03-23:
Did you try this driver?

https://github.com/ulli-kroll/mt7612u

LorenzoBianconi, 2018-03-23:
>
> Did you try this driver?
>
> https://github.com/ulli-kroll/mt7612u
>
> <https://github.com/openwrt/mt76/issues/139#issuecomment-375620200>, or mute
> the thread

I started adding usb support to mt76/mt76x2 driver, I will keep you updated.
Regards,

Lorenzo

LorenzoBianconi, 2018-05-07:
https://github.com/LorenzoBianconi/mt76/tree/mt76x2u
Here I added mt76x2u support to mt76 driver. What are the idVendor and idProduct of your device?
At the moment I added just the support for Netgear A6210

LorenzoBianconi, 2018-05-08:
@cyangy: could you please provide me the idVendor and idProduct used by your device? (just plug-in the usb dongle and type lsusb). Please note fw images are not included yet, you can use the ones from mtk driver. Moreover you need to change openwrt mt76 package Makefile in order to compile mt76x2u module

cyangy, 2018-05-08:
I use the NetGear A6210 , and I found that the  idVendor and idProduct has been included in the 
`{ USB_DEVICE(0x0846, 0x9053) },		/* Netgear A6210 */`
Now  I am modfying the OpenWRT  mt76 package Makefile to compile the module , I will tell you as soon as I finish it.

cyangy, 2018-05-09:
@LorenzoBianconi : I had tested the driver on  Linksys EA4500 v1 with latest LEDE  use my  Netgear A6210  and the 2.4G STA mode works fine , except  a little problem described [here](https://github.com/cyangy/openwrt-mt7612u)  . I'll be appreciated if these issues can be solved. Thanks  again.
also these are known   idVendor and idProduct from [here](https://github.com/genodeftest/Netgear-A6210/blob/master/common/rtusb_dev_id.c)

	{USB_DEVICE(0x0846, 0x9014)}, /* Netgear WNDA3100v3 */
	{USB_DEVICE(0x0B05, 0x180B)}, /* ASUS USB-N53 */
	{USB_DEVICE(0x0B05, 0x17EB)}, /* ASUS USB-AC55 */
	{USB_DEVICE(0x045e, 0x02e6)}, /* Microsoft XBox One Wireless Adapter */

LorenzoBianconi, 2018-05-09:
Cool, thx @cyangy for testing. For the moment ap mode is not supported.
Does boot hang happen with dongle configured in ap or in sta?

Orfait, 2018-05-09:
May i suggest you to look at this [list](https://wikidevi.com/wiki/Special:Ask?title=Special%3AAsk&q=%5B%5BChip1+model::~MT7612U*%5D%5D&po=%3FInterface%0D%0A%3FForm+factor=FF%0D%0A%3FInterface+connector+type=USB+conn.%0D%0A%3FFCC+ID%0D%0A%3FManuf%0D%0A%3FManuf+product+model=Manuf.+mdl%0D%0A%3FVendor+ID%0D%0A%3FDevice+ID%0D%0A%3FChip1+model%0D%0A%3FSupported+802dot11+protocols=PHY+modes%0D%0A%3FMIMO+config%0D%0A%3FOUI%0D%0A%3FEstimated+year+of+release=Est.+year&eq=yes&p%5Bformat%5D=broadtable&order%5B0%5D=ASC&sort_num=&order_num=ASC&p%5Blimit%5D=500&p%5Boffset%5D=&p%5Blink%5D=all&p%5Bsort%5D=&p%5Bheaders%5D=show&p%5Bmainlabel%5D=&p%5Bintro%5D=&p%5Boutro%5D=&p%5Bsearchlabel%5D=%E2%80%A6+further+results&p%5Bdefault%5D=&p%5Bclass%5D=sortable+wikitable+smwtable).
You will find a list of all know MT7612U vendorid and deviceid :

`{USB_DEVICE(0x0e8d, 0x7612)}, // ALFA Network AWUS036ACM`
`{USB_DEVICE(0x0b05, 0x1833)}, // ASUS USB-AC54`
`{USB_DEVICE(0x0b05, 0x17eb)}, // ASUS USB-AC55`
`{USB_DEVICE(0x0b05, 0x180b)}, // ASUS USB-N53 B1`
`{USB_DEVICE(0x0e8d, 0x7612)}, // EDUP EP-AC1601`
`{USB_DEVICE(0x0e8d, 0x7612)}, // EDUP EP-AC1605`
`{USB_DEVICE(0x7392, 0xb711)}, // Edimax EW-7722UAC`
`{USB_DEVICE(0x0846, 0x9053)}, // Netgear A6210`

LorenzoBianconi, 2018-05-09:
@orfai: sure, for the moment I added just the device I was able to test. Are you able to test other devices?

Orfait, 2018-05-09:
@LorenzoBianconi : I was try to help for the list of @cyangy. I only own a A6210 and I am still learning how to compile and add modules to openwrt. So, I can help for testing as soon as I can build by myself :)

cyangy, 2018-05-09:
@LorenzoBianconi  boot never hang, but if I keep USB insert , then  power on the router, error message appears like
     mt76x2u 1-1:1.0: MAC RX failed to stop
     mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
and the router become very slow 
I wonder if the timeout value  too small , so after I use this patch , 
`
--- a/mt76x2u_mcu.c	2018-05-07 04:47:54.000000000 +0800
+++ b/mt76x2u_mcu.c	2018-05-09 15:00:56.063299831 +0800
@@ -59,7 +59,7 @@
 
 	for (i = 0; i < 5; i++) {
 		if (!wait_for_completion_timeout(&dev->mcu.resp_cmpl,
-						 msecs_to_jiffies(300)))
+						 msecs_to_jiffies(30000)))
`
the problem seems solved. Since I'm  a newbee  , I don't know if it's  **really** solved

LorenzoBianconi, 2018-05-09:
@cyangy ok, thx for the info. What is the mode configured in '/etc/config/wireless'? 30s seems to much, what about 500?

@Orfait any tests is very welcome. You can use the Makefile provided by @cyangy

cyangy, 2018-05-09:
my  /etc/config/wireless  part related to  mt7612u is

config wifi-device 'radio2'
	option type 'mac80211'
	option channel 'auto'
	option hwmode '11a'
	option path 'platform/ocp@f1000000/f1050000.ehci/usb1/1-1/1-1:1.0'
	option htmode 'HT40'
	option disabled '0'

config wifi-iface 
        option device 'radio2'
        option network 'wan'
        option mode 'sta'
        option ssid 'CUG'
        option encryption 'none'
	option disabled '0' 

which CUG is a router that provide both 5G and 2.4G signal. BUT it seems the frequence alaways  2.4G band
the   `iwconfig`   shows

          IEEE 802.11  ESSID:"CUG"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: 38:ED:18:CF:91:E0   
          Bit Rate=144.4 Mb/s   Tx-Power=19 dBm   
          RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=45/70  Signal level=-65 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:1  Invalid misc:383   Missed beacon:0

LorenzoBianconi, 2018-05-09:
@cyangy ok, thx. Does it hang at bootstrap with this configuration?
2.4GHz band only is something related to openwrt wpa_supplicant since plugging the dongle in my laptop works fine @5GHz as well.

cyangy, 2018-05-09:
@LorenzoBianconi  this configuration  works fine except 5ghz not work.

LorenzoBianconi, 2018-05-09:
@cyangy: I mean, does it hang at bootstrap if the dongle is plugged-in? The issue is that wpa_supplicant does not scan @5GHz. I will work on it

cyangy, 2018-05-09:
@LorenzoBianconi   NO, bootstrap is work well .

cyangy, 2018-05-09:
@LorenzoBianconi  sorry for that, I mean the route become very slow like it hangs but not really . 
If I use standard fw (mcu timeout set to 300ms) , and  keep  the dongle   plugged-in , power on it , then the message
      mt76x2u 1-1:1.0: MAC RX failed to stop
      mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
will appear  continuously ,   the other process  still  output log  to stdout, if  mcu timeout set to 30000ms ,it works well.

Orfait, 2018-05-09:
@LorenzoBianconi, @cyangy : I don't have such output in dmesg, but I am not able to join any network (the page /cgi-bin/luci/admin/network/wireless_join is empty).

cyangy, 2018-05-09:
@Orfait  as the `iwlist  wlanx scanning` not work ,you should add  wireless config  manually like I posted before.

LorenzoBianconi, 2018-05-09:
@Orfait @cyangy : please use iw instead

iw dev wlanX scan

cyangy, 2018-05-09:
OK.

Orfait, 2018-05-09:
@LorenzoBianconi : Client mode seems to be working for me.

In fact, I am waiting for AP 😊 , but I can test client mode...

LorenzoBianconi, 2018-05-09:
>
> @Orfait <https://github.com/Orfait> : Me too , does 5ghz work for you ?
>
> <https://github.com/openwrt/mt76/issues/139#issuecomment-387736858>, or mute
> the thread
@cyangy @Orfait <https://github.com/Orfait>: 5GHz works if you run the scan
manually (e.g; iw dev wlanX scan). It is something related to
wpa_supplicant not only driver related since running the diver on a laptop
works properly @5GHz

Lorenzo

Orfait, 2018-05-09:
@cyangy : I cannot test 5 GHz, my actual router is 2.4 GHz only.

At the moment, I have 144 Mbits/s in RX, only 6.5 Mbits/s in TX (why ?)

cyangy, 2018-05-09:
@Orfait  :  Actually, It doesn't matter, On my router there is  a wlan  interface (not  A6210)  which only  shows  1.0 Mbit/s  Tx rate ,  but it work well. 

The A6210 on my router is 

         Rx  300.0 Mbit/s, 40MHz, MCS 15, Short GI
         Tx   216.0 Mbit/s, 40MHz, MCS 13

By the way , the [firmware](https://github.com/cyangy/openwrt-mt7612u/tree/master/src/firmware) has been updated, you shoud try it.

cyangy, 2018-05-09:
@LorenzoBianconi : Thanks for your great work. I am  live  in GMT+8 ，so I have to go to sleep now. Bye @Orfait

araujorm, 2018-05-09:
Hi.

I've been following this thread. I've got a EP-AC1601 dongle also. Managed to compile this driver on Fedora 26, kernel 4.15.15-200.fc26.x86_64, and I'm testing it.

For the device to be recognized, I had to had the 0e8d:7612 ID to the code, I ended up using:

```
diff --git a/mt76x2_usb.c b/mt76x2_usb.c
index 3cbd3ee..69a1b60 100644
--- a/mt76x2_usb.c
+++ b/mt76x2_usb.c
@@ -21,6 +21,11 @@
 
 static const struct usb_device_id mt76x2u_device_table[] = {
        { USB_DEVICE(0x0846, 0x9053) },         /* Netgear A6210 */
+       { USB_DEVICE(0x0b05, 0x1833) },         /* ASUS USB-AC54 */
+       { USB_DEVICE(0x0b05, 0x17eb) },         /* ASUS USB-AC55 */
+       { USB_DEVICE(0x0b05, 0x180b) },         /* ASUS USB-N53 B1 */
+       { USB_DEVICE(0x0e8d, 0x7612) },         /* EDUP EP-AC1601, EDUP EP-AC1605, ALFA Network AWUS036ACM */
+       { USB_DEVICE(0x7392, 0xb711) },         /* Edimax EW-7722UAC */
        { },
 };
```

As for the tests, the driver doesn't seem to support AC radio as it doesn't see my 5 GHz networks. Please tell me if you add anything you want me to test on that regard.

Regarding 2.4 GHz, I was able to achieve peaks of 60 Mbps on speedtest dot net, ending with 50 Mbps reported speed, which isn't bad at all (my Internet speed is 100 Mbps fibre).

Problems found:
- if the dongle is already inserted and the module is inserted, hangs with errors like "mt76x2u 2-1.5:1.0: rx urb failed: -71"
- when trying to remove the module, I also have a "MAC RX failed to stop" error

Anything else you need me to test, please feel free to ask. If you prefer to discuss it on another issue on your repo (since I'm not using the USB driver with openwrt) please say, although I don't think it will make much difference.

cyangy, 2018-05-10:
@araujorm  Are you use the firmwre  [THIS COMMIT](https://github.com/cyangy/openwrt-mt7612u/commit/873bcaa76f30f9527167c55dc883cc2806805813) , Sorry for that , It seems the latest firmware have some problem ,I just restore to previous version firmware，you could try it

LorenzoBianconi, 2018-05-10:
@cyangy , @araujorm , @Orfait could you please try:
[mt76x2u_testing](https://github.com/LorenzoBianconi/mt76/tree/mt76x2u_testing) commit 2e04878bc5ebb5cc56257a0e879d744a8e506adc.

5GHz scanning should be fixed now

Orfait, 2018-05-10:
@LorenzoBianconi : I cannot test 5 GHZ as I don't have any AP in this band.

cyangy, 2018-05-10:
@LorenzoBianconi  I'm testing...

cyangy, 2018-05-10:
@LorenzoBianconi  @Orfait   : I have test the 5GHz, now even from the LUCI  web interface , I can  scan  and join a network no matter it is 2.4GHz or  5GHz. Let me show  you the evidence
[image]
`iwlist wlan2 channel`  
shows
    wlan2     32 channels in total; available frequencies :
    Channel 01 : 2.412 GHz
    Channel 02 : 2.417 GHz
    Channel 03 : 2.422 GHz
    Channel 04 : 2.427 GHz
    Channel 05 : 2.432 GHz
    Channel 06 : 2.437 GHz
    Channel 07 : 2.442 GHz
    Channel 08 : 2.447 GHz
    Channel 09 : 2.452 GHz
    Channel 10 : 2.457 GHz
    Channel 11 : 2.462 GHz
    Channel 12 : 2.467 GHz
    Channel 13 : 2.472 GHz
    Channel 14 : 2.484 GHz
    Channel 36 : 5.18 GHz
    Channel 40 : 5.2 GHz
    Channel 44 : 5.22 GHz
    Channel 48 : 5.24 GHz
    Channel 52 : 5.26 GHz
    Channel 56 : 5.28 GHz
    Channel 60 : 5.3 GHz
    Channel 64 : 5.32 GHz
    Channel 100 : 5.5 GHz
    Channel 104 : 5.52 GHz
    Channel 108 : 5.54 GHz
    Channel 112 : 5.56 GHz
    Channel 116 : 5.58 GHz
    Channel 120 : 5.6 GHz
    Channel 124 : 5.62 GHz
    Channel 128 : 5.64 GHz
    Channel 132 : 5.66 GHz
    Channel 136 : 5.68 GHz
    Current Frequency:5.18 GHz (Channel 36)

LorenzoBianconi, 2018-05-10:
@cyangy thx for testing, I will push the commits to [mt76x2u](https://github.com/LorenzoBianconi/mt76/tree/mt76x2u)

Orfait, 2018-05-10:
@cyangy : Do you plan to make also AP mode ?

cyangy, 2018-05-10:
@Orfait  : Sorry, I'm newbee in linux, I just can write some short  and basic function , so I can not do something like write driver for a device . I hope someday I can write a driver for a device after learning .

araujorm, 2018-05-10:
> @araujorm <https://github.com/araujorm> if you prefer to test the
> driver on your pc you can use
> https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x2u
> <mt76x2u>

Great, I will try later. Thanks.

Orfait, 2018-05-10:
@cyangy : Sorry, I wrote to you by mistake, my question about AP mode was for @LorenzoBianconi 😉

Orfait, 2018-05-10:
@cyangy : Sorry, the question about AP mode was for @LorenzoBianconi 😉

pellerb, 2018-05-10:
Hi,
My EDUP EP-AC1605 WiFi adapter has MediaTek MT7612U controller. Its VendorID=0E8D and ProductID=7612.
Thanks for your help, regards,
Balázs

LorenzoBianconi, 2018-05-10:
@pellerb it is already supported

pellerb, 2018-05-10:
Ok, I can see that it's the same as Aukey USB-AC1200. Thanks, I will try it with EDUP EP-AC1605.

araujorm, 2018-05-10:
Hello.

I confirm that 5 GHz networks are now visible and connectable.

However after a while it all stops working, as if the already established connection freezes for about 10 seconds or more, until it starts works again. A ping test even reported 15000 ms response time once, so it doesn't seem to be packet loss but something that hangs for a while. Nothing on dmesg while it happens. Is there any debug setting one should activate to try to understand what's going wrong?

Also, after a cold boot, the problem I mentioned earlier with the module hanging after inserted still persists. I have to unplug the device for modprobe to return. Here's journalctl -f output while it happened (kept going until I unplugged the device):
```
May 10 21:00:02 zoo.localdomain kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
May 10 21:00:02 zoo.localdomain kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
May 10 21:00:02 zoo.localdomain kernel: platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
May 10 21:00:02 zoo.localdomain kernel: cfg80211: failed to load regulatory.db
May 10 21:00:02 zoo.localdomain kernel: usb 2-1.5: reset high-speed USB device number 6 using ehci-pci
May 10 21:00:02 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: ASIC revision: 76120044
May 10 21:00:02 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: ROM patch build: 20140408060640a
May 10 21:00:05 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:07 off:09a8 failed:-110
May 10 21:00:09 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:06 off:09a8 failed:-110
May 10 21:00:12 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0230 failed:-110
May 10 21:00:15 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0232 failed:-110
May 10 21:00:18 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0234 failed:-110
May 10 21:00:21 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0236 failed:-110
May 10 21:00:25 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:07 off:09a8 failed:-110
May 10 21:00:28 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:06 off:09a8 failed:-110
May 10 21:00:31 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0230 failed:-110
May 10 21:00:34 zoo.localdomain kernel: mt76x2u 2-1.5:1.0: vendor request req:42 off:0232 failed:-110
```

Plugging the device with the module already loaded, it started working (with the occasional freezings I mentioned first).

Finally, when doing rmmod mt76x2u mt76x2_common mt76 I get:

```
May 10 21:39:43 zoo.localdomain kernel: rx urb mismatch
May 10 21:39:43 zoo.localdomain kernel: WARNING: CPU: 0 PID: 0 at /home/rod/mt76/mt76/usb.c:345 mt76_usb_complete_rx+0xc9/0xe0 [mt76]
May 10 21:39:43 zoo.localdomain kernel: Modules linked in: mt76x2u(OE-) mt76x2_common(OE) mt76(OE) ccm arc4 mac80211 cfg80211 rfcomm ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables cmac bnep sunrpc vfat fat fuse nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) btusb btrtl btbcm btintel bluetooth snd_usb_audio ecdh_generic snd_usbmidi_lib snd_rawmidi intel_rapl snd_hda_codec_hdmi x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_hda_codec_realtek kvm snd_hda_codec_generic
May 10 21:39:43 zoo.localdomain kernel:  snd_hda_intel irqbypass snd_hda_codec drm_kms_helper snd_hda_core crct10dif_pclmul crc32_pclmul snd_hwdep iTCO_wdt ghash_clmulni_intel iTCO_vendor_support snd_seq drm intel_cstate snd_seq_device intel_uncore eeepc_wmi snd_pcm asus_wmi intel_rapl_perf ipmi_devintf ipmi_msghandler snd_timer sparse_keymap rfkill snd mei_me wmi_bmof i2c_i801 soundcore mei shpchp lpc_ich binfmt_misc mxm_wmi firewire_ohci r8169 firewire_core crc32c_intel serio_raw mii crc_itu_t wmi video uas usb_storage analog gameport joydev vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) [last unloaded: mt76]
May 10 21:39:43 zoo.localdomain kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: P        W  OE    4.15.15-200.fc26.x86_64 #1
May 10 21:39:43 zoo.localdomain kernel: Hardware name: System manufacturer System Product Name/P8P67-M PRO, BIOS 3701 11/30/2012
May 10 21:39:43 zoo.localdomain kernel: RIP: 0010:mt76_usb_complete_rx+0xc9/0xe0 [mt76]
May 10 21:39:43 zoo.localdomain kernel: RSP: 0018:ffff9ba51ec03ea0 EFLAGS: 00010086
May 10 21:39:43 zoo.localdomain kernel: RAX: 0000000000000000 RBX: ffff9ba491f51520 RCX: ffffffffa2253848
May 10 21:39:43 zoo.localdomain kernel: RDX: 0000000000000001 RSI: 0000000000000086 RDI: 0000000000000087
May 10 21:39:43 zoo.localdomain kernel: RBP: ffff9ba491f52310 R08: 00000000000004ab R09: ffffffffa295b600
May 10 21:39:43 zoo.localdomain kernel: R10: ffff9ba51ec03f20 R11: 0000000000000000 R12: ffff9ba4921d3600
May 10 21:39:43 zoo.localdomain kernel: R13: 0000000000000000 R14: ffff9ba514c459a4 R15: ffff9ba4921d3600
May 10 21:39:43 zoo.localdomain kernel: FS:  0000000000000000(0000) GS:ffff9ba51ec00000(0000) knlGS:0000000000000000
May 10 21:39:43 zoo.localdomain kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... cut ...]
May 10 21:39:43 zoo.localdomain kernel: Code: d4 18 6e c0 e8 a9 e7 ea e0 eb 81 80 3d 64 51 00 00 00 75 d5 48 c7 c7 aa 18 6e c0 48 89 34 24 c6 05 50 51 00 00 01 e8 67 a8 9c e0 <0f> 0b 48 8b 34 24 eb b6 0f 1f 44 00 00 66 2e 0f 1f 84 00 00 00 
```

After that, trying to modprobe mt76x2u again doesn't hang, but still has the connection freezings issue.

All this was tried the new firmware, the old and the one from https://github.com/ulli-kroll/mt7612u, I didn't notice any difference.

Any thing I should try? Remember, Fedora 26 x86_64 here.

Thanks and best regards.

LorenzoBianconi, 2018-05-10:
@araujorm thx a lot for testing
1- Are you using the kernel source from my github repo or have you backported just the mt76 driver? Huge ping delay seems related to this [issue](https://github.com/openwrt/mt76/issues/152) 
2- If I understood correctly, if the dongle is plugged in your pc at device bootstrap it will properly work just once, if you reboot the pc without unplugging the dongle it will not be loaded properly. Correct?
3- Before rmmod mt76x2u mt76x2_common, have you stopped wpa_supplicant? (I agree it should not crash even if wpa_supplicant is still running)

araujorm, 2018-05-10:
@LorenzoBianconi:
> 1- Are you using the kernel source from my github repo or have you backported just the mt76 driver? Huge ping delay seems related to this issue

Compiled the mt76 driver only for my kernel, taken from your mt76/tree/mt76x2u. Kernel is Fedora's 4.15.15-200.fc26.x86_64.

> 2- If I understood correctly, if the dongle is plugged in your pc at device bootstrap it will properly work just once, if you reboot the pc without unplugging the dongle it will not be loaded properly. Correct?

I've done some more testing. It seems to only happen if I power off the machine, wait a while and then power it up, then the first time the module is inserted with the dongle already plugged in the firmware upload hangs with the mentioned errors. If I just reboot having previously used the device, even if I reboot from Windows, it doesn't hang, at least it didn't just now. I may also mention that the device works without issues from Windows non-stop, so it doesn't seem to be a hardware failure, and USB support seems stable for other devices on my machine on both Windows and Linux.

> 3- Before rmmod mt76x2u mt76x2_common, have you stopped wpa_supplicant? (I agree it should not crash even if wpa_supplicant is still running)

Well, wpa_supplicant was running. I did some more testing and disconnected, and if wpa_supplicant is running, even if it (supposedly) isn't doing anything, I get that crash when doing rmmod. But if I stop wpa_supplicant, I don't get the crash.

Are there any printk I could add anywhere, or any debug mode I could activate to provide more info? Any other suggestions or anything, please feel free to ask.

cyangy, 2018-05-11:
@LorenzoBianconi :  When bootup , ` rx urb mismatch`  problem    bother me  days. if I reboot the router  without unplugging the dongle , sometimes(about 90%) the problem happens,and sometimes is just fine 

when the issuse happens , the log realated it is
 
```
[    8.603478] usb 1-1: reset high-speed USB device number 2 using orion-ehci
[    8.804468] mt76x2u 1-1:1.0: ASIC revision: 76120044
[    8.836158] mt76x2u 1-1:1.0: ROM patch build: 20140408060640a
[    8.910710] ------------[ cut here ]------------
[    8.915372] WARNING: CPU: 0 PID: 0 at /home/myname/LEDE/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/mt76-mt76x2u-758f64a1/usb.c:345 mt76_usb_complete_rx+0x90/0x100 [mt76]
[    8.931345] rx urb mismatch
[    8.931349] Modules linked in: mt76x2u(+) mt76x2_common mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack 
xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_
ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netpor
t ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap
_ipmac
[    9.006136]  ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun xhci_plat_hcd xhci_pci xhci_hcd ehci_platform
[    9.024149] CPU: 0 PID: 0 Comm: swapper Not tainted 4.14.37 #0
[    9.030004] Hardware name: Marvell Kirkwood (Flattened Device Tree)
[    9.036291] Backtrace: 
[    9.038765] [<c0105e1c>] (dump_backtrace) from [<c0106104>] (show_stack+0x18/0x1c)
[    9.046367]  r7:bf3a30f0 r6:00000000 r5:bf3a4f30 r4:c0801d90
[    9.052064] [<c01060ec>] (show_stack) from [<c05886c0>] (dump_stack+0x20/0x28)
[    9.059328] [<c05886a0>] (dump_stack) from [<c01103d8>] (__warn+0xe0/0x10c)
[    9.066320] [<c01102f8>] (__warn) from [<c0110444>] (warn_slowpath_fmt+0x40/0x48)
[    9.073837]  r9:00000040 r8:ffffe000 r7:60000093 r6:c6969b90 r5:c69f0080 r4:bf3a4f20
[    9.081623] [<c0110408>] (warn_slowpath_fmt) from [<bf3a30f0>] (mt76_usb_complete_rx+0x90/0x100 [mt76])
[    9.091054]  r3:bf3a610c r2:bf3a4f20
[    9.094636]  r4:c6969460
[    9.097196] [<bf3a3060>] (mt76_usb_complete_rx [mt76]) from [<c0428f5c>] (__usb_hcd_giveback_urb+0xac/0x104)
[    9.107067]  r7:c6a558e0 r6:00000000 r5:60000013 r4:c69f0080
[    9.112757] [<c0428eb0>] (__usb_hcd_giveback_urb) from [<c0429044>] (usb_giveback_urb_bh+0x90/0xd4)
[    9.121838]  r7:c6a558e0 r6:00000000 r5:c0801df8 r4:c6a558dc
[    9.127526] [<c0428fb4>] (usb_giveback_urb_bh) from [<c01141b8>] (tasklet_action+0x84/0xcc)
[    9.135911]  r7:00000100 r6:00000000 r5:c0808fa0 r4:00000000
[    9.141598] [<c0114134>] (tasklet_action) from [<c01014d0>] (__do_softirq+0x118/0x2fc)
[    9.149550]  r7:00000100 r6:00000018 r5:00000006 r4:c084d138
[    9.155236] [<c01013b8>] (__do_softirq) from [<c0113d98>] (irq_exit+0xa4/0xf8)
[    9.162492]  r10:00000000 r9:c0801f00 r8:c7809400 r7:00000001 r6:00000000 r5:c0847824
[    9.170351]  r4:00000000
[    9.172903] [<c0113cf4>] (irq_exit) from [<c013f548>] (__handle_domain_irq+0x8c/0xa8)
[    9.180771] [<c013f4bc>] (__handle_domain_irq) from [<c010138c>] (orion_handle_irq+0x74/0xa0)
[    9.189333]  r9:c0801f00 r8:00000001 r7:c087a87c r6:c781a01c r5:00080000 r4:00000013
[    9.197112] [<c0101318>] (orion_handle_irq) from [<c0106be8>] (__irq_svc+0x68/0x84)
[    9.204797] Exception stack(0xc0801f00 to 0xc0801f48)
[    9.209869] 1f00: 00000000 00000000 00000000 60000013 00000000 ffffe000 c0803088 c083ebf0
[    9.218082] 1f20: c080c2fa c0669220 00000000 c0801f5c c0801f50 c0801f50 c0102f60 c05a4464
[    9.226292] 1f40: 60000013 ffffffff
[    9.229794]  r10:00000000 r9:c0800000 r8:c080c2fa r7:c0801f34 r6:ffffffff r5:60000013
[    9.237653]  r4:c05a4464
[    9.240208] [<c05a442c>] (default_idle_call) from [<c0138c28>] (do_idle+0x84/0x144)
[    9.247895] [<c0138ba4>] (do_idle) from [<c0138f48>] (cpu_startup_entry+0x14/0x18)
[    9.255500]  r10:00735474 r9:c0736a20 r8:c7ffca60 r7:00000000 r6:c0803020 r5:ffffffff
[    9.263360]  r4:c080a90c r3:40000013
[    9.266948] [<c0138f34>] (cpu_startup_entry) from [<c059f238>] (rest_init+0x74/0x94)
[    9.274729] [<c059f1c4>] (rest_init) from [<c0700d54>] (start_kernel+0x3a4/0x428)
[    9.282240]  r5:ffffffff r4:c084caa0
[    9.285828] [<c07009b0>] (start_kernel) from [<00008048>] (0x8048)
[    9.292031] ---[ end trace 0e412fa249281f90 ]---
[    9.354707] mt76x2u 1-1:1.0: Firmware Version: 0.0.00
[    9.359789] mt76x2u 1-1:1.0: Build: 1
[    9.363534] mt76x2u 1-1:1.0: Build Time: 201406241830____
[   10.299574] usbcore: registered new interface driver mt76x2u
                       ............................

[   23.451087] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[   25.043805] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[   26.623541] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[   29.743608] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[   31.303541] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
                       ............................
```
even I  increase the value of   ` MT_VEND_REQ_MAX_RETRY`   and  `MT_VEND_REQ_TOUT_MS`  in 
  **usb.c** can not solve it.
In  my humble opinion, is it something related to the USB init ?  How about init the USB device more than one time to avoid this issuse?

LorenzoBianconi, 2018-05-11:
>
> @LorenzoBianconi <https://github.com/LorenzoBianconi> : When bootup , rx
> urb mismatch problem bother me days. if I reboot the router without
> unplugging the dongle , sometimes(about 90%) the problem happens,and
> sometimes is just fine
>
> when the issuse happens , the log realated it is
>
> [    8.603478] usb 1-1: reset high-speed USB device number 2 using orion-ehci
> [    8.804468] mt76x2u 1-1:1.0: ASIC revision: 76120044
> [    8.836158] mt76x2u 1-1:1.0: ROM patch build: 20140408060640a
> [    8.910710] ------------[ cut here ]------------
> [    8.915372] WARNING: CPU: 0 PID: 0 at /home/myname/LEDE/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/mt76-mt76x2u-758f64a1/usb.c:345 mt76_usb_complete_rx+0x90/0x100 [mt76]
> [    8.931345] rx urb mismatch
> [    8.931349] Modules linked in: mt76x2u(+) mt76x2_common mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack
> xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_
> ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netpor
> t ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap
> _ipmac
> [    9.006136]  ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun xhci_plat_hcd xhci_pci xhci_hcd ehci_platform
> [    9.024149] CPU: 0 PID: 0 Comm: swapper Not tainted 4.14.37 #0
> [    9.030004] Hardware name: Marvell Kirkwood (Flattened Device Tree)
> [    9.036291] Backtrace:
> [    9.038765] [<c0105e1c>] (dump_backtrace) from [<c0106104>] (show_stack+0x18/0x1c)
> [    9.046367]  r7:bf3a30f0 r6:00000000 r5:bf3a4f30 r4:c0801d90
> [    9.052064] [<c01060ec>] (show_stack) from [<c05886c0>] (dump_stack+0x20/0x28)
> [    9.059328] [<c05886a0>] (dump_stack) from [<c01103d8>] (__warn+0xe0/0x10c)
> [    9.066320] [<c01102f8>] (__warn) from [<c0110444>] (warn_slowpath_fmt+0x40/0x48)
> [    9.073837]  r9:00000040 r8:ffffe000 r7:60000093 r6:c6969b90 r5:c69f0080 r4:bf3a4f20
> [    9.081623] [<c0110408>] (warn_slowpath_fmt) from [<bf3a30f0>] (mt76_usb_complete_rx+0x90/0x100 [mt76])
> [    9.091054]  r3:bf3a610c r2:bf3a4f20
> [    9.094636]  r4:c6969460
> [    9.097196] [<bf3a3060>] (mt76_usb_complete_rx [mt76]) from [<c0428f5c>] (__usb_hcd_giveback_urb+0xac/0x104)
> [    9.107067]  r7:c6a558e0 r6:00000000 r5:60000013 r4:c69f0080
> [    9.112757] [<c0428eb0>] (__usb_hcd_giveback_urb) from [<c0429044>] (usb_giveback_urb_bh+0x90/0xd4)
> [    9.121838]  r7:c6a558e0 r6:00000000 r5:c0801df8 r4:c6a558dc
> [    9.127526] [<c0428fb4>] (usb_giveback_urb_bh) from [<c01141b8>] (tasklet_action+0x84/0xcc)
> [    9.135911]  r7:00000100 r6:00000000 r5:c0808fa0 r4:00000000
> [    9.141598] [<c0114134>] (tasklet_action) from [<c01014d0>] (__do_softirq+0x118/0x2fc)
> [    9.149550]  r7:00000100 r6:00000018 r5:00000006 r4:c084d138
> [    9.155236] [<c01013b8>] (__do_softirq) from [<c0113d98>] (irq_exit+0xa4/0xf8)
> [    9.162492]  r10:00000000 r9:c0801f00 r8:c7809400 r7:00000001 r6:00000000 r5:c0847824
> [    9.170351]  r4:00000000
> [    9.172903] [<c0113cf4>] (irq_exit) from [<c013f548>] (__handle_domain_irq+0x8c/0xa8)
> [    9.180771] [<c013f4bc>] (__handle_domain_irq) from [<c010138c>] (orion_handle_irq+0x74/0xa0)
> [    9.189333]  r9:c0801f00 r8:00000001 r7:c087a87c r6:c781a01c r5:00080000 r4:00000013
> [    9.197112] [<c0101318>] (orion_handle_irq) from [<c0106be8>] (__irq_svc+0x68/0x84)
> [    9.204797] Exception stack(0xc0801f00 to 0xc0801f48)
> [    9.209869] 1f00: 00000000 00000000 00000000 60000013 00000000 ffffe000 c0803088 c083ebf0
> [    9.218082] 1f20: c080c2fa c0669220 00000000 c0801f5c c0801f50 c0801f50 c0102f60 c05a4464
> [    9.226292] 1f40: 60000013 ffffffff
> [    9.229794]  r10:00000000 r9:c0800000 r8:c080c2fa r7:c0801f34 r6:ffffffff r5:60000013
> [    9.237653]  r4:c05a4464
> [    9.240208] [<c05a442c>] (default_idle_call) from [<c0138c28>] (do_idle+0x84/0x144)
> [    9.247895] [<c0138ba4>] (do_idle) from [<c0138f48>] (cpu_startup_entry+0x14/0x18)
> [    9.255500]  r10:00735474 r9:c0736a20 r8:c7ffca60 r7:00000000 r6:c0803020 r5:ffffffff
> [    9.263360]  r4:c080a90c r3:40000013
> [    9.266948] [<c0138f34>] (cpu_startup_entry) from [<c059f238>] (rest_init+0x74/0x94)
> [    9.274729] [<c059f1c4>] (rest_init) from [<c0700d54>] (start_kernel+0x3a4/0x428)
> [    9.282240]  r5:ffffffff r4:c084caa0
> [    9.285828] [<c07009b0>] (start_kernel) from [<00008048>] (0x8048)
> [    9.292031] ---[ end trace 0e412fa249281f90 ]---
> [    9.354707] mt76x2u 1-1:1.0: Firmware Version: 0.0.00
> [    9.359789] mt76x2u 1-1:1.0: Build: 1
> [    9.363534] mt76x2u 1-1:1.0: Build Time: 201406241830____
> [   10.299574] usbcore: registered new interface driver mt76x2u
>                        ............................
>
> [   23.451087] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
> [   25.043805] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[... 11 lines cut ...]
[... cut ...]
b/drivers/net/wireless/mediatek/mt76/mt76x2u_init.c
index f253648844e0..e38893256f95 100644
--- a/drivers/net/wireless/mediatek/mt76/mt76x2u_init.c
+++ b/drivers/net/wireless/mediatek/mt76/mt76x2u_init.c
@@ -276,11 +276,11 @@ int mt76x2u_register_device(struct mt76x2_dev *dev)
  if (err < 0)
  return err;

- err = mt76x2u_alloc_queues(dev);
+ err = mt76x2u_init_hardware(dev);
  if (err < 0)
  goto fail;

- err = mt76x2u_init_hardware(dev);
+ err = mt76x2u_alloc_queues(dev);
  if (err < 0)
  goto fail;

LorenzoBianconi, 2018-05-11:
> @LorenzoBianconi <https://github.com/LorenzoBianconi>:
>
> 1- Are you using the kernel source from my github repo or have you
> backported just the mt76 driver? Huge ping delay seems related to this [...]

Have you backported mac80211 stack? Could you please try to compile the
kernel from my github tree since I run a pretty long test (few hours) and
the connection was stable (~200M TCP)

> 2- If I understood correctly, if the dongle is plugged in your pc at
> device bootstrap it will properly work just once, if you reboot the pc
> without unplugging the dongle it will not be loaded [...]

Could you please try the patch I posted to @cyangy?

> 3- Before rmmod mt76x2u mt76x2_common, have you stopped wpa_supplicant? (I
> agree it should not crash even if wpa_supplicant is still running)
>
> Well, wpa_supplicant was running. I did some more testing and
> disconnected, and if wpa_supplicant is running, even if it (supposedly)
> isn't doing anything, I get that crash when doing rmmod. But if I stop
> wpa_supplicant, I don't get the crash.
>
> Are there any printk I could add anywhere, or any debug mode I could
> activate to provide more info? Any other suggestions or anything, please
> feel free to ask.
>
> <https://github.com/openwrt/mt76/issues/139#issuecomment-388217553>, or mute
> the thread

I will try to reproduce it but it is not a big issue if the fw is running
when you try to re-insert the module (it is the WARN_ON() in usb.c:345)

cyangy, 2018-05-11:
@LorenzoBianconi : I'm trying your patch now . 
By the way, after hours  work, I found that the issuse only happen if  function `mt76_usb_rx_tasklet(unsigned long data) `  been called while  the  `for (i = 0; i < q->ndesc; i++)`  cycle  of  function `mt76_usb_alloc_rx(struct mt76_dev *dev)`  not finish yet. for example,this  is what  the output  of   **rx urb mismatch**  issuse  happen   after I add debug message in usb.c
```
[   10.444421] mt76x2u 1-1:1.0: ASIC revision: 76120044
[   10.465555] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=0 err=0
[   10.472853] mt76x2u 1-1:1.0: mt76_usb_rx_tasklet(unsigned long data) been called err=0 
[   10.482571] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=1 err=0
[   10.489920] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=2 err=0
                          ...............................................
```
But in this  situation ,for example 
```
[   10.444421] mt76x2u 1-1:1.0: ASIC revision: 76120044
[   10.465555] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=0 err=0
[   10.482571] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=1 err=0
                           ...............................................
[   10.489920] mt76x2u 1-1:1.0: mt76_usb_alloc_rx(struct mt76_dev *dev) been called  i=15 err=0  // for (i = 0; i < q->ndesc; i++)   cycle  in  mt76_usb_alloc_rx(struct mt76_dev *dev)  finished
[   11.521452] mt76x2u 1-1:1.0: mt76_usb_rx_tasklet(unsigned long data) been called err=0 
                          ...............................................
```
all will be fine.

So, my opinion is : make sure the function `mt76_usb_rx_tasklet`  not been called before the  function `mt76_usb_alloc_rx` finish it's job.

LorenzoBianconi, 2018-05-11:
@cyangy thx a lot for the bug report I will work on it

cyangy, 2018-05-11:
@LorenzoBianconi : I use the patch  of  mt76x2u_init.c you privide , after  at least  50 times  reboot (with power cut off and not cut off ),  the `rx urb mismatch` issuse disappear,but these issuse happen
sometime 
```
[   13.603496] mt76x2u 1-1:1.0: error: mt76x2u_mcu_wait_resp timed out
[   27.619708] mt76x2u: probe of 1-1:1.0 failed with error -110
```
or sometimes 
```
[   10.083453] mt76x2u 1-1:1.0: firmware upload timed out
[   25.118571] mt76x2u 1-1:1.0: MAC RX failed to stop
[   25.129494] mt76x2u: probe of 1-1:1.0 failed with error -5
```
or sometimes
```
[    8.983391] mt76x2u 1-1:1.0: Build Time: 201406241830____
[   10.083453] mt76x2u 1-1:1.0: firmware upload timed out
[   25.118571] mt76x2u 1-1:1.0: MAC RX failed to stop
[   25.129494] mt76x2u: probe of 1-1:1.0 failed with error -5
```
or sometimes 

```
[    8.844699] mt76x2u 1-1:1.0: ROM patch build: 20140408060640a
[   12.093501] mt76x2u 1-1:1.0: vendor request req:07 off:09a8 failed:-110
[   15.293486] mt76x2u 1-1:1.0: vendor request req:06 off:09a8 failed:-110
```
Only 20% of these testes  work well . So the patch may not a solution.
As  I mentioned before,  maybe the problem  is just  because of  **the function `mt76_usb_rx_tasklet`  been called before the  function `mt76_usb_alloc_rx` finish it's job** . Is this  the key point ?

cyangy, 2018-05-11:
@araujorm : This is my patch to track function excute in usb.c ,you may have a try to figure out what you issue relate to.
``` diff
diff -Nur a/usb.c b/usb.c
--- a/usb.c	2018-05-10 23:09:20.000000000 +0800
+++ b/usb.c	2018-05-11 19:54:18.011881920 +0800
@@ -357,6 +357,7 @@
 	struct mt76_usb_buf *buf;
 	int err;
 
+	static int mt76_usb_rx_tasklet_counter = 0;
 	rcu_read_lock();
 
 	while (true) {
@@ -371,6 +372,7 @@
 		err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
 					  buf, GFP_ATOMIC,
 					  mt76_usb_complete_rx, dev);
+	       dev_err(dev->dev, "mt76_usb_rx_tasklet(unsigned long data) called,err=%d mt76_usb_rx_tasklet_counter=%d\n",err,mt76_usb_rx_tasklet_counter); 
 		if (err < 0)
 			break;
 	}
@@ -400,6 +402,7 @@
 		err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
 					  &q->entry[i].ubuf, GFP_KERNEL,
 					  mt76_usb_complete_rx, dev);
+		dev_err(dev->dev, "mt76_usb_alloc_rx(struct mt76_dev *dev) called  i=%d err=%d\n",i, err);
 		if (err < 0)
 			return err;
 	}
@@ -416,6 +419,7 @@
 		err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
 					  &q->entry[i].ubuf, GFP_KERNEL,
 					  mt76_usb_complete_rx, dev);
+		dev_err(dev->dev, "mt76_usb_submit_rx_buffers(struct mt76_dev *dev) called  i=%d err=%d\n",i, err); //this functions seem never been called
 		if (err < 0)
 			return err;
 	}
```

LorenzoBianconi, 2018-05-11:
> @araujorm : This is my patch to track function excute in usb.c ,you may have a try to figure out what you issue relate to.
>
> diff -Nur a/usb.c b/usb.c
> --- a/usb.c 2018-05-10 23:09:20.000000000 +0800
> +++ b/usb.c 2018-05-11 19:54:18.011881920 +0800
> @@ -357,6 +357,7 @@
>   struct mt76_usb_buf *buf;
>   int err;
>
> + static int mt76_usb_rx_tasklet_counter = 0;
>   rcu_read_lock();
>
>   while (true) {
> @@ -371,6 +372,7 @@
>   err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
>    buf, GFP_ATOMIC,
>    mt76_usb_complete_rx, dev);
> +       dev_err(dev->dev, "mt76_usb_rx_tasklet(unsigned long data) called,err=%d mt76_usb_rx_tasklet_counter=%d\n",err,mt76_usb_rx_tasklet_counter);
>   if (err < 0)
>   break;
>   }
> @@ -400,6 +402,7 @@
>   err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
>    &q->entry[i].ubuf, GFP_KERNEL,
>    mt76_usb_complete_rx, dev);
> + dev_err(dev->dev, "mt76_usb_alloc_rx(struct mt76_dev *dev) called  i=%d err=%d\n",i, err);
>   if (err < 0)
>   return err;
>   }
> @@ -416,6 +419,7 @@
>   err = mt76_usb_submit_buf(dev, USB_DIR_IN, MT_EP_IN_PKT_RX,
>    &q->entry[i].ubuf, GFP_KERNEL,
>    mt76_usb_complete_rx, dev);
> + dev_err(dev->dev, "mt76_usb_submit_rx_buffers(struct mt76_dev *dev) called  i=%d err=%d\n",i, err); //this functions seem never been called
>   if (err < 0)
>   return err;
>   }

@cyangy Could you please test the following patch that make usb urb
submission atomic please.

https://pastebin.com/mgQiYCCa

cyangy, 2018-05-11:
@LorenzoBianconi : I'll do it right now.

cyangy, 2018-05-11:
@LorenzoBianconi : :+1:  Problem solved. After apply  the patch you posted to [mt76x2u](https://github.com/LorenzoBianconi/mt76/tree/mt76x2u), just the  patch, no other code modfied. Now  after 30 times reboot (with power cut off and not cut off ), pluge and unpluge ,every thing work well except when 
`rmmod mt76x2u`  when the USB pluged-in 
```
[   78.995889] usbcore: deregistering interface driver mt76x2u
[   79.001560] wlan0: deauthenticating from 14:75:8a:44:46:e2 by local choice (Reason: 3=DEAUTH_LEAVING)
[   79.075222] ------------[ cut here ]------------
[   79.079893] WARNING: CPU: 0 PID: 3417 at /home/myname/LEDE/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/mt76-mt76x2u-758f64a1/usb.c:345 mt76_usb_complete_rx+0x90/0x100 [mt76]
[   79.096125] rx urb mismatch
[   79.096129] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async rtlwifi pppox ppp_generic nf_conntrack_ipv6 mwl8k mt76x2u(-) mt76x2_common mt76 mac80211 iptable_nat ipt_REJECT ipt_MASt
[   79.169793]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nm
[   79.197676] CPU: 0 PID: 3417 Comm: sh Not tainted 4.14.37 #0
[   79.203356] Hardware name: Marvell Kirkwood (Flattened Device Tree)
[   79.209643] Backtrace: 
[   79.212108] [<c0105e1c>] (dump_backtrace) from [<c0106104>] (show_stack+0x18/0x1c)
[   79.219711]  r7:bf3a30ec r6:00000000 r5:bf3a4f30 r4:c61f7e40
[   79.225407] [<c01060ec>] (show_stack) from [<c05886c0>] (dump_stack+0x20/0x28)
[   79.232671] [<c05886a0>] (dump_stack) from [<c01103d8>] (__warn+0xe0/0x10c)
[   79.239664] [<c01102f8>] (__warn) from [<c0110444>] (warn_slowpath_fmt+0x40/0x48)
[   79.247182]  r9:00000040 r8:ffffe000 r7:60000093 r6:c6ce9b90 r5:c6998700 r4:bf3a4f20
[   79.254967] [<c0110408>] (warn_slowpath_fmt) from [<bf3a30ec>] (mt76_usb_complete_rx+0x90/0x100 [mt76])
[   79.264399]  r3:bf3a610c r2:bf3a4f20
[   79.267990]  r4:c6ce9460
[   79.270548] [<bf3a305c>] (mt76_usb_complete_rx [mt76]) from [<c0428f5c>] (__usb_hcd_giveback_urb+0xac/0x104)
[   79.280420]  r7:c6a558e0 r6:00000000 r5:a0000013 r4:c6998700
[   79.286109] [<c0428eb0>] (__usb_hcd_giveback_urb) from [<c0429044>] (usb_giveback_urb_bh+0x90/0xd4)
[   79.295192]  r7:c6a558e0 r6:00000000 r5:c61f7ea8 r4:c6a558dc
[   79.300880] [<c0428fb4>] (usb_giveback_urb_bh) from [<c01141b8>] (tasklet_action+0x84/0xcc)
[   79.309265]  r7:00000100 r6:00000000 r5:c0808fa0 r4:00000000
[   79.314951] [<c0114134>] (tasklet_action) from [<c01014d0>] (__do_softirq+0x118/0x2fc)
[   79.322903]  r7:00000100 r6:00000018 r5:00000006 r4:c084d138
[   79.328588] [<c01013b8>] (__do_softirq) from [<c0113d98>] (irq_exit+0xa4/0xf8)
[   79.335846]  r10:00000000 r9:c61f7fb0 r8:c7809400 r7:00000001 r6:00000000 r5:c0847824
[   79.343705]  r4:00000000
[   79.346257] [<c0113cf4>] (irq_exit) from [<c013f548>] (__handle_domain_irq+0x8c/0xa8)
[   79.354123] [<c013f4bc>] (__handle_domain_irq) from [<c010138c>] (orion_handle_irq+0x74/0xa0)
[   79.362687]  r9:c61f7fb0 r8:00000001 r7:c087a87c r6:c781a01c r5:00080000 r4:00000013
[   79.370466] [<c0101318>] (orion_handle_irq) from [<c0106f78>] (__irq_usr+0x58/0x80)
[   79.378150] Exception stack(0xc61f7fb0 to 0xc61f7ff8)
[   79.383220] 7fa0:                                     0000000a b6faf76a 0000006f 0006e803
[   79.391436] 7fc0: 0000000a 0166250e 0166250c 0008332c 00000004 00000000 00000000 bea8f654
[   79.399648] 7fe0: 00000300 bea8f5e8 000351cc 000351f8 20000010 ffffffff
[   79.406295]  r10:00000000 r9:00000000 r8:00053977 r7:0005397f r6:ffffffff r5:20000010
[   79.414154]  r4:000351f8
[   79.416691] ---[ end trace b4be17775b2a01bd ]---
```
but I don't think it's a problem(because if `insmod mt76x2u` , it works again). If someone  wants to remove the device, just unpluge it .

LorenzoBianconi, 2018-05-11:
@cyangy do you mean this [patch](https://pastebin.com/mgQiYCCa), right?
If you confirm I will push it. I will look to the rmmod issue during the we

Thx for testing

cyangy, 2018-05-11:
@LorenzoBianconi : Yes this [patch](https://pastebin.com/mgQiYCCa). Good night.

LorenzoBianconi, 2018-05-11:
@cyangy ok, cool, I will push it. Btw I will stabilize STA mode and push it upstream first, then I will switch to AP mode (I am not 100% of hw capabilities). Thx a lot for testing it

araujorm, 2018-05-11:
Hi.

First of all, I can also confirm that with the latest patch tested by cyangy I'm no longer having problems inserting the module after a cold boot, which is great. Moving on.

I was yet unable to compile your full kernel, but still planning to try it later when possible.

Meanwhile, after looking at the issue you referred, I've been doing some checks and trying to do some debugging and I think I might have found a clue to the culprit for the "pauses".

Take a look at the functions mt76_txq_schedule_list and mt76_txq_send_burst in tx.c. I placed some printks over there near both the ```return -EBUSY``` parts and noticed that everytime I have those pauses, it's hitting there.

I then commented both of those checks, the ones like this, there's one on each of those functions:
```
                if (test_bit(MT76_SCANNING, &dev->state) ||
                    test_bit(MT76_RESET, &dev->state))
                        return -EBUSY;
```

and then I no longer get those huge ping time values. I however don't always have brilliant results on speed test, and the ping values seem irregular oscillating between 1ms and 50 ms, when before they were between 1ms and 2ms (when not suffering from the "pauses", because if so I get 10000ms to 15000ms as I said before). This means something that must be taken care of isn't being, but at least I don't get those mood-killing "pauses".

I've searched for MT76_RESET on the code and I only get other references of that being set or checked in mt7603_mac.c, which isn't used for mt76x2u. So, obviously, something is missing and not being handled for mt76x2u (and I believe it's the same case for mt76x2e). A mystery is: what is setting it? Is there any clashing somewhere?

I then tried to comment the MT76_RESET parts on tx.c, leaving the MT76_SCANNING checks only, and vice-versa, but the huge 15 seconds ping values still occur like before. The "pauses" only stop occurring when I remove both of the -EBUSY returns.

Any clue on what is missing to handle those states correctly? And what about the MT76_RESET, is that really the same state on mt76x2[eu] that what you have on mt7603?

If there's anything you want me to try, please feel free to ask. Meanwhile I'll continue to test usage and stability without both of those -EBUSY returns.

LorenzoBianconi, 2018-05-12:
>
> Hi.
>
> First of all, I can also confirm that with the latest patch tested by
> cyangy I'm no longer having problems inserting the module after a cold
> boot, which is great. Moving on.
>
> I was yet unable to compile your full kernel, but still planning to try it
> later when possible.
What is the kernel version you are running? mt76 tx side is based on
mac80211 queue management so first you need to be sure you are running an
updated mac80211 stack.
To confirm this I have never faced that issue running mt76x2u on my laptop
or on openwrt. @cyangy what about you?

> Meanwhile, after looking at the issue you referred, I've been doing some
> checks and trying to do some debugging and I think I might have found a
> clue to the culprit for the "pauses".
>
> Take a look at the functions mt76_txq_schedule_list and
> mt76_txq_send_burst in tx.c. I placed some printks over there near both the return
> -EBUSY parts and noticed that everytime I have those pauses, it's hitting
> there.
>
> I then commented both of those checks, the ones like this, there's one on
> each of those functions:
>
>                 if (test_bit(MT76_SCANNING, &dev->state) ||
>                     test_bit(MT76_RESET, &dev->state))
>                         return -EBUSY;
>
> and then I no longer get those huge ping time values. I however don't
> always have brilliant results on speed test, and the ping values seem
> irregular oscillating between 1ms and 50 ms, when before they were between
> 1ms and 2ms (when not suffering from the "pauses", because if so I get
> 10000ms to 15000ms as I said before). This means something that must be
> taken care of isn't being, but at least I don't get those mood-killing
> "pauses".
>
> I've searched for MT76_RESET on the code and I only get other references
> of that being set or checked in mt7603_mac.c, which isn't used for mt76x2u.
> So, obviously, something is missing and not being handled for mt76x2u (and
> I believe it's the same case for mt76x2e). A mystery is: what is setting
> it? Is there any clashing somewhere?
>
> I then tried to comment the MT76_RESET parts on tx.c, leaving the
> MT76_SCANNING checks only, and vice-versa, but the huge 15 seconds ping
> values still occur like before. The "pauses" only stop occurring when I
> remove both of the -EBUSY returns.
>
> Any clue on what is missing to handle those states correctly? And what
> about the MT76_RESET, is that really the same state on mt76x2[eu] that what
> you have on mt7603?
>
> If there's anything you want me to try, please feel free to ask. Meanwhile
> I'll continue to test usage and stability without both of those -EBUSY
> returns.
>
> <https://github.com/openwrt/mt76/issues/139#issuecomment-388504185>, or mute
> the thread
I guess first we need to be on the same page running the same sw. I guess
it will pretty easy to use my GH tree, just copy fedora default config and
fix new stuff typing 'make oldconfig'. I am currently using
wireless-drivers-next tree on my feodra27 btw. Thx a lot for testing

Regards,
Lorenzo

[29 more replies]

closed 2018-09-29 by nbd168