openwrt/mt76 #141: TSF Field not set in beacon for 7612E chip in 5GHz

mdsohailahmad, 2018-01-18:
I am observing client disconnections. One of the reasons identified is TSF remains set to 0.

nbd168, 2018-01-18:
Which TSF field? I just did a capture and looked at the TSF field in beacons and it appears to be set properly.

mdsohailahmad, 2018-01-19:
Please see my observation below. With Mediatek stock firmware, TSF field is present and values are also monotonically increasing. But with LEDE/OpenWrt firmware the field is present but the values are always 0.
================================================================
Case1: Mediatek original driver
================================================================
ra0       Link encap:Ethernet  HWaddr 68:F0:BC:00:D0:18
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:4

rai0      Link encap:Ethernet  HWaddr 68:F0:BC:00:D0:1B
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:13

# iwconfig rai0
rai0      RTWIFI SoftAP  ESSID:"WIFIUSB_5G-00d018"
          Mode:Managed  Channel=64  Access Point: 68:F0:BC:00:D0:1B
          Bit Rate=300 Mb/s

****************************************************************
Sniffer output
Run1:
iw dev wlan0 scan | grep -i 68:F0 -A 1
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 251496023 usec (0d, 00:04:11)

Run2:
iw dev wlan0 scan | grep -i 68:F0 -A 1
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 279902121 usec (0d, 00:04:39)
*****************************************************************

=================================================================
Case 2: LEDE Firmware with latest mt76 driver
=================================================================
wlan-hs24 Link encap:Ethernet  HWaddr 68:F0:BC:00:D0:18
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:23 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:3810 (3.7 KiB)

wlan-hs50 Link encap:Ethernet  HWaddr 68:F0:BC:00:D0:1B
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:3748 (3.6 KiB)

root@:~#
root@:~# iw dev wlan-hs24 info
Interface wlan-hs24
        ifindex 24
        wdev 0x100000004
        addr 68:f0:bc:00:d0:18
        ssid WIFI-2G
        type AP
        wiphy 1
        channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
        txpower 20.00 dBm
root@:~# iw dev wlan-hs50 info
Interface wlan-hs50
        ifindex 23
        wdev 0x4
        addr 68:f0:bc:00:d0:1b
        ssid WIFI-5G
        type AP
        wiphy 0
        channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
        txpower 18.00 dBm

****************************************************************
Sniffer output
Run1:
iw dev wlan0 scan | grep -i 68:F0 -A 1
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 0 usec (0d, 00:00:00)

Run2:
iw dev wlan0 scan | grep -i 68:F0 -A 1
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 0 usec (0d, 00:00:00)

Run3: (All WiFi in the range)
iw dev wlan0 scan | grep -i TSF -B 1
BSS 70:d3:79:2b:45:4f(on wlan0)
        TSF: 88431891223 usec (1d, 00:33:51)
--
BSS 18:64:72:ee:88:f0(on wlan0)
        TSF: 66548570991 usec (0d, 18:29:08)
--
BSS 18:64:72:ee:88:f1(on wlan0)
        TSF: 66548571387 usec (0d, 18:29:08)
--
BSS 00:a0:0a:23:08:60(on wlan0)
        TSF: 52011661180 usec (0d, 14:26:51)
--
BSS 00:a0:0a:23:08:61(on wlan0)
        TSF: 52011661805 usec (0d, 14:26:51)
--
BSS 78:d3:8d:ed:ac:14(on wlan0)
        TSF: 64308721509 usec (0d, 17:51:48)
--
BSS 00:a0:0a:22:52:60(on wlan0)
        TSF: 241076950753 usec (2d, 18:57:56)
--
BSS 00:a0:0a:22:52:61(on wlan0)
        TSF: 241076952443 usec (2d, 18:57:56)
--
BSS 3a:56:5a:9a:db:97(on wlan0)
        TSF: 456277244 usec (0d, 00:07:36)
--
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 0 usec (0d, 00:00:00)

*****************************************************************

nbd168, 2018-01-19:
I looked into this. It seems that beacons had the proper TSF value set, but probe responses didn't. I've pushed a commit to master that fixes this.

closed 2018-01-19 by nbd168

mdsohailahmad, 2018-01-20:
Verified. The issue is resolved. Thanks.
iw dev wlan0 scan | grep -i TSF -B 1
BSS 68:f0:bc:00:d0:1b(on wlan0)
        TSF: 525440316 usec (0d, 00:08:45)