openwrt/mt76 #142: Too many MCU messages

mdsohailahmad, 2018-01-22:
Hi I am getting to many these messages. I will appreciate if one can point me to the exact fix.
[  627.722655] MCU message 31 (seq 4) timed out
[  628.443951] MCU message 31 (seq 5) timed out
[  629.161805] MCU message 31 (seq 6) timed out
[  629.886893] MCU message 31 (seq 7) timed out
[  630.608587] MCU message 31 (seq 8) timed out
[  631.328344] MCU message 31 (seq 9) timed out
[  632.048735] MCU message 31 (seq 10) timed out
[  632.768757] MCU message 31 (seq 11) timed out
[  633.493920] MCU message 31 (seq 12) timed out
[  634.216493] MCU message 31 (seq 13) timed out
[  634.940194] MCU message 31 (seq 14) timed out
[  635.670144] MCU message 31 (seq 15) timed out
[  636.392369] MCU message 31 (seq 1) timed out
[  637.125573] MCU message 31 (seq 2) timed out
[  637.887099] MCU message 31 (seq 3) timed out
[  638.612095] MCU message 31 (seq 4) timed out
[  639.336311] MCU message 31 (seq 5) timed out
[  640.090168] MCU message 31 (seq 6) timed out
[  640.815029] MCU message 31 (seq 7) timed out
[  641.577323] MCU message 31 (seq 8) timed out
[  642.296622] MCU message 31 (seq 9) timed out
[  643.207889] MCU message 31 (seq 10) timed out
[  643.970049] MCU message 31 (seq 11) timed out
[  644.693206] MCU message 31 (seq 12) timed out
[  645.415252] MCU message 31 (seq 13) timed out
[  646.138283] MCU message 31 (seq 14) timed out
[  646.863635] MCU message 31 (seq 15) timed out
[  647.586981] MCU message 31 (seq 1) timed out
[  648.492355] MCU message 31 (seq 2) timed out
[  649.214227] MCU message 31 (seq 3) timed out
[  649.936770] MCU message 31 (seq 4) timed out

ngkaho1234, 2018-01-25:
Hi all,
I can confirm this on my R6220.

Build: [    0.000000] Linux version 4.9.67 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5484-69d22a6) ) #0 SMP Fri Dec 8 21:34:31 2017
[r6220.log](https://github.com/openwrt/mt76/files/1664156/r6220.log)

Whenever this happens, the wireless connection drops and I have to resort to rebooting the device.

Mushoz, 2018-03-23:
Seeing the same thing on my Dir-860l B1 router running Lede 17.01.4. It was caused by heavy WLAN activity. Setting igmp_snooping to 1 in my lan bridge in /etc/config/network fixed the issue by removing the heavy load from the WLAN. I am not sure whether these messages are normal under these circumstances, but if either of you are using multicast traffic on your network, please try enabling igmp_snooping to see if this fixes the issue.

MaFeSa, 2018-04-14:
Hi, I also have the same problem with DIR-860L B1 and OpenWrt SNAPSHOT r6671-f5195e7 / LuCI Master (git-18.101.21196-d488681).

```
[  898.861139] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[179095.316804] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[179097.396788] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[179125.796796] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[179127.876773] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[179129.956764] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[179216.276758] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
```
When it happens also the traffic of the 2.4 stops.

@Mushoz thanks I will try with `option igmp_snooping 1`

MaFeSa, 2018-04-17:
For me is fixed, thanks nbd!

easyteacher, 2018-06-14:
The latest version of mt76 driver are getting  these messages again. 2 months ago that was fine.

MaFeSa, 2018-06-14:
I confirm, with OpenWrt 18.06-SNAPSHOT r7009 and DIR-860L B1 I have some messages (5 days uptime)
[120309.687068] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[195420.772135] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[195480.692097] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[306688.529047] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out

Not happend with Xiaomi 3G but with this router I have other problem (2.4 GHz unstable).

nbd168, 2018-06-18:
Please try the latest version, it should be better now.

easyteacher, 2018-06-18:
5G wifi is very stable now. Thanks nbd!

closed 2018-06-18 by nbd168

MaFeSa, 2018-06-18:
I have no more messages, too.
Thanks for your work nbd.

MaFeSa, 2018-06-27:
DIR-860L B1 
OpenWrt 18.06-SNAPSHOT r7102-3f3a2c9 / LuCI openwrt-18.06 branch (git-18.163.60998-c379f00) 

```
[   27.842429] br-lan: port 2(wlan0) entered blocking state
[   27.853080] br-lan: port 2(wlan0) entered forwarding state
[  353.525389] pppoe-wan: renamed from ppp0
[44137.037406] mt76x2e 0000:01:00.0: MCU message 31 (seq 13) timed out
[59562.303038] mt76x2e 0000:01:00.0: MCU message 31 (seq 14) timed out
[71716.776350] mt76x2e 0000:01:00.0: MCU message 31 (seq 13) timed out
[78258.375002] mt76x2e 0000:01:00.0: MCU message 31 (seq 13) timed out
[100555.161997] mt76x2e 0000:01:00.0: MCU message 31 (seq 6) timed out
[103645.960428] mt76x2e 0000:01:00.0: MCU message 31 (seq 3) timed out
[103648.040449] mt76x2e 0000:01:00.0: MCU message 31 (seq 4) timed out
[112719.622382] pppoe-wan: renamed from ppp0
[114089.558261] mt76x2e 0000:01:00.0: MCU message 31 (seq 10) timed out
[115170.278582] mt76x2e 0000:01:00.0: MCU message 31 (seq 11) timed out
[133442.744973] mt76x2e 0000:01:00.0: MCU message 31 (seq 1) timed out
[133533.224826] mt76x2e 0000:01:00.0: MCU message 31 (seq 11) timed out
[135753.941800] mt76x2e 0000:01:00.0: MCU message 31 (seq 14) timed out
[138094.258941] mt76x2e 0000:01:00.0: MCU message 31 (seq 12) timed out
[140105.695584] mt76x2e 0000:01:00.0: MCU message 31 (seq 9) timed out
[143946.810552] mt76x2e 0000:01:00.0: MCU message 31 (seq 1) timed out
[155530.220758] mt76x2e 0000:01:00.0: MCU message 31 (seq 10) timed out
[156100.141235] mt76x2e 0000:01:00.0: MCU message 31 (seq 8) timed out
[166153.943267] mt76x2e 0000:01:00.0: MCU message 31 (seq 4) timed out
[173206.254818] mt76x2e 0000:01:00.0: MCU message 31 (seq 15) timed out
[173208.334815] mt76x2e 0000:01:00.0: MCU message 31 (seq 1) timed out
[174616.409558] mt76x2e 0000:01:00.0: MCU message 31 (seq 3) timed out
[174618.489561] mt76x2e 0000:01:00.0: MCU message 31 (seq 4) timed out
[176386.083836] mt76x2e 0000:01:00.0: MCU message 31 (seq 6) timed out
```

jkimmel, 2018-11-30:
ZBT-APE522ii
OpenWrt SNAPSHOT unknown / LuCI Master (git-18.333.34665-4f2b80e)
same messages here

Yohan54, 2020-01-25:
Issues have re-appeared again
Firmware Version: OpenWrt SNAPSHOT unknown / LuCI Master git-20.022.75744-8947526
Kernel Version: 4.14.166

mdiehm, 2020-10-19:
I have the same wifi issues happening once every few days and can only recover when rebooting.

loglines: mt76_wmac ... timed out...

any ideas for a workaround or even fix?

```
SoC Type: MediaTek MT7628AN ver:1 eco:2
MIPS: machine is GL-MT300N-V2
[   12.351439] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[   13.390246] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[   13.396271] mt76_wmac 10300000.wmac: Build Time: 20151201183641

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.4'
DISTRIB_REVISION='r11208-ce6496d796'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'

Linux md-home-wifirouter 4.14.195 #0 Sun Sep 6 16:19:39 2020 mips GNU/Linux
```

ptpt52, 2020-12-04:
same issue on latest HEAD code:
```
[28186.958887] mt76_wmac 10300000.wmac: MCU message 8 (seq 2) timed out
[28190.030876] mt76_wmac 10300000.wmac: MCU message 8 (seq 3) timed out
[28193.102877] mt76_wmac 10300000.wmac: MCU message 8 (seq 4) timed out
[28196.174877] mt76_wmac 10300000.wmac: MCU message 8 (seq 5) timed out
[28199.246893] mt76_wmac 10300000.wmac: MCU message 8 (seq 6) timed out
[28202.318876] mt76_wmac 10300000.wmac: MCU message 8 (seq 7) timed out
[28205.390878] mt76_wmac 10300000.wmac: MCU message 8 (seq 8) timed out
[28208.462881] mt76_wmac 10300000.wmac: MCU message 8 (seq 9) timed out
[28211.534877] mt76_wmac 10300000.wmac: MCU message 8 (seq 10) timed out
[28214.606877] mt76_wmac 10300000.wmac: MCU message 8 (seq 11) timed out
[28217.678877] mt76_wmac 10300000.wmac: MCU message 8 (seq 12) timed out
[28221.006876] mt76_wmac 10300000.wmac: MCU message 8 (seq 13) timed out
```

Linaro1985, 2020-12-04:
same on 19.07.4 code (MT7628):
[40729.144156] mt76_wmac 10300000.wmac: MCU message 8 (seq 7) timed out
[40732.184026] mt76_wmac 10300000.wmac: MCU message 8 (seq 8) timed out
[40735.223911] mt76_wmac 10300000.wmac: MCU message 8 (seq 9) timed out
[40738.263744] mt76_wmac 10300000.wmac: MCU message 8 (seq 10) timed out
[40741.303655] mt76_wmac 10300000.wmac: MCU message 8 (seq 11) timed out
[40744.343485] mt76_wmac 10300000.wmac: MCU message 8 (seq 12) timed out
[40747.383365] mt76_wmac 10300000.wmac: MCU message 8 (seq 13) timed out
[40750.423367] mt76_wmac 10300000.wmac: MCU message 8 (seq 14) timed out
[40753.463063] mt76_wmac 10300000.wmac: MCU message 8 (seq 15) timed out
[40756.502923] mt76_wmac 10300000.wmac: MCU message 8 (seq 1) timed out
[40759.542790] mt76_wmac 10300000.wmac: MCU message 8 (seq 2) timed out
[40762.582655] mt76_wmac 10300000.wmac: MCU message 8 (seq 3) timed out
[40765.622521] mt76_wmac 10300000.wmac: MCU message 8 (seq 4) timed out
[40768.902376] mt76_wmac 10300000.wmac: MCU message 8 (seq 5) timed out

sanitariu, 2021-08-10:
Anyone knows how to fix this with some settings except making shapers for downloading ?

maulvi, 2022-01-29:
I got r6220 tried with 19.07, 21.02.1 and have the same problem with single, dual band wifi running after a while.

```
[  236.735995] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[  637.198889] mt7603e 0000:02:00.0: MCU message 8 (seq 2) timed out
[  639.269198] br-lan: port 2(wlan0-1) entered disabled state
[  642.318843] mt7603e 0000:02:00.0: MCU message 8 (seq 3) timed out
[  647.438796] mt7603e 0000:02:00.0: MCU message 8 (seq 4) timed out
[  652.558748] mt7603e 0000:02:00.0: MCU message 8 (seq 5) timed out
[  657.678705] mt7603e 0000:02:00.0: MCU message 8 (seq 6) timed out
[  662.798662] mt7603e 0000:02:00.0: MCU message 8 (seq 7) timed out
[  667.918619] mt7603e 0000:02:00.0: MCU message 8 (seq 8) timed out
[  673.038583] mt7603e 0000:02:00.0: MCU message 8 (seq 9) timed out
[  676.419968] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[  678.158549] mt7603e 0000:02:00.0: MCU message 8 (seq 10) timed out
[  683.278518] mt7603e 0000:02:00.0: MCU message 8 (seq 11) timed out
[  688.398495] mt7603e 0000:02:00.0: MCU message 8 (seq 12) timed out
[  693.518421] mt7603e 0000:02:00.0: MCU message 8 (seq 13) timed out
[  698.638382] mt7603e 0000:02:00.0: MCU message 8 (seq 14) timed out
[  703.758336] mt7603e 0000:02:00.0: MCU message 8 (seq 15) timed out
[  708.878314] mt7603e 0000:02:00.0: MCU message 8 (seq 1) timed out
[  713.998266] mt7603e 0000:02:00.0: MCU message 8 (seq 2) timed out
[  719.118253] mt7603e 0000:02:00.0: MCU message 8 (seq 3) timed out
```

I've tried @Mushoz solution I think its fixed