openwrt/mt76 #143: Kernel Panic on version mt76-9f685fea[mt7603e]

kofec, 2018-01-25:
[ 7225.417782] ------------[ cut here ]------------
[ 7225.422442] WARNING: CPU: 0 PID: 0 at /mnt/openwrt/OpenwrtXiaomiDVB/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-12-17-9f685fea/mt7603_mac.c:415 mt7603_mac_fill_rx+0x16c/0x414 [mt7603e]
[ 7225.440555] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat hid evdev i2c_dev ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi
[ 7225.511273]  snd_seq_device snd_hwdep snd soundcore vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc32c_generic crc32_generic
[ 7225.540759] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.77 #0
[ 7225.546648] Stack : 00000000 00000000 805b7b4a 00000033 8043e364 00000000 00000000 805b0000
[ 7225.554993]         805381bc 80537da7 804c4a84 00000000 00000000 805b3824 00000000 8e8b5a00
[ 7225.563337]         00000000 8006a9e8 00000001 805b0000 8053dec4 8053dec8 804c9678 8fc0dd7c
[ 7225.571681]         00000003 800a78a0 00000000 8e8b5a00 00000000 00000000 00000000 00c0dd7c
[ 7225.580024]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 7225.588366]         ...
[ 7225.590802] Call Trace:
[ 7225.593260] [<8000f714>] show_stack+0x54/0x88
[ 7225.597609] [<801dabcc>] dump_stack+0x8c/0xd0
[ 7225.601947] [<8002adc4>] __warn+0xe4/0x118
[ 7225.606024] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[ 7225.611059] [<8e93c13c>] mt7603_mac_fill_rx+0x16c/0x414 [mt7603e]
[ 7225.617129] [<8e93b020>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
[ 7225.623116] [<8e8d0d7c>] mt76_dma_attach+0xaac/0xcc0 [mt76]
[ 7225.628731] ---[ end trace ba72a6894473f2f5 ]---
[34439.839292] ------------[ cut here ]------------
[34439.844072] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4306 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[34439.855105] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 84 (0x54)
[34439.864487] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG wireguard usbhid slhc nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat hid evdev i2c_dev ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi
[34439.935199]  snd_seq_device snd_hwdep snd soundcore vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc32c_generic crc32_generic
[34439.964668] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.77 #0
[34439.971766] Stack : 00000000 00000000 805b7b4a 00000041 8043e364 00000000 00000000 805b0000
[34439.980112]         805381bc 80537da7 804c4a84 00000000 00000000 805b3824 00000000 000009a8
[34439.988456]         8ce30dd0 8006a9e8 00000001 805b0000 8053dec4 8053dec8 804c9678 8fc0dc8c
[34439.996799]         00000003 800a78a0 00000000 000009a8 81211e20 00000000 00000002 00c0dc8c
[34440.005142]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[34440.013485]         ...
[34440.015921] Call Trace:
[34440.018379] [<8000f714>] show_stack+0x54/0x88
[34440.022727] [<801dabcc>] dump_stack+0x8c/0xd0
[34440.027064] [<8002adc4>] __warn+0xe4/0x118
[34440.031142] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[34440.036147] [<8e9a5c20>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[34440.042251] [<8e8d1b78>] mt76_rx_complete+0x18c/0x234 [mt76]
[34440.047888] [<8e8d1cb4>] mt76_rx_poll_complete+0x94/0x114 [mt76]
[34440.053870] [<8e8d0e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
[34440.059497] ---[ end trace ba72a6894473f2f6 ]---

MaFeSa, 2018-01-25:
I have many of these warnings too, OpenWrt SNAPSHOT r5739-18f4944 - Xiaomi 3G

```
[55651.487725] ------------[ cut here ]------------
[55651.492376] WARNING: CPU: 0 PID: 0 at /home/massy/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-12-17-68b0cf17/mt7603_mac.c:392 mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[55651.509369] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit
[55651.579881]  act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc32c_generic
[55651.622307] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.75 #0
[55651.628195] Stack : 00000000 00000000 80537b4a 00000033 803f49c4 00000000 00000000 80530000
[55651.636542]         804d61bc 804d5da7 8046ea24 00000000 00000000 80533824 811b0400 8e8a9a00
[55651.644888]         00000000 8006a9c8 00000001 80530000 804dbec4 804dbec8 80473630 8fc0dd84
[55651.653233]         00000003 800a7670 811b0400 8e8a9a00 00000000 00000000 00000000 00c0dd84
[55651.661578]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[55651.669921]         ...
[55651.672358] Call Trace:
[55651.674814] [<8000f884>] show_stack+0x54/0x88
[55651.679162] [<801da1fc>] dump_stack+0x8c/0xd0
[55651.683500] [<8002adc4>] __warn+0xe4/0x118
[55651.687578] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[55651.692612] [<8ea3c00c>] mt7603_mac_fill_rx+0x11c/0x330 [mt7603e]
[55651.698683] [<8ea3af40>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
[55651.704669] [<8eb2cd6c>] mt76_dma_attach+0xa9c/0xc94 [mt76]
[55651.710310] ---[ end trace 711c4d9d7b9a394f ]---
```

kofec, 2018-02-10:
24h and no stacktrace so close this issue

closed 2018-02-10 by kofec