openwrt/mt76 #145: No connection with 2.4 GHz

MaFeSa, 2018-02-05:
Hi,
I compiled now with this commit https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=fe1244e04d45872f09f47b49465ca9367b4c8bd0

I have a problem with all my (5 five) Galaxy Tab A 2016, after only a minute that the wireless clients remain connected but there is no access to the network. There are no errors in the kernel and in the system log.
No problem with other 2.4 GHz devices.

Thanks

lukasz1992, 2018-02-05:
I confirm on my notebook with Intel card.
After a while (1 - 5 minutes) communication stops working.

kofec, 2018-02-05:
On lede forum there was: I applied option disassoc_low_ack ‘0’ and option legacy_rates ‘0’ but I have no internet warnings on my android phone often.

lumixen, 2018-02-05:
I can confirm that with the latest openwrt snapshot my Xiaomi Router 3G struggles with 2.4G connectivity  issues as well. Periodically (1-5 mins), on my laptop with `Intel-AC8260`, `Windows 10`, 2.4 network connection just hangs. Some time after (1-5 mins) it may recover itself.
Meantime 5G works pretty stable.

Also I managed to do some iperf testing during the window when 2.4G network was accessible. Following setup: `Laptop (client) --> 2.4G ---> Xiaomi router 3G --> 5G --> PC (Server)`.

`Connecting to host 192.XXX.XXX.XXX, port 5201                                                          
[  4] local 192.XXX.XXX.XXX port 59241 connected to 192.XXX.XXX.XXX port 5201                           
[ ID] Interval           Transfer     Bandwidth                                                       
[  4]   0.00-1.00   sec  9.75 MBytes  81.7 Mbits/sec                                                  
[  4]   1.00-2.00   sec  6.50 MBytes  54.5 Mbits/sec                                                  
[  4]   2.00-3.00   sec  3.38 MBytes  28.3 Mbits/sec                                                  
[  4]   3.00-4.00   sec  9.62 MBytes  80.9 Mbits/sec                                                  
[  4]   4.00-5.00   sec  9.88 MBytes  82.8 Mbits/sec                                                  
[  4]   5.00-6.00   sec  10.1 MBytes  84.9 Mbits/sec                                                  
[  4]   6.00-7.00   sec  13.1 MBytes   110 Mbits/sec                                                  
[  4]   7.00-8.00   sec  10.5 MBytes  87.9 Mbits/sec                                                  
[  4]   8.00-9.00   sec  14.5 MBytes   122 Mbits/sec                                                  
[  4]   9.00-10.00  sec  16.0 MBytes   134 Mbits/sec                                                  
[  4]  10.00-11.00  sec  15.0 MBytes   126 Mbits/sec                                                  
[  4]  11.00-12.00  sec  16.6 MBytes   139 Mbits/sec                                                  
[  4]  12.00-13.00  sec  15.5 MBytes   130 Mbits/sec                                                  
[  4]  13.00-14.00  sec  15.5 MBytes   130 Mbits/sec                                                  
[  4]  14.00-15.00  sec  15.9 MBytes   133 Mbits/sec                                                  
[  4]  15.00-16.00  sec  16.8 MBytes   141 Mbits/sec                                                  
[  4]  16.00-17.00  sec  17.1 MBytes   144 Mbits/sec                                                  
[  4]  17.00-18.00  sec  15.9 MBytes   133 Mbits/sec                                                  
[  4]  18.00-19.00  sec  16.0 MBytes   134 Mbits/sec                                                  
[  4]  19.00-20.00  sec  16.2 MBytes   136 Mbits/sec                                                  
[  4]  20.00-21.00  sec  15.5 MBytes   130 Mbits/sec                                                  
[  4]  21.00-22.00  sec  16.0 MBytes   134 Mbits/sec                                                  
[  4]  22.00-23.00  sec  15.4 MBytes   129 Mbits/sec                                                  
[  4]  23.00-24.00  sec  15.8 MBytes   132 Mbits/sec                                                  
[  4]  24.00-25.00  sec  16.8 MBytes   140 Mbits/sec                                                  
[  4]  25.00-26.00  sec  15.2 MBytes   128 Mbits/sec                                                  
[  4]  26.00-27.00  sec  16.4 MBytes   137 Mbits/sec                                                  
[  4]  27.00-28.01  sec  15.6 MBytes   130 Mbits/sec                                                  
[  4]  28.01-29.00  sec  6.62 MBytes  56.2 Mbits/sec                                                  
[  4]  29.00-30.00  sec  9.38 MBytes  78.6 Mbits/sec                                                  
[  4]  30.00-31.00  sec  8.38 MBytes  70.3 Mbits/sec                                                  
[  4]  31.00-32.00  sec  14.6 MBytes   123 Mbits/sec                                                  
[  4]  32.00-33.00  sec  16.4 MBytes   137 Mbits/sec                                                  
[  4]  33.00-34.00  sec  16.4 MBytes   137 Mbits/sec                                                  
[  4]  34.00-35.00  sec  9.25 MBytes  77.6 Mbits/sec                                                  
[  4]  35.00-36.00  sec   512 KBytes  4.19 Mbits/sec                                                  
[  4]  36.00-37.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  37.00-38.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  38.00-39.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  39.00-40.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  40.00-41.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  41.00-42.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  42.00-43.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  43.00-44.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[  4]  44.00-45.00  sec  0.00 Bytes  0.00 bits/sec                                                    
[... cut ...]
- - - - - - - - - - - - - - - - - - - - - - - - -                                                                                                     
[ ID] Interval           Transfer     Bandwidth                                                       
[  4]   0.00-60.00  sec   472 MBytes  66.0 Mbits/sec                  sender                          
[  4]   0.00-60.00  sec   472 MBytes  66.0 Mbits/sec                  receiver                        
`
As you can see, when 2.4G is not frozen, I can get quite awesome speeds.

`OpenWrt SNAPSHOT r6009-a8b0232`, `mt76 - 8693864`

MaFeSa, 2018-02-06:
> Some time after (1-5 mins) it may recover itself.
Not for me...

The problem seems only with 7603

pparent76, 2018-02-06:
I seem to have a similar problem with latest commit, even if it's hard for me to understand what happens. I seems very random.

MaFeSa, 2018-02-07:
I will test all commits to find the cause of the problem with 7603

MaFeSa, 2018-02-07:
@nbd168 I have find it, before this commit https://github.com/openwrt/mt76/commit/b0508d3044a2d469946347dfc020afb5cd0bac25

2.4 GHz is working fine on 7603 and 7628

pparent76, 2018-02-07:
Thanks a lot for your work!

nbd168, 2018-02-07:
Thanks for tracking it down. I will look into it.

lumixen, 2018-02-07:
@MaFeSa Thanks for the hint! Built locally with 58a41f1 and hangs are gone too.

AspirinJunkie, 2018-02-08:
I have no ability to compile a version on my own. Is it expected that the fix will be included in a daily snapshot soon?

pparent76, 2018-02-08:
Question: How do you do to modify this file to compile a specific commit?

https://github.com/lede-project/source/blob/master/package/kernel/mt76/Makefile

I.e how do you calculate PKG_MIRROR_HASH for a specific commit?

lumixen, 2018-02-08:
@AspirinJunkie  could u try to use script from appropriate lede topic: https://forum.lede-project.org/t/xiaomi-wifi-router-3g/5377/544 ?
It uses for build forked mt76 driver, with excluded all recent commits, related to CCMP PM validation.

AspirinJunkie, 2018-02-08:
@lumixen thanks! i will give it a try. But before i have to get a linux to work on it ;-)

But anyway: We can expect that the fix will be included in the snapshots soon?

lumixen, 2018-02-08:
@pparent76 I use sha256sum utility with downloaded package. Firstly, you change PKG_SOURCE_VERSION, then execute make ..path to package../prepare, then find the downloaded package in dl directory and calculate the new sha256sum. 
P.S. At least, I did so, though not sure if it is an correct approach -)

nbd168, 2018-02-09:
Fixed in the latest version

closed 2018-02-09 by nbd168

MaFeSa, 2018-02-09:
@nbd168 Thanks!

lumixen, 2018-02-09:
@nbd168 works for me, thanks!

pparent76, 2018-02-12:
Thanks a lot!

AspirinJunkie, 2018-02-12:
Same here - now it works.
Thanks!

heguihua, 2019-10-10:
I have the same issue too with the latest 18.06.1 & 18.06.4 on my Xiaomi-mini router。The wifi stops working after some minutes, both 2.4G & 5G.

I had to downgrade to an old version (PandoraBox - base on OpenWRT 14.09) , now it seems stable.

Could anyone reopen the issue?

kofec, 2019-10-10:
Xiaomi mini is using diffrent driver for 2,4G. I have the same router and I'm using master branch
(https://downloads.openwrt.org/snapshots/targets/ramips/mt7620/openwrt-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin) but compiled by myself. And I can recomend this build
```
Hostname	XiaomiMiniAP
Model	Xiaomi MiWiFi Mini
Architecture	MediaTek MT7620A ver:2 eco:6
Firmware Version	OpenWrt SNAPSHOT r11175-700e7a2eb9 / LuCI Master (git-19.281.57027-a154533)
Kernel Version	4.14.148
```
You can also try future stable branch:
https://dl.eko.one.pl/luci/openwrt-19.07/ramips/mt7620/luci-19.07-snapshot-r10532-cf3b50377e-ramips-mt7620-miwifi-mini-squashfs-sysupgrade.bin

heguihua, 2019-10-10:
Hi Mr. Kofec, many thanks to your detailed instructions, I will stay tune to the future builds.

But I will stick to (PandoraBox - base on OpenWRT 14.09) for time being, it outperforms the original Xiaomi mini router firmware,  it's stable and the signal is significantly great.