openwrt/mt76 #146: Device disconnect issue on MT76

MangeshBhamre, 2018-02-05:
I have been using latest trunk imagebuilder to create firmware for TL-WR841n v13 (ramips/mt76x8).  We have installed it on multiple routers.

There is an issue of device disconnection reported. The user needs to reboot the router to use the Internet. Wi-Fi SSID does not up. High severity issue.

I see there was check-in on mt76 and then reverted. I hope some work-in-progress. When can this be available for me to test? Is above-reported problem part of the fix?

Unfortunately, there are no logs available. 

Regards,
Mangesh

MangeshBhamre, 2018-02-05:
Manually running below command makes Wi-Fi work without a reboot.
wifi down; wifi up

barhom, 2018-02-08:
@nbd168 

Device used is the ZBT-WE3526.

The log pasted below happens while my wife is browsing facebook on the latest mt76 drivers on her Galaxy S7.
Happens on both 2.4 & 5ghz and has increased in occurences since the [previous](https://github.com/openwrt/openwrt/commit/6271539fc58d6268f4198002c53c792712f0449a) mt76 driver.

What can I do to help?

> Thu Feb  8 22:07:28 2018 daemon.info hostapd: wlan0: STA 8c:f5:a3:a4:8a:e8 IEEE 802.11: authenticated
> Thu Feb  8 22:07:28 2018 daemon.info hostapd: wlan0: STA 8c:f5:a3:a4:8a:e8 IEEE 802.11: associated (aid 2)
> Thu Feb  8 22:07:28 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:f5:a3:a4:8a:e8
> Thu Feb  8 22:07:28 2018 daemon.info hostapd: wlan0: STA 8c:f5:a3:a4:8a:e8 WPA: pairwise key handshake completed (RSN)
> Thu Feb  8 22:07:28 2018 daemon.info dnsmasq-dhcp[1646]: DHCPDISCOVER(br-lan) 8c:f5:a3:a4:8a:e8
> Thu Feb  8 22:07:28 2018 daemon.info dnsmasq-dhcp[1646]: DHCPOFFER(br-lan) 192.168.20.224 8c:f5:a3:a4:8a:e8
> Thu Feb  8 22:07:28 2018 daemon.info dnsmasq-dhcp[1646]: DHCPREQUEST(br-lan) 192.168.20.224 8c:f5:a3:a4:8a:e8
> Thu Feb  8 22:07:28 2018 daemon.info dnsmasq-dhcp[1646]: DHCPACK(br-lan) 192.168.20.224 8c:f5:a3:a4:8a:e8 Samsung-Galaxy-S7
> Thu Feb  8 22:11:34 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:f5:a3:a4:8a:e8
> Thu Feb  8 22:11:34 2018 daemon.info hostapd: wlan0: STA 8c:f5:a3:a4:8a:e8 IEEE 802.11: disassociated
> Thu Feb  8 22:11:35 2018 daemon.info hostapd: wlan0: STA 8c:f5:a3:a4:8a:e8 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

nbd168, 2018-02-09:
Please try the latest version

iancd, 2018-02-09:
@nbd168 Just loaded the latest image from buildbot and I have immediately experienced multiple connectivity drops with my device on mt7603. mt76x2e is stable as usual. Thanks for trying to fix.

```
Fri Feb  9 15:05:04 2018 kern.notice kernel: [    0.000000] Linux version 4.9.77 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5996-256477f) ) #0 SMP Mon Feb 5 11:58:27 2018

Fri Feb  9 18:29:15 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Fri Feb  9 18:29:15 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Fri Feb  9 18:29:15 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Fri Feb  9 18:29:15 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 3)
Fri Feb  9 18:29:15 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Fri Feb  9 18:29:15 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
```

MaFeSa, 2018-02-09:
I'm using mt7603 with [this](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=175538ffdb6605f7485750f9058f3815eb07bb89) and all seems ok, are you sure that in your build there is this commit?

iancd, 2018-02-09:
@MaFeSa Yes, I'm sure. The build I just flashed says OpenWrt SNAPSHOT r6022-175538f which corresponds to the [buildbot log](http://phase1.builds.lede-project.org/builders/ramips%2Fmt7621/builds/665#changes-) showing the latest mt76 commit.

Here's my wifi config for mt7603 radio in case that provides any clues.
```
# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11g'
wireless.radio0.path='pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.country='GB'
wireless.radio0.txpower='20'
wireless.radio0.htmode='HT20'
wireless.radio0.channel='3'
wireless.radio0.legacy_rates='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.encryption='psk2+ccmp'
wireless.default_radio0.key='XXX'
wireless.default_radio0.ssid='mblan'
```

barhom, 2018-02-09:
@iancd for me the latest mt76 drivers did help. Compare my config to yours.
The obvious difference is disassoc_low_ack.

@nbd168 what is your take on disassoc_low_ack, should that be 0 or 1 for mt7603?

```wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='0'
wireless.radio0.country='SE'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.disassoc_low_ack='0'
wireless.default_radio0.encryption='psk2'

iancd, 2018-02-09:
@barhom Thanks. The disassoc_low_ack is making a difference in the last 20mins of use. In testing my iPhone 6S device is two meters with line of sight to the AP so this should not be a signal issue. Could this behaviour be related to devices with aggressive power saving modes? I've never used the disassoc_low_ack value with the mt76x2e radio and it always worked fine when the mt7603 drops . I'll keep testing for longer term and report back. For anyone else trying the command is...
```uci set wireless.default_radio0.disassoc_low_ack='0'```

MaFeSa, 2018-02-10:
@barhom @iancd @MangeshBhamre 
Devices with disconnection problem are iPad or iPhone?

I ask you why I have ± 10 different devices connected to 2.4 GHz without problems and without disassoc_low_ack='0' but none is iPhone or iPad

lfiebach, 2018-02-10:
@MaFeSa 
i have also some ESP8266 IoT devices with the same Problem.

http://sonoff.itead.cc/en/

barhom, 2018-02-10:
@nbd168, what is your take on disassoc_low_ack? Should we enable or disable it?
@MaFeSa which target do you use?

I've been using the WE3526 for about 24 hours now. Whilst there are no connection drops I feel like my latency is high sometimes.

iancd, 2018-02-11:
I've been testing today using an iPhone 6S and a MacbookAir with latest OS - two meters from AP around -40db signal. Apple's WiFi behaviour is to prefer 5Ghz whenever possible so I must disable the mt76x2 radio for testing.

@MaFeSa  Adding disassoc_low_ack='0' to my config change has stopped the iPhone connectivity drop effectively. Before this change a drop happened very quickly when I suspended the device. **I've been unable to get the iPhone to disconnect today** - great thanks @barhom. There is one caveat if I take the iPhone further from the AP and beyond Apple's [trigger threshold](https://support.apple.com/en-gb/HT2030680) of -70db the iPhone will lose connectivity during a suspend about 50% of the time. The iPhone then stays disconnected until it is disassociated with the AP (shown in logs) and then connectivity is restored. I believe this may be a design choice on Apple's part and I've seen this reported elsewhere. There are too many flaky 2.4Ghz networks out there.

@barhom. On my MacbookAir latency to the mt7603 2.4 network is double that of the mt76x2 5G. I measured using mtr traceroute and swapped 2.4 channels to check behaviour was consistent. I'll probably not know for certain without a professional radio spectrum analyser. 5Ghz is the way to go it's just that I have some IOT devices using 2.4ghz.
Like you I would also be interested to know what the default behaviour and if this could be improved for other mt7603 users.

I'd like to use 802.11r roaming but I've hit the same issue [as others have reported](https://forum.openwrt.org/viewtopic.php?pid=371932#p371932). Hostapd creates two processes on the AP and my iPhone will drop down to using 2.4Ghz with higher latency even when 5G 802.11ac is available. Without 802.11r configured the iPhone makes the best choice and uses 5G.

dpeddi, 2018-04-02:
Disassoc on netgear r6220 as well (128mb ram) 
-----------
Please edit your posts with the device model and system memory amount

dpeddi, 2018-06-13:
issue seems to persist on mt76 bundled with openwrt
PKG_SOURCE_DATE:=2018-05-20
PKG_SOURCE_VERSION:=792dbe02613299f67e92815f0b2b7584dc951ffd

laoshaw, 2018-07-25:
Can we claim that mt7603en is a "bad" 2.4Ghz chip, that MT76 won't ever be able to fix its issues(performance drops suddenly, signal is weak, and worse, connection drops). I feel it is not a software/driver issue anymore.

slthomason, 2018-07-25:
No.  We have recently gotten it much more stable than before.

dangowrt, 2018-07-25:
@laoshaw other than your feelings, is there anything that makes you assume that this is a hardware issue (which cannot be worked-around in software)?

slthomason, 2018-07-26:
> @slthomason <https://github.com/slthomason> how recently is that? running
> 18.06-rc here and the same issues such as slow speed, dropping connections
> once a while persist.

We are currently in the middle of verifying it, but somewhere in the last
round of changes to the mt76 we have gotten a lot more stability out of it.

Spencer

ahmed-anas, 2018-09-08:
I'm facing connection drops (internet stops working but wifi remains connected) on iphone X even with  `disassoc_low_ack` set to 0. 

Do I need to build the latest openwrt or is using the latest snapshot sufficient?

Driver: mt76x8
Router: TP-Link mr3420 v5

xavier83, 2018-09-12:
I'm facing the same issue as wifi becomes unresponsive. could this be a related issue?
https://forum.openwrt.org/t/solved-wifi-disconnects-using-asus-rt-n56u-and-18-06-rc2/17225/4
Router:	ZBT-WE1326
kmod-mt76-core - 4.14.63+2018-07-13-67803752-1
kmod-mt7603 - 4.14.63+2018-07-13-67803752-1
kmod-mt76x2 - 4.14.63+2018-07-13-67803752-1
OpenWrt - 18.06.1

xavier83, 2018-09-17:
@laoshaw  to quote [Openwrt wiki's Recommended routers](https://wiki.openwrt.org/toh/recommended_routers) page about MT7621.

> A quite well supported platform, it has 2 cores and 2 threads (somewhat similar to Hyperthreading on x86) along with a more powerful core (MIPS 1004Kc) which makes this SoC about about twice as fast as the AR934X-series (depending on application). These are typically shipped with 11ac hardware and the driver (mt76) works fairly well, it's still under development but is more open than the ath10k driver making it **a more interesting choice for the Open Source community**. 

So don't lose hope yet. :wink: 

BTW. I was able to partially mitigate this issue by adopting suggestions from https://bugs.openwrt.org/index.php?do=details&task_id=1679 or https://forum.openwrt.org/t/mt7621-wg3526-multicore-support/6918/17

nbd168, 2019-01-25:
Fixed in the current versions

closed 2019-01-25 by nbd168

bond2bill, 2019-04-18:
hi @sotux 

I also faulted like you, I'm using Newifi D2, and OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d); I use IPx (ios 12.2) and some wifi switches (sonoff).

Did you fix the above error, please help me!