openwrt/mt76 #147: STA Mode problem

GBert, 2018-02-10:
I'm using latest trunk OpenWrt SNAPSHOT, r6022-175538f with my Omega2+ (MediaTek MT7688 ver:1 eco:2).  Scanning seems to work most of the time but using client mode doesn't work:
```
[ 6493.562736] wlan0: authenticate with c0:25:06:a2:ab:58
[ 6493.582919] wlan0: send auth to c0:25:06:a2:ab:58 (try 1/3)
[ 6493.593336] wlan0: authenticated
[ 6493.611402] wlan0: associate with c0:25:06:a2:ab:58 (try 1/3)
[ 6493.624286] wlan0: RX AssocResp from c0:25:06:a2:ab:58 (capab=0x431 status=0 aid=1)
[ 6493.632341] wlan0: associated
[ 6497.602960] wlan0: authenticate with c0:25:06:a2:ab:58
[ 6497.626826] wlan0: send auth to c0:25:06:a2:ab:58 (try 1/3)
[ 6497.637235] wlan0: authenticated
[ 6497.641666] wlan0: associate with c0:25:06:a2:ab:58 (try 1/3)
[ 6497.654457] wlan0: RX AssocResp from c0:25:06:a2:ab:58 (capab=0x431 status=0 aid=1)
[ 6497.662535] wlan0: associated
```
that's repeated all the time (every 3-4 seconds).

Here the settings
```
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/10300000.wmac'
wireless.radio0.legacy_rates='1'
wireless.radio0.country='DE'
wireless.radio0.htmode='HT20'
wireless.radio0.disassoc_low_ack='0'
wireless.radio0.channel='8'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].network='wwan'
wireless.@wifi-iface[0].ssid='Fischernetz'
wireless.@wifi-iface[0].device='radio0'
wireless.@wifi-iface[0].mode='sta'
wireless.@wifi-iface[0].bssid='C0:25:06:A2:AB:58'
wireless.@wifi-iface[0].key='geheim'
wireless.@wifi-iface[0].encryption='psk2'

```

cdrose, 2018-03-21:
Not sure if I should open a new issue but I am running similar hardware (MT7688 Linkit Smart) and also having trouble in STA mode.

I started with the LEDE 17 release (Oct '17) and was getting a good connection for 5 minutes or so before it dropped and then kernel panic if I tried to restart with 'ip link set wlan0 down/up'.

With the latest OpenWRT snapshot (March '18) I get a good connection for 5 minutes and then it drops, however now I can restart with 'ip link set wlan0 down/up' _BUT_ when the link comes back up it seems to only connect to local addresses eg. ping 192.168.1.xx works, but ping 8.8.8.8 does not.

Please let me know what info I can provide to help with debugging this.

cdrose, 2018-03-21:
So after some more testing, restarting the networking with '/etc/init.d/network restart' works to restore Wifi once the connection has been dropped.

Also running a continuous ping seems to keep the Wifi up. I have a script with 'ping -c 1 8.8.8.8' looping every 5 seconds and it seems to keep the wireless connection active.

benVolatiles, 2018-03-23:
For me sta mode works rather well with the GL-MT300N-V2 [MT7628AN],  I'm on r6289-2d8eeca.

Only about once a day I will the connection drop after seeing `MCU message 8 (seq 9) timed out` in `dmesg`.

`/etc/init.d/network restart` would not get the link up again, a reboot did fix it.

```
[   50.985136] random: crng init done
[17352.742588] MCU message 8 (seq 9) timed out
[17352.747847] br-lan: port 2(wlan0-1) entered disabled state
[17353.862622] MCU message 8 (seq 10) timed out
[17354.902556] MCU message 8 (seq 11) timed out
[17355.942570] MCU message 8 (seq 12) timed out
[17356.982585] MCU message 8 (seq 13) timed out
[17358.022603] MCU message 8 (seq 14) timed out
[17359.062626] MCU message 8 (seq 15) timed out
[17360.102646] MCU message 8 (seq 1) timed out
[17361.142664] MCU message 8 (seq 2) timed out
[17362.182684] MCU message 8 (seq 3) timed out
[17363.222712] MCU message 8 (seq 4) timed out
```

Next time I see that I shall also see if unloading the module already helps.

benpicco, 2018-03-26:
^
This is still happening with the latest version from git.

wdu, 2018-04-06:
I see the same behaviour. STA mode is unreliable on MT7688. Both with or without WPA, associating with the access point fails more often than it succeeds, and when it succeeds it gets disassociated fairly quickly (every few minutes) and reassociating may or may not be successful.

Both errors (MCU message and the frequent reassociation messages) occur in my setup.

I only briefly tested AP mode so I may have been lucky, but I had the impression that AP mode works.

Please let me know if you need specific debuging info, or provide hints on how this may be fixed.

benVolatiles, 2018-07-18:
^
Try OpenWRT 18.06, it has a much more recent version of this driver.

https://downloads.openwrt.org/releases/18.06.0-rc2/targets/ramips/mt76x8/

dam2k, 2018-07-18:
Thank you. I'm goin to give it a try.

dam2k, 2018-07-18:
WOW!!! That worked great! After I upgraded to OpenWRT 18.06 the wifi STA connection is working and seems to be stable with no packet loss. Thank you Benjamin.

closed 2018-10-02 by nbd168