openwrt/mt76 #149: MT7603 (2.4ghz) - Main Issue

barhom, 2018-02-12:
Hello,

I thought that maybe we should try to put together all the different issues that everyone is facing with 2.4ghz on MT7603 (specifically this chipset) into this main issue that everyone can easily follow.

I suggest closing the following issues:
#26, #66, #90, #107, #121, #136, #146

MaFeSa, 2018-02-12:
For me only these warnings, connection is stable with ~ 10 devices 
I'm NOT using option disassoc_low_ack='0'

[78328.598527] WARNING: CPU: 1 PID: 17853 at /home/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-02-09-246d548b/mt7603_mac.c:1168 mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[78328.615965] Modules linked in: pppoe ppp_async option usb_wwan qmi_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter
[78328.686964]  ipt_ECN ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common mii crc32c_generic
[78328.735998] CPU: 1 PID: 17853 Comm: kworker/u8:1 Not tainted 4.9.77 #0
[78328.742513] Workqueue: phy0 mt7603_mac_work [mt7603e]
[78328.747541] Stack : 8e51f800 804c0000 80460614 8ea6ebc8 00000009 00000490 8e51f818 080ffff3
[78328.755892]         00000064 8006a9e8 80460614 00000001 000045bd 80523824 80460580 82791d3c
[78328.764239]         804c0000 800a78a0 00000001 80520000 804cdec4 804cdec8 80465220 82791d3c
[78328.772583]         804c0000 8004235c 8e51f818 82791d74 0000053e 00000000 00000001 00791d5c
[78328.780931]         8ea6d5dc 8f323b00 8f323a00 30796870 00000000 00000000 00000000 00000000
[78328.789278]         ...
[78328.791715] Call Trace:
[78328.794176] [<8000f714>] show_stack+0x54/0x88
[78328.798535] [<801da45c>] dump_stack+0x8c/0xd0
[78328.802878] [<8002adc4>] __warn+0xe4/0x118
[78328.806959] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[78328.811995] [<8ea6d908>] mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[78328.817829] [<8004013c>] process_one_work+0x200/0x348
[78328.822860] [<8004105c>] worker_thread+0x2fc/0x488
[78328.827642] [<80045c40>] kthread+0xe0/0xf4
[78328.831720] [<8000a458>] ret_from_kernel_thread+0x14/0x1c
[78328.837207] ---[ end trace e9efb0f734778945 ]---

barhom, 2018-02-12:
@MaFeSa 
Can you update your comment to show which device you are using?

MaFeSa, 2018-02-25:
OpenWrt SNAPSHOT r6297 Xiaomi 3G

```
[  453.855017] ------------[ cut here ]------------
[  453.859673] WARNING: CPU: 0 PID: 0 at /home/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-02-24-5ba59953/mt7603_mac.c:441 mt7603_mac_fill_rx+0x178/0x520 [mt7603e]
[  453.876914] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[  453.947851]  ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[  453.996660] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.82 #0
[  454.002549] Stack : 00000000 00000000 80537b4a 00000033 803e89e4 00000000 00000000 80530000
[  454.010897]         804ca1bc 804c9da7 80461724 00000000 00000000 80533824 4e000058 8eab5a00
[  454.019244]         00000000 8006aaa8 00000001 80530000 804cfec4 804cfec8 80466330 8fc0dd7c
[  454.027590]         00000003 800a79e0 4e000058 8eab5a00 00000000 00000000 804c0000 00c0dd7c
[  454.035936]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  454.044278]         ...
[  454.046715] Call Trace:
[  454.049174] [<8000f714>] show_stack+0x54/0x88
[  454.053520] [<801dab6c>] dump_stack+0x8c/0xd0
[  454.057859] [<8002adc4>] __warn+0xe4/0x118
[  454.061935] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[  454.066972] [<8eb6c148>] mt7603_mac_fill_rx+0x178/0x520 [mt7603e]
[  454.073041] [<8eb6b020>] mt7603_queue_rx_skb+0x9c/0xe4 [mt7603e]
[  454.079030] [<8ea18d7c>] mt76_dma_attach+0xaac/0xcc0 [mt76]
[  454.084642] ---[ end trace 53bd671b42898ca6 ]---
[ 3158.988958] ------------[ cut here ]------------
[ 3158.993745] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4325 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 3159.004768] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 84 (0x54)
[ 3159.014132] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[ 3159.085033]  ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[ 3159.133824] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.82 #0
[ 3159.140925] Stack : 00000000 00000000 80537b4a 00000041 803e89e4 00000000 00000000 80530000
[ 3159.149275]         804ca1bc 804c9da7 80461724 00000000 00000000 80533824 00000074 0000096c
[ 3159.157621]         8cac55d0 8006aaa8 00000001 80530000 804cfec4 804cfec8 80466330 8fc0dc7c
[ 3159.165968]         00000003 800a79e0 00000074 0000096c 81212e20 00000000 00000002 00c0dc7c
[ 3159.174313]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3159.182656]         ...
[ 3159.185092] Call Trace:
[ 3159.187551] [<8000f714>] show_stack+0x54/0x88
[ 3159.191899] [<801dab6c>] dump_stack+0x8c/0xd0
[ 3159.196238] [<8002adc4>] __warn+0xe4/0x118
[ 3159.200314] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[ 3159.205318] [<8eb25c8c>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 3159.211439] [<8ea19b78>] mt76_rx_complete+0x18c/0x278 [mt76]
[ 3159.217081] [<8ea19e24>] mt76_rx_poll_complete+0x1c0/0x260 [mt76]
[ 3159.223150] [<8ea18e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
[ 3159.228783] ---[ end trace 53bd671b42898ca7 ]---
```

MaFeSa, 2018-02-25:
With build r6297 I have many warnings and router (Xiaomi 3G) it has restarted itself twice in 24 hours

```
[ 4871.047869] ------------[ cut here ]------------
[ 4871.052663] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4333 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 4871.063689] Rate marked as a VHT rate but data is invalid: MCS: 92, NSS: 0
[ 4871.070527] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[ 4871.141473]  ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[ 4871.190241] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.82 #0
[ 4871.197340] Stack : 00000000 00000000 80537b4a 00000041 803e89e4 00000000 00000000 80530000
[ 4871.205688]         804ca1bc 804c9da7 80461724 00000000 00000000 80533824 00000064 0000096c
[ 4871.214034]         8f7505d0 8006aaa8 00000001 80530000 804cfec4 804cfec8 80466330 8fc0dc7c
[ 4871.222379]         00000003 800a79e0 00000064 0000096c 81212e20 00000000 00000002 00c0dc7c
[ 4871.230725]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 4871.239068]         ...
[ 4871.241504] Call Trace:
[ 4871.243963] [<8000f714>] show_stack+0x54/0x88
[ 4871.248311] [<801dab6c>] dump_stack+0x8c/0xd0
[ 4871.252651] [<8002adc4>] __warn+0xe4/0x118
[ 4871.256727] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[ 4871.261741] [<8eb25c8c>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[ 4871.267844] [<8ea19b78>] mt76_rx_complete+0x18c/0x278 [mt76]
[ 4871.273483] [<8ea19e24>] mt76_rx_poll_complete+0x1c0/0x260 [mt76]
[ 4871.279552] [<8ea18e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
[ 4871.285177] ---[ end trace 53bd671b42898ca8 ]---
[ 6094.793325] Data buffer not 16 bytes aligned: 8c8ec588
[ 6094.805746] Data buffer not 16 bytes aligned: 8c8ec088
[ 6094.817335] Data buffer not 16 bytes aligned: 8c8ec418
[ 6094.825916] Data buffer not 16 bytes aligned: 8c8ec308
[ 6094.834453] Data buffer not 16 bytes aligned: 8c8ec2c8
[ 6094.846612] Data buffer not 16 bytes aligned: 8c8ec038
[ 6094.857252] Data buffer not 16 bytes aligned: 8c8ec128
[ 6094.865041] Data buffer not 16 bytes aligned: 8c8ec418
[ 6094.876828] Data buffer not 16 bytes aligned: 8c8ec3b8
[ 6094.885334] Data buffer not 16 bytes aligned: 8c8ec048
[ 6094.892635] Data buffer not 16 bytes aligned: 8c8ec168
[ 6094.900116] Data buffer not 16 bytes aligned: 8c8ec218
[ 6094.910978] Data buffer not 16 bytes aligned: 8c8ec168
[ 6094.965759] Data buffer not 16 bytes aligned: 8c8ec068
[ 7786.278815] ------------[ cut here ]------------
[ 7786.283471] WARNING: CPU: 2 PID: 13013 at /home/massy/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-02-24-5ba59953/mt7603_mac.c:1168 mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[ 7786.300897] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[ 7786.371796]  ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[ 7786.420692] CPU: 2 PID: 13013 Comm: kworker/u8:1 Tainted: G        W       4.9.82 #0
[ 7786.428418] Workqueue: phy0 mt7603_mac_work [mt7603e]
[... cut ...]
[ 7786.480138] [<8000f714>] show_stack+0x54/0x88
[ 7786.484493] [<801dab6c>] dump_stack+0x8c/0xd0
[ 7786.488841] [<8002adc4>] __warn+0xe4/0x118
[ 7786.492920] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[ 7786.497954] [<8eb6d908>] mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[ 7786.503807] [<8004013c>] process_one_work+0x200/0x348
[ 7786.508852] [<8004105c>] worker_thread+0x2fc/0x488
[ 7786.513632] [<80045c40>] kthread+0xe0/0xf4
[ 7786.517710] [<8000a458>] ret_from_kernel_thread+0x14/0x1c
[ 7786.523218] ---[ end trace 53bd671b42898ca9 ]---
```

MaFeSa, 2018-02-27:
I noticed that the warning "Workqueue: phy0 mt7603_mac_work [mt7603e]" is obtained every time I transfer large files (2.4 GHz) between clients and can also be obtained with iperf...

MaFeSa, 2018-02-27:
Router (Xiaomi 3G with r6297) has restarted itself again after in 19 hours.
Kernel crashlog is empty, do I have to enable the log in the kernel_menuconfig before compile?
Thanks

barhom, 2018-02-28:
Router: WE3526
Wifi MT7603 (2.4ghz)

Issue: Wifi stays connected but cannot ping router.
Solution: Turn wifi on client off/on

Log has nothing relevant.

MaFeSa, 2018-03-01:
@barhom are you using option disassoc_low_ack='0' ?

slthomason, 2018-03-01:
@MaFeSa - we were, but in our testing it didn't make any difference.

CR-Ryan, 2018-03-01:
@MaFeSa, @barhom  - I am working with @slthomason on our MT7603 issues, and have some additional information.

We are testing the throughput on MT7603 & MT7612 - on the ZBT WE1326. We are running on the latest commit.

I am using Iperf 3.1 to test this. During my last test, I had two iperf clients running simultaneously against MT7603 - and eventually one of them "locked up".  One machine was connected, but unable to communicate with the AP _(same issue as described by @barhom)_ . There were two other devices also connected to this radio, and they were working fine. To be clear, here is what I saw on the machine having issues:

Windows shows as connected, but offline.
[image]

Pings and Tracert failed.

[image]

Here is the station dump, with some useful information. The device having issues was:  60:67:20:2a:f2:ec
Please compare this device, to the other two. Specifically, note the "expected throughput" field, and the txbitrate. 

root@ph:/proc# iw wlan0 station dump
Station 48:d2:24:73:c3:ab (on wlan0)
        inactive time:  1380 ms
        rx bytes:       11278691782
        rx packets:     7385360
        tx bytes:       439246083
        tx packets:     3487452
        tx retries:     54837
        tx failed:      558
        rx drop misc:   1
        signal:         -41 [-41, -42] dBm
        signal avg:     -40 [-40, -42] dBm
        tx bitrate:     270.0 MBit/s MCS 14 40MHz short GI
        rx bitrate:     1.0 MBit/s
        expected throughput:    55.846Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 3796 seconds

Station 54:8c:a0:79:52:6b (on wlan0)
        inactive time:  20320 ms
        rx bytes:       708996
        rx packets:     6218
        tx bytes:       100955998
        tx packets:     48583
        tx retries:     1682
        tx failed:      0
        rx drop misc:   430
        signal:         -39 [-48, -39] dBm
        signal avg:     -39 [-45, -39] dBm
        tx bitrate:     120.0 MBit/s MCS 5 40MHz short GI
        rx bitrate:     1.0 MBit/s
        expected throughput:    42.388Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 2669 seconds

Station 60:67:20:2a:f2:ec (on wlan0)
        inactive time:  950 ms
        rx bytes:       56314
        rx packets:     923
        tx bytes:       82454
        tx packets:     413
        tx retries:     2598
        tx failed:      229
        rx drop misc:   0
        signal:         -34 [-37, -34] dBm
        signal avg:     -30 [-31, -32] dBm
        tx bitrate:     6.5 MBit/s MCS 0
        rx bitrate:     6.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 621 seconds

-----------------

Here is an important observation:  **During this behavior, "expected throughput" disappears.**

Another observation:  **The txbit rate always returns to "6.5 MBit/s MCS 0" during the issue**

For comparison, here is the station dump when this device is connected properly:

Station 60:67:20:2a:f2:ec (on wlan0)
        inactive time:  850 ms
        rx bytes:       153980
        rx packets:     572
        tx bytes:       133523
        tx packets:     356
        tx retries:     86
        tx failed:      2
        rx drop misc:   0
        signal:         -30 [-30, -45] dBm
        signal avg:     -33 [-33, -41] dBm
        tx bitrate:     144.4 MBit/s MCS 15 short GI
        rx bitrate:     144.4 MBit/s MCS 15 short GI
        expected throughput:    37.994Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 56 seconds

We have been able to "fix" this behavior two ways:

1. doing a wifi up (or wifi reboot)

2. Running this command to specifically kick the affected client, and make it reconnect:

_ubus call hostapd.wlan0 del_client '{"addr":"60:67:20:2a:f2:ec", "reason":1, "deauth":true, "ban_time":10}'_

This command kicks the client, and the client immediately reconnects successfully. 

However, this fix only seems to last about 5 - 10 minutes, before this same behavior returns.

I have wireless logging set to its most verbose setting, and see nothing in logread or dmesg to indicate an issue.

barhom, 2018-03-01:
@CR-Ryan Thats a good report. I hope it helps @nbd168 with finding this bug.

MaFeSa, 2018-03-03:
r6350 with this (https://github.com/openwrt/mt76/commit/cda627fd8feffdadfcd692fbb6d823d455aa19ab) for me is very stable, I have transferred more than 40 GB with 2.4GHz and 60GB with 5GHz without any kernel warning.
Thanks

ashbenz, 2018-03-07:
@MaFeSa 
Do you use any special settings for the wireless? Can you post your 2.4Ghz config?
I'm running r6380 (which includes the commit you referenced) and still see kernel traces..

MaFeSa, 2018-03-09:
No, my config is simple:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country '00'
	option noscan '1'
	option channel '1'
	option htmode 'HT40'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'psk2+ccmp'
	option key 'password'
	option disassoc_low_ack '0'
	option hidden '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country '00'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option key 'password'
	option ssid 'LEDE 5GHz'
	option hidden '1'
```
After 5 days uptime and many data over wireless I have **only** this warning

```
[167940.170419] ------------[ cut here ]------------
[167940.175281] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4347 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[167940.186392] Rate marked as an HT rate but passed status->rate_idx is not an MCS index [0-76]: 126 (0x7e)
[167940.195927] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[167940.266956]  ip_tables crc_ccitt compat cdc_wdm fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[167940.315749] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.85 #0
[167940.321725] Stack : 00000000 00000000 80537b4a 00000033 803eaa04 00000000 00000000 80530000
[167940.330158]         804cc1bc 804cbda7 80463a94 00000000 00000000 80533824 00000000 00000000
[167940.338587]         00000000 8006aaa8 00000001 80530000 804d1ec4 804d1ec8 804686a0 8fc0dc7c
[167940.347017]         00000003 800a79e0 00000000 00000000 00000000 00000000 00000002 00c0dc7c
[167940.355446]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[167940.363875]         ...
[167940.366397] Call Trace:
[167940.368942] [<8000f714>] show_stack+0x54/0x88
[167940.373380] [<801dcb2c>] dump_stack+0x8c/0xd0
[167940.377803] [<8002adc4>] __warn+0xe4/0x118
[167940.381967] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
[167940.387056] [<8eb25d84>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
[167940.393246] [<8e9b9b80>] mt76_rx_complete+0x18c/0x278 [mt76]
[167940.398971] [<8e9b9e2c>] mt76_rx_poll_complete+0x1c0/0x258 [mt76]
[167940.405125] [<8e9b8e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
[167940.410835] ---[ end trace a5fc2674efe5e966 ]---
```

No more Workqueue: phy0 mt7603_mac_work [mt7603e] warnings.

ashbenz, 2018-03-10:
I see. I don't have option noscan, will add it now and see how it goes.. My config now looks like:

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT20'
        option country '00'
        option channel '6'
        option txpower '20'
        option legacy_rates '0'
        option noscan '1'

MaFeSa, 2018-03-10:
Try tomorrow snapshot, probabily ndb had fixed these warnings.
Now I'm using r6420...

baohavan, 2018-03-12:
Crash with multiple SSID
Device: Tplink Archer C50 v3
LEDE: 87606e25afac6776d1bbc67ed284434ec5a832b4
kmod-mt76: 
kmod-mt76 - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt76-core - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt7603 - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt76x2 - 4.9.86+2018-02-27-ffa2069e-1

Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.246406] ------------[ cut here ]------------
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.251214] WARNING: CPU: 0 PID: 0 at net/core/skbuff.c:4347 skb_try_coalesce+0x20c/0x31c
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.259517] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables cmac leds_gpio gpio_button_hotplug crypto_hash
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.318811] CPU: 0 PID: 0 Comm: swapper Not tainted 4.9.86 #0
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.324634] Stack : 8042751a 00000031 00000000 00000001 803d7594 803d7207 8038a990 00000000
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.333162]         80423660 000010fb 803efcd8 82cb1680 fffffffe 8004dba0 8039017c 803d0000
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.341681]         00000003 000010fb 8038e338 8380d7e4 fffffffe 8007bddc 8042751a 0000004d
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.350200]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.358708]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.367207]         ...
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.369703] Call Trace:
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.372209] [<8000e458>] show_stack+0x54/0x88
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.376650] [<800242d0>] __warn+0xe4/0x118
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.380829] [<80024398>] warn_slowpath_null+0x1c/0x34
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.385968] [<80216770>] skb_try_coalesce+0x20c/0x31c
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.391130] [<80270748>] tcp_try_coalesce+0x4c/0xd0
^CMon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.396087] [<80270cf4>] tcp_queue_rcv+0x60/0x148
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.400882] [<802760b4>] tcp_rcv_established+0x300/0x628
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.406285] [<8027ed30>] tcp_v4_do_rcv+0x90/0x1bc
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.411076] [<80281330>] tcp_v4_rcv+0x6dc/0x9ec
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.415692] [<8025d4a4>] ip_local_deliver_finish+0xd4/0x134
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.421365] [<8025d9fc>] ip_local_deliver+0xc8/0xe0
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.426321] [<8025dccc>] ip_rcv+0x2b8/0x32c
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.430591] [<80226920>] __netif_receive_skb_core+0x620/0x708
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.436444] [<802f6678>] br_pass_frame_up+0xa4/0x114
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.441510] [<802f6ba8>] br_handle_frame_finish+0x4c0/0x524
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.447171] [<802f6fbc>] br_handle_frame+0x358/0x3ec
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.452232] [<80226698>] __netif_receive_skb_core+0x398/0x708
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.458067] [<802280e4>] napi_gro_receive+0x6c/0xd0
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.463174] [<832a01c4>] ieee80211_attach_ack_skb+0xfcc/0x193c [mac80211]
Mon Mar 12 03:10:38 2018 kern.warn kernel: [ 2522.470135] ---[ end trace 3b17b51855c8991b ]---

MaFeSa, 2018-03-12:
With 7603 (Xiaomi 3G) there are no errors but it is not possible to connect to the new virtual wireless network.

lukasz1992, 2018-03-13:
I suggest separating and tagging MT7603 and MT7628.
@baohavan tells about MT7628

slthomason, 2018-03-14:
@nbd168 Any ideas on this one?  We are able to readily reproduce this behavior.  We have written a script which consistently bumps the client and works for a short time, but then the problem occurs again.  But it is that readily reproducible - even on the latest build.  Any ideas?

`#!/usr/bin/php-cli
<?php

define('MODEL', trim(shell_exec("cat /tmp/sysinfo/model")));

file_put_contents('/tmp/run/wifi_babysitter.pid',getmypid());

$ap = "";
if(stripos(MODEL, 'WE1326') !== false) {
	$ap = "wlan0"; //for pro
} else if (stripos(MODEL, 'WG3526') !== false){
	$ap = "ap2"; //for sb
}

//loop forever
while (true) {

	exec("iw ".$ap." station dump | grep Station | awk '{print $2}'", $connected_macs);
	foreach($connected_macs as $connected_mac) {
		exec('date', $dt);
		echo $dt[0]." - checking: ".$connected_mac.PHP_EOL;
		
		$tx_bitrate = shell_exec('iw '.$ap.' station get '.$connected_mac.' | grep "tx bitrate" | grep "6.5 MBit"');
		$expected_throughput = shell_exec('iw '.$ap.' station get '.$connected_mac.' | grep "expected throughput"');

		if($tx_bitrate != "" && $expected_throughput == "") {
			//we have a problem.  kick the wifi, and log:

			shell_exec('ubus call hostapd.'.$ap.' del_client \'{"addr":"'.$connected_mac.'", "reason":1, "deauth":true, "ban_time":1}\'');

			unset($dt);
			exec('date', $dt);
			file_put_contents('/phscripts/wifi_log.txt',$dt[0]." - had to kick: ".$connected_mac.'***********'.PHP_EOL, FILE_APPEND);
			echo $dt[0]." - had to kick: ".$connected_mac.PHP_EOL;
		}
	}

	unset($dt);
	unset($connected_macs);
	sleep(20);
}
`

slthomason, 2018-03-19:
@nbd168 is no one else seeing this?

barhom, 2018-03-20:
I am seeing this on a daily basis.
My laptop can be connected to wifi but nothing goes through. Unfortunately I have been unable to run an "iw wlan0 station dump" when this happens as I dont have a wired computer at this moment.

slthomason, 2018-03-20:
If you have php, you can run that script and it will log times that it has
happened. If you don't have php, that script can easily be written to an
ash script.  But it will log on the router the times that this happens.

@nbd168 <https://github.com/nbd168> - do you want @CR-Ryan
<https://github.com/cr-ryan> to split this out into a separate issue?  I
know this thread has gotten muddy with other issues.

Thanks!

Spencer

Rising-Sun, 2018-03-20:
> @nbd168 is no one else seeing this?

No not anymore, disabled 2.4Ghz because of this problem. :p It is unusable right now for my iPhone7, trying every new commits. (Not tried the new driver of March 19.)
So using my wzr-hp g300nh as a backup for the 2.4Ghz...

The same? bug happens in 5Ghz too, but far less often.

pparent76, 2018-04-17:
Just compile 2d08440e907829a512e3b3cce3a22de27cf28046 , same issue as before:

> <4>[ 2180.464380] ------------[ cut here ]------------
> <4>[ 2180.469054] WARNING: CPU: 0 PID: 0 at mm/slub.c:3843 ksize+0xe0/0xf8
> <4>[ 2180.475460] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie sch_gred act_ipt em_cmp sch_prio sch_codel em_text em_meta sch_sfq act_police sch_fq
> <4>[ 2180.546058]  sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
> <4>[ 2180.617931] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.77 #0
> <4>[ 2180.623822] Stack : 00000000 00000000 80537b4a 00000033 803e89e4 00000000 00000000 80530000
> <4>[ 2180.632168]         804ca1bc 804c9da7 80461734 00000000 00000000 80533824 02080020 8e94005c
> <4>[ 2180.640511]         804c75fc 8006a9e8 00000001 80530000 804cfec4 804cfec8 80466340 8fc0dcd4
> <4>[ 2180.648854]         00000003 800a78a0 02080020 8e94005c 804c75fc 00000000 8eef6400 00c0dcd4
> <4>[ 2180.657200]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
> <4>[ 2180.665542]         ...
> <4>[ 2180.667977] Call Trace:
> <4>[ 2180.670438] [<8000f714>] show_stack+0x54/0x88
> <4>[ 2180.674787] [<801da45c>] dump_stack+0x8c/0xd0
> <4>[ 2180.679126] [<8002adc4>] __warn+0xe4/0x118
> <4>[ 2180.683203] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
> <4>[ 2180.688237] [<800e737c>] ksize+0xe0/0xf8
> <4>[ 2180.692150] [<802ce058>] pskb_expand_head+0x98/0x23c
> <4>[ 2180.697098] [<8031d8d4>] ip_forward+0x1a0/0x3ac
> <4>[ 2180.701606] [<8031c508>] ip_rcv+0x2b8/0x320
> <4>[ 2180.705777] [<802dead0>] __netif_receive_skb_core+0x718/0x970
> <4>[ 2180.711500] [<802e0aac>] process_backlog+0x9c/0x160
> <4>[ 2180.716354] [<802e084c>] net_rx_action+0x138/0x2fc
> <4>[ 2180.721129] [<8002e924>] __do_softirq+0x294/0x2e0
> <4>[ 2180.725811] [<8002ec18>] irq_exit+0x7c/0x98
> <4>[ 2180.729980] [<80209670>] plat_irq_dispatch+0xb4/0xdc
> <4>[ 2180.734922] [<8000a968>] except_vec_vi_end+0xb8/0xc4
> <4>[ 2180.739863] [<8000c324>] r4k_wait_irqoff+0x18/0x20
> <4>[ 2180.744658] [<80064134>] cpu_startup_entry+0x104/0x170
> <4>[ 2180.749797] [<804efbe4>] start_kernel+0x47c/0x49c
> <4>[ 2180.754662] ---[ end trace bdeffdc844c29999 ]---
> <4>[ 2180.760660] Kernel bug detected[#1]:
> <4>[ 2180.764266] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G        W       4.9.77 #0
> <4>[ 2180.771547] task: 8fc50bc0 task.stack: 8fc62000
> <4>[ 2180.776050] $ 0   : 00000000 00000001 810268e0 00000001
> <4>[ 2180.781269] $ 4   : 00000000 810268f3 000f0000 811c7500
> <4>[ 2180.786486] $ 8   : 804c0000 00000002 00000000 00000000
> <4>[ 2180.791702] $12   : 00000000 00000000 3fff0000 ffffffff
> <4>[ 2180.796916] $16   : 8e41b3c0 8ff80308 8ff80000 000000e1
> <4>[ 2180.802131] $20   : 000001d8 0000003f 8ff80308 00000018
> <4>[ 2180.807346] $24   : 40000000 0d4ad800                  
> <4>[ 2180.812561] $28   : 8fc62000 8fc63d10 ffffffff 802cadc4
> <4>[ 2180.817777] Hi    : 00000004
> <4>[ 2180.820637] Lo    : 00000000
> <4>[ 2180.823532] epc   : 800e86e0 kfree+0x78/0x1a4
> <4>[ 2180.827868] ra    : 802cadc4 __kfree_skb+0x14/0xa4
> <4>[ 2180.832631] Status: 11008403      KERNEL EXL IE 
> <4>[ 2180.836806] Cause : 50800034 (ExcCode 0d)
> <4>[ 2180.840792] PrId  : 0001992f (MIPS 1004Kc)
> <4>[ 2180.844863] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie sch_gred act_ipt em_cmp sch_prio sch_codel em_text em_meta sch_sfq act_police sch_fq
[... cut ...]
> <4>[ 2181.076867] [<80045c40>] kthread+0xe0/0xf4
> <4>[ 2181.080951] [<8000a458>] ret_from_kernel_thread+0x14/0x1c
> <4>[ 2181.086322] Code: 30630001  24030001  38630001 <00030336> 8c430000  7c630380  10600003  00000000  10000002 
> <4>[ 2181.096060] 
> <4>[ 2181.097882] ---[ end trace bdeffdc844c2999a ]---
> 
> ===================================
> Time: 1523972268.599079
> <4>[ 2181.096060] 
> <4>[ 2181.097882] ---[ end trace bdeffdc844c2999a ]---
> <0>[ 2181.110390] Kernel panic - not syncing: Fatal exception in interrupt

pparent76, 2018-04-23:
On recent lede version with mt76 e2eedc9229dad9b9d653ad0abb4f3571d1676148

```
<1>[24429.007499] CPU 3 Unable to handle kernel paging request at virtual address 07404000, epc == 801084b0, ra == 80108394
<4>[24429.018091] Oops[#1]:
<4>[24429.020358] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.34 #0
<4>[24429.026338] task: 8fc44b00 task.stack: 8fc70000
<4>[24429.030842] $ 0   : 00000000 00000001 00000000 814c16b0
<4>[24429.036053] $ 4   : 805561b8 00000001 00000001 07404000
<4>[24429.041263] $ 8   : 00099193 00099192 8f0a4820 8f044000
<4>[24429.046482] $12   : 80550000 ffffffff 8e7fd130 00000000
<4>[24429.051701] $16   : 8fc02a00 01090220 8d984000 80316b6c
<4>[24429.056910] $20   : 00000800 8fd1705c 01080020 01080020
<4>[24429.062121] $24   : 0000000a 80370f74                  
<4>[24429.067331] $28   : 8fc70000 8fc15bc8 805568e8 80108394
<4>[24429.072542] Hi    : 00000606
<4>[24429.075413] Lo    : 90a24000
<4>[24429.078310] epc   : 801084b0 __kmalloc_track_caller+0x1d4/0x228
<4>[24429.084208] ra    : 80108394 __kmalloc_track_caller+0xb8/0x228
<4>[24429.090009] Status: 11007c03      KERNEL EXL IE 
<4>[24429.094181] Cause : 40800008 (ExcCode 02)
<4>[24429.098168] BadVA : 07404000
<4>[24429.101030] PrId  : 0001992f (MIPS 1004Kc)
<4>[24429.105101] Modules linked in: xt_coova rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async rtlwifi pppox ppp_generic nf_conntrack_ipv6 mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie
<4>[24429.175559]  sch_gred act_ipt em_cmp sch_multiq sch_prio em_text em_meta sch_codel sch_sfq act_police sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug
<4>[24429.245886]  ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[24429.254933] Process swapper/3 (pid: 0, threadinfo=8fc70000, task=8fc44b00, tls=00000000)
<4>[24429.262980] Stack : 00000000 ffffffff 8fc15bfc 8ece0cb8 8eadb300 00000000 01080020 80314c80
<4>[24429.271318]         00000000 805561b8 8f1463a0 80550000 8eadb300 8ede2d80 00000000 00000740
<4>[24429.279657]         000005dc 80316b6c 8e99d918 8ece0d48 8eadb300 b3f6da0b 8eadb300 8eadb300
<4>[24429.287992]         8ede2d80 80576880 80570000 000005dc 8fd1705c 01080020 00000000 803711e4
<4>[24429.296327]         00000004 80367834 8c0ca864 8fd17000 00000052 800655a0 8fc38802 8fd17000
<4>[24429.304666]         ...
<4>[24429.307112] Call Trace:
<4>[24429.309572] [<801084b0>] __kmalloc_track_caller+0x1d4/0x228
<4>[24429.315157] [<80314c80>] __kmalloc_reserve.isra.7+0x40/0xa4
<4>[24429.320714] [<80316b6c>] pskb_expand_head+0x8c/0x31c
<4>[24429.325671] [<803711e4>] ip_forward+0x270/0x3ec
<4>[24429.330181] [<8036fd2c>] ip_rcv+0x2a0/0x324
<4>[24429.334373] [<80327718>] __netif_receive_skb_core+0xa94/0xc5c
<4>[24429.340097] [<8032d568>] netif_receive_skb_internal+0xd8/0xf0
<4>[24429.345826] [<8041da30>] br_pass_frame_up+0xe8/0x154
<4>[24429.350767] [<8041e020>] br_handle_frame_finish+0x52c/0x570
<4>[24429.356322] [<8041e394>] br_handle_frame+0x330/0x3dc
<4>[24429.361266] [<8032740c>] __netif_receive_skb_core+0x788/0xc5c
<4>[24429.367003] [<8032a1d0>] process_backlog+0x98/0x160
<4>[24429.371871] [<8032d9d4>] net_rx_action+0x150/0x30c
<4>[24429.376643] [<8044ea90>] __do_softirq+0x128/0x2ec
<4>[24429.381337] [<800328e0>] irq_exit+0x98/0xcc
<4>[24429.385526] [<8023af5c>] plat_irq_dispatch+0xfc/0x138
<4>[24429.390562] [<8000b5a8>] except_vec_vi_end+0xb8/0xc4
<4>[24429.395511] [<8000cf70>] r4k_wait_irqoff+0x1c/0x24
<4>[24429.400306] [<8006620c>] do_idle+0xe4/0x168
<4>[24429.404494] [<80066488>] cpu_startup_entry+0x24/0x2c
<4>[24429.409441] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[24429.419166] 
<4>[24429.420940] ---[ end trace fd2deeb77d8f283c ]---

===================================
```

pparent76, 2018-04-24:
With latest version it crashes 2-3 times a day. I use hostapd not wpad-mini.
```
<1>[ 5725.787196] CPU 1 Unable to handle kernel paging request at virtual address 07406800, epc == 801071a4, ra == 80314d58
<4>[ 5725.797816] Oops[#1]:
<4>[ 5725.800091] CPU: 1 PID: 16223 Comm: chilli Not tainted 4.14.34 #0
<4>[ 5725.806162] task: 8dccbe80 task.stack: 8f252000
<4>[ 5725.810672] $ 0   : 00000000 00000001 00000000 814a6600
<4>[ 5725.815891] $ 4   : 805561b8 00000001 00000001 07406800
<4>[ 5725.821118] $ 8   : 00001424 00001423 811d9a00 00007d0c
<4>[ 5725.826331] $12   : 7f9b53e8 77f872c0 77ee1280 77ee1280
<4>[ 5725.831545] $16   : 8fc20e00 014000c0 8e800000 8fc20e00
<4>[ 5725.836756] $20   : 00000001 8fb34f20 8e77ff08 00000000
<4>[ 5725.841966] $24   : 77ea3dd9 800e5368                  
<4>[ 5725.847176] $28   : 8f252000 8f253d30 00000001 80314d58
<4>[ 5725.852396] Hi    : 00039fff
<4>[ 5725.855264] Lo    : 2e496cf7
<4>[ 5725.858161] epc   : 801071a4 kmem_cache_alloc+0x128/0x17c
<4>[ 5725.863561] ra    : 80314d58 __alloc_skb+0x74/0x180
<4>[ 5725.868412] Status: 11007c03      KERNEL EXL IE 
<4>[ 5725.872585] Cause : 40800008 (ExcCode 02)
<4>[ 5725.876572] BadVA : 07406800
<4>[ 5725.879434] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 5725.883505] Modules linked in: xt_coova rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async rtlwifi pppox ppp_generic nf_conntrack_ipv6 mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie
<4>[ 5725.953928]  sch_gred act_ipt em_cmp sch_multiq sch_prio em_text em_meta sch_codel sch_sfq act_police sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug
<4>[ 5726.024262]  ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[ 5726.033317] Process chilli (pid: 16223, threadinfo=8f252000, task=8dccbe80, tls=77f88dc0)
<4>[ 5726.041453] Stack : 8f253d74 8f253d88 00000000 00000005 8eb8e200 00000100 014000c0 80314d58
<4>[ 5726.049790]         00000200 00000100 811d9100 000001fc 8dade7fc 8eede720 8dade00c 8eb8e200
<4>[ 5726.058125]         ffffff84 8e800600 8eb8e2f4 8fc248d0 8fb34f20 8039232c 811d9114 00000000
<4>[ 5726.066462]         8dade000 802a000c 8f253e3c 8eb8e200 ffffff84 8eb8e29c 00000000 80381b50
<4>[ 5726.074798]         00100000 8dccbe80 77f7d000 8f253ec0 77f7d000 00000000 00000001 8eb8e200
<4>[ 5726.083135]         ...
<4>[ 5726.085573] Call Trace:
<4>[ 5726.088033] [<801071a4>] kmem_cache_alloc+0x128/0x17c
<4>[ 5726.093093] [<80314d58>] __alloc_skb+0x74/0x180
<4>[ 5726.097610] [<8039232c>] tcp_send_fin+0x1e8/0x214
<4>[ 5726.102319] [<80381b50>] tcp_close+0x1a8/0x414
<4>[ 5726.106757] [<803adbac>] inet_release+0x60/0x78
<4>[ 5726.111300] [<80309578>] sock_release+0x2c/0xe0
<4>[ 5726.115817] [<8030963c>] sock_close+0x10/0x20
<4>[ 5726.120162] [<8010e308>] __fput+0x10c/0x1fc
<4>[ 5726.124339] [<80049864>] task_work_run+0xc4/0x100
<4>[ 5726.129033] [<800307d4>] do_exit+0x3dc/0x9a0
<4>[ 5726.133292] [<80031c40>] SyS_exit_group+0x0/0x14
<4>[ 5726.137886] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
```

My hostapd conf file:
```
country_code=FR
ieee80211n=1
ieee80211d=1
hw_mode=g
interface=wlan0
channel=6
ssid=hello world
bridge=br-lan
ap_isolate=1
```

Rising-Sun, 2018-04-25:
The 2.4Ghz wifi runs now much better with the lastest commits. Thank you nbd, updatete for your endless effort to improve the driver. 😃 

Thank you openwrt team for this nice peace of software.

MaFeSa, 2018-04-26:
My Xiaomi 3G has eight days of uptime (from [this](https://github.com/openwrt/mt76/commit/e2eedc9229dad9b9d653ad0abb4f3571d1676148)) with no warnings and problems.
Never seen so stable, thanks!

pparent76, 2018-04-26:
@MaFeSa can you please give the details of your setup? What commit? What software do you use (hotsapd, wpad-mini)? What configuration file?

MaFeSa, 2018-04-26:
@pparent76 
Xiaomi Router 3G compiled on April 18th with default wpad-mini.
My configuration
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country '00'
	option noscan '1'
	option channel '1'
	option legacy_rates '1'
	option htmode 'HT40'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'LEDE'
	option encryption 'psk2+ccmp'
	option key 'password'
	option hidden '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country '00'
	option legacy_rates '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option key 'password'
	option ssid 'LEDE 5GHz'
	option hidden '1'
```

2.4 GHz - 10 clients
5 GHz - 2 clients
[image]

Rising-Sun, 2018-04-27:
My iPhone7 locked up again. Hmm.. have the feeling it runs more reliable when more clients are conencted.

MaFeSa, 2018-04-28:
Now I also have disconnections using iPad connected to 2.4
`Sat Apr 28 13:04:34 2018 daemon.info hostapd: wlan0: STA 98:fe:94:95:f0:17 IEEE 802.11: disconnected due to excessive missing ACKs`

iPad remains connected without data exchange, other devices work well.
I will try with `option disassoc_low_ack 0` and if I don't solve I will use 5 ghz for iPad.

MaFeSa, 2018-04-30:
`option disassoc_low_ack 0` did not solve the problem.

With r6787-097f3aa there are many "disconnected due to excessive missing ACKs" with all 2.4 clients.

```
Mon Apr 30 08:41:39 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 08:41:39 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:41:39 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 08:41:39 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 08:41:39 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:41:39 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 08:42:24 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 08:42:24 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:42:24 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 08:42:24 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 08:42:25 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:42:25 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 08:54:21 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 08:54:21 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:54:22 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 08:54:22 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 08:54:22 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:54:22 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 08:54:49 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 08:54:49 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:54:49 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 08:54:49 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 08:54:49 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 08:54:49 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 09:03:57 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 09:03:57 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:03:57 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 09:03:57 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 09:03:57 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:03:57 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 09:08:37 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 09:08:37 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:08:37 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 09:08:37 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 09:08:37 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:08:37 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
Mon Apr 30 09:13:12 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Apr 30 09:13:12 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:13:12 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: authenticated
Mon Apr 30 09:13:12 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 IEEE 802.11: associated (aid 2)
Mon Apr 30 09:13:12 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:ca:a0:09:72:94
Mon Apr 30 09:13:12 2018 daemon.info hostapd: wlan0: STA a4:ca:a0:09:72:94 WPA: pairwise key handshake completed (RSN)
```

Xiaomi Router 3G

cheikemate, 2018-04-30:
Hi.

I have two Archer C50 v3 and I am using latest snapshots (currently OpenWrt SNAPSHOT r6761-ddb70e7 / LuCI Master (git-18.115.55974-e56d9f4)). 

Wifi 2.4 is really unstable. Adding the option "disassoc_low_ack 0" increased stability but my two routers need to be restarted at least twice a day. The log shows behaviors similar to the one described by @MaFeSa .

`2018-04-30T08:52:26+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DEAUTHENTICATE.indication(00:08:22:62:55:39, 3)
2018-04-30T08:52:26+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: authentication OK (open system)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-AUTHENTICATE.indication(00:08:22:62:55:39, OPEN_SYSTEM)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: did not acknowledge authentication response
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: association OK (aid 1)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: associated (aid 1)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-ASSOCIATE.indication(00:08:22:62:55:39)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: binding station to interface 'wlan0'
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: event 1 notification
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: start authentication
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: unauthorizing port
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: sending 1/4 msg of 4-Way Handshake
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: did not Ack EAPOL-Key frame (unicast index=48)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: received EAPOL-Key frame (2/4 Pairwise)
2018-04-30T09:00:29+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: sending 3/4 msg of 4-Way Handshake
2018-04-30T09:00:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: did not Ack EAPOL-Key frame (unicast index=48)
2018-04-30T09:00:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: received EAPOL-Key frame (4/4 Pairwise)
2018-04-30T09:00:30+01:00 ap2 hostapd: wlan0: AP-STA-CONNECTED 00:08:22:62:55:39
2018-04-30T09:00:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: authorizing port
2018-04-30T09:00:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: pairwise key handshake completed (RSN)
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: AP-STA-DISCONNECTED 00:08:22:62:55:39
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: event 3 notification
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: unauthorizing port
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: deauthenticated
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DEAUTHENTICATE.indication(00:08:22:62:55:39, 3)
2018-04-30T09:10:12+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: authentication OK (open system)
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-AUTHENTICATE.indication(00:08:22:62:55:39, OPEN_SYSTEM)
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: did not acknowledge authentication response
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: association OK (aid 1)
2018-04-30T11:03:39+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: did not acknowledge association response
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: authentication OK (open system)
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: event 0 notification
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-AUTHENTICATE.indication(00:08:22:62:55:39, OPEN_SYSTEM)
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: did not acknowledge authentication response
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: association OK (aid 1)
2018-04-30T11:07:30+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: did not acknowledge association response
2018-04-30T11:08:28+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: authentication OK (open system)
2018-04-30T11:08:28+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: event 0 notification
2018-04-30T11:08:28+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-AUTHENTICATE.indication(00:08:22:62:55:39, OPEN_SYSTEM)
[... cut ...]
2018-04-30T11:08:38+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 WPA: event 3 notification
2018-04-30T11:08:38+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.1X: unauthorizing port
2018-04-30T11:08:38+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 IEEE 802.11: deauthenticated
2018-04-30T11:08:38+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DEAUTHENTICATE.indication(00:08:22:62:55:39, 3)
2018-04-30T11:08:38+01:00 ap2 hostapd: wlan0: STA 00:08:22:62:55:39 MLME: MLME-DELETEKEYS.request(00:08:22:62:55:39)
`

pparent76, 2018-05-07:
@MaFeSa just to understand, where did you get your config file from? The default one is like below which is quite different (for exemple htmode):

```

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'
```

MaFeSa, 2018-05-07:
@pparent76 

> where did you get your config file from? 

I'm using LuCI for wireless configuration and `option legacy_rates '1'` is added automatically.

pparent76, 2018-05-09:
I can confirm that even with wpad-mini and your conf file I get the following crash:

```
<1>[102073.665578] CPU 1 Unable to handle kernel paging request at virtual address 07405800, epc == 801084b0, ra == 80108394
<4>[102073.676262] Oops[#1]:
<4>[102073.678619] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.34 #0
<4>[102073.684687] task: 8fc43e80 task.stack: 8fc64000
<4>[102073.689282] $ 0   : 00000000 00000001 00000000 814a56b0
<4>[102073.694584] $ 4   : 805561b8 00000001 00000001 07405800
<4>[102073.699883] $ 8   : 0013fb3c 0013fb3b 00000000 00000001
<4>[102073.705180] $12   : 00000000 8f2de5e8 00000001 00000002
<4>[102073.710478] $16   : 8fc02a00 01090220 8edc0000 80316b6c
<4>[102073.715777] $20   : 00000720 8f1a8bc0 8dd38800 01080020
<4>[102073.721073] $24   : 00000005 00000003                  
<4>[102073.726372] $28   : 8fc64000 8fc0d998 00000014 80108394
<4>[102073.731670] Hi    : 0000424d
<4>[102073.734620] Lo    : 00000020
<4>[102073.737606] epc   : 801084b0 __kmalloc_track_caller+0x1d4/0x228
<4>[102073.743585] ra    : 80108394 __kmalloc_track_caller+0xb8/0x228
<4>[102073.749475] Status: 11007c03     KERNEL EXL IE 
<4>[102073.753733] Cause : 40800008 (ExcCode 02)
<4>[102073.757807] BadVA : 07405800
<4>[102073.760756] PrId  : 0001992f (MIPS 1004Kc)
<4>[102073.764912] Modules linked in: xt_coova rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async rtlwifi pppox ppp_generic nf_conntrack_ipv6 mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter sch_teql em_nbyte cls_basic sch_dsmark sch_pie
<4>[102073.835404]  sch_gred act_ipt em_cmp sch_multiq sch_prio em_text em_meta sch_codel sch_sfq act_police sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug
<4>[102073.905796]  ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[102073.914912] Process swapper/1 (pid: 0, threadinfo=8fc64000, task=8fc43e80, tls=00000000)
<4>[102073.923045] Stack : 00000000 00000000 bb01e09e 11b6a8c0 8f134300 00000000 01080020 80314c80
<4>[102073.931468]         8f1a8c54 8fd78098 8f1a8c5c 8edc104a 8f134300 8f4af480 00000000 00000660
<4>[102073.939891]         8ee21e80 80316b6c 8f134358 8f4af480 8e908d80 8f2857a0 8f1a8bc0 8f1a8bc0
<4>[102073.948313]         8f4af480 8f4af000 8e077800 8ee21e80 8f1a8bc0 8dd38800 8f1a8bc0 8f2acdd4
<4>[102073.956735]         8f4af000 8f134300 8f4af480 8f4af000 8f134300 8f2b0fbc 00000ffc 00000000
<4>[102073.965154]         ...
<4>[102073.967677] Call Trace:
<4>[102073.970206] [<801084b0>] __kmalloc_track_caller+0x1d4/0x228
<4>[102073.975868] [<80314c80>] __kmalloc_reserve.isra.7+0x40/0xa4
<4>[102073.981505] [<80316b6c>] pskb_expand_head+0x8c/0x31c
<4>[102073.986688] [<8f2acdd4>] ieee80211_beacon_get_tim+0x244/0xebc [mac80211]
<4>[102073.993477] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[102074.003288] 
<4>[102074.005124] ---[ end trace 7608142ef65405d6 ]---

===================================
Time: 1525843292.317169
<4>[102074.005124] ---[ end trace 7608142ef65405d6 ]---
<0>[102074.015137] Kernel panic - not syncing: Fatal exception in interrupt
```

I'm not sure why you don't have it. In order to make tests I put the router on a public space. So many different kind of devices may access it. You may not have the same kind of devices connected to it.

MaFeSa, 2018-05-09:
> I'm not sure why you don't have it. 

I do not know, do you have a Xiaomi Router 3G too?

pparent76, 2018-05-09:
No ZBT wg-3526

slthomason, 2018-05-09:
We also are using the ZBT WG3526 and a variant of it - ZBT WE1326.  Both
have the 2.4Ghz as mt7603 and 5Ghz as MT7612e.  I believe these are the
same as a Xiaomi.

slthomason, 2018-05-09:
> @pparent76 <https://github.com/pparent76> I see that you are using coova,
> have you tried to disable it?

We aren't using it, and have still had the problem.

MaFeSa, 2018-05-09:
With Xiaomi Router 3G I've never had kernel panics.
Can it be a specific problem with ZBT WG3526?

I think, however, that this and [this](https://github.com/openwrt/mt76/issues/167) issues are to be closed.
It is better to open an issue for every problem and for yours there is [this](https://github.com/openwrt/mt76/issues/121).

slthomason, 2018-05-09:
https://github.com/openwrt/mt76/issues/167 <-- That thread has not been
resolved.  We are still having this issue.  Please don't close it.  We can
consistently reproduce it.  Setting to Legacy is not a resolution - we
would really like to get the speed of HT40.  Legacy is extremely slow.

pparent76, 2018-05-09:
>Can it be a specific problem with ZBT WG3526?
Each time i've tested, with wifi disabled no crashes were recorded.
Plus Xiaomi Router 3G seems to have the exact same hardware as 3526

slthomason, 2018-05-09:
It is not a crash.  Please read the original post.

slthomason, 2018-05-09:
Others are reporting the same issue with the Xiaomi.  You as well said you
could repro if not set to Legacy.

On Wed, May 9, 2018 at 8:05 AM, Spencer Thomason <spencer.thomason@gmail.com

MaFeSa, 2018-05-09:
@slthomason you're confusing, it's a different problem than that of @pparent76 
This is why it is better to keep separate issues!
@pparent76 has kernel panic and it's a different thing.

> Each time i've tested, with wifi disabled no crashes were recorded.

I repeat @pparent76 I see that you are using coova, have you tried to disable it?

slthomason, 2018-05-09:
Ok.  Just please don't close https://github.com/openwrt/mt76/issues/167
then.  You said earlier you thought it was closed.  This issue (#149) is
resolved for us.

pparent76, 2018-05-09:
>I repeat @pparent76 I see that you are using coova, have you tried to disable it?

No but it is not that easily doable. As I'm testing in real world situation, and in a public space (testing manually is way too long as crashes happen once per day on average). I will see if I can find a way to test without coova-chilli.

Although with coova-chilli and no wifi, there are no crashes.

jsantala, 2018-05-10:
I'm experiencing these problems, mainly 20% or more packet loss, on VoCore2 with the mt7603e chip. Apparently it also works with a binary driver from the manufacturer, but it's for an older kernel and I'd rather not use it anyway.

laadayush, 2018-05-18:
Try this build. It has MTK proprietary driver integrated with all lede versions. Pretty much stable. No crash observed so far from last 2 weeks.

https://gitlab.com/kyson/openwrt-teton

jsantala, 2018-05-18:
@laadayush AFAIK the proprietary driver does not support adhoc so it's not a viable alternative for us. Do correct me if I'm wrong. Thanks.

laadayush, 2018-05-18:
I haven't tried it. But as per https://github.com/dcboy/mt7628-p4rev-120395/blob/master/src/embedded/doc/README_STA_usb , i think it supports. Probably you could try once.

pparent76, 2018-05-18:
But that proprietary version is vulnerable to KRACK ATTACKS isn't it?

pparent76, 2018-06-06:
Not sure if it is due to mt76 driver, but I tested latest version. After a while the wifi is not emitted anymore.

Then any network command like ifconfig or hostapd or /etc/init.d/network restart, ... is freezing the console. (Even ctrl+c does not reply). Although, the network itself is still working. If you want to know more I still have my router in this freezy state and I can try other commands.

I've already seen this, but I cannot remember when, and what was the solution.

pparent76, 2018-06-25:
With commit 792dbe02613299f67e92815f0b2b7584dc951ffd

```
<1>[79032.004078] CPU 1 Unable to handle kernel paging request at virtual address 07407800, epc == 80108880, ra == 80108764
<4>[79032.014717] Oops[#1]:
<4>[79032.017009] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.44 #0
<4>[79032.023010] task: 8fc43e80 task.stack: 8fc64000
<4>[79032.027541] $ 0   : 00000000 00000001 00000000 814a56f0
<4>[79032.032780] $ 4   : 8058e1b8 00000001 00000001 07407800
<4>[79032.038020] $ 8   : 000c6a65 000c6a64 8f0bc830 8f3e8000
<4>[79032.043263] $12   : 80590000 ffffffff 8e21c930 00000000
<4>[79032.048506] $16   : 8fc02a00 01090220 8c89a000 8032f4d8
<4>[79032.053759] $20   : 00000800 8fc1e05c 01080020 01080020
<4>[79032.059005] $24   : 0000000a 8038af24                  
<4>[79032.064271] $28   : 8fc64000 8fc0dbc8 8058e8f0 80108764
<4>[79032.069525] Hi    : 00001ef9
<4>[79032.072422] Lo    : f704a000
<4>[79032.075348] epc   : 80108880 __kmalloc_track_caller+0x1d4/0x228
<4>[79032.081294] ra    : 80108764 __kmalloc_track_caller+0xb8/0x228
<4>[79032.087127] Status: 11007c03      KERNEL EXL IE 
<4>[79032.091316] Cause : 40800008 (ExcCode 02)
<4>[79032.095307] BadVA : 07407800
<4>[79032.098176] PrId  : 0001992f (MIPS 1004Kc)
<4>[79032.102257] Modules linked in: xt_coova rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async rtlwifi pppox ppp_generic nf_conntrack_ipv6 mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter em_cmp sch_multiq sch_teql em_nbyte sch_pie
<4>[79032.173110]  sch_gred sch_dsmark act_police cls_basic act_ipt sch_prio em_meta em_text sch_codel sch_sfq sch_fq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug
<4>[79032.243700]  ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
<4>[79032.252751] Process swapper/1 (pid: 0, threadinfo=8fc64000, task=8fc43e80, tls=00000000)
<4>[79032.260805] Stack : 00000000 00000001 805ee694 8f11e5a0 8f5ec840 00000000 01080020 8032d5b0
<4>[79032.269150]         8f2bc000 8f11e5e0 8f11e638 00000014 8f5ec840 8d9c1b00 00000000 00000740
<4>[79032.277496]         000005dc 8032f4d8 8e9f9918 8f2bcd48 8f5ec840 ac07f535 8f5ec840 8f5ec840
<4>[79032.285839]         8d9c1b00 805af320 805b0000 000005dc 8fc1e05c 01080020 00000000 8038b22c
<7>[79032.292476] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:04:79:70:f4:99:49:08:00 SRC=192.168.182.19 DST=216.58.198.202 LEN=83 TOS=0x00 PREC=0x00 TTL=64 ID=47908 DF PROTO=TCP SPT=60174 DPT=443 SEQ=3708027957 ACK=2252467739 WINDOW=1520 RES=0x00 ACK PSH URGP=0 OPT (0101080A00B331006563CC36) MARK=0x100 
<7>[79032.298805] dotspot-7617IN=br-lan OUT= MAC=78:a3:51:2b:f4:1a:04:79:70:f4:99:49:08:00 SRC=192.168.182.19 DST=216.58.198.202 LEN=83 TOS=0x00 PREC=0x00 TTL=64 ID=55728 DF PROTO=TCP SPT=60175 DPT=443 SEQ=2581358021 ACK=3043162869 WINDOW=1520 RES=0x00 ACK PSH URGP=0 OPT (0101080A00B33101D50CA9A6) MARK=0x100 
<4>[79032.347627]         00000004 803815d4 8e75a064 8fc0dca4 8f180b80 800657f0 80590002 8fc1e000
<4>[79032.355978]         ...
<4>[79032.358425] Call Trace:
<4>[79032.360900] [<80108880>] __kmalloc_track_caller+0x1d4/0x228
<4>[79032.366488] [<8032d5b0>] __kmalloc_reserve.isra.7+0x40/0xa4
<4>[79032.372053] [<8032f4d8>] pskb_expand_head+0x8c/0x31c
<4>[79032.377024] [<8038b22c>] ip_forward+0x308/0x48c
<4>[79032.381538] [<80389ce0>] ip_rcv+0x2a0/0x320
<4>[79032.385719] [<80340078>] __netif_receive_skb_core+0xa94/0xc5c
<4>[79032.391463] [<80345fe8>] netif_receive_skb_internal+0xd8/0xf0
<4>[79032.397186] [<8044ba70>] br_pass_frame_up+0xe8/0x154
<4>[79032.402136] [<8044c070>] br_handle_frame_finish+0x53c/0x580
<4>[79032.407683] [<8044c3e4>] br_handle_frame+0x330/0x3dc
<4>[79032.412630] [<8033fd6c>] __netif_receive_skb_core+0x788/0xc5c
<4>[79032.418387] [<80342be4>] process_backlog+0x98/0x160
<4>[79032.423258] [<80346454>] net_rx_action+0x150/0x30c
<4>[79032.428054] [<8047cb90>] __do_softirq+0x128/0x2ec
<4>[79032.432777] [<80032910>] irq_exit+0x98/0xcc
<4>[79032.436972] [<8024a6cc>] plat_irq_dispatch+0xfc/0x138
<4>[79032.442012] [<8000b5a8>] except_vec_vi_end+0xb8/0xc4
<4>[79032.446958] [<8000cf70>] r4k_wait_irqoff+0x1c/0x24
<4>[79032.451757] [<8006645c>] do_idle+0xe4/0x168
<4>[79032.455934] [<800666d8>] cpu_startup_entry+0x24/0x2c
<4>[79032.460884] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[79032.470617] 
<4>[79032.472480] ---[ end trace 8ecf7d1623c6d529 ]---

```

psyborg55, 2018-07-08:
you should've used mt7602 chipset. is better than 7603

slthomason, 2018-07-09:
Haha ... nice, but already invested a lot in this hardware, and it has
7603.

Spencer

zdila, 2018-08-08:
I have massive Wifi disconnection problem with Archer C50 and OpenWrt 18.06 on both 2.4 and 5 GHz. Option `option disassoc_low_ack '0'` doesn't fix it. I observe the problem only on Lenovo Yoga Tablet 2 which has no problems when using different router. On Xiaomi Mi 4, iPad, Runbo X6 ang HP laptop there is no such problem.

```
...
Wed Aug  8 21:45:08 2018 daemon.notice hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: did not acknowledge authentication response
Wed Aug  8 21:45:20 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: authenticated
Wed Aug  8 21:45:20 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: associated (aid 2)
Wed Aug  8 21:45:20 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:20 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 WPA: pairwise key handshake completed (RSN)
Wed Aug  8 21:45:21 2018 daemon.info dnsmasq-dhcp[1521]: DHCPREQUEST(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:21 2018 daemon.info dnsmasq-dhcp[1521]: DHCPACK(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9 android-920fcb9d4829c68b
Wed Aug  8 21:45:50 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: disconnected due to excessive missing ACKs
Wed Aug  8 21:45:50 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:54 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: authenticated
Wed Aug  8 21:45:54 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: associated (aid 2)
Wed Aug  8 21:45:54 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:54 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 WPA: pairwise key handshake completed (RSN)
Wed Aug  8 21:45:55 2018 daemon.info dnsmasq-dhcp[1521]: DHCPREQUEST(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:55 2018 daemon.info dnsmasq-dhcp[1521]: DHCPACK(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9 android-920fcb9d4829c68b
Wed Aug  8 21:45:58 2018 daemon.info hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: disconnected due to excessive missing ACKs
Wed Aug  8 21:45:58 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:45:59 2018 daemon.notice hostapd: wlan1: STA 6c:5f:1c:db:12:f9 IEEE 802.11: did not acknowledge authentication response
...
Wed Aug  8 21:47:03 2018 daemon.info hostapd: wlan0: STA 6c:5f:1c:db:12:f9 IEEE 802.11: disconnected due to excessive missing ACKs
Wed Aug  8 21:47:03 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:47:09 2018 daemon.info hostapd: wlan0: STA 6c:5f:1c:db:12:f9 IEEE 802.11: authenticated
Wed Aug  8 21:47:09 2018 daemon.info hostapd: wlan0: STA 6c:5f:1c:db:12:f9 IEEE 802.11: associated (aid 1)
Wed Aug  8 21:47:09 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 6c:5f:1c:db:12:f9
Wed Aug  8 21:47:09 2018 daemon.info hostapd: wlan0: STA 6c:5f:1c:db:12:f9 WPA: pairwise key handshake completed (RSN)
Wed Aug  8 21:47:10 2018 daemon.info dnsmasq-dhcp[1521]: DHCPREQUEST(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9
Wed Aug  8 21:47:10 2018 daemon.info dnsmasq-dhcp[1521]: DHCPACK(br-lan) 192.168.1.112 6c:5f:1c:db:12:f9 android-920fcb9d4829c68b
Wed Aug  8 21:47:14 2018 daemon.info hostapd: wlan0: STA 6c:5f:1c:db:12:f9 IEEE 802.11: disconnected due to excessive missing ACKs
Wed Aug  8 21:47:14 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 6c:5f:1c:db:12:f9
...
```

```
root@OpenWrt:~# lsmod | grep mt7
cfg80211              202032  5 rt2x00lib,mt76x2e,mt7603e,mt76,mac80211
mac80211              390272  7 rt2800lib,rt2x00soc,rt2x00pci,rt2x00lib,mt76x2e,mt7603e,mt76
mt76                   18880  2 mt76x2e,mt7603e
mt7603e                28288  0 
mt76x2e                42944  0 
```

[2 more replies]

closed 2018-08-12 by barhom