openwrt/mt76 #150: 5g wifi packet stuck issue

updatede, 2018-02-13:
I use Xiaomi Router 3G , encounter packet stuck for 1-3 seconds at 5g wifi. when switch to another openwert router wndr3800, the stuck disappear.

iancd, 2018-02-14:
Hi @updatede 
Please include your TARGET device and confirm you are in the latest trunk version?
Could you edit your issue description to include steps to reproduce the problem?

updatede, 2018-02-15:
new discovery:
stuck not from the start of wlan, it occur when run some time of wlan, and after restart wlan of openwrt, stuck disappear, will stuck again after some time of running wlan

iancd, 2018-02-15:
@updatede 
Could you run this command and report back if this improves?
`uci set wireless.@wifi-iface[0].disassoc_low_ack=‘0’ && uci commit`

updatede, 2018-02-15:
it seems improve. the stuck time change from about 5 mins after turning on wifi to 10 mins.
but it stuck any way, and after stuck the packet order mess, the action of game strange. must restart wifi to correct this.  
and I check the hostapd-phyx.conf files in /var/run of  wndr3800, disassoc_low_ack=1

iancd, 2018-02-15:
@updatede I also have used a wndr3800 for many years and the 2.4G radio worked well with Apple devices without having to add the disassoc_low_ack config. There does seem to be an issue with this and mt7603 drivers.  So my next suggestion is to keep the iPad very close (line of sight) with your router and see if it is stable. Then experiment moving the device further away from the router.  You should see rx and tx -dBm values increase.  Apple considers -70dBm or greater to be a slow network and the iPad will look for another wireless network and may not rejoin your network without restarting the network. Consider changing the wifi channel frequency and see if that makes a difference.

```
opkg update && opkg install iwinfo procps-ng-watch
watch -c -n 1 'iwinfo wlan0 assoclist'
```
or for 5G devices:
```
watch -c -n 1 'iwinfo wlan1 assoclist'
```
Please report back if and close issue if this is related to signal strength, router distance.

updatede, 2018-02-16:
It is not related to signal strength, router distance.
In previous test, I did not use your command, my 5g is wlan1, so I directly edit /etc/config/wireless and add option disassoc_low_ack '0' to appropriate wifi interface. After restart wifi I check /var/run/hostapd-phy1.conf and disassoc_low_ack=0.
Today, I sit just beside mir3g router, and the problem is same . watch -c -n 1 'iwinfo wlan1 assoclist',
display signal vary from -35dm to -44dbm, and always in assoclist. 
the problem is obivous, when play my favorite game, it stuck, after  1-3 seconds, packet arrives, altogether and misorder. for example, correct process is : a packet make a sound, 0.5s later a packet make a action, stuck process is: stuck for 3 seconds, sound packet and action packet arrives almost at the same time, but action first sound later, wrong time and wrong order. After that Th whole process loop again, stuck first, many packets arrive misorder. At this time, you can see wifi is still connect to correct SSID in ipad, and you turn off wif then turn on wifi in ipad can't stop this stuck--mass arrive loop, you must type "wifi down" then "wifi" on router to stop this stuck loop.
In my imagination, it look likes mt76 wrongly cached packet in some conditons. and must restart wifi interface of router to clear this cache.

updatede, 2018-02-16:
By the way , channel 100 does not work at all, the tx-power always 22dbm despite whatever you select, and once you try to connect , the  wlan interface disabled. I try many channels from 52 to 140 , all the same.
So only channel 36-48 is good, others are bad. This is another bug.

iancd, 2018-02-16:
I'm unable to replicate the problems you describe @updatede with my mt7621 routers. Looking at the LEDE forums the problems you describe are Xiaomi device and firmware specific. I think it might be better to check out solutions in that forum and close this github issue.
https://forum.lede-project.org/t/xiaomi-wifi-router-3g/5377/159

updatede, 2018-02-16:
seems no solution at forum. the link you give me is about 2.4g and old snapshot, i’m using 5g and latest snapshot. 
even the problem is xiaomi specific , still need modify driver code to solve it. if you aren’t interested in it, maybe someone else will be.

updatede, 2018-02-26:
same issue as #152 ,and descriped more clearly there.

closed 2018-02-26 by updatede

macrokernel, 2020-01-22:
Looks like the problem still exists with OpenWrt 19.07.0, r10860-a3ffeb413b on mir3g usb variant.
[image]