openwrt/mt76 #151: mt76 upload/download performance

lukasz1992, 2018-02-13:
I have R3G router. Some tests with Intel AC 3160 (1x1):

mt76x2e, VHT80, client -> AP
[  5]   0.00-6.00   sec   170 MBytes   238 Mbits/sec                  receiver
mt76x2e, VHT80, AP -> client
[  4]   0.00-10.00  sec  69.2 MBytes  58.0 Mbits/sec                  receiver
mt7603e, HT40, client -> AP
[  5]   0.00-10.03  sec   126 MBytes   105 Mbits/sec                  receiver
mt7603e, HT40, AP -> client
[  4]   0.00-10.00  sec  54.1 MBytes  45.4 Mbits/sec                  receiver

Speeds seems to be stable, pings are low, even with crowded 2.4GHz network.
I observe it on both master and lede-17.01 branch.

Is it possible to optimize mt76 for the opposite direction?

nbd168, 2018-03-08:
Please try the latest version

lukasz1992, 2018-04-11:
It is a little better on mt7612e device:
81.9 Mbits/sec instead of 58.0Mbits/sec, but still far than optimal
client->AP remains very similar, I find it OK.

It is a really much worse on mt7603e:
19.2 Mbits/sec instead of 45.4 Mbits/sec with AP->client and
65.5 Mbits/sec instead of 105 Mbits/sec

In case of mt7603e I see huge differences, like that:

> [  5]  38.00-39.00  sec  9.27 MBytes  77.9 Mbits/sec
> [  5]  39.00-40.00  sec  8.19 MBytes  68.6 Mbits/sec
> [  5]  40.00-41.00  sec  4.00 MBytes  33.6 Mbits/sec
> [  5]  41.00-42.00  sec  8.38 MBytes  70.2 Mbits/sec
> [  5]  42.00-43.00  sec  10.9 MBytes  91.2 Mbits/sec
> [  5]  43.00-44.00  sec  6.68 MBytes  56.2 Mbits/sec
> [  5]  44.00-45.00  sec  7.98 MBytes  66.9 Mbits/sec
> [  5]  45.00-46.00  sec  3.38 MBytes  28.3 Mbits/sec
> [  5]  46.00-47.00  sec  8.87 MBytes  74.4 Mbits/sec
> [  5]  47.00-48.00  sec  6.57 MBytes  55.2 Mbits/sec

I also observe problems described in #26 with my notebook.
I have a next wireless adapter and can capture traffic in monitor mode, if it helps.

psyborg55, 2018-07-08:
you should've used 2x2 card for your tests, sir!

lukasz1992, 2018-11-15:
In this case I was asking about mt76 common part, and it seems that the issue with low TX speed has been fixed. I will close this issue.

Stats from today test on https://github.com/openwrt/mt76/commit/88f68838d91e7b00e2abb5638cf40a514837b7da:

```
MT7603, HT20

sta -> ap		0.00-30.05  sec   207 MBytes  57.7 Mbits/sec                  receiver
sta -> ap		0.00-30.05  sec   206 MBytes  57.4 Mbits/sec                  receiver
sta -> ap		0.00-30.06  sec   207 MBytes  57.7 Mbits/sec                  receiver

ap -> sta		0.00-30.04  sec   200 MBytes  55.8 Mbits/sec    0             sender
ap -> sta		0.00-30.05  sec   201 MBytes  56.0 Mbits/sec    0             sender
ap -> sta		0.00-30.04  sec   198 MBytes  55.3 Mbits/sec    0             sender

MT7603, HT40

sta -> ap		0.00-30.05  sec   384 MBytes   107 Mbits/sec                  receiver
sta -> ap		0.00-30.05  sec   386 MBytes   108 Mbits/sec                  receiver
sta -> ap		0.00-30.05  sec   389 MBytes   108 Mbits/sec                  receiver

ap -> sta		0.00-30.16  sec   356 MBytes  98.9 Mbits/sec    0             sender
ap -> sta		0.00-30.05  sec   361 MBytes   101 Mbits/sec    0             sender
ap -> sta		0.00-30.04  sec   366 MBytes   102 Mbits/sec    0             sender
```

closed 2018-11-15 by lukasz1992