openwrt/mt76 #153: TX power unexpected behavior

MrBan, 2018-02-21:
Test environment
-----------------------
-   1 AP + 1 STA connected with coax cable + static attenuation
-   channel 5745@20MHz
-   fixed MCS-0
-   no data traffic

Test
-----
Set fixed TX power on the AP with `iw dev wlan0 set txpower fixed X` and measure the corresponding RSSI reported by the STA with `iw dev wlan0 station dump`.
The AP TX power has been increased from 0 to 30 dBm, in steps of 1 dBm. The test has been repeated on different frequencies and with different MCS values, almost with the same outcome.

I expected to see the RSSI on the STA growing almost proportionally, as long as the country regulation limit wasn't reached.
The actual behavior is reported in the following chart.

Test result
--------------
[image]

LorenzoBianconi, 2018-02-22:
Hi Gaetano,

could you provide more info about your test environment please:
- commit id
- same behaviour on all channels/channel widths?
- what about calibration data?

Moreover in order to get more precise measurements you have to tx data with
duty cycle close (best as you can) to 100% and sniff rx data at the
receiver (rx radiotap contains rx pwr info)

Regards
Lorenzo

Test environment

   - 1 AP + 1 STA connected with coax cable + static attenuation
   - channel 5745@20MHz
   - fixed MCS-0
   - no data traffic

Test

Set fixed TX power on the AP with iw dev wlan0 set txpower fixed X and
measure the corresponding RSSI reported by the STA with iw dev wlan0
station dump.
The AP TX power has been increased from 0 to 30 dBm, in steps of 1 dBm. The
test has been repeated on different frequencies and with different MCS
values, almost with the same outcome.

I expected to see the RSSI on the STA growing almost proportionally, as
long as the country regulation limit wasn't reached.
The actual behavior is reported in the following chart.
Test result

[image: chart]
<https://user-images.githubusercontent.com/7239570/36488760-92204542-1724-11e8-9798-78cdcf1f4bb1.png>

MrBan, 2018-02-22:
Here some other test results. This time there was background traffic from AP to STA and 
the RSSI value has been extracted from packets' radiotap on the receiver side (average value over 1000 packets).
For each channel the test has been repeated with different MCS values.

commit id: e50465605266118060df1b86eb7641c65fd29c65
chip: mt7612e

5754@40MHz
-------------------
[image]

5180@20MHz
-------------------
[image]

5500@20MHz
-------------------
[image]

LorenzoBianconi, 2018-02-23:
Hi Gaetano,

thx for your tests. Have you tried to bisect in order to verify if this is a regression?
Regards,
Lorenzo

nbd168, 2018-02-24:
Fixed in the latest version, please test.

MrBan, 2018-02-26:
Definitely better!

Thanks guys.

[image]

closed 2018-02-26 by nbd168