openwrt/mt76 #154: mt76_wcid_key_setup warning

simonsmh, 2018-02-27:
I'm using latest snapshot of openwrt on NETGEAR R6220, related infomation could be found on the internet.
Here is the log trace.
```
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.405702] ------------[ cut here ]------------
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.414927] WARNING: CPU: 0 PID: 2458 at backports-2017-11-01/net/mac80211/key.c:1010 mt76_wcid_key_setup+0x6c/0x138 [mt76]
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.437116] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.566157] CPU: 0 PID: 2458 Comm: hostapd Tainted: G        W       4.9.82 #0
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.580512] Stack : 00000000 00000000 80567b4a 00000042 80410c04 00000000 00000000 80560000
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.597151]         86e9a65c 804f7ea7 8048ddf4 00000000 0000099a 80563824 87044ba0 00000004
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.613787]         00000000 8006ac58 00000001 80560000 804fe004 804fe008 80492a00 873319d4
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.630424]         00000003 800a7b90 87044ba0 00000004 81111e20 00000000 00000001 003319d4
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.647056]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.663693]         ...
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.668545] Call Trace:
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.673406] [<8000f714>] show_stack+0x54/0x88
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.682064] [<801e64ac>] dump_stack+0x8c/0xd0
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.690722] [<8002adc4>] __warn+0xe4/0x118
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.698858] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.708902] [<870b95a4>] mt76_wcid_key_setup+0x6c/0x138 [mt76]
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.720501] [<871e8d30>] 0x871e8d30
Tue Feb 27 12:47:41 2018 kern.warn kernel: [ 6842.727595] ---[ end trace 430c0840304b2b78 ]---
```

nanpuyue, 2018-03-27:
```
[   63.571041] ------------[ cut here ]------------
[   63.575767] WARNING: CPU: 0 PID: 1546 at backports-2017-11-01/net/mac80211/key.c:1010 mt76_wcid_key_setup+0x6c/0x210 [mt76]
[   63.587118] Modules linked in: pppoe ppp_async pptp pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 ax88179_178a xt_time xt_tcpudp xt_state xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_esp xt_conntrack xt_comment xt_TRACE xt_TCPMSS xt_REDIRECT xt_LOG xt_CT usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ip6table_raw ip_tables crc_ccitt compat fuse xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport
[   63.659878]  ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_gre ip_gre gre sit ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel tun af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic sha1_generic jitterentropy_rng drbg md5 hmac echainiv des_generic deflate zlib_inflate zlib_deflate cbc authenc usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc16 mii aead crypto_null cryptomgr crc32c_generic crypto_hash
[   63.748852] CPU: 0 PID: 1546 Comm: hostapd Not tainted 4.9.87 #0
[   63.754948] Stack : 8044751a 00000034 00000000 00000001 87d970ec 803f52e7 803a77e0 0000060a
[   63.763456]         80443660 000003f2 8708cb80 00000004 00000000 8004cc38 803acfc4 803f0000
[   63.771964]         00000003 000003f2 803ab180 873a19e4 00000000 8007a9dc 8044751a 0000006f
[   63.780463]         00000000 00000008 00000000 00000000 00000000 00000000 00000000 00000000
[   63.788969]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   63.797474]         ...
[   63.799956] Call Trace:
[   63.802477] [<8000e220>] show_stack+0x70/0x8c
[   63.806916] [<80023a34>] __warn+0x110/0x118
[   63.811178] [<80023ad0>] warn_slowpath_null+0x1c/0x2c
[   63.816310] [<871515c0>] mt76_wcid_key_setup+0x6c/0x210 [mt76]
[   63.822249] [<870a0c94>] 0x870a0c94
[   63.825787] ---[ end trace 1a391bb242f6a068 ]---
```

LorenzoBianconi, 2018-03-27:
Have you tried this commit? 249fa2deda40600cf2d9198199f805f5b806b422

simonsmh, 2018-03-27:
@nanpuyue 你是什么路由出现的这个问题？

nanpuyue, 2018-03-28:
@simonsmh 极路由 Go

nanpuyue, 2018-03-28:
@LorenzoBianconi
I will test it tonight.

nanpuyue, 2018-03-28:
@LorenzoBianconi 
```
[   63.899074] ------------[ cut here ]------------
[   63.903800] WARNING: CPU: 0 PID: 1515 at backports-2017-11-01/net/mac80211/key.c:1010 mt76_wcid_key_setup+0x6c/0x210 [mt76]
[   63.915170] Modules linked in: pppoe ppp_async pptp pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 ax88179_178a xt_time xt_tcpudp xt_state xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_esp xt_conntrack xt_comment xt_TRACE xt_TCPMSS xt_REDIRECT xt_LOG xt_CT usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ip6table_raw ip_tables crc_ccitt compat fuse xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport
[   63.987841]  ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_gre ip_gre gre sit ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel tun af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic sha1_generic jitterentropy_rng drbg md5 hmac echainiv des_generic deflate zlib_inflate zlib_deflate cbc authenc crypto_acompress
[   64.060492]  usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc16 mii aead crypto_null cryptomgr crc32c_generic crypto_hash
[   64.080916] CPU: 0 PID: 1515 Comm: hostapd Not tainted 4.14.29 #0
[   64.087089] Stack : 00000000 00000000 80497502 00000035 87ce5354 8043f947 803ec3bc 000005eb
[   64.095582]         80493650 000003f2 8709cb80 00000004 00000000 00000001 8695f978 2b902291
[   64.104074]         00000000 00000000 80490000 000000d3 00000000 65746e69 2e342064 322e3431
[   64.112566]         7263206c fffffffc d8a58c41 6f747079 80000000 00000000 871315b0 871dacc0
[   64.121057]         00000009 000003f2 8709cb80 00000004 00000000 801dcb08 00000000 80490000
[   64.129548]         ...
[   64.132026] Call Trace:
[   64.134528] [<8000e374>] show_stack+0x74/0x104
[   64.139064] [<800244f0>] __warn+0x110/0x118
[   64.143310] [<8002458c>] warn_slowpath_null+0x1c/0x30
[   64.148449] [<871315b0>] mt76_wcid_key_setup+0x6c/0x210 [mt76]
[   64.154375] [<870a8c54>] 0x870a8c54
[   64.157923] ---[ end trace bf1486304ab62c3e ]---
```

LorenzoBianconi, 2018-03-28:
Are you sure you are using the latest mt76 commit?
(238f3f1c2d702e3b6b8a30b862b400293cdd4510). At the moment you need to
update mt76 pkg Makefile:

-PKG_SOURCE_VERSION:=ffa2069e78cf267ce79d7a8820aa0368161ff6e1
+#PKG_SOURCE_VERSION:=ffa2069e78cf267ce79d7a8820aa0368161ff6e1
+PKG_SOURCE_VERSION:=238f3f1c2d702e3b6b8a30b862b400293cdd4510

latest mt76 version falls-back to sw encryption for unsupported
cipher, so that line should not be hit

Regards,
Lorenzo

nanpuyue, 2018-03-28:
@LorenzoBianconi
[image]
I have tested both kernel 4.9 and 4.14.

LorenzoBianconi, 2018-03-28:
Does the earning come from mt76x2 (5GHz) or mt7603 (2.4GHz)? I guess the
latter since we probably need to add the same fix for mt7603 as well

nanpuyue, 2018-03-28:
2.4G

LorenzoBianconi, 2018-03-28:
Could you please try the attached patch (just compiled, not tested)

Lorenzo

nanpuyue, 2018-03-28:
I have tesed, it works!

nanpuyue, 2018-03-28:
```diff
--- mt7603_main.c	2018-03-28 20:40:03.826751531 +0800
+++ mt7603_main.c	2018-03-28 20:30:38.896832607 +0800
@@ -343,6 +343,18 @@
 	struct mt76_wcid *wcid = msta ? &msta->wcid : &mvif->wcid;
 	int idx = key->keyidx;
 
+	/* fall back to sw encryption for unsupported ciphers */
+	switch (key->cipher) {
+	case WLAN_CIPHER_SUITE_WEP40:
+	case WLAN_CIPHER_SUITE_WEP104:
+	case WLAN_CIPHER_SUITE_TKIP:
+	case WLAN_CIPHER_SUITE_CCMP:
+		break;
+	default:
+		return -EOPNOTSUPP;
+	}
+
+
 	/*
 	 * The hardware does not support per-STA RX GTK, fall back
 	 * to software mode for these.
```

nbd168, 2018-03-28:
Patch applied

closed 2018-03-28 by nbd168