openwrt/mt76 #155: BUG: sleeping function called from invalid context at kernel/workqueue.c:2816

easyteacher, 2018-02-27:
Device: Newifi D1

```
[61804.131204] br-lan: port 3(wlan1) entered disabled state
[61804.138759] BUG: sleeping function called from invalid context at kernel/workqueue.c:2816
[61804.147019] in_atomic(): 1, irqs_disabled(): 0, pid: 22020, name: hostapd
[61804.153790] INFO: lockdep is turned off.
[61804.157784] CPU: 1 PID: 22020 Comm: hostapd Tainted: G        W       4.14.20 #0
[61804.165172] Stack : 00000000 00000000 00000000 00000000 80d2bcc2 00000044 00000000 00000000
[61804.173563]         8fecd194 8078ac87 806e6380 00000001 00005604 8054c004 8c007818 b3f6da3d
[61804.181936]         00000000 00000000 80d30000 009eb000 00000000 80016ef8 80016ef8 00000000
[61804.190309]         00000000 00000000 00002c21 00002c20 80000000 00000000 80790000 00000000
[61804.198678]         00000001 80780000 00000000 8de90000 00000003 80351e70 00000004 80840004
[61804.207046]         ...
[61804.209497] Call Trace:
[61804.211970] [<80011614>] show_stack+0x64/0x10c
[61804.216432] [<8054c004>] dump_stack+0x94/0xd0
[61804.220780] [<8006128c>] ___might_sleep+0x10c/0x12c
[61804.225647] [<80050238>] flush_work+0x2a0/0x304
[61804.230167] [<80052538>] __cancel_work_timer+0x150/0x1e0
[61804.235479] [<8d4542cc>] mt76_rx_aggr_stop+0xf4/0x140 [mt76]
[61804.241168] [<8d440adc>] mt76x2_ampdu_action+0xc8/0x198 [mt76x2e]
[61804.248083] [<8de06ec0>] drv_ampdu_action+0x168/0x2f8 [mac80211]
[61804.254769] [<8de1a5e4>] ___ieee80211_stop_rx_ba_session+0xe8/0x1b8 [mac80211]
[61804.262625] [<8de1727c>] ieee80211_sta_tear_down_BA_sessions+0x84/0x124 [mac80211]
[61804.270894] [<8de079a0>] __sta_info_destroy_part1+0xa4/0x890 [mac80211]
[61804.278301] [<8de0da28>] __sta_info_flush+0xd4/0x1d4 [mac80211]
[61804.284902] [<8de2cafc>] ieee80211_stop_ap+0x11c/0x46c [mac80211]
[61804.292052] [<8d542de8>] __cfg80211_stop_ap+0x178/0x554 [cfg80211]
[61804.298861] [<8d543218>] cfg80211_stop_ap+0x54/0x8c [cfg80211]
[61804.304941] [<80462490>] genl_rcv_msg+0x308/0x3d0
[61804.309650] [<8046124c>] netlink_rcv_skb+0xcc/0x17c
[61804.314517] [<80462170>] genl_rcv+0x3c/0x54
[61804.318690] [<804607c4>] netlink_unicast+0x174/0x290
[61804.323636] [<80460d94>] netlink_sendmsg+0x3b0/0x450
[61804.328602] [<803f7564>] ___sys_sendmsg+0x1bc/0x2cc
[61804.333471] [<803f86ac>] __sys_sendmsg+0x58/0x88
[61804.338098] [<8001b778>] syscall_common+0x34/0x58
[61804.720996] device wlan0 left promiscuous mode
[61804.725813] br-lan: port 2(wlan0) entered disabled state
```

nbd168, 2018-03-04:
Pushed a fix, please test.

slthomason, 2018-03-05:
Will that help the problem that  @CR-Ryan <https://github.com/cr-ryan> reported
in the other thread?  Or is that something different?

easyteacher, 2018-03-10:
After upgrading to the latest git, my router cannot boot up. I'm going to use the default build config and try again later.

easyteacher, 2018-03-11:
It's solved, thank you!

easyteacher, 2019-10-03:
Now my router's dmesg flushes with that message again!

```
root@ASUS-PC:~# opkg list-installed|grep kmod-mt76
kmod-mt76 - 4.14.146+2019-09-29-6f7d0f50-1
kmod-mt76-core - 4.14.146+2019-09-29-6f7d0f50-1
kmod-mt7603 - 4.14.146+2019-09-29-6f7d0f50-1
kmod-mt76x02-common - 4.14.146+2019-09-29-6f7d0f50-1
kmod-mt76x2 - 4.14.146+2019-09-29-6f7d0f50-1
kmod-mt76x2-common - 4.14.146+2019-09-29-6f7d0f50-1
```

> [ 7533.231095] BUG: sleeping function called from invalid context at kernel/workqueue.c:2816
> [ 7533.239650] in_atomic(): 1, irqs_disabled(): 0, pid: 10962, name: kworker/u8:2
> [ 7533.246895] CPU: 2 PID: 10962 Comm: kworker/u8:2 Tainted: G        W       4.14.146 #0
> [ 7533.254930] Workqueue: phy1 ieee80211_ibss_leave [mac80211]
> [ 7533.260481] Stack : 80549910 8e0cdbdc 805c0000 8c9a5500 805499a4 00000000 fffffffe 00000001
> [ 7533.268821]         8e439050 80075218 8fdf9c7c 00000000 805c0000 00000001 8e0cdba0 ac07f58b
> [ 7533.277159]         00000000 00000000 00000000 0001a978 00000000 0000138e 00000008 00000000
> [ 7533.285493]         00000000 805c0000 0003e3d2 00000000 00000000 00000000 805e0000 00000000
> [ 7533.293828]         fffffffe 00000001 8e439050 8e96d294 00000002 802bea20 00000008 80720008
> [ 7533.302166]         ...
> [ 7533.304604] Call Trace:
> [ 7533.307064] [<8000caa0>] show_stack+0x58/0x100
> [ 7533.311497] [<80485fbc>] dump_stack+0x9c/0xe0
> [ 7533.315856] [<80053be4>] ___might_sleep+0xf8/0x118
> [ 7533.320630] [<80049a44>] flush_work+0x178/0x27c
> [ 7533.325144] [<80049d80>] __cancel_work_timer+0x144/0x1d4
> [ 7533.330449] [<8ede53e8>] mt76_rx_aggr_stop+0x50/0x298 [mt76]
> [ 7533.336111] [<8e85083c>] mt76x02_ampdu_action+0xbc/0x178 [mt76x02_lib]
> [ 7533.342714] [<8e912cbc>] ___ieee80211_stop_rx_ba_session+0xc8/0x17c [mac80211]
> [ 7533.350008] [<8e912dc8>] __ieee80211_stop_rx_ba_session+0x58/0x80 [mac80211]
> [ 7533.357193] [<8e910f4c>] ieee80211_process_delba+0x30/0x44 [mac80211]
> [ 7533.363714] [<8e917798>] ieee80211_ibss_leave+0x48c/0x19e8 [mac80211]
> [ 7533.370220] [<804a06f8>] __schedule+0x7d8/0x890

easyteacher, 2019-10-12:
New warning messages, not sure if it's related to mt76.

> [  131.283761] BUG: sleeping function called from invalid context at mm/slab.h:419
> [  131.291125] in_atomic(): 1, irqs_disabled(): 0, pid: 7660, name: iptables
> [  131.298040] CPU: 0 PID: 7660 Comm: iptables Tainted: G        W       4.14.148 #0
> [  131.305536] Stack : 00000000 00000000 8d664f98 80073e88 805e0000 80583fa8 00000000 00000000
> [  131.313956]         8054e4ec 8eaf5bec 8fe8c83c 805bea07 80549008 00000001 8eaf5b90 ac07f58d
> [  131.322335]         00000000 00000000 80720000 00000000 80728370 00000161 00000008 00000000
> [  131.330698]         00000000 00000000 00048c38 ffffffff 80000000 00000000 805e0000 8e839f80
> [  131.339070]         8e840000 8d664fd8 8eaf5e38 00000000 00000000 802bebf0 00000000 80720000
> [  131.347453]         ...
> [  131.349907] Call Trace:
> [  131.352386] [<8000c6d4>] show_stack+0x58/0x100
> [  131.356833] [<80485bcc>] dump_stack+0x9c/0xe0
> [  131.361208] [<80053bd4>] ___might_sleep+0xf8/0x118
> [  131.366004] [<80118a88>] kmem_cache_alloc+0x44/0x1c0
> [  131.370982] [<8e83a6dc>] 0x8e83a6dc

closed 2019-10-12 by easyteacher