openwrt/mt76 #156: Data buffer not 16 bytes aligned

MaFeSa, 2018-02-28:
Xiaomi 3G - OpenWrt SNAPSHOT r6297-9a0cc49 

Every time I transfer large files using 5 GHz between clients speed transfert decreases and on kernel log there are many "Data buffer not 16 bytes aligned"
Thanks

```
[ 9576.288069] Data buffer not 16 bytes aligned: 8cb004b8
[ 9576.337872] Data buffer not 16 bytes aligned: 8cb001f8
[ 9576.347530] Data buffer not 16 bytes aligned: 8cb00328
[ 9576.361787] Data buffer not 16 bytes aligned: 8cb00068
[ 9576.444747] Data buffer not 16 bytes aligned: 8cb00078
[ 9576.453296] Data buffer not 16 bytes aligned: 8cb000b8
[ 9576.462692] Data buffer not 16 bytes aligned: 8cb00288
[ 9576.472247] Data buffer not 16 bytes aligned: 8cb00068
[ 9576.488434] Data buffer not 16 bytes aligned: 8cb00268
[ 9576.496996] Data buffer not 16 bytes aligned: 8cb002f8
[ 9576.505507] Data buffer not 16 bytes aligned: 8cb000e8
[ 9576.519563] Data buffer not 16 bytes aligned: 8cb000d8
[ 9576.530198] Data buffer not 16 bytes aligned: 8cb003d8
[ 9576.552339] Data buffer not 16 bytes aligned: 8cb00078
[ 9576.581297] Data buffer not 16 bytes aligned: 8cb00038
[ 9576.610400] Data buffer not 16 bytes aligned: 8cb00158
[ 9576.623424] Data buffer not 16 bytes aligned: 8cb00068
[ 9576.637247] Data buffer not 16 bytes aligned: 8cb00188
[ 9576.683518] Data buffer not 16 bytes aligned: 8cb00008
[ 9576.707916] Data buffer not 16 bytes aligned: 8cb00128
[ 9576.726215] Data buffer not 16 bytes aligned: 8cb003c8
[ 9576.740223] Data buffer not 16 bytes aligned: 8cb000f8
[ 9576.767276] Data buffer not 16 bytes aligned: 8cb00328
[ 9576.793536] Data buffer not 16 bytes aligned: 8cb001d8
[ 9576.806612] Data buffer not 16 bytes aligned: 8cb00198
[ 9576.820377] Data buffer not 16 bytes aligned: 8cb00008
[ 9576.832427] Data buffer not 16 bytes aligned: 8cb00048
[ 9576.841974] Data buffer not 16 bytes aligned: 8cb001c8
[ 9576.857564] Data buffer not 16 bytes aligned: 8cb001d8
[ 9576.871624] Data buffer not 16 bytes aligned: 8cb00298
[ 9576.891191] Data buffer not 16 bytes aligned: 8cb00238
[ 9576.899652] Data buffer not 16 bytes aligned: 8cb001a8
[ 9576.915445] Data buffer not 16 bytes aligned: 8cb00198
[ 9576.934921] Data buffer not 16 bytes aligned: 8cb00048
[ 9576.946731] Data buffer not 16 bytes aligned: 8cb00048
[ 9576.955219] Data buffer not 16 bytes aligned: 8cb00238
[ 9576.968130] Data buffer not 16 bytes aligned: 8cb00038
[ 9576.978717] Data buffer not 16 bytes aligned: 8cb004e8
[ 9576.988364] Data buffer not 16 bytes aligned: 8cb00028
[ 9576.996807] Data buffer not 16 bytes aligned: 8cb00068
[ 9577.005331] Data buffer not 16 bytes aligned: 8cb00038
[ 9577.013832] Data buffer not 16 bytes aligned: 8cb00018
[ 9577.026701] Data buffer not 16 bytes aligned: 8cb00308
[ 9577.046807] Data buffer not 16 bytes aligned: 8cb00258
[ 9577.066075] Data buffer not 16 bytes aligned: 8cb00178
[ 9577.080095] Data buffer not 16 bytes aligned: 8cb00068
[ 9577.093140] Data buffer not 16 bytes aligned: 8cb00038
[ 9577.101566] Data buffer not 16 bytes aligned: 8cb00278
[ 9577.135286] Data buffer not 16 bytes aligned: 8cb00178
[ 9577.144939] Data buffer not 16 bytes aligned: 8cb00018
[ 9577.163347] Data buffer not 16 bytes aligned: 8cb00378
[ 9577.175214] Data buffer not 16 bytes aligned: 8cb00298
[ 9577.184864] Data buffer not 16 bytes aligned: 8cb00178
[ 9577.194630] Data buffer not 16 bytes aligned: 8cb00278
[ 9577.203132] Data buffer not 16 bytes aligned: 8cb000a8
[ 9577.215043] Data buffer not 16 bytes aligned: 8cb00198
[ 9577.242162] Data buffer not 16 bytes aligned: 8cb00268
[ 9577.256327] Data buffer not 16 bytes aligned: 8cb00088
[ 9577.275686] Data buffer not 16 bytes aligned: 8cb00028
[ 9577.290823] Data buffer not 16 bytes aligned: 8cb001b8
[ 9577.313727] Data buffer not 16 bytes aligned: 8cb003c8
[ 9577.336948] Data buffer not 16 bytes aligned: 8c9c8748
[ 9577.344247] Data buffer not 16 bytes aligned: 8c9c8718
[ 9577.396888] Data buffer not 16 bytes aligned: 8c9c8568
[... 309 lines cut ...]
[ 9643.629687] Data buffer not 16 bytes aligned: 8c320238
[ 9643.641051] Data buffer not 16 bytes aligned: 8c320748
[ 9643.649387] Data buffer not 16 bytes aligned: 8c320038
[ 9643.656706] Data buffer not 16 bytes aligned: 8c320178
[ 9643.663948] Data buffer not 16 bytes aligned: 8c320148
[ 9643.671165] Data buffer not 16 bytes aligned: 8c320238
[ 9643.680470] Data buffer not 16 bytes aligned: 8c3203b8
[ 9643.687753] Data buffer not 16 bytes aligned: 8c320528
[ 9643.701142] Data buffer not 16 bytes aligned: 8c320668
[ 9643.709476] Data buffer not 16 bytes aligned: 8c320138
[ 9643.716693] Data buffer not 16 bytes aligned: 8c3202f8
[ 9643.724449] Data buffer not 16 bytes aligned: 8c3202c8
[ 9643.731813] Data buffer not 16 bytes aligned: 8c320298
[ 9643.745227] Data buffer not 16 bytes aligned: 8c320528
[ 9643.762893] Data buffer not 16 bytes aligned: 8c320378
[ 9643.785138] Data buffer not 16 bytes aligned: 8c320088
[ 9643.815075] Data buffer not 16 bytes aligned: 8c320058
[ 9643.824615] Data buffer not 16 bytes aligned: 8c3200d8
[ 9643.838615] Data buffer not 16 bytes aligned: 8c320078
[ 9643.849423] Data buffer not 16 bytes aligned: 8c320138
[ 9643.864348] Data buffer not 16 bytes aligned: 8c320028
[ 9643.889810] Data buffer not 16 bytes aligned: 8c3200c8
[ 9643.900510] Data buffer not 16 bytes aligned: 8c320368
[ 9643.908999] Data buffer not 16 bytes aligned: 8c320518
```

nbd168, 2018-02-28:
If I grep for this error message, I end up in a printk in the nand driver. I don't think this is mt76 related at all.

closed 2018-02-28 by nbd168

MaFeSa, 2018-02-28:
OK, thanks for your help!

kofec, 2018-02-28:
@MaFeSa 
mcaptur - compile latest nbd168 branch where: 
Here find a build with latest master (https://github.com/openwrt/openwrt/commit/d8655868ca04d12ad5ed36e26fc63d61b076e33d9)
latest MT76 commit (https://github.com/openwrt/mt76/commit/cda627fd8feffdadfcd692fbb6d823d455aa19ab12)
using kernel 4.14 and Gcc 7.3 (O2 optimization)
You can test HW offload by using (can add to firewall.user)

iptables -I FORWARD 1 -m conntrack --ctstate RELATED,ESTABLISHED -j FLOWOFFLOAD

Maybe you want to try ?