openwrt/mt76 #157: MT7612 (5ghz) - Main issue

barhom, 2018-03-06:
Device: WE3526
Running commit: 9046e921aff8947600567134ad10d6451a1adaeb

I have noticed this issue several times over the past week.
It seems that at random the wifi can become quite weird in a way that ICMP goes through but I am unable to connect via TCP to anything.

I can ping the router
I can ping 8.8.8.8
but I cannot ssh to the router nor can I ssh to any of my servers over wan.

The only solution is to reconnect to wifi.

Please feel free to use this issue to explain any other issues you might have noticed with the mt7612

updatede, 2018-03-07:
I encounter this either, high ping value, high packet loss,frequently make network unusable unless restart wifi.

Rising-Sun, 2018-05-31:
Had a similiar problem. Device was connected but no data is tansferred, like happens with 7603e.
Had these entries at the time of occuring in my log but dont know if it has any correlation.

```
Thu May 31 12:21:42 2018 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 6c:ab:31:59:XX:XX
Thu May 31 12:22:07 2018 daemon.notice hostapd: wlan1: STA 6c:ab:31:59:XX:XX IEEE 802.11: Could not set STA to kernel driver
Thu May 31 12:27:15 2018 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 6c:ab:31:59:XX:XX
Thu May 31 12:28:51 2018 daemon.notice hostapd: wlan1: STA 6c:ab:31:59:XX:XX IEEE 802.11: Could not set STA to kernel driver
Thu May 31 12:33:56 2018 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 6c:ab:31:59:XX:XX
Thu May 31 12:38:05 2018 daemon.notice hostapd: wlan1: STA 6c:ab:31:59:XX:XX IEEE 802.11: Could not set STA to kernel driver
```

nbd168, 2018-06-15:
Please try the latest version (master, not 18.06)

barhom, 2018-07-08:
I dont see a problem anymore.
If I stumble upon something I will open a new ticket.

closed 2018-07-08 by barhom