openwrt/mt76 #158: MT7628 multi SSID crash

baohavan, 2018-03-13:
Device: Tplink C50 v3
Lede rev: r6433-4fd8722056

kernel - 4.9.86-1-b8a218be09752a08276d68d9847edd43
kmod-mt76 - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt76-core - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt7603 - 4.9.86+2018-02-27-ffa2069e-1
kmod-mt76x2 - 4.9.86+2018-02-27-ffa2069e-1

[ 8888.455686] ------------[ cut here ]------------
[ 8888.460423] WARNING: CPU: 0 PID: 1955 at /Volumes/Build/gateway/c50/master/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2018-02-27-ffa2069e/mt7603_mac.c:441 mt7603_mac_fill_rx+0x1b0/0x530 [mt7603e]
[ 8888.480108] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables cmac leds_gpio gpio_button_hotplug crypto_hash
[ 8888.539444] CPU: 0 PID: 1955 Comm: kworker/u2:0 Not tainted 4.9.86 #0
[ 8888.545995] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 8888.551116] Stack : 4fb102fe 00000000 831d1ba0 8004caa8 8382d1fc 803cf207 80383900 000007a3
[ 8888.559640]         8038386c 8380ddac 803d0000 8007a7fc 831d1ba0 8004caa8 803890ec 803d0000
[ 8888.568153]         00000003 8380ddac 803d0000 800386c8 831d1ba0 8380dde4 000000ff 00000007
[ 8888.576669]         00000000 00000007 8323d48c 83b81f00 83b81e00 30796870 00000000 00000000
[ 8888.585165]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 8888.593678]         ...
[ 8888.596175] Call Trace:
[ 8888.598680] [<8000e220>] show_stack+0x70/0x8c
[ 8888.603119] [<80023a34>] __warn+0x110/0x118
[ 8888.607385] [<80023ad0>] warn_slowpath_null+0x1c/0x2c
[ 8888.612523] [<8323c018>] mt7603_mac_fill_rx+0x1b0/0x530 [mt7603e]
[ 8888.618738] [<8323aedc>] mt7603_queue_rx_skb+0xb8/0xe4 [mt7603e]
[ 8888.624843] [<831d8dbc>] mt76_dma_attach+0xafc/0xce0 [mt76]
[ 8888.630516] ---[ end trace ef40fdf3c5d20b48 ]---

loyasto, 2018-03-23:
I met the same problem. Did you solve it？

ffrediani, 2018-11-10:
Have exactly the same issue with the same radio. I initially thought it was something broader related to 2.4 Ghz radio but from the recent tests indicate it's related to multi SSID.

nbd168, 2018-11-20:
Please try the latest version from OpenWrt master, I pushed some more fixes.

closed 2019-01-25 by nbd168

ffrediani, 2019-01-29:
Hello. Has this been fixed completely ? Since my last test it was broken. Has anyone else tested after the latest commit if multiple SSID works fine now without any issues ?

goraelec, 2019-01-29:
Hello.

Unfortunately, I have no way to tell. I reverted to stock fw, which has
superior Wi-Fi performance