openwrt/mt76 #159: WARNING net/ipv4/tcp.c:1701 tcp_recvmsg+0x288/0x908

MaFeSa, 2018-03-14:
I don't know if it's related to latest mt76 commits, if not sorry nbd.

Xiaomi Mi Router 3G - OpenWrt SNAPSHOT r6437-1cd76e2 / LuCI Master (git-18.071.69548-de88c2e)

Transferring a file between clients connected to 5 ghz, the transfer hangs and I always have this warning.

```
[64918.940052] WARNING: CPU: 1 PID: 11063 at net/ipv4/tcp.c:1701 tcp_recvmsg+0x288/0x908
[64918.948257] recvmsg bug: copied 29D02D97 seq 29D09687 rcvnxt 29E5D4D7 fl 0
[64918.955611] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN
[64919.026898]  ip_tables crc_ccitt compat cdc_wdm fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[64919.075968] CPU: 1 PID: 11063 Comm: smbd Tainted: G        W       4.9.86 #0
[64919.082999] Stack : 00000000 00000000 80537b4a 00000040 80533824 00000000 00000000 80530000
[64919.091366]         8e8037fc 804d1da7 8046af14 00000001 00002b37 80533824 8199540c ffff9710
[64919.099762]         88793dd0 8006afa4 00000001 80530000 804d7f04 804d7f08 8046fb20 88793c64
[64919.108167]         00000003 800a7850 8199540c ffff9710 8121ee20 00000000 00000002 00793c64
[64919.116558]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[64919.124924]         ...
[64919.127364] Call Trace:
[64919.129853] [<8000f58c>] show_stack+0x70/0x8c
[64919.134223] [<801dab64>] dump_stack+0x94/0xd0
[64919.138566] [<8002bce4>] __warn+0x110/0x118
[64919.142751] [<8002bd1c>] warn_slowpath_fmt+0x30/0x3c
[64919.147706] [<80332fb0>] tcp_recvmsg+0x288/0x908
[64919.152367] [<8035c0a0>] inet_recvmsg+0x90/0xac
[64919.156909] [<802cc194>] sock_read_iter+0x90/0xc4
[64919.161599] [<800edcdc>] do_readv_writev+0x170/0x32c
[64919.166565] [<800edee0>] vfs_readv+0x48/0x64
[64919.170822] [<800edf60>] do_readv+0x64/0xd0
[64919.175023] [<80016598>] syscall_common+0x34/0x58
[64919.179923] ---[ end trace 374680a68428ae44 ]---
[64919.184806] ------------[ cut here ]------------
```

Thanks

MaFeSa, 2018-03-14:
The problem occurs only during the transfer of .mkv files.
I've never had this problem, it started when I updated to r6437.

MaFeSa, 2018-03-15:
@nbd168 
I have done some tests and the problem is in this commit: https://github.com/openwrt/openwrt/commit/6011f7bcf0bfab0d65fb0522de0c210565fe4e74

Going back to https://github.com/openwrt/openwrt/commit/b7b14fd64e09b523b2da6a9db6d7ff300964b955 the problem disappears.

Thanks.

nbd168, 2018-03-19:
Please try the latest version

MaFeSa, 2018-03-20:
Solved, thanks again.

closed 2018-03-20 by MaFeSa