openwrt/mt76 #160: insmod mt76.ko/mt7603e.ko/76x2e.ko  error

somme8023, 2018-03-17:
cpu:mt7621 kernel:3.18.92
The mt76 driver was compiled and an error occurred while installing the driver. What can I do to solve this problem?

/lib/modules/3.18.92 # insmod mt76.ko 
[10931.840000] mt76: Unknown symbol ieee80211_sta_uapsd_trigger (err 0)
[10931.840000] mt76: Unknown symbol ieee80211_sta_pspoll (err 0)
[10931.850000] mt76: Unknown symbol wiphy_read_of_freq_limits (err 0)
[10931.850000] mt76: Unknown symbol ieee80211_tx_dequeue (err 0)
[10931.860000] mt76: Unknown symbol backport_debugfs_create_devm_seqfile (err 0)
[10931.870000] mt76: Unknown symbol ieee80211_rx_napi (err 0)
insmod: can't insert 'mt76.ko': unknown symbol in module or invalid parameter
/lib/modules/3.18.92 # 
/lib/modules/3.18.92 # 
/lib/modules/3.18.92 # insmod mt7603e.ko 
[10955.140000] mt7603e: Unknown symbol mt76_txq_remove (err 0)
[10955.140000] mt7603e: Unknown symbol mt76_release_buffered_frames (err 0)
[10955.150000] mt7603e: Unknown symbol mt76_wake_tx_queue (err 0)
[10955.160000] mt7603e: Unknown symbol ieee80211_tx_status_ext (err 0)
[10955.160000] mt7603e: Unknown symbol mt76_stop_tx_queues (err 0)
[10955.170000] mt7603e: Unknown symbol mt76_dma_cleanup (err 0)
[10955.170000] mt7603e: Unknown symbol mt76_register_debugfs (err 0)
[10955.180000] mt7603e: Unknown symbol mt76_tx (err 0)
[10955.190000] mt7603e: Unknown symbol mt76_tx_queue_skb (err 0)
[10955.190000] mt7603e: Unknown symbol mt76_wcid_alloc (err 0)
[10955.200000] mt7603e: Unknown symbol mt76_wcid_key_setup (err 0)
[10955.200000] mt7603e: Unknown symbol mt76_rx (err 0)
[10955.210000] mt7603e: Unknown symbol mt76_eeprom_init (err 0)
[10955.210000] mt7603e: Unknown symbol mt76_register_device (err 0)
[10955.220000] mt7603e: Unknown symbol __mt76_poll_msec (err 0)
[10955.230000] mt7603e: Unknown symbol mt76_mmio_init (err 0)
[10955.230000] mt7603e: Unknown symbol mt76_eeprom_override (err 0)
[10955.240000] mt7603e: Unknown symbol ieee80211_alloc_hw_nm (err 0)
[10955.240000] mt7603e: Unknown symbol mt76_unregister_device (err 0)
[10955.250000] mt7603e: Unknown symbol mt76_txq_init (err 0)
[10955.250000] mt7603e: Unknown symbol mt76_txq_schedule_all (err 0)
[10955.260000] mt7603e: Unknown symbol mt76_rx_aggr_start (err 0)
[10955.270000] mt7603e: Unknown symbol __mt76_poll (err 0)
[10955.270000] mt7603e: Unknown symbol mt76_rx_aggr_stop (err 0)
[10955.280000] mt7603e: Unknown symbol mt76_dma_attach (err 0)
[10955.280000] mt7603e: Unknown symbol backport_debugfs_create_devm_seqfile (err 0)
insmod: can't insert 'mt7603e.ko': unknown symbol in module or invalid parameter

lukasz1992, 2018-03-18:
You try to mix incompatible (older?) version of mac80211 stack and this driver.
Please compile both mac80211 and mt76 from sources.

somme8023, 2018-03-19:
@lukasz1992 For some reason, I had to use the latest mt76 drive on the 3.18 kernel, I compiled the mac80211 and mt76 drive source code, and there was an error insmod ko files. I also tried to compile the old version of mac80211, and the same error occurred insmod ko file. The WiFi-related configuration in the kernel is configured. Is the kernel version too old?

lukasz1992, 2018-03-19:
> [10931.870000] mt76: Unknown symbol ieee80211_rx_napi (err 0)

You see: https://elixir.bootlin.com/linux/v3.18.100/ident/ieee80211_rx_napi  - this function is not present on 3.18 kernel. I don't know if (or which) compat-wireless version did you use, but it is probably too old.

somme8023, 2018-03-20:
@lukasz1992 Thank you, the problem has been solved.

closed 2018-03-20 by nbd168