openwrt/mt76 #161: Error installing ko file(probe of xxx failed with error xxx)

somme8023, 2018-03-22:
There is no problem installing Compat.ko/cfg80211.ko/mac80211.ko/mt76.ko, mt76x2e.ko/mt7603e.ko error occurred in installation.
Why is there an error?

/lib/modules/3.18.92 # insmod mt76x2e.ko 
[   57.460000] PCI: Enabling device 0000:00:00.0 (0000 -> 0002)
[   57.490000] mt76x2e 0000:01:00.0: Refused to change power state, currently in D3
[   57.510000] mt76x2e 0000:01:00.0: ASIC revision: 00000000
[   57.520000] mt76x2e 0000:01:00.0: EEPROM data check failed: ffff
[   57.520000] mt76x2e 0000:01:00.0: Invalid MAC address, using random address 0e:83:d5:f8:eb:f7
[   62.540000] mt76x2e: probe of 0000:01:00.0 failed with error -145
/lib/modules/3.18.92 # 
/lib/modules/3.18.92 # 
/lib/modules/3.18.92 # 
/lib/modules/3.18.92 # insmod mt7603e.ko 
[  132.570000] PCI: Enabling device 0000:00:01.0 (0000 -> 0002)
[  132.610000] mt7603e 0000:02:00.0: Refused to change power state, currently in D3
[  132.630000] mt7603e 0000:02:00.0: ASIC revision: 0000
[  132.640000] mt7603e 0000:02:00.0: Invalid MAC address, using random address 3e:66:98:29:8d:03
[  132.650000] mt7603e 0000:02:00.0: Firmware Version: _e1_drv_tv
[  132.650000] mt7603e 0000:02:00.0: Build Time: 20140313215548
[  133.680000] mt7603e 0000:02:00.0: Timeout waiting for ROM code to become ready
[  133.680000] mt7603e: probe of 0000:02:00.0 failed with error -5

lukasz1992, 2018-03-22:
Both seem to be timeouts, probably incorrect (too old) fw or/and driver version.

Please use the newest build, describe your hardware and then we could help you.

somme8023, 2018-03-27:
@lukasz1992 
cpu:mtk mt7621 
wifi:mt7603e/7612e 
flash:32M 
RAM:512M

I have used a newer kernel version(4.9.73). But there was an error insmod ko files.
Insmod mt7603e.ko,no mac address was obtained.
Insmod mt76x2e.ko,direct firmware load for mt7662.bin failed with error -2.

/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # insmod compat.ko 
[  308.189522] Loading modules backported from Linux version wt-2017-11-01-0-gfe248fc2c180
[  308.197558] Backport generated by backports.git v4.14-rc2-1-31-g86cf0e5d
/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # insmod cfg80211.ko 
/lib/modules/4.9.73 # insmod mac80211.ko 
/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # insmod mt76.ko 
/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # insmod mt7603e.ko 
[  328.363487] mt7603e 0000:02:00.0: ASIC revision: 76030010
[  328.372634] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
[  328.378279] mt7603e 0000:02:00.0: Build Time: 20160107100755
[  328.405043] firmware init done
/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # insmod mt76x2e.ko 
[  333.000932] mt76x2e 0000:01:00.0: ASIC revision: 76120044
[  333.025063] mt76x2e 0000:01:00.0: ROM patch already applied
[  333.030769] mt76x2e 0000:01:00.0: **Direct firmware load for mt7662.bin failed with error -2**
[  333.039070] mt76x2e 0000:01:00.0: Falling back to user helper
[  397.385325] mt76x2e: probe of 0000:01:00.0 failed with error -11

/lib/modules/4.9.73 # 
/lib/modules/4.9.73 # ifconfig wlan0
wlan0     Link encap:Ethernet  **HWaddr FF:FF:FF:FF:FF:FF**  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lukasz1992, 2018-03-27:
-2 = No such file or directory

somme8023, 2018-03-28:
@lukasz1992 
In "/lib/firmware" path, there are 5 files(mt7603_e1.bin/mt7603_e2.bin/mt7662.bin/mt7662_rom_patch.bin/regulatory.db).Why can't I find a file?

somme8023, 2018-03-28:
The problem has been solved. Found mt7662.bin files are linked.

closed 2018-03-28 by nbd168