openwrt/mt76 #164: Cannot change mac address using uci (or similar tools) in openwrt for mt76x2e on the Xiaomi 3G device

d1b, 2018-03-31:
Hi, it appears that configuring a mac address for the mt76x2e interface with openwrt[0] on the Xiaomi 3G device currently requires [modifying the Factory partition](https://forum.lede-project.org/t/xiaomi-wifi-router-3g/5377/621) in order to allow clients to connect.

[0] snapshot release r6565-fd588db

nasbdh9, 2018-05-07:
You can use breed ( a bootloaderHe can modify many of R3G's parameters, including mac address ) After refreshing, you need to set a `xiaomi.r3g.bootfw` environment variable in the breath and set the value to `2` to specify boot from kernel1.

d1b, 2018-05-20:
@nasbdh9 okay.

d1b, 2018-06-10:
The issue appears to still be present.

psyborg55, 2018-07-08:
how does your issue relates to the wifi driver?

d1b, 2018-08-12:
Assuming that there isn't a bug in uci (and similar tools) then there is a bug in the wifi driver.

d1b, 2019-12-27:
It doesn't appear to have been fixed in 19.07.0-rc2.

d1b, 2022-01-02:
This seemingly has been resolved.

closed 2022-01-02 by d1b