openwrt/mt76 #169: MT7628: Clients keep disconnecting.

yujincheng08, 2018-05-06:
I have a HC5661A using MT7628. I've just test the latest commit e2eedc92 and it's very unstable.
I connect my xperia phone to the wifi and it keeps disconnected around every 10 minutes. And the log keeps showing:
```
WPA: group key handshake completed (RSN)
```
I google for a solution and I added the followings to `/etc/config/wireless`
```
option wmm '0'
option wpa_group_rekey '0'
option wpa_pair_rekey '0'
option wpa_master_rekey '0'
```
Things become better. But after 18-20 hours, clients disconnect again and the AP disappears from the air. I reference to the log and it shows:
```
hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs
```
I asked google for help again and added the following line to `/etc/config/wireless`:
```
option disassoc_low_ack '0'
```
After a few days testing, clients also keep disconnected, the log did not show something related to ACKs but:
```
hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: disassociated due to inactivity
daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
```
BTW, when clients disconnected, only rebooting recover the AP.

The full `/etc/config/wireless`:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option htmode 'HT20'
        option country '00'
        option legacy_rates '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option wmm '0'
        option encryption 'psk-mixed'
        option key 'keep it secret'
        option wpa_group_rekey '0'
        option wpa_pair_rekey '0'
        option wpa_master_rekey '0'
        option disassoc_low_ack '0'
```

XobQuit, 2018-05-19:
Same on TL-WR841ND v13

nbd168, 2018-05-19:
I've pushed some more fixes for MT7603 and MT7628, please test the latest version from mt76.git (not pushed to OpenWrt yet).

yujincheng08, 2018-05-20:
@nbd168 Thx for your fixes. I've been tried it for half a day and clients still disconnect.
And the system log is
```
Sun May 20 19:02:46 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Sun May 20 19:02:46 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated due to inactivity
Sun May 20 19:02:47 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```
I didn't change any configuration in `/etc/config/wireless` so it's the same as when I post this issue.
If you want some more information, please tell me and I am glad to provide.

XobQuit, 2018-05-20:
Problem still present.

ingamedeo, 2018-05-26:
Issue confirmed on hardware with the same chipset

ekenchan, 2018-06-08:
Same issue on MI-Nano ( MT7628AN )

yujincheng08, 2018-06-28:
Thing seems to be better after testing for several days on 65e161b. I did not suffer from disconnecting for these days. Thx for fixing!

outbackdingo, 2018-08-31:
no idea how you think its fixed, i am seeing it on both the NANO and 841 devices still

john3voltas, 2018-09-04:
And I'm still seeing this on a very different piece of hardware, the TP-Link TL-WR1043ND V2 with a Qualcomm Atheros QCA9558 chipset and I'm on OpenWrt 18.06.1 r7258-5eb055306f.
`Tue Sep  4 12:25:35 2018 daemon.info dnsmasq-dhcp[1417]: DHCPINFORM(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc
Tue Sep  4 12:25:35 2018 daemon.info dnsmasq-dhcp[1417]: DHCPACK(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc MyLAPTOP
Tue Sep  4 12:26:09 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 30:52:cb:aa:bb:cc
Tue Sep  4 12:26:09 2018 daemon.info hostapd: wlan0: STA 30:52:cb:aa:bb:cc IEEE 802.11: disassociated
Tue Sep  4 12:26:10 2018 daemon.info hostapd: wlan0: STA 30:52:cb:aa:bb:cc IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Sep  4 12:26:21 2018 daemon.info dnsmasq-dhcp[1417]: DHCPREQUEST(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc
Tue Sep  4 12:26:21 2018 daemon.info dnsmasq-dhcp[1417]: DHCPACK(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc MyLAPTOP
Tue Sep  4 12:26:25 2018 daemon.info dnsmasq-dhcp[1417]: DHCPINFORM(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc
Tue Sep  4 12:26:25 2018 daemon.info dnsmasq-dhcp[1417]: DHCPACK(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc MyLAPTOP
Tue Sep  4 12:27:43 2018 daemon.info dnsmasq-dhcp[1417]: DHCPINFORM(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc
Tue Sep  4 12:27:43 2018 daemon.info dnsmasq-dhcp[1417]: DHCPACK(br-lan) 10.11.11.30 30:52:cb:aa:bb:cc MyLAPTOP`

yujincheng08, 2018-09-04:
Reopen this issue because problem still presents.

dibg, 2018-10-18:
Same issue on MI-Nano (MT7628AN)

julianocss, 2018-10-18:
Same issue on WR840 V4 (MT7628AN).

sanitariu, 2018-10-19:
Same issue on WR840 V5 (MT7628AN).
Router is unusable.

yujincheng08, 2018-10-29:
It seems that the latest version(fdc63f16) performs better now. I have tested it for two days and no disconnection has occurred. But I didn't connect the clients for a continuous long period so I am not sure the issue was completely settled down. I will keep this issue opened and if no other feedback saying this issue persists, I will then close it.

julianocss, 2018-10-30:
I´m testing on my WR840 V4 too.

I give a feedback in the next days.

Thanks.
Juliano

ffrediani, 2018-11-07:
Hi
Just want to report that similar to other people reporting stability issues when using TP-Link C20 v4 which has this same radio has also been very unstable. The symptoms are: when you boot up the router the SSID is advertised but after a couple of minutes it becomes unstable and disconnects clients which, although the SSID keeps been advertised, cannot connect anymore until reboot.
Tried version 18.06.1 and also Snapshot.  For the Snapshot (from last 30th October) it has been worst and most of the time the SSID is not even advertised most of the time and therefore clients cannot connect.

Tried to change the TX power (18, 19 and 20) and also to use in different fixed channels and also in Auto mode. Made no difference.

**The configuration for /etc/config/wireless is default as follow:**

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option htmode 'HT20'
        option country 'US'
        option legacy_rates '1'
        option txpower '20'
        option channel 'auto'

Find attached some logs for 2 scenarios: 1) Radio Channel configured in Auto and then in Channel 6. Both scenarios the client's behaviour is the same as reported initially.

[Logs-MT7628AN-C20-v4.txt](https://github.com/openwrt/mt76/files/2555932/Logs-MT7628AN-C20-v4.txt)

ffrediani, 2018-11-08:
Hello again.

I have a significant feedback to give. I have been running the same router with MT7628AN for the last two days with a single SSID and it seems more stable. Haven't had any disconnects or SSID disappearing as the previous times running with two SSIDs.
Perhaps multiple SSIDs is something to be looked into more deep in order to confirm these issue.

I will keep it this way and if there is anything new I will report again.

julianocss, 2018-11-09:
Hello!

I left a feedback in the issue #188 about latest commit.
There was a huge improvement in speed on my WR-840 v4 (MT7628AN).

Juliano

ido1990, 2018-12-28:
Same issue on my GL-MT300N-V2

HunterDG, 2019-01-08:
Same issue on my HooToo TM06 (Using GL MT300N build).  un-checking Disassociate On Low Acknowledgement (default checked) seems to help.  Agree that multiple SSIDs may exaggerate the issue (I use wifi>wifi AND WAN>wifi routing).  Disabling the client/STA SSID and using only ethernet for WAN seems to allow the router to function a bit longer than using wifi>wifi routing, before connectivity dies.

nbd168, 2019-01-25:
Please try the latest version, with improved tx status handling, things should be better now.

joshuisken, 2019-02-14:
I have the same issues on a Archer C7 v2, actually upgraded from lede 17.01.3 due to this issue and it still exists in 18.06.2 (SoC: Qualcomm Atheros QCA9558 ver 1 rev 0)
Various options suggested at the start of this thread did not help. Except that with these options:

Thu Feb 14 21:08:15 2019 daemon.info hostapd: wlan1: STA 00:26:e8:d3:43:45 IEEE 802.11: authenticated
Thu Feb 14 21:08:15 2019 daemon.info hostapd: wlan1: STA 00:26:e8:d3:43:45 IEEE 802.11: associated (aid 2)
Thu Feb 14 21:08:19 2019 daemon.info hostapd: wlan1: STA d0:fc:cc:db:2f:32 IEEE 802.11: authenticated
Thu Feb 14 21:08:19 2019 daemon.info hostapd: wlan1: STA d0:fc:cc:db:2f:32 IEEE 802.11: associated (aid 2)
Thu Feb 14 21:08:27 2019 daemon.info hostapd: wlan1: STA d0:fc:cc:db:2f:32 IEEE 802.11: deauthenticated due to local deauth request

I do have 2 radios switched on with same SSID + pwd (2.4G and 5G) and enabled those options on both interfaces.

joshuisken, 2019-02-14:
Now I disabled "Allow legacy 802.11b rates" on both radios. This seems to help! Without reboot I haven't seen a single disconnect during more than half an hour whereas before is was happening every minute or so.

ingamedeo, 2019-02-14:
@joshuisken Completely unrelated! It's a different chipset!!

joshuisken, 2019-02-14:
Maybe I'm suggesting that it is not chipset related? Actually I did realize
it is a completely different chipset, as mentioned.

ingamedeo, 2019-02-14:
@joshuisken Thank you. The issue you have reported can be caused by many different factors and it is actually a known issue on the mt76, therefore it clearly is chipset related. It may not be on your unit however.

joshuisken, 2019-02-15:
I was facing many "deauthenticated due to inactivity (timer DEAUTH/REMOVE)" in my Archer C7 v2, and since last night they are completely gone after disabling "Allow legacy 802.11b rates" on both radios.
I'm running 18.06.2 happily now.

john3voltas, 2019-02-17:
@joshuisken 
Yep.
To me, OpenWRT's wifi is broken for way more than an year and no one seems to care about it...

markuznw, 2019-03-07:
I've got the same issue of
IEEE 802.11: disassociated due to inactivity
on my wife Note9. It appears to happen on 5ghz network and every 2-3 minutes at max.

I've got both disassoc_low_ack and legacy rates options deactivated.

Hope @nbd168 can look into it because in the current state she got to use LTE to use phone as mir3g wifi is TOTALLY unusable with that phone :(

Btw using stock rom (no modifications of any kind) so I guess it's really related to a bug in the driver :/

All looks fine on other phones, but same issue keeps happening with a Samsung Smart TV on the other room. It all goes fine as soon as we reboot the router but stalls like said after a while

nbd168, 2019-03-07:
@markuznw: for the 5ghz network that's a different issue (mt76x2e driver, not mt7603e). What openwrt version are you using? If it's older than about 5 days (or based on the 18.06 branch), you should update to latest master and try again.

markuznw, 2019-03-07:
Thanks for the fast reply Felix. I'm using master (and I even tried e72376d2f03a5b4b19b00bbefd80023de6d218f6 on this repo to see if there were any differences) but no luck. Can't get why it is all fine on my oneplus 3t and other devices but on Note9 with better antennas and hardware so many problems

lukasz1992, 2019-03-24:
@markuznw @yujincheng08 it is probably finally fixed in master.

socomsystems, 2019-03-25:
This is not fixed... Thought it would be if after upgrading from .1
Model	MikroTik RBM33G
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.051.55698-76cf653)
Kernel Version	4.14.95

nbd168, 2019-03-25:
You need to use a current snapshot, the fix was committed very recently.

markuznw, 2019-03-25:
@nbd168 my issue with note9 seems to be solved on latest Master and your last push of mt76.
Not a single disconnection in the past 16 hours. 

Thanks for your hard work on making this kernel module available and fixing all quirks :)

closed 2019-03-27 by nbd168

socomsystems, 2019-03-28:
@nbd168 forgive my ignorance, I'm not sure how to go about applying latest commits.  I'm on an RBM33G board MediaTek MT7621 ver:1 eco:3 using R11e-2HPnD radio cards.  Thanks so much for your help, creation and fixing of this issue.

agunzagun, 2019-04-08:
I hate to just "me too" this (or if I should have filed the report on its own), but I'm also experiencing issues - namely on the 5GHz band. I found this thread trying to Google solutions.

Logread comes up

```
root@OpenWrt:~# logread | tail
Mon Apr  8 02:35:37 2019 daemon.info dnsmasq-dhcp[3138]: DHCPACK(br-lan) X.X.X.X xx:xx:xx:xx:xx:xx WirelessClient  
Mon Apr  8 02:35:38 2019 daemon.warn odhcpd[1202]: DHCPV6 CONFIRM IA_NA from xxxxxxxxxxxxxxxxxxxxxxxxxxxx on lan: ok xxxx:xxxx:xxxx::xxx/xxx  
Mon Apr  8 02:35:45 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs  
Mon Apr  8 02:35:45 2019 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED xx:xx:xx:xx:xx:xx 1  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED xx:xx:xx:xx:xx:xx 2  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
```

And more or less repeats as such. No outstanding issues (so far) on 2.4GHz.

The build used is based on the latest git as of this post. The profile is based on RAMIPS>MTC WR1201.

```
Arch: MediaTek MT7621 ver:1 eco:3
Radio0 Type: MediaTek MT76x2E 802.11ac
```

lukasz1992, 2019-04-08:
@agunzagun If you have this issue on 5GHz, please open another ticket, as this is about MT7628 (2.4GHz radio)

ffrediani, 2019-04-08:
@agunzagun for the 5 Ghz if you you Wifi chip is MT7610E please report all in issues #227

NischalBhandari, 2020-02-23:
This issue exists in MediaTek MT7620N ver:2 eco:6 also . Have not been able to fix this issue learned a lot trying to debug this but many having tried it with different device with MediaTek MT7620N ver:2 eco:6 and different firmware still brings up this issue . While analyzing wifi from a mobile device the signal seem to drop sharply and then again restart again .

ykzj, 2020-03-06:
same here on Nexx 3020F

lukasz1992, 2020-03-06:
@ykzj @NischalBhandari But MT7620N has nothing to do with mt76 driver.
MT7620N wifi is driven by rt2800soc, not mt76

ykzj, 2020-03-06:
you are right. I finally figure out that is caused by high cpu temperature. I decrease cpu frequency to low cpu temperature, and the symptoms are gone.

发自我的iPhone

@ykzj @NischalBhandari But MT7620N has nothing to do with mt76 driver.
MT7620N wifi is driven by rt2800soc, not mt76

brentonv, 2020-03-27:
https://github.com/openwrt/mt76/issues/379

db260179, 2020-11-22:
So for the mt76 driver, there is an unknown issue with setting the country code. So for now remove

`option country`

From your wireless config, and let the driver decide what it needs to set.

brentonv, 2020-11-25:
@db260179 sorry mate deleting config `option country` doesn't fix anything. This crontab is the only thing I've found to keep the device reliable  `0 */1 * * * ifdown wwan && ifup wwan`

trunneml, 2024-02-22:
Don't know if it helps but I have the same problem with a banana pi r3 (MediaTek MT7986) on the 5GHz Network with two iphones.

Hemi03, 2024-12-04:
@trunneml got the same issue on the same router but with a Pixel 9 Pro. Did you manage to fix the issue?

chirantan300, 2025-01-16:
> [@trunneml](https://github.com/trunneml) enabling wnm sleep mode seems to fix it

chirantan300, 2025-01-16:
@Hemi03 enabling wnm sleep mode seems to fix it for me