openwrt/mt76 #172: MT7612 - TX Power Issue + Solution

CR-Ryan, 2018-05-16:
We have had TX power issues on the 7612 for ages - and no amount of goofing with the wireless config helped.  Setting a TX value anywhere between 11 - 30dBm made no difference to the actual observed signal strength. 

Here are some examples of what we tried:
Setting various values for  "option txpower"
Trying all of the available htmodes
Trying different country codes
Trying various channels  (using "iw list" as a reference)
We are on 73edb2

So we (myself and @slthomason) had a contractor write this patch for us:

[002-debugfs-api-for-target-power.txt](https://github.com/openwrt/mt76/files/2010980/002-debugfs-api-for-target-power.txt)
 
We recognize that it's a bit of a hack, but it works, and has been for sometime. When we include that patch in our build, we see a ~9dBm improvement in observed signal strength. (-63 vs -72dBm @ 15m) 

Also, _without_ our tx patch in place, iwconfig shows this:

wlan1     IEEE 802.11  Mode:Master  Tx-Power=9 dBm
          RTS thr:off   Fragment thr:off
          Power Management:off

So it appears to be hitting some kind of wall around 9 - 11dBm in stock form. 

We were hoping to see this issue fixed correctly, so that we don't have to use a band-aid.

barhom, 2018-05-26:
May we know which target it is you are testing?

barhom, 2018-06-19:
@nbd168 
You have one of these ZBT devices. What is your comment regarding this patch?

Mushoz, 2018-06-19:
Is this a related problem? If not, please let me know so I can open a new issue. I am using a Dir-860l.

When I double-check whether my regulatory domain is correctly set to The Netherlands, we can see that this is correct:

```
root@LEDE:~# iw reg get
global
country NL: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

I can see that on channels 100-140 I should be able to use up to 27 dBm:

```
		Frequencies:
			* 5180 MHz [36] (20.0 dBm)
			* 5200 MHz [40] (20.0 dBm)
			* 5220 MHz [44] (20.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (radar detection)
			* 5280 MHz [56] (20.0 dBm) (radar detection)
			* 5300 MHz [60] (20.0 dBm) (radar detection)
			* 5320 MHz [64] (20.0 dBm) (radar detection)
			* 5500 MHz [100] (27.0 dBm) (radar detection)
			* 5520 MHz [104] (27.0 dBm) (radar detection)
			* 5540 MHz [108] (27.0 dBm) (radar detection)
			* 5560 MHz [112] (27.0 dBm) (radar detection)
			* 5580 MHz [116] (27.0 dBm) (radar detection)
			* 5600 MHz [120] (27.0 dBm) (radar detection)
			* 5620 MHz [124] (27.0 dBm) (radar detection)
			* 5640 MHz [128] (27.0 dBm) (radar detection)
			* 5660 MHz [132] (27.0 dBm) (radar detection)
			* 5680 MHz [136] (27.0 dBm) (radar detection)
			* 5700 MHz [140] (27.0 dBm) (radar detection)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
```

However, setting anything higher than 20 dBm still shows only 20 dBm in Luci:

```
Mode: Master | SSID: XXXX
BSSID: XX:XX:XX:XX:XX:XX | Encryption: WPA2 PSK (CCMP)
Channel: 100 (5.500 GHz) | Tx-Power: 20 dBm
Signal: -28 dBm | Noise: 0 dBm
Bitrate: 300.0 Mbit/s | Country: NL
```

Setting the tx power to something lower than 20 does correctly work, so it's not the setting of power itself that seems bugged:

```
Mode: Master | SSID: XXXXX
BSSID: XX:XX:XX:XX:XX:XX | Encryption: WPA2 PSK (CCMP)
Channel: 100 (5.500 GHz) | Tx-Power: 19 dBm
Signal: -34 dBm | Noise: 0 dBm
Bitrate: 144.4 Mbit/s | Country: NL
```

Kherby, 2018-07-05:
I have the same problem with my DIR-860L @ OpenWRT 18.06rc1.

> root@860L-AP1:~# iw list | grep -i dbm
>                         * 2412 MHz [1] (20.0 dBm)
>                         * 2417 MHz [2] (20.0 dBm)
>                         * 2422 MHz [3] (20.0 dBm)
>                         * 2427 MHz [4] (20.0 dBm)
>                         * 2432 MHz [5] (20.0 dBm)
>                         * 2437 MHz [6] (20.0 dBm)
>                         * 2442 MHz [7] (20.0 dBm)
>                         * 2447 MHz [8] (20.0 dBm)
>                         * 2452 MHz [9] (20.0 dBm)
>                         * 2457 MHz [10] (20.0 dBm)
>                         * 2462 MHz [11] (20.0 dBm)
>                         * 2467 MHz [12] (20.0 dBm)
>                         * 2472 MHz [13] (20.0 dBm)
>                         * 5180 MHz [36] (20.0 dBm)
>                         * 5200 MHz [40] (20.0 dBm)
>                         * 5220 MHz [44] (20.0 dBm)
>                         * 5240 MHz [48] (20.0 dBm)
>                         * 5260 MHz [52] (20.0 dBm) (radar detection)
>                         * 5280 MHz [56] (20.0 dBm) (radar detection)
>                         * 5300 MHz [60] (20.0 dBm) (radar detection)
>                         * 5320 MHz [64] (20.0 dBm) (radar detection)
>                         * 5500 MHz [100] (26.0 dBm) (radar detection)
>                         * 5520 MHz [104] (26.0 dBm) (radar detection)
>                         * 5540 MHz [108] (26.0 dBm) (radar detection)
>                         * 5560 MHz [112] (26.0 dBm) (radar detection)
>                         * 5580 MHz [116] (26.0 dBm) (radar detection)
>                         * 5600 MHz [120] (26.0 dBm) (radar detection)
>                         * 5620 MHz [124] (26.0 dBm) (radar detection)
>                         * 5640 MHz [128] (26.0 dBm) (radar detection)
>                         * 5660 MHz [132] (26.0 dBm) (radar detection)
>                         * 5680 MHz [136] (26.0 dBm) (radar detection)
>                         * 5700 MHz [140] (26.0 dBm) (radar detection)
>                         * 5745 MHz [149] (13.0 dBm)
>                         * 5765 MHz [153] (13.0 dBm)
>                         * 5785 MHz [157] (13.0 dBm)
>                         * 5805 MHz [161] (13.0 dBm)
>                         * 5825 MHz [165] (13.0 dBm)

So first of all it should be 27dBm @ channel 100-140 in my country and i'm using channel 100 right now but the Tx Power is still @20dBm and a manuel change doesn't work either.... It's really weird !
Hopefully this will be fixed soon...

barhom, 2018-07-08:
Has this been incorporated ?
@nbd168

psyborg55, 2018-07-08:
this half-assed solution should not be incorporated anywhere. mt76 chip max output is 23.5dBm and even at full power you need hell of a calibration and offload routines that this driver don't provide. this could help in a case your device has ext RF components which the mentioned devices do not have at least according to logs and wiki images of the board (so the bug is somewhere else) but even in such a case that is not a proper way of setting output power to these chips and can only result in further performance degradation

slthomason, 2018-07-09:
We initially introduced it as a "hack".  However, the power, range, and
throughput all dramatically go up when we ad this patch.  I am all ears on
how to do this.

Spencer

psyborg55, 2018-07-09:
then you have dramatically buggy board as i've said and need to look elsewhere (you did check eeprom?). on average boards this would cause problems.

nbd168, 2018-07-09:
From what I can see, the vendor driver also has some code to allow boosting the tx power beyond the EEPROM values. However, I think simply exposing this via debugfs is too risky. In my opinion, an acceptable solution would be to allow adding a power offset or overriding the target power via device tree

psyborg55, 2018-07-09:
overriding the target power via device tree would be better solution

Mushoz, 2018-07-09:
I am not sure I am following. So the fact "iw list" shows 27 dBm on some channels, but unable being able to set anything higher than 20 dBm is not a bug? Or am I misunderstanding things?

camelcamro, 2018-07-10:
is this also related to 2,4GHZ ?
and is there the same situation, that it is not supported to setup TX power ?

slthomason, 2018-07-10:
This patch (hack) is just for the mt7612 (5g)

xavier83, 2019-03-24:
@slthomason do you think I'm having the same issue (https://github.com/openwrt/mt76/issues/199) described here?

nbd168, 2019-06-27:
Please try the latest version, it should work better now

closed 2020-02-20 by nbd168

storm1er, 2020-11-08:
Hi,

Could this issue cause this kind of problem? 
https://forum.openwrt.org/t/rpi4-and-netgear-a6210-mt7612u-long-distance-problem/78682

> As soon as I am at 15m (~50 feets) away from the router, my devices disconnect instantly without any reason, the quality connection is good (like 4/5 before disconnect)