openwrt/mt76 #173: OpenWRT build issues - Can't find MT76 wireless drivers

CR-Ryan, 2018-05-22:
We just pulled the latest OpenWRT code, and ran into build issues right away. 

- Both radios will not come up
- The default wireless config is incomplete

Here is the default wireless config that is generated:

"config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'"

We replaced that config with a known good one, and when we do a wifi reboot, we get these errors:

_Tue May 22 16:23:38 2018 daemon.notice netifd: radio1 (1746): Could not find PHY for device 'radio1'
Tue May 22 16:23:38 2018 daemon.notice netifd: radio1 (1761): WARNING: Variable 'data' does not exist or is not an array/object
Tue May 22 16:23:39 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Tue May 22 16:23:40 2018 daemon.err hostapd: Could not set interface wlan1 flags (UP): Operation timed out
Tue May 22 16:23:40 2018 daemon.err hostapd: nl80211: Could not set interface 'wlan1' UP
Tue May 22 16:23:40 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Tue May 22 16:23:40 2018 daemon.err hostapd: nl80211 driver initialization failed.
Tue May 22 16:23:40 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Tue May 22 16:23:40 2018 daemon.notice hostapd: wlan1: AP-DISABLED
Tue May 22 16:23:40 2018 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Tue May 22 16:23:40 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Tue May 22 16:23:40 2018 kern.err kernel: [ 1318.987207] mt76x2e 0000:01:00.0: MCU message 20 (seq 3) timed out
Tue May 22 16:23:40 2018 daemon.notice netifd: radio0 (1745): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Tue May 22 16:23:40 2018 daemon.notice netifd: radio0 (1745): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path ()
Tue May 22 16:23:40 2018 daemon.notice netifd: radio0 (1745): Command failed: Invalid argument
Tue May 22 16:23:40 2018 daemon.notice netifd: radio0 (1745): Device setup failed: HOSTAPD_START_FAILED_

We interpreted these errors to mean that the wireless drivers are missing - is that correct? Is anyone else having issues on the latest code? (18.06)

slthomason, 2018-05-23:
@ndb168 is that part of this: https://forum.lede-project.org/t/mt76-stopped-working-with-latest-updates/14623

intriguedlife, 2018-05-23:
I have no problem with the latest openwrt in combination with mt76 master or 0b8d1ddec926bf212f6d665d2dcfb0effd1b3c66 (What openwrt master currently uses).

So this issue has likely been fixed. Or you had some local build problem.

slthomason, 2018-05-23:
Are you using 4.14 kernel?  Or 4.9?

https://forum.lede-project.org/t/mt76-stopped-working-with-latest-updates/14623/15

intriguedlife, 2018-05-23:
I'm using openwrt master. That is currently kernel 4.14.

slthomason, 2018-05-23:
Hrm, we pulled OpenWRT on a brand new pull on the very latest master
(pulled the whole thing new) and selected the WE1326 and it is definitely
broken.

$ git log -1
commit a03035dad198cd4b51645ceb43c1170f9cf95f16

CR-Ryan, 2018-05-25:
Still having issues with this. 

See this thread: [https://github.com/openwrt/mt76/issues/174#issuecomment-392168725](url)

barhom, 2018-05-26:
Same issues here on WE3526 (zbt) target.

barhom, 2018-05-27:
Just double verified this by making a make clean, make distclean and rebuild everything from scratch.
Target WE3526 has completely broken wifi.

I am unsure if it has something to do with the drivers and/or openwrt master.

intriguedlife, 2018-05-28:
Perhaps you could flash a snapshot image from https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/

Just to rule out any build issues.

karlp, 2018-05-29:
@intriguedlife I also have failures with 2.4g missing, on an updated branch today.
mt76-2018-05-18-0b8d1dde.tar.xz is the broken one, going back to mt76-2018-04-15-a8e89215.tar.xz works, but I'm still bisecting.

slthomason, 2018-05-29:
This looks to be the offending commit: https://github.com/openwrt/openwrt/commit/5f7396ebef09b224edf08b0bda113613a42f0928

Anyone here know why this change was made and/or how to fix it?

I found it from this: https://github.com/openwrt/mt76/issues/174#issuecomment-392820725

intriguedlife, 2018-05-29:
Great to hear that the problem has been found. I happen to have a MIR3G so that explains why I didn't have any problems.

karlp, 2018-05-29:
Well, mir3g.dts _has_ the lines, so you probably _should_ have been affected.  there's reports of MIR3G users both having two radios, and having one radio, so... *shrugs* who knows what's going on there!

intriguedlife, 2018-05-31:
> @slthomason <https://github.com/slthomason> Do we need to revert to that
> commit, or do we need to add the correct annotations in our .dts files to
> work with the changes?

You need to remove some pcie lines from the dts file. But a fix has already
been commited to master for a lot of, maybe even all, devices affected.

nickberry17, 2018-06-01:
Do you know which commit has the fix?

nickberry17, 2018-06-01:
Found it, I believe

commit 02f815d1907cdd7e042415a2b4a749c819087168
Author: Rosen Penev <rosenp@gmail.com>
Date:   Tue May 29 11:03:03 2018 -0700

    ramips: Fix WiFi after 5f7396ebef09b224edf08b0bda113613a42f0928
    
    That commit exposed a bug in the DTS files used by mt7621 where the wrong
    reg value for pcie1 (and potentially pcie2) was being used. This was
    causing WiFi failures for interfaces in pcie1.
    
    eg. 2.4GHz working but not 5GHz.
    
    As all of these dts entries are already specified in mt7621.dtsi, remove
    them.
    
    Signed-off-by: Rosen Penev <rosenp@gmail.com>

mrkiko, 2018-06-01:
Could you try again? Some reverts happened. My Netgear R6220 is experiencing 
issues too, see issue #176. But I suspect it has nothing to do with mt76 
directly.

mrkiko, 2018-06-01:
I mean, nothing to do with mt76 itself. vSorry....

nickberry17, 2018-06-01:
There is more info here https://forum.lede-project.org/t/mt7621-pcie-dts-issues-mt76-stopped-working-with-latest-updates/14623/66

nbd168, 2018-06-07:
The broken commits have been reverted

closed 2018-06-07 by nbd168