openwrt/mt76 #174: MCU timeout with latest mt76

kristrev, 2018-05-24:
After bumping mt76 to the latest version (792dbe02613299f67e92815f0b2b7584dc951ffd), wifi fails during boot on my devices. The devices are ZBT WG3526 and WE1326, so mt7603/mt7612. In dmesg, I see the following:

```
[   13.044046] mt7603e 0000:02:00.0: ASIC revision: 76030010
[   13.094339] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
[   13.099943] mt7603e 0000:02:00.0: Build Time: 20160107100755
[   13.426631] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[   14.117415] MCU message -1 (seq 1) timed out
[   14.121670] mt7603e 0000:02:00.0: Download request failed
[   14.127225] mt7603e: probe of 0000:02:00.0 failed with error -145
[   14.136457] mt76x2e 0000:01:00.0: ASIC revision: 76120044
[   14.168273] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
[   14.177999] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[   14.183480] mt76x2e 0000:01:00.0: Build: 1
[   14.187574] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[   14.207434] mt76x2e 0000:01:00.0: Firmware running!
[   15.237412] mt76x2e 0000:01:00.0: MCU message 1 (seq 1) timed out
[   15.244649] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
```

I am currently trying to bisect which commit causes issues and will update when I find it.

slthomason, 2018-05-24:
We were having similar problems, and tried to get latest on OpenWRT, and
not can't get that to book on either of the same models.

What commit of OpenWRT are you at?

kristrev, 2018-05-24:
Head as of now.

I have done some testing and I dont think the error is in mt76, but rather related to other ramips changes. Closing issue for now.

closed 2018-05-24 by kristrev

kristrev, 2018-05-25:
Just to summarize. I compiled new, fresh images this morning and installed them on my router showing this problem. Wifi-problem is now gone, so maybe there was something else wrong in my build-directory. Will keep an eye out for message, should it appear again.

CR-Ryan, 2018-05-25:
Both myself and @slthomason are still seeing this on the ZBT WE1326 - 7603/7612

We deleted our entire build folder, and pulled new again.  Still seeing this in logread:

Fri May 25 20:05:38 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Fri May 25 20:05:39 2018 daemon.err hostapd: Could not set interface wlan1 flags (UP): Operation timed out
Fri May 25 20:05:39 2018 daemon.err hostapd: nl80211: Could not set interface 'wlan1' UP
Fri May 25 20:05:39 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Fri May 25 20:05:39 2018 daemon.err hostapd: nl80211 driver initialization failed.
Fri May 25 20:05:39 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Fri May 25 20:05:39 2018 daemon.notice hostapd: wlan1: AP-DISABLED
Fri May 25 20:05:39 2018 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Fri May 25 20:05:39 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
**Fri May 25 20:05:39 2018 kern.err kernel: [  583.624013] mt76x2e 0000:01:00.0: MCU message 20 (seq 2) timed out**
Fri May 25 20:05:39 2018 daemon.notice netifd: radio0 (1573): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Fri May 25 20:05:39 2018 daemon.notice netifd: radio0 (1573): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path ()
Fri May 25 20:05:39 2018 daemon.notice netifd: radio0 (1573): Command failed: Invalid argument
Fri May 25 20:05:39 2018 daemon.notice netifd: radio0 (1573): Device setup failed: HOSTAPD_START_FAILED
Fri May 25 20:05:48 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Fri May 25 20:05:49 2018 daemon.err hostapd: Could not set interface wlan1 flags (UP): Operation timed out
Fri May 25 20:05:49 2018 daemon.err hostapd: nl80211: Could not set interface 'wlan1' UP
Fri May 25 20:05:49 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Fri May 25 20:05:49 2018 daemon.err hostapd: nl80211 driver initialization failed.
Fri May 25 20:05:49 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Fri May 25 20:05:49 2018 daemon.notice hostapd: wlan1: AP-DISABLED
Fri May 25 20:05:49 2018 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Fri May 25 20:05:49 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
**Fri May 25 20:05:49 2018 kern.err kernel: [  593.383697] mt76x2e 0000:01:00.0: MCU message 20 (seq 3) timed out**
Fri May 25 20:05:49 2018 daemon.notice netifd: radio0 (1707): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Fri May 25 20:05:49 2018 daemon.notice netifd: radio0 (1707): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path ()
Fri May 25 20:05:49 2018 daemon.notice netifd: radio0 (1707): Command failed: Invalid argument
Fri May 25 20:05:49 2018 daemon.notice netifd: radio0 (1707): Device setup failed: HOSTAPD_START_FAILED

and dmesg

[  583.624013] mt76x2e 0000:01:00.0: MCU message 20 (seq 2) timed out
[  593.383697] mt76x2e 0000:01:00.0: MCU message 20 (seq 3) timed out
[  902.415546] mt76x2e 0000:01:00.0: MCU message 20 (seq 4) timed out

The errors occur when using the default wireless config, or when using our previously known-good config.  An interesting note that might be related, is that the default wireless config that is generated is missing radio 1 altogether:

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

We have tried this config (after setting disabled 0), and many other good wireless configs, and none of them change the outcome.

barhom, 2018-05-25:
Seeing same problem. Just pulled master and this happened immediately.
Problem might not be in mt76.

kristrev, 2018-05-29:
The offending commit (in OpenWRT) is 5f7396ebef09b224edf08b0bda113613a42f0928.

I don't know the implications of a manual revert on all mt7621 targets, but I do not see any regressions on the 1326 at least.