openwrt/mt76 #178: two issues

dimatto, 2018-06-07:
1. 2.4g is very laggy, back to normal after reverting commit 792dbe0.
2. frequently 100% packet loss on 5g, back to normal after reverting commit b2ecc52.

dimatto, 2018-06-07:
I  usually use it behind a wall which makes signal down to -78db~-82db as luci display, but it works very good and connect and transfer data stably unless apply those two commits.

ingamedeo, 2018-06-13:
Just out of curiosity... on which device is this?

nbd168, 2018-06-15:
Please try the latest version (master, not 18.06)

dimatto, 2018-06-16:
2.4g is better, but I don’t know if there are other factors, because this update is all about mt76x2.
5g is still suffering frequently packet loss up to 30%, but a little bit better than 100%, it is 0% before this update and those buggy commits.

nbd168, 2018-06-16:
Please show me the output of `cat /sys/kernel/debug/ieee80211/phy*/mt76/agc` while it’s running normally vs. while there’s packet loss on 5G

dimatto, 2018-06-16:
result after 0% packet loss:
avg_rssi: -77
low_gain: 0
false_cca: 38505
agc_gain_adjust: 0
result after 46% packet loss:(I can’t guarantee correctness because I can’t run command when packet is losing. I think packet loss is over when I can run command)
avg_rssi: -73
low_gain: 1
false_cca: 38328
agc_gain_adjust: 4

nbd168, 2018-06-18:
Please try the latest version from master, it should be better now.

dimatto, 2018-06-18:
it’s better, less packet loss happening ,but still happen.
after a 68% packet loss, I get this:
avg_rssi: -52
low_gain: 2
false_cca: 0
agc_gain_adjust: 0
at luci I see two clients associated, one signal is -52 and the other is -80 which suffered packet loss.
when there is no packet loss, I cat the agc file, avg_rssi is -79

dimatto, 2018-06-19:
By the way, I revert commit 65e161b, packet loss all disappeared, maybe issue is not in commit b2ecc52 , maybe it has been fixed in commit 2cbaa57.
the agc file after this revert:
avg_rssi: -85
low_gain: 0
false_cca: 19028
agc_gain_adjust: 4
the avg_rssi is always about -85, no matter how many clients associated and whether its signal is -48 or -80.

nbd168, 2018-06-19:
Now that I know that you're using two clients (one with strong signal, one with weak signal), I've finally figured out what's going on.
The global averaging of RSSI doesn't work for your setup, because it keeps switching back between high-RSSI and low-RSSI mode, causing your intermittent packet loss.
The reason why it worked before commit 65e161b is that it was globally underestimating the average RSSI, so the driver would never switch to high-RSSI mode.
I've pushed a commit to mt76.git (not to OpenWrt yet), which switches to per-client RSSI tracking and uses the lowest RSSI value as reference for the gain adjustment. Hopefully it should work for you now.
Thanks for your detailed testing and feedback.

nbd168, 2018-06-19:
The formular for calculating the average is not really the issue, it's what you calculate the average from. Even the vendor driver has some workaround to traverse the MAC table and override the RSSI value if there is a station connected with low RSSI. In the latest commit, I just made a cleaner implementation that better suites low RSSI clients.

MaFeSa, 2018-06-20:
@nbd168 does this commit https://github.com/openwrt/mt76/commit/1d4ca10a562b6d9eaec3aaa5f3905804070ae743 also apply to the 2.4 network?
I ask because with the dir-860l I noticed a considerable improvement during streaming with 6 connected devices, some are far from the router.
Thanks

slthomason, 2018-06-21:
What about mt7603?

dimatto, 2018-06-25:
There’s no packet loss for a week.
I still wonder before commit 65e161b why vendor driver’s formula underestimate rssi, what you calculate rssi from is different from vendor driver , or vendor driver underestimate too ?

closed 2018-10-02 by nbd168