openwrt/mt76 #179: Occasional failing to connect and receiving "Wrong password" response

intriguedlife, 2018-06-10:
I occasionally get a "Wrong password" response when attempting to connect with Xiaomi 3G router on an Iphone. This affects both 2.4G MT7603 and 5G MT7612. This has happend when I was far away and close to the station. Tested with latest version. 792dbe02613299f67e92815f0b2b7584dc951ffd

Rising-Sun, 2018-06-10:
I do not have your problems with 5Ghz anymore, at least 5Ghz runs pretty well since 2days. (Using mir3g with master-branch)
2.4Ghz is completly unusable with my iPhone 7. Usually having this bug: https://github.com/openwrt/mt76/issues/167
But yours occured often too, when trying to reconnect with the AP after no data was transfered.

intriguedlife, 2018-06-10:
> I do not have your problems with 5Ghz anymore, at least 5Ghz runs pretty well since 2days. (Using mir3g with master-branch)
Does that mean that you have experienced the issue of this topic (#179) with 5G in the past?

> 2.4Ghz is completly unusable with my iPhone 7. Usually having this bug: #167
I also have the issues described in #167. To keep things clear: #167 is likely a different issue because #179 happens on 2.4G and 5G.

nbd168, 2019-01-26:
Fixed in current versions

closed 2019-01-26 by nbd168