openwrt/mt76 #180: Wifi not working at all on TL-WR840N V5

duvi, 2018-06-11:
I'm new to this device, bought it just last week for testing.
Unfortunately the wlan does not seem to work at all.
I installed the latest trunk image from https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/openwrt-ramips-mt76x8-tl-wr840n-v5-squashfs-sysupgrade.bin and also created my own image in the 18.06 branch.
Either way, there seem to be no errors in the log, but when I enable the wireless interface, it's not working. Most of the time the SSID is not even visible to other devices. Sometimes it is for some seconds, but then it's still impossible to make the connection.
Iwinfo shows TX power as 0 dBm, not sure if that's a good sign.
The router log shows nothing when I accidentally see the SSID on my laptop / phone and try to connect.

duvi, 2018-06-17:
Can someone please confirm if this is a known issue, or am I the only one experiencing it, so my device might be faulty?

robfitsy, 2018-06-18:
I have a mt7612 pcie and have similar issues. Built openwrt image myself. SSID is intermittently barely visible, when it is, phone gets stuck at 'obtaining IP address' and doesn't get any further.

sanitariu, 2018-07-03:
Same result on zbt3805

xfengfeng, 2018-07-27:
Same result on zbt-wg3526

huafu, 2018-07-28:
It's **mostly** working for me, but...

My router is also `TL-WR840N V5`, I've build the image myself with the latest RC tag of `18.06` (I think it's rc2), after cherry-picking the [commit](https://github.com/openwrt/openwrt/commit/ce31bdc20c3f2eb387a84c277dff76324a8e39c7) that allows my router as a target in menuconfig

The wifi does work, it is visible, BUT even with trying most configurations I've got this:
- connecting from my nexus 5x, no problem
- connecting from an iphone (I think it's a 6 or 7): it works very close to the router, but from farer it tries to connect and finally ask for the wifi password again. The logs shows:
```
hostapd: wlan0: STA 70:70:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
```

milankragujevic, 2018-08-26:
Is this fixed? It wasn't working the last time I flashed my router which was a month ago.

nbd168, 2018-08-26:
@huafu The bug you're describing should be fixed now, please try the latest master or 18.06 branch

huafu, 2018-08-26:
@nbd168 thanks but it was for someone who couldn't wait, so I went to exchange the router to another one :-/

closed 2018-10-02 by nbd168

milankragujevic, 2018-10-02:
well, it still doesn't work, i tried yesterday, wifi works but as soon as i pull more than 20 Mbps for a few seconds it disconnects. And sometimes the router freezes so I have to reboot it manually.

milankragujevic, 2018-10-02:
@sanitariu Agreed. With regards to the router loosing configuration, that's because the LUCI interface uses so much space on it's tiny 4MB flash memory that it doesn't have any space left to store configuration. You can see in shell that overlayfs is mounted in /tmp, meaning it will be lost after RAM is cleared, i.e. after reboot. You cannot use LuCI on it, you must remove it and use shell only.

sanitariu, 2018-10-02:
About space i think its a bug. I did a small test. I removed wifi driver from menuconfig. Compiled with luci and again i changed only password but after restart password is lost. This is the space:

root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.3M      2.3M         0 100% /rom
tmpfs                    29.4M    420.0K     29.0M   1% /tmp
tmpfs                    29.4M     48.0K     29.4M   0% /tmp/root
overlayfs:/tmp/root      29.4M     48.0K     29.4M   0% /
tmpfs                   512.0K         0    512.0K   0% /dev

This is the size of flash file:

-rw-r--r--  1 dri  staff  3867334 Oct  2 14:51 openwrt-ramips-mt76x8-tl-wr840n-v5-squashfs-sysupgrade.bin

I think there is free space up to 4MB we need to store settings.

milankragujevic, 2018-10-02:
It is true what I said, because as you can see overlayfs is mounted on /tmp meaning that everything you change is stored in RAM (HINT: 32 MB RAM is free for tmp). 3.8 MiB is 3.99 MB, so you are pretty much full on the storage side, and you cannot store settings. It's supposed to look like this: (from a different router with 8MB FLASH).

```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.5M      2.5M         0 100% /rom
tmpfs                    29.2M    552.0K     28.7M   2% /tmp
/dev/mtdblock4            3.8M      1.6M      2.2M  42% /overlay
overlayfs:/overlay        3.8M      1.6M      2.2M  42% /
tmpfs                   512.0K         0    512.0K   0% /dev
root@OpenWrt:~#
```

araujorm, 2018-10-02:
I've been through this a lot of times too and I can assure that what
Milan Kragujević said is absolutely correct.

What one should do is recompile the image and somehow reduce it's size.
If you want to be able to keep storing the settings on a router with
only 4 MB of flash space, the image should have a size of about 3.6 MiB
of size at most, but even then it can vary - depends on the block size
of the NAND and other reserved partitions sizes, so YMMV.

Unfortunately, if you are on the limits, some small changes on the new
OpenWRT versions (not necessarily the mt76 driver) may be the "missing
drop of water" for the new image's size to increase too much and not
leaving space for the jffs partition to be created.

My advice (that's what I've been doing with success): try removing
unnecessary packages when recompiling (not necessary removing luci
completely, but some of it's modules and other stuff you don't need),
and change any compiling options that allow you to "optimize for size" -
in particular, on master version of OpenWRT you now have options to
compile the kernel and choose "optimize for size" that should help.
Also, see if there are any options mentioning debug symbols or
information that you won't need on you final image and deselect them too
(note: if you are trying to diagnose something, they might come to be
useful, so if you need them in the meantime then just remove other stuff
until you're happy with your image).

In my case, I also remove IPv6 stuff (yeah, I know... it's the future...
but I still live in the present and I'm limited on flash space), and
everything iptables/netfilter related (since I only want a dumb AP - if
you need the firewall funcionality on that router, don't do that). With
this, I'm able to include other stuff I find more useful for my setup
(like iperf3 or tcpdump-mini). Please note that without
iptables/netfilter you may get errors when compiling some of the drivers
- I already posted that as a bug for one I found in OpenWRT Flyspray,
but it was ignored so far and I don't know if what I proposed will ever
be accepted. Should you be interested, see:

https://bugs.openwrt.org/index.php?do=details&task_id=1800

Anyway, that part of the issue you're having would be a general
OpenWRT/LEDE question, so please post that on OpenWRT/LEDE forums if you
still need help. This repo is for mt76 wifi drivers only, let's keep it
clean and separated. Best of luck.

Having said that, I must say I'm also having serious issues with 2.4 Ghz
Wifi on latest mt76 drivers. It used to work at least a few days
(tipically 4) before starting to ask for the password every time
anything tried to connect, at least with drivers of last month (can't
remember the commit, sorry...), but I worked around it by rebooting the
router via cron every 2 days (yeah, ugly solution). With latest versions
however, it just works for a few hours, then nothing is able to connect.

@nbd168: Unfortunately I still can't find a consistent way to trigger
the problem, but I can do any tests or collect information when it
happens again. Any steps you need me to take?

Is there any steps w

sanitariu, 2020-06-03:
Wifi is working but it still has too low tx-power. You can use it only if you are next to the router.
Something is not correctly setup.