openwrt/mt76 #183: mt7603: tx power issue

kysonlok, 2018-06-19:
@hauke 

I use the latest 18.06 [snapshot firmware](https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/openwrt-ramips-mt76x8-gl-mt300n-v2-squashfs-sysupgrade.bin). There is a problem with setting wireless's txpower value. Regardless of any value I setting, *iwinfo* tool always show its value is zero, besides, *cat /sys/kernel/debug/ieee80211/phy0/power* always show 20. I think it is a buggy for setting txpower.

lukasz1992, 2018-06-19:
Getting txpower is not implemented in mt7603 driver.

kysonlok, 2018-06-19:
@lukasz1992 Thanks for your reply. It means txpower is fixed to 20dBm for mt7603 driver? Any advice to change txpower?

lukasz1992, 2018-06-19:
```
static int mt7603_get_txpower(struct ieee80211_hw *hw, struct ieee80211_vif *vif, int *dbm)
{
	return -EINVAL;
}
```

https://github.com/openwrt/mt76/search?utf8=%E2%9C%93&q=IEEE80211_CONF_CHANGE_POWER&type=

mt7603 doesn't support power settings at all - 20dBm is just a placeholder.
Maybe ask @nbd168 to implement it.

kysonlok, 2018-06-19:
Thanks a lot! It seems that the url is mt76x2 txpower setting implement.

closed 2018-06-27 by kysonlok

christiansasig, 2020-03-11:
Hello, I installed the compatible firmware for the zbt-wg108 router "openwrt_ZBT-WE1326_V7" and the 2.4Ghz and 5Ghz wifi modules work correctly. This firmware runs on Linux kernel 3.10.
[https://drive.google.com/drive/folders/1ZFfQQ-7KYNgKc53z_G3_8BOW8Lcl0oEp](url)

Extract the drivers (.ko) of the wifi modules to test in the openwrt 4.14 kernel "openwrt-19.07.2-ramips-mt7621-zbt-we1326-squashfs-sysupgrade.bin" and the problem of tx power remains.

I am attaching the firmware files for possible help to solve this problem.
regards