openwrt/mt76 #184: mt7603: only one device works at a time in adhoc

jsantala, 2018-06-19:
I'll start a separate ticket for the problem I'm having with VoCore2. I have a setup with adhoc + batman on top. I can only have one device with mt7603 active in the adhoc network at a time. If I activate another mt7603 they both stop working. I can have any number of RT5572 devices in the adhoc network and they all work fine together and with one mt7603. Add another mt7603 and the RT5572 keep working, but mt7603 do not. Just two or more mt7603 alone in the adhoc will not work either.

jsantala, 2018-08-13:
Actually, it now (as of last week) seems that you can have two devices in adhoc, but not more. A third one will stop all from working. Will check again this week if the situation is still the same.

jsantala, 2018-08-13:
Same thing today at openwrt commit 0578993. The second device will work barely. A large transfer will drop it offline. Only one an be online realiably at a time.

jsantala, 2018-11-23:
Situation remains the same with OpenWrt SNAPSHOT, r8483-952b117.

sgruszka, 2019-01-23:
What you mean by one device ? Do you create more interfaces by iw dev interface add ?

jsantala, 2019-01-23:
I mean actual multiple physical devices connected to the same IBSS. Only one interface per device.

sgruszka, 2019-01-25:
I checked two MT76 USB devices (MT7612U and MT7610U) in IBSS mode with my patches: 
https://lore.kernel.org/linux-wireless/1548344649-10404-1-git-send-email-sgruszka@redhat.com/
And IBSS work. It should work with MT7628AN as well, but I tested on PC platform , it's possible that this is OpenWRT specific problem. Anyway please recheck that with latest mt76 driver.

sgruszka, 2019-01-25:
You don't need them, they are only for USB, PCe/SoC devices should work without any extra patches.

jsantala, 2019-01-27:
The situation as of 2019-01-26 11:58 openwrt master commit 0465e4 seems to be worse than ever. Also tried 2019-01-25 10:55 commit f01044 before that and it seemed to be just as bad. There is almost zero IBSS connectivity between two VoCore2 boards using the mt76 driver. I will recompile again now, but I don't see any obvious changes in files that would make a difference.

`root@tolppaB8D812x67_27402:~ 130# iw dev
phy#0
	Interface wlan0
		ifindex 5
		wdev 0x2
		addr b8:d8:12:67:6b:0a
		ssid a13-uqeEzrAb
		type IBSS
		channel 13 (2472 MHz), width: 20 MHz (no HT), center1: 2472 MHz
		txpower 20.00 dBm
root@tolppaB8D812x67_27402:~ 0# ifconfig wlan0 192.168.42.1
root@tolppaB8D812x67_27402:~ 0# arp
IP address       HW type     Flags       HW address            Mask     Device
192.168.0.1      0x1         0x2         34:4b:50:b7:ef:08     *        usb0
10.103.98.37     0x1         0x2         b2:1e:24:79:66:d1     *        bat0
172.27.41.192    0x1         0x2         a8:60:b6:08:74:2f     *        eth0
root@tolppaB8D812x67_27402:~ 0# batctl n
[B.A.T.M.A.N. adv openwrt-2018.4-0, MainIF/MAC: eth0/b8:d8:12:67:6b:0b (bat0/3e:f6:4b:3d:d8:83 BATMAN_V)]
IF             Neighbor              last-seen
b8:d8:12:67:62:25   27.250s (        1.0) [     wlan0]
root@tolppaB8D812x67_27402:~ 1# ip a show dev bat0
6: bat0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN qlen 1000
    link/ether 3e:f6:4b:3d:d8:83 brd ff:ff:ff:ff:ff:ff
    inet 10.103.107.10/8 scope global bat0
       valid_lft forever preferred_lft forever
    inet 10.0.0.80/8 scope global secondary bat0
       valid_lft forever preferred_lft forever
    inet6 fe80::3cf6:4bff:fe3d:d883/64 scope link 
       valid_lft forever preferred_lft forever

---cut---

root@tolppaB8D812x67_25125:/ 1# iw dev
phy#0
	Interface wlan0
		ifindex 6
		wdev 0x3
		addr b8:d8:12:67:62:25
		ssid a13-uqeEzrAb
		type IBSS
		channel 13 (2472 MHz), width: 20 MHz (no HT), center1: 2472 MHz
		txpower 20.00 dBm
root@tolppaB8D812x67_25125:/ 0# ifconfig wlan0 192.168.42.2
root@tolppaB8D812x67_25125:/ 0# ping -c 10 192.168.42.1
PING 192.168.42.1 (192.168.42.1): 56 data bytes

--- 192.168.42.1 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
root@tolppaB8D812x67_25125:/ 1# ping -c 10 10.0.0.80
PING 10.0.0.80 (10.0.0.80): 56 data bytes

--- 10.0.0.80 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
`

sgruszka, 2019-01-28:
What is (sub) driver for your hardware  mt76x2e or mt7603 ?

jsantala, 2019-01-28:
Uh-oh. Thanks for that question. Looks like my config had gotten badly out of date and I may not have had all the modules compiled. Now I do. I think I have a mt7603e, but how do I make sure?

Anyway, looks like the situation has improved! I just got two boards working and I'll be testing more. Thanks a lot!

jsantala, 2019-01-28:
Ok, so the limit now seems to be two boards, a situation which I've seen before. I've now been testing with three VoCore2, numbered 27402, 38217 and 38442. I can have either 38217 or 38442 in the same network with 27402 and the IBSS network works, but if I have all three on at the same time, neither 38217 nor 38442 works. They show up in batman and arp table too, but that's all. No replies to pings. 27402 functions as the gateway, so that one I can always connect to.
```
root@tolppaB8D812x67_27402:~ 0# batctl n
[B.A.T.M.A.N. adv openwrt-2018.4-0, MainIF/MAC: eth0/b8:d8:12:67:6b:0b (bat0/2a:2a:92:66:bf:7e BATMAN_V)]
IF             Neighbor              last-seen
b8:d8:12:67:95:49    0.460s (        1.0) [     wlan0]
b8:d8:12:67:96:2a    1.740s (        1.0) [     wlan0]
root@tolppaB8D812x67_27402:~ 0# arp -a
IP address       HW type     Flags       HW address            Mask     Device
192.168.0.1      0x1         0x2         34:4b:50:b7:ef:08     *        usb0
10.103.150.42    0x1         0x2         d2:ed:90:1c:18:20     *        bat0
10.63.118.200    0x1         0x2         d2:ed:90:1c:18:20     *        bat0
10.157.251.96    0x1         0x0         ca:7c:27:80:cc:e8     *        bat0
10.103.149.73    0x1         0x2         ca:7c:27:80:cc:e8     *        bat0
root@tolppaB8D812x67_27402:~ 0# ping -c10 10.103.150.42
PING 10.103.150.42 (10.103.150.42): 56 data bytes

--- 10.103.150.42 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
root@tolppaB8D812x67_27402:~ 1# ping -c10 10.63.118.200
PING 10.63.118.200 (10.63.118.200): 56 data bytes

--- 10.63.118.200 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
root@tolppaB8D812x67_27402:~ 1# ping -c10 10.157.251.96
PING 10.157.251.96 (10.157.251.96): 56 data bytes

--- 10.157.251.96 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
root@tolppaB8D812x67_27402:~ 1# ping -c10 10.103.149.73
PING 10.103.149.73 (10.103.149.73): 56 data bytes

--- 10.103.149.73 ping statistics ---
10 packets transmitted, 0 packets received, 100% packet loss
root@tolppaB8D812x67_27402:~ 1# 
```

sgruszka, 2019-01-28:
You can use  lspci and dmesg to see what HW do you have and lsmod to see what modules are loaded. I would check first if IBSS network works without batman to see if problem is not batman specific.

jsantala, 2019-01-28:
It doesn't seem to work without batman. Right now it doesn't work with just two boards either. Very unstable and generally random. Boards that worked earlier today don't work now. I'll try removing any extra drivers and adding lspci, which I didn't have. lsmod and dmesg didn't really reveal anything other than that it's a MT7628AN.

jsantala, 2019-01-28:
´´´
root@tolppaB8D812x67_27402:~ 0# dmesg | grep -i mt76
[    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
[    0.148519] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.154113] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.159734] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.334812] spi-mt7621 10000b00.spi: sys_freq: 193333333
[   13.011078] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[   14.243534] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[   14.249625] mt76_wmac 10300000.wmac: Build Time: 20151201183641
[   14.269507] mt76_wmac 10300000.wmac: firmware init done
root@tolppaB8D812x67_27402:~ 0# lsmod | grep -i mt76
cfg80211              231456  6 batman_adv,rt2x00lib,mt76x02_lib,mt7603e,mt76,mac80211
mac80211              452385  7 rt2800lib,rt2x00usb,rt2x00lib,mt76x2e,mt76x02_lib,mt7603e,mt76
mt76                   25118  4 mt76x2e,mt76x2_common,mt76x02_lib,mt7603e
mt7603e                37675  0 
mt76x02_lib            37766  2 mt76x2e,mt76x2_common
mt76x2_common          11368  1 mt76x2e
mt76x2e                10358  0 
root@tolppaB8D812x67_27402:~ 0# lspci
root@tolppaB8D812x67_27402:~ 0# 
´´´

jsantala, 2019-01-29:
Hmmh. I'm now testing with LinkIt Smart 7688 Duo (MT7668AN) and they seem to work ok, at least with just two of them. Add a VoCore2 to the network and it won't work, but the LinkIts keep working. They all see each other in batman though. I'll try with some Omega2+ (also MT7688AN) next.

jsantala, 2019-01-29:
Ok, so I now have 3x Omega2+ and 2x LinkIt7688Duo, both MT7688AN, plus a device with RT5370 wifi as a gateway, in the same adhoc network with batman on top and they seem to work ok. I was able to download a 10M file simultaneously on all five devices with no issue.

jsantala, 2019-01-29:
Also, adding a VoCore2/MT7628AN to this network screws things up somehow, because it causes all the other hosts' pings to outside network go through the roof, from XX ms to XXXX ms, even causing the others to drop out.
```
64 bytes from 8.8.8.8: seq=518 ttl=122 time=29.030 ms
64 bytes from 8.8.8.8: seq=519 ttl=122 time=43.744 ms
...
64 bytes from 8.8.8.8: seq=535 ttl=122 time=188.255 ms
64 bytes from 8.8.8.8: seq=536 ttl=122 time=249.604 ms
...
64 bytes from 8.8.8.8: seq=612 ttl=122 time=6019.799 ms
64 bytes from 8.8.8.8: seq=613 ttl=122 time=6155.796 ms
```

sgruszka, 2019-01-30:
> 64 bytes from 8.8.8.8: seq=612 ttl=122 time=6019.799 ms
> 64 bytes from 8.8.8.8: seq=613 ttl=122 time=6155.796 ms

I'm able to reproduce this on USB hardware , for me it's like on the beginning of ping. After few ping's then time values get back to normal. I'm going to investigate this. First I'll try this commit:
https://github.com/openwrt/mt76/commit/1c751f391f7a99047d34835fa5c048ff17e17028
perhaps it fixes the issue.

jsantala, 2019-01-30:
Yes, if I just use VoCore2/MT7628AN, the pings can be bad right away for me as well.

I just realized that MT7688AN devices might have been fine all along, I just never thought to test them with no VoCore2/MT7628AN devices around messing things up.

sgruszka, 2019-01-30:
Indeed commit 1c751f3 fixed enormous ping time-outs for me.

sgruszka, 2019-01-30:
@jsantala I'm lost reading your comments, what do not exact not work .

jsantala, 2019-01-30:
VoCore2 (MT7628AN) does not work.

Omega2+ and LinkIt 7688 Duo (both MT7688AN) both work

Btw, according to https://wikidevi.com/wiki/MediaTek the 7628 has a 2x2:2 radio and 7688 has a 1x1:1. Could that give a clue as to why they act differently or is it something else?

jsantala, 2019-05-23:
At openwrt master head commit cde52c at least five devices work ok. Closing ticket.

closed 2019-05-23 by jsantala