openwrt/mt76 #185: Clients Remaining Associated

Mushoz, 2018-06-22:
Clients that have been turned off hours ago or left the house sometimes remain associated. It seems client specific to which clients it happens and which not: Two android phones do trigger this bug sometimes, while 1 other does not. My TV is another device that also triggers the bug.

It happens both with 2.4ghz and 5ghz WiFi. My logs are spammed with the following stream of messages with the MAC addresses of the stuck clients:

```
Fri Jun 22 08:56:10 2018 daemon.notice hostapd: wlan0-1: AP-STA-POLL-OK AA:BB:CC:DD:EE:FF
Fri Jun 22 08:57:19 2018 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 00:11:22:33:44:55
```

This issue did not happen on Lede 17.01.4.

Mushoz, 2018-06-22:
Another interesting tidbit: After many hours, they finally disassociated with the following messages in the log:

```
Fri Jun 22 11:18:06 2018 daemon.info hostapd: wlan0: STA 00:11:22:33:44:55 WPA: group key handshake failed (RSN) after 4 tries
Fri Jun 22 11:18:06 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:11:22:33:44:55
Fri Jun 22 11:18:07 2018 daemon.info hostapd: wlan0-1: STA AA:BB:CC:DD:EE:FF WPA: group key handshake failed (RSN) after 4 tries
Fri Jun 22 11:18:07 2018 daemon.notice hostapd: wlan0-1: AP-STA-DISCONNECTED AA:BB:CC:DD:EE:FF
Fri Jun 22 11:18:11 2018 daemon.info hostapd: wlan0: STA 00:11:22:33:44:55 IEEE 802.11: deauthenticated due to local deauth request
Fri Jun 22 11:18:12 2018 daemon.info hostapd: wlan0-1: STA AA:BB:CC:DD:EE:FF IEEE 802.11: deauthenticated due to local deauth request
```

Why did this take so long? On 17.01.4 they would disassociate almost immediately when no longer in range/turned on. One important little detail is that I am running wpad instead of wpad-mini for 802.11r. I do have 802.11r disabled, to rule that out as a potential cause. But could the change of wpad-mini to wpad also give rise to the observed behavior?

MaFeSa, 2018-06-26:
Same problem with DIR-860L B1 but when a client (5 GHz) try to connect again after one hour I have these errors.

OpenWrt 18.06-SNAPSHOT r7102-3f3a2c9 / LuCI openwrt-18.06 branch (git-18.163.60998-c379f00) 
```
Tue Jun 26 19:54:56 2018 daemon.notice hostapd: handle_auth_cb: STA 54:60:09:38:cb:ea not found
Tue Jun 26 19:54:56 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: Station tried to associate before authentication (aid=-1 flags=0x0)
Tue Jun 26 19:54:56 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: authenticated
Tue Jun 26 19:54:56 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: associated (aid 1)
Tue Jun 26 19:54:56 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 54:60:09:38:cb:ea
Tue Jun 26 19:54:56 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea WPA: pairwise key handshake completed (RSN)
Tue Jun 26 19:54:56 2018 daemon.info dnsmasq-dhcp[3452]: DHCPREQUEST(br-lan) 192.168.11.146 54:60:09:38:cb:ea
Tue Jun 26 19:54:56 2018 daemon.info dnsmasq-dhcp[3452]: DHCPACK(br-lan) 192.168.11.146 54:60:09:38:cb:ea Chromecast
Tue Jun 26 19:55:59 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: authenticated
Tue Jun 26 19:55:59 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: Station tried to associate before authentication (aid=-1 flags=0x0)
Tue Jun 26 19:55:59 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: authenticated
Tue Jun 26 19:55:59 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: associated (aid 1)
Tue Jun 26 19:55:59 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 54:60:09:38:cb:ea
Tue Jun 26 19:55:59 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea WPA: pairwise key handshake completed (RSN)
Tue Jun 26 19:55:59 2018 daemon.info dnsmasq-dhcp[3452]: DHCPREQUEST(br-lan) 192.168.11.146 54:60:09:38:cb:ea
Tue Jun 26 19:55:59 2018 daemon.info dnsmasq-dhcp[3452]: DHCPACK(br-lan) 192.168.11.146 54:60:09:38:cb:ea Chromecast
Tue Jun 26 19:56:16 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: authenticated
Tue Jun 26 19:56:16 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: Station tried to associate before authentication (aid=-1 flags=0x0)
Tue Jun 26 19:56:16 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: authenticated
Tue Jun 26 19:56:16 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea IEEE 802.11: associated (aid 1)
Tue Jun 26 19:56:16 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 54:60:09:38:cb:ea
Tue Jun 26 19:56:16 2018 daemon.info hostapd: wlan0: STA 54:60:09:38:cb:ea WPA: pairwise key handshake completed (RSN)
Tue Jun 26 19:56:17 2018 daemon.info dnsmasq-dhcp[3452]: DHCPREQUEST(br-lan) 192.168.11.146 54:60:09:38:cb:ea
Tue Jun 26 19:56:17 2018 daemon.info dnsmasq-dhcp[3452]: DHCPACK(br-lan) 192.168.11.146 54:60:09:38:cb:ea Chromecast
Tue Jun 26 19:56:34 2018 daemon.notice hostapd: handle_auth_cb: STA 54:60:09:38:cb:ea not found
```

nbd168, 2018-10-25:
Please try the latest version (master only, not 18.06).

nbd168, 2018-10-27:
Fix confirmed in another ticket

closed 2018-10-27 by nbd168