openwrt/mt76 #186: mt7603 automatic channel selection fails

amaumene, 2018-06-26:
Hi,

On a Netgear R6220, when I set channel to auto on wlan0 (mt7603), the wifi doesn't come up and logread shows this:
```
Tue Jun 26 11:38:17 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.900674] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.928368] br-lan: port 2(wlan0) entered blocking state
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.939162] br-lan: port 2(wlan0) entered disabled state
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.950398] device wlan0 entered promiscuous mode
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.972657] Data buffer not 16 bytes aligned: 87168178
Tue Jun 26 11:38:17 2018 kern.info kernel: [25594.994549] Data buffer not 16 bytes aligned: 8779c2c8
Tue Jun 26 11:38:17 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Jun 26 11:38:17 2018 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Tue Jun 26 11:38:17 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Tue Jun 26 11:38:17 2018 daemon.notice hostapd: wlan0: ACS-STARTED
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.049443] Data buffer not 16 bytes aligned: 864e0038
Tue Jun 26 11:38:17 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.076880] Data buffer not 16 bytes aligned: 8716c0e8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.089763] Data buffer not 16 bytes aligned: 8716c0b8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.105559] Data buffer not 16 bytes aligned: 849c4288
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.126106] Data buffer not 16 bytes aligned: 849c43b8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.141423] Data buffer not 16 bytes aligned: 849c46d8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.155807] Data buffer not 16 bytes aligned: 849c4118
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.168546] Data buffer not 16 bytes aligned: 849c41b8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.181381] Data buffer not 16 bytes aligned: 849c4188
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.198699] Data buffer not 16 bytes aligned: 849c45e8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.211359] Data buffer not 16 bytes aligned: 849c45b8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.227260] Data buffer not 16 bytes aligned: 849c4718
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.241183] Data buffer not 16 bytes aligned: 849c4018
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.254560] Data buffer not 16 bytes aligned: 849c40c8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.267502] Data buffer not 16 bytes aligned: 849c40d8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.285476] Data buffer not 16 bytes aligned: 849c4228
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.307829] Data buffer not 16 bytes aligned: 849c4098
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.320748] Data buffer not 16 bytes aligned: 849c42b8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.335729] Data buffer not 16 bytes aligned: 849c43d8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.348859] Data buffer not 16 bytes aligned: 849c43a8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.370535] Data buffer not 16 bytes aligned: 849c42c8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.397459] Data buffer not 16 bytes aligned: 849c4178
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.409973] Data buffer not 16 bytes aligned: 849c42f8
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.424424] Data buffer not 16 bytes aligned: 849c4158
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.440236] Data buffer not 16 bytes aligned: 849c4458
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.453007] Data buffer not 16 bytes aligned: 849c4428
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.467093] Data buffer not 16 bytes aligned: 849c4258
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.479800] Data buffer not 16 bytes aligned: 849c4448
Tue Jun 26 11:38:17 2018 kern.info kernel: [25595.491726] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.510998] Data buffer not 16 bytes aligned: 849c4558
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.522771] br-lan: port 3(wlan1) entered blocking state
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.533475] br-lan: port 3(wlan1) entered disabled state
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.544642] device wlan1 entered promiscuous mode
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.554288] br-lan: port 3(wlan1) entered blocking state
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.564928] br-lan: port 3(wlan1) entered forwarding state
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.588951] Data buffer not 16 bytes aligned: 8779c698
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ACS
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: wlan1: ACS-STARTED
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.614014] Data buffer not 16 bytes aligned: 8779c468
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.626779] Data buffer not 16 bytes aligned: 8779c638
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.640872] Data buffer not 16 bytes aligned: 8779c0e8
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.653770] Data buffer not 16 bytes aligned: 8779c288
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.666437] Data buffer not 16 bytes aligned: 8779c258
Tue Jun 26 11:38:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Tue Jun 26 11:38:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: handle_probe_req: send failed
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.680584] Data buffer not 16 bytes aligned: 8779c228
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.693664] Data buffer not 16 bytes aligned: 8779c1f8
Tue Jun 26 11:38:18 2018 kern.info kernel: [25595.717455] Data buffer not 16 bytes aligned: 8779c008
Tue Jun 26 11:38:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Tue Jun 26 11:38:18 2018 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Tue Jun 26 11:38:18 2018 daemon.notice hostapd: handle_probe_req: send failed
[... 48 lines cut ...]
	option channel 'auto'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'SG'
	option disabled '0'
```

iwinfo shows this:
```
wlan0     ESSID: "my_ssid"
          Access Point: 8C:3B:AD:DE:9A:20
          Mode: Master  Channel: unknown (2.407 GHz)
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

Let me know if you want more details. This is with OpenWrt SNAPSHOT, r7314-c4aadbdaf6.

LorenzoBianconi, 2018-06-26:
AFAIK automatic channel selection is not supported by mt7603 driver yet

amaumene, 2018-06-29:
Oh ok ! I didn't know that. I let you choose if you want to close this issue then. Thanks :)

barhom, 2018-07-08:
Please close this issue
Maybe we should not show "Auto" in Luci for MT7603?

nbd168, 2018-08-24:
Fixed in the latest version

closed 2018-08-24 by nbd168